lwfinger/rtw88 #200: Rtw_8821cu driver loading exception.

xiaomeng9597, 2024-06-11:
In addition, downloading drv rsvd page and firmware failed, and I don't know where they were downloaded. The following is the error log.
`Tue Jun 11 22:53:02 2024 cron.err crond[3670]: time disparity of 5993642 minutes detected
Tue Jun 11 22:54:50 2024 daemon.info hostapd: wlan0: STA 46:7c:6b:cc:de:b0 IEEE 802.11: authenticated
Tue Jun 11 22:54:50 2024 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 46:7c:6b:cc:de:b0 1
Tue Jun 11 22:54:50 2024 daemon.info hostapd: wlan0: STA 46:7c:6b:cc:de:b0 IEEE 802.11: associated (aid 1)
Tue Jun 11 22:54:50 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 46:7c:6b:cc:de:b0
Tue Jun 11 22:54:50 2024 daemon.info hostapd: wlan0: STA 46:7c:6b:cc:de:b0 RADIUS: starting accounting session F9726CC73B9B71ED
Tue Jun 11 22:55:19 2024 kern.err kernel: [  247.716534] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:19 2024 kern.err kernel: [  247.717835] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:19 2024 kern.err kernel: [  247.973109] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:19 2024 kern.err kernel: [  247.974331] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:20 2024 kern.err kernel: [  248.323236] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:20 2024 kern.err kernel: [  248.324255] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:20 2024 kern.err kernel: [  248.584985] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:20 2024 kern.err kernel: [  248.586260] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:20 2024 kern.err kernel: [  248.854157] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:20 2024 kern.err kernel: [  248.856091] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:21 2024 kern.err kernel: [  249.107109] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:21 2024 kern.err kernel: [  249.108255] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:21 2024 kern.err kernel: [  249.370235] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:21 2024 kern.err kernel: [  249.371457] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:21 2024 kern.err kernel: [  249.651360] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:21 2024 kern.err kernel: [  249.652457] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:22 2024 kern.err kernel: [  250.083864] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:22 2024 kern.err kernel: [  250.084881] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:22 2024 kern.err kernel: [  250.338110] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:22 2024 kern.err kernel: [  250.339260] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:22 2024 kern.err kernel: [  250.591241] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:22 2024 kern.err kernel: [  250.592457] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:23 2024 kern.err kernel: [  251.237941] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:23 2024 kern.err kernel: [  251.239457] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:24 2024 kern.err kernel: [  252.037985] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:24 2024 kern.err kernel: [  252.039211] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:24 2024 kern.err kernel: [  252.750865] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:24 2024 kern.err kernel: [  252.752083] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:25 2024 kern.err kernel: [  253.268180] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:25 2024 kern.err kernel: [  253.269461] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:25 2024 kern.err kernel: [  253.529614] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:25 2024 kern.err kernel: [  253.530832] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:26 2024 kern.err kernel: [  254.597459] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:26 2024 kern.err kernel: [  254.599430] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:26 2024 kern.err kernel: [  254.869985] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:26 2024 kern.err kernel: [  254.871207] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:27 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 46:7c:6b:cc:de:b0
Tue Jun 11 22:55:27 2024 kern.err kernel: [  255.574360] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:27 2024 kern.err kernel: [  255.576016] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:27 2024 kern.warn kernel: [  255.775895] rtw_8821cu 1-1.1:1.2: timed out to flush queue 1
Tue Jun 11 22:55:27 2024 kern.warn kernel: [  255.975762] rtw_8821cu 1-1.1:1.2: timed out to flush queue 3
Tue Jun 11 22:55:28 2024 kern.err kernel: [  256.236365] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:28 2024 kern.err kernel: [  256.237589] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:30 2024 kern.err kernel: [  259.013611] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:30 2024 kern.err kernel: [  259.014832] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:37 2024 kern.err kernel: [  265.283611] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:37 2024 kern.err kernel: [  265.284832] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:43 2024 kern.err kernel: [  271.485530] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:43 2024 kern.err kernel: [  271.487008] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:49 2024 kern.err kernel: [  277.710486] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:49 2024 kern.err kernel: [  277.711708] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:55:55 2024 kern.err kernel: [  283.978659] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:55:55 2024 kern.err kernel: [  283.979959] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:56:02 2024 kern.err kernel: [  290.186906] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:56:02 2024 kern.err kernel: [  290.188007] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:56:08 2024 kern.err kernel: [  296.405911] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:56:08 2024 kern.err kernel: [  296.407130] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:56:14 2024 kern.err kernel: [  302.611659] rtw_8821cu 1-1.1:1.2: error beacon valid
Tue Jun 11 22:56:14 2024 kern.err kernel: [  302.612756] rtw_8821cu 1-1.1:1.2: failed to download drv rsvd page
Tue Jun 11 22:56:18 2024 daemon.err uwsgi[7442]: *** has_emperor mode detected (fd: 7) ***
Tue Jun 11 22:56:18 2024 daemon.err uwsgi[7442]: [uWSGI] getting INI configuration from /etc/uwsgi/vassals/luci-webui.ini
Tue Jun 11 22:56:18 2024 daemon.err uwsgi[7442]: Tue Jun 11 22:56:18 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini has been spawned
Tue Jun 11 22:56:18 2024 daemon.info uwsgi-luci: spawned uWSGI master process (pid: 12591)
[... 52 lines cut ...]
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949): Usage: ubus [<options>] <command> [arguments...]
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949): Options:
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949):  -s <socket>:		Set the unix domain socket to connect to
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949):  -S:			Use simplified output (for scripts)
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949):  -v:			More verbose output
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949):  -m <type>:		(for monitor): include a specific message type
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949): 			(can be used more than once)
[... cut ...]
Tue Jun 11 22:56:59 2024 daemon.notice netifd: radio0 (12949): Device setup failed: HOSTAPD_START_FAILED
Tue Jun 11 22:56:59 2024 daemon.notice netifd: Wireless device 'radio0' set retry=0
Tue Jun 11 22:56:59 2024 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Tue Jun 11 22:56:59 2024 daemon.notice netifd: Wireless device 'radio0' is now down
Tue Jun 11 22:57:02 2024 daemon.info [7395]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/fe380000.usb/usb1/1-1/1-1.1': not supported by any plugin`

xiaomeng9597, 2024-06-12:
> RTL8821C, RTL8822C, and RTL8822B with USB or SDIO interface are not working well in AP mode with this driver.

哦哦，谢谢解答

gtisan, 2024-06-17:
I applied the patch [1] (wifi: rtw88: usb: unbreak multicast) from mailing list on this repo and AP is working now on my USB stick  with 8822CU chipset.

[1] https://lore.kernel.org/linux-wireless/CA+GA0_vcUneM4Uo8fg0=4sCK049uhjv+NskotRbNg9Em0cnyUQ@mail.gmail.com/T/#m2fec56f2958519c40621c3bbd97ce3d06daa5095

kimocoder, 2024-06-17:
https://patchwork.kernel.org/project/linux-wireless/patch/cb46ea35-7e59-4742-9c1f-01ceeaad36fb@gmail.com/

kimocoder, 2024-06-17:
There is active work on the USB part at this point, I see;

https://patchwork.kernel.org/project/linux-wireless/list/?series=&submitter=&state=*&q=rtw88&archive=&delegate=

xiaomeng9597, 2024-06-17:
> I applied the patch [1] (wifi: rtw88: usb: unbreak multicast) from mailing list on this repo and AP is working now on my USB stick with 8822CU chipset.
>
> [1] [...]

Can it be used in openwrt kernel version 5.10 or above? Now my error is that on the openwrt kernel version 5.10.201, the 8821cu module can normally receive WiFi when the board is turned on, but it is useless to disconnect and restart WiFi after a while.

xiaomeng9597, 2024-06-17:
> There is active work on the USB part at this point, I see;
> 
> https://patchwork.kernel.org/project/linux-wireless/list/?series=&submitter=&state=*&q=rtw88&archive=&delegate=

Thank you for your reply. Can this patch be used above openwrt kernel 5.10?

dubhater, 2024-07-25:
This problem may be [fixed](https://github.com/lwfinger/rtw88/commit/4a1ee64475672a9186e10d87ac5e251448f86914) now.

xiaomeng9597, 2024-07-25:
> This problem may be [fixed](https://github.com/lwfinger/rtw88/commit/4a1ee64475672a9186e10d87ac5e251448f86914) now.

Ok, I'll try it when I'm free. Thank you.

closed 2024-07-25 by xiaomeng9597