lwfinger/rtw88 #205: Connection breaks after a while

stkw0, 2024-06-23:
Sometimes my wifi connection suddenly disconnects. When it happens, `dmesg` shows the next message recurrently: "rtw_8821au 1-8:1.0: MAC has not been powered on yet". No matter what I try, seems it's only resolved by rebooting the computer.
This message also shows the first time I boot up the computer, but it connects properly the first time.
May also important to note that before using rtw88 source I used aircrack-ng/rtl8812au driver. With those drivers, I also had random disconnections (maybe due to the AP¿?) but after restarting iwd or NetworkManager it recovered the connection. This does not happen now.

Module: rtw_8821au 
Hardware: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Linux: 6.9.5

dubhater, 2024-06-23:
How long does it usually take to lose the connection like that?

Please attach the full journalctl output from a boot where your connection broke.

stkw0, 2024-06-23:
From a couple of hours (say ~3h) to 1-2 days. I don't use systemd. I will add the relevant logs once it happens again. 

Thank you.

dubhater, 2024-06-24:
It looks like the firmware stops working, no idea why.

"MAC has not been powered on yet" is printed when the chip is being powered on.

Instead of rebooting, have you tried reloading rtw_8821au? Have you tried unplugging the device?

Also, I see that your computer was suspended. Does the problem happen if the computer stays awake the whole time?

stkw0, 2024-06-29:
I waited ~50 hours without suspending it, it didn't failed. Then I suspended it two times leaving some hours in between, without failures. I don't know if some changed that I made to the kernel (for other reasons) could affect this issue. For the past 5 days I didn't had this issue again. 

Will report back if at some point I have more clues :/

dubhater, 2024-06-29:
Did you update rtw88 since your original report? I pushed some changes recently.

stkw0, 2024-06-29:
No. I updated now, if it happens again I will report back. 
I also found weird it switch automatically from 2.4 GHz to 5 GHz network (it being a desktop), I don't know if that could be related.

dubhater, 2024-06-29:
It must be switching because the signal strength varies over time. The switching could be related. If you can give the 2.4 GHz and 5 GHz networks separate SSIDs, you could try to make it switch from one to the other in a loop using nmcli/iwctl. Maybe give it a few seconds after each switch.

stkw0, 2024-06-30:
No. I updated yesterday. For now there are no problems. If I have time I will try to build a test script. Thank you

stkw0, 2024-07-04:
I had some disconnects but now seems it reconnects correctly without hanging forever. I guess this issue can be closed as it could not be reproduced.

tratum, 2024-07-08:
I am having the same problems but it was automatically resolved when I rebooted my system but now I am  having the same problems again today

dubhater, 2024-07-08:
I'm trying to reproduce it now:
```sh
for i in {001..100}; do nmcli connection down 64e4328c-6606-4648-93bc-247763c3bc5a; sleep 10; nmcli connection up 64e4328c-6606-4648-93bc-247763c3bc5a; sleep 10; done
```

dubhater, 2024-07-08:
Still works.

stkw0, 2024-07-08:
I am and also using iwd backend instead of wpa_supplicant

stkw0, 2024-07-08:
It happened again now. rmmod rtw_8821au and modprobe again fixed the issue. Here is the log of the failure and the recovery:

```
[ 7860.564657] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:50:72:2f:39:35
[ 8613.130077] wlan0: disconnect from AP 10:50:72:2f:39:35 for new auth to 10:50:72:2f:39:31
[ 8613.731694] wlan0: authenticate with 10:50:72:2f:39:31 (local address=98:48:27:20:4c:84)
[ 8613.839173] wlan0: send auth to 10:50:72:2f:39:31 (try 1/3)
[ 8613.844410] wlan0: authenticated
[ 8613.846222] wlan0: associate with 10:50:72:2f:39:31 (try 1/3)
[ 8613.851052] wlan0: RX ReassocResp from 10:50:72:2f:39:31 (capab=0x411 status=0 aid=3)
[ 8613.855522] wlan0: associated
[ 8623.976358] rtw_8821au 1-8:1.0: write register 0x8c4 failed with -71
[ 8623.976478] rtw_8821au 1-8:1.0: read register 0x848 failed with -71
[ 8623.976596] rtw_8821au 1-8:1.0: write register 0x848 failed with -71
[ 8623.976718] rtw_8821au 1-8:1.0: read register 0xc00 failed with -71
[ 8623.976838] rtw_8821au 1-8:1.0: read register 0x8b0 failed with -71
[ 8623.976956] rtw_8821au 1-8:1.0: write register 0x8b0 failed with -71
[ 8627.011948] wlan0: disconnect from AP 10:50:72:2f:39:31 for new auth to 10:50:72:2f:39:35
[ 8627.516349] rtw_8821au 1-8:1.0: failed to get tx report from firmware
[ 8627.623288] wlan0: authenticate with 10:50:72:2f:39:35 (local address=98:48:27:20:4c:84)
[ 8627.796034] wlan0: send auth to 10:50:72:2f:39:35 (try 1/3)
[ 8628.300319] rtw_8821au 1-8:1.0: failed to get tx report from firmware
[ 8628.836342] wlan0: send auth to 10:50:72:2f:39:35 (try 2/3)
[ 8629.340327] rtw_8821au 1-8:1.0: failed to get tx report from firmware
[ 8629.860645] wlan0: send auth to 10:50:72:2f:39:35 (try 3/3)
[ 8630.364338] rtw_8821au 1-8:1.0: failed to get tx report from firmware
[ 8630.884403] wlan0: authentication with 10:50:72:2f:39:35 timed out
[ 8670.516554] rtw_8821au 1-8:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 8681.133490] usbcore: deregistering interface driver rtw_8821au
[ 8681.154267] rtw_8821au 1-8:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 8681.295984] usb 1-8: reset high-speed USB device number 3 using xhci_hcd
[ 8685.031857] Loading firmware: rtw88/rtw8821a_fw.bin
[ 8685.032059] rtw_8821au 1-8:1.0: Firmware version 42.4.0, H2C version 0
[ 8685.073861] 00000000: 29 81 00 7c 01 00 01 00 4c 00 04 00 10 00 00 00  )..|....L.......
[ 8685.073863] 00000010: 25 26 26 27 27 27 2e 2e 2e 2e 2e ff ff ff ff ff  %&&'''..........
[ 8685.073864] 00000020: ff ff 1d 1b 19 19 1b 19 17 16 15 15 17 16 16 16  ................
[ 8685.073865] 00000030: fd ff ff ff ff ff 10 ff ff ff ff ff ff ff ff ff  ................
[ 8685.073866] 00000040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073867] 00000050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073868] 00000060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073869] 00000070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073870] 00000080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073870] 00000090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073871] 000000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073872] 000000b0: ff ff ff ff ff ff ff ff ba 27 1e 00 01 00 00 08  .........'......
[ 8685.073873] 000000c0: ff 09 00 ff 00 00 00 55 00 ff ff ff ff ff ff ff  .......U........
[ 8685.073874] 000000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073874] 000000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073875] 000000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073876] 00000100: 57 23 20 01 ff ff 03 98 48 27 20 4c 84 0a 03 52  W# .....H' L...R
[ 8685.073877] 00000110: 65 61 6c 74 65 6b 20 18 03 38 30 32 2e 31 31 61  ealtek ..802.11a
[ 8685.073878] 00000120: 63 20 57 4c 41 4e 20 41 64 61 70 74 65 72 20 00  c WLAN Adapter .
[ 8685.073879] 00000130: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073880] 00000140: ff ff ff ff ff ff ff 0f ff ff ff ff ff ff ff ff  ................
[ 8685.073880] 00000150: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073881] 00000160: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073882] 00000170: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073883] 00000180: ff ff ff ff ff ff ff ff 83 ab 99 2d 03 93 98 a0  ...........-....
[ 8685.073884] 00000190: fc 8c 00 11 9b c4 00 ff ff ff ff ff ff ff ff ff  ................
[ 8685.073884] 000001a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073885] 000001b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073886] 000001c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073887] 000001d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073888] 000001e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.073888] 000001f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[ 8685.075057] usbcore: registered new interface driver rtw_8821au
[ 8687.195869] wlan0: authenticate with 10:50:72:2f:39:35 (local address=98:48:27:20:4c:84)
[ 8687.367864] wlan0: send auth to 10:50:72:2f:39:35 (try 1/3)
[ 8687.369748] wlan0: authenticated
[ 8687.370819] wlan0: associate with 10:50:72:2f:39:35 (try 1/3)
[ 8687.372767] wlan0: RX AssocResp from 10:50:72:2f:39:35 (capab=0x11 status=0 aid=4)
[ 8687.377249] wlan0: associated
[ 8687.466278] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:50:72:2f:39:35
```

dubhater, 2024-07-11:
I pushed something that may help. Maybe it won't. Please pull and test.

tratum, 2024-07-11:
Here's something silly that works for me whenever my connection breaks
For Fedora
```
sudo systemctl restart NetworkManager
sudo systemctl restart NetworkManager.service
sudo reboot
```

dubhater, 2024-07-13:
> I pushed something that may help. Maybe it won't. Please pull and test.

Well, I ran into the disconnection problem again yesterday, and today too. I'm thinking it's somehow caused by my torrent client. @stkw0 and @tratum were you downloading or uploading a lot of Linux ISOs when the connection died? :)

When the connection died yesterday, qBittorrent was showing over 5 GiB downloaded and about the same uploaded. Today it showed 22 GiB downloaded and 7 uploaded.

I tried to trigger the disconnection using iperf3, but it downloaded and uploaded a lot with no issues.

I was wrong earlier, the firmware doesn't die. Everything keeps working, except the driver doesn't receive anything from the chip anymore. I can see it transmitting probe requests on channels 48 and 149, so it's switching the channel and transmitting fine.

tratum, 2024-07-13:
@dubhater, I mean I was in the process of downloading the Windows ISO to set up a dual-boot configuration. However, I encountered frequent network disconnections randomly even before initiating the download. Additionally, upon switching to the Windows operating system, I faced another issue where I was unable to establish a connection to my Wi-Fi network even in the Windows OS.

I've been thinking about it, and I don't think the torrent client is the root cause of the disconnections

tratum, 2024-07-13:
Now these frequent disconnects have stopped for me and here are my system details
```    
         .';:cccccccccccc:;,.
      .;cccccccccccccccccccccc;.         OS: Fedora Linux 40 (Workstation Edition) x86_64 
    .:cccccccccccccccccccccccccc:.       Host: TUF Gaming FX505DT_FX505DT 1.0 
  .;ccccccccccccc;.:dddl:.;ccccccc;.     Kernel: 6.9.8-200.fc40.x86_64 
 .:ccccccccccccc;OWMKOOXMWd;ccccccc:.    
.:ccccccccccccc;KMMc;cc;xMMc:ccccccc:.   
,cccccccccccccc;MMM.;cc;;WW::cccccccc,   Shell: bash 5.2.26 
:cccccccccccccc;MMM.;cccccccccccccccc:  
:ccccccc;oxOOOo;MMM0OOk.;cccccccccccc:   DE: GNOME 46.3.1 
cccccc:0MMKxdd:;MMMkddc.;cccccccccccc;   WM: Mutter 
ccccc:XM0';cccc;MMM.;cccccccccccccccc'   
ccccc;MMo;ccccc;MMW.;ccccccccccccccc;    
ccccc;0MNc.ccc.xMMd:ccccccccccccccc;     
cccccc;dNMWXXXWM0::cccccccccccccc:,      Terminal: gnome-terminal 
cccccccc;.:odl:.;cccccccccccccc:,.       CPU: AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx (8) @ 2.300GHz 
:cccccccccccccccccccccccccccc:'.         GPU: NVIDIA GeForce GTX 1650 Mobile / Max-Q 
.:cccccccccccccccccccccc:;,..            GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series 
  '::cccccccccccccc::;,.                 Memory: 7958MiB / 30007MiB 
```

stkw0, 2024-07-13:
> were you downloading or uploading a lot of Linux ISOs when the connection died? :)

I had qbittorrent opened, but it was not transmitting a high amount of bandwidth. If it's related with that, maybe the problem is more about opening and closing connections (the DHT and so) than bandwidth.

dubhater, 2024-07-18:
Next time it happens, before you do anything else, please gather some information with these simple steps:

1. Mount debugfs: ``# mount -t debugfs none /sys/kernel/debug``
2. Prepare this command but don't run it yet: ``# cat /sys/kernel/debug/ieee80211/phy0/rtw88/{mac_{0..2},mac_{4..7},bb_{8,9},bb_{a..f}} > registers.txt`` On your system it may not be phy0.
3. When you see the LED blinking, press enter to run the command. That's when the chip is definitely powered on, because it's transmitting probe requests while scanning.

If registers.txt is filled mostly with ``eaeaeaea    eaeaeaea    eaeaeaea    eaeaeaea`` it means you missed (the chip was powered off) and need to run cat again.

stkw0, 2024-07-28:
It happened again but seems it's far less common now (still using rtw88 commit 5db150854a0d4dde0c121ad10486c3854b5139d1). rmmod & modprobe workaround the problem. I could not gather the information requested since I didn't have debugfs enabled. Will do it now and update to latest master.

stkw0, 2024-08-08:
> Next time it happens, before you do anything else, please gather some information with these simple steps:
>
>     1. Mount debugfs: `# mount -t debugfs none /sys/kernel/debug`
>
>     2. Prepare this [...]

Tried to follow this instructions but it changed from phy5 (on my machine) to phy6 when re-loading the module. Is there a way to fix it to have a predictable number? Also, would it be fine if I use `tail -f ` to not miss anything?

dubhater, 2024-08-08:
I don't think you can make it more predictable. ``tail -f`` doesn't work for this case.

dubhater, 2024-08-09:
I got another idea. Please apply this patch and let me know if you see the error message "rtw_usb: probably just ran out of RX URBs" when the connection dies:
```diff
diff --git a/drivers/net/wireless/realtek/rtw88/usb.c b/drivers/net/wireless/realtek/rtw88/usb.c
index bf55360f9daf..149a200ffe19 100644
--- a/drivers/net/wireless/realtek/rtw88/usb.c
+++ b/drivers/net/wireless/realtek/rtw88/usb.c
@@ -671,6 +671,9 @@ static void rtw_usb_read_port_complete(struct urb *urb)
 		}
 		if (skb)
 			dev_kfree_skb_any(skb);
+		rtwusb->skipped_resubmit++;
+		if (rtwusb->skipped_resubmit >= RTW_USB_RXCB_NUM)
+			pr_err_once("rtw_usb: probably just ran out of RX URBs\n");
 	}
 }
 
diff --git a/drivers/net/wireless/realtek/rtw88/usb.h b/drivers/net/wireless/realtek/rtw88/usb.h
index 86697a5c0103..85bcb09b7997 100644
--- a/drivers/net/wireless/realtek/rtw88/usb.h
+++ b/drivers/net/wireless/realtek/rtw88/usb.h
@@ -82,6 +82,7 @@ struct rtw_usb {
 	struct rx_usb_ctrl_block rx_cb[RTW_USB_RXCB_NUM];
 	struct sk_buff_head rx_queue;
 	struct work_struct rx_work;
+	int skipped_resubmit;
 };
 
 static inline struct rtw_usb_tx_data *rtw_usb_get_tx_data(struct sk_buff *skb)
```

I would test it myself but my RTL8812AU just died and I don't want to reload the driver until I'm sure I don't need any more information from it.

dubhater, 2024-08-10:
Haha, after RTL8812AU died I plugged RTL8811AU and it also died a few hours later. Only when I'm not trying to make it happen...

I got the register contents from both. I confirmed that rtw88 is not even receiving messages from the firmware (this is the cause of the "failed to get tx report from firmware" errors).

If you haven't started yet, here is a better patch which shows the error code:
```diff
diff --git a/drivers/net/wireless/realtek/rtw88/usb.c b/drivers/net/wireless/realtek/rtw88/usb.c
index bf55360f9daf..4dbcc276a76c 100644
--- a/drivers/net/wireless/realtek/rtw88/usb.c
+++ b/drivers/net/wireless/realtek/rtw88/usb.c
@@ -664,7 +664,6 @@ static void rtw_usb_read_port_complete(struct urb *urb)
 		case -ECOMM:
 		case -EOVERFLOW:
 		case -EINPROGRESS:
-			break;
 		default:
 			rtw_err(rtwdev, "status %d\n", urb->status);
 			break;
```
If this is indeed the right direction, you will see something like ``rtw_8821au 1-8:1.0: status -XYZ``. The connection will break when this appears for the fourth time.

stkw0, 2024-08-11:
In linux 6.10.3 seems that this patch is already applied? Will try with the latest Linux kernel and the latest master of this repo

dubhater, 2024-08-11:
Nevermind, I tried it today. The problem is not there.

stkw0, 2024-09-08:
I don't really know if it's related. But I noticed an incredibly higher number of those failures doing intensive I/O with a USB 3 port in which a HDD is connected. 

After runnig rmmod && modprobe I dumped the registers in case it helps: 
[registers.txt](https://github.com/user-attachments/files/16923591/registers.txt)

dubhater, 2024-09-20:
Today I ran into this problem with RTL8811CU. I'm so glad it's not just RTL8821AU/RTL8812AU.

dubhater, 2024-09-24:
@stkw0 Do you feel like trying some other random things?

Something simple:
```sh
iw wlan0 set power_save off
```

And disabling some code which helps make the wifi fast but isn't strictly necessary:

```diff
diff --git a/phy.c b/phy.c
index 35e05fa..dc5196f 100644
--- a/phy.c
+++ b/phy.c
@@ -774,6 +774,7 @@ static void rtw_phy_ra_track(struct rtw_dev *rtwdev)
 
 void rtw_phy_dynamic_mechanism(struct rtw_dev *rtwdev)
 {
+	return;
 	/* for further calculation */
 	rtw_phy_statistics(rtwdev);
 	rtw_phy_dig(rtwdev);
```

stkw0, 2024-10-06:
In 1 day and 5 hours of network bound activity (downloading big torrents and seeding them) it didn't crash so hard. There was just some minors issues but the driver could recover autonomously (see next dmesg log):

```
[19936.993478] wlan0: disconnect from AP 10:50:72:2f:39:31 for new auth to 10:50:72:2f:39:35
[19937.598483] wlan0: authenticate with 10:50:72:2f:39:35 (local address=98:48:27:20:4c:84)
[19937.770116] wlan0: send auth to 10:50:72:2f:39:35 (try 1/3)
[19937.772213] wlan0: authenticated
[19937.772456] wlan0: associate with 10:50:72:2f:39:35 (try 1/3)
[19937.773950] wlan0: RX ReassocResp from 10:50:72:2f:39:35 (capab=0x11 status=0 aid=2)
[19937.778683] wlan0: associated
[19937.851603] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:50:72:2f:39:35
[26125.116797] wlan0: disconnect from AP 10:50:72:2f:39:35 for new auth to 10:50:72:2f:39:31
[26125.710847] wlan0: authenticate with 10:50:72:2f:39:31 (local address=98:48:27:20:4c:84)
[26125.820297] wlan0: send auth to 10:50:72:2f:39:31 (try 1/3)
[26125.825818] wlan0: authenticated
[26125.826154] wlan0: associate with 10:50:72:2f:39:31 (try 1/3)
[26125.830629] wlan0: RX ReassocResp from 10:50:72:2f:39:31 (capab=0x411 status=0 aid=5)
[26125.835193] wlan0: associated
[29690.802966] TCP: request_sock_TCP: Possible SYN flooding on port 192.168.0.17:62735. Sending cookies.
[74431.407372] wlan0: disconnect from AP 10:50:72:2f:39:31 for new auth to 10:50:72:2f:39:35
[74432.027728] wlan0: authenticate with 10:50:72:2f:39:35 (local address=98:48:27:20:4c:84)
[74432.201877] wlan0: send auth to 10:50:72:2f:39:35 (try 1/3)
[74432.202684] wlan0: authenticated
[74432.203387] wlan0: associate with 10:50:72:2f:39:35 (try 1/3)
[74432.204688] wlan0: RX ReassocResp from 10:50:72:2f:39:35 (capab=0x11 status=0 aid=5)
[74432.209507] wlan0: associated
[74432.227936] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:50:72:2f:39:35
[74873.439924] wlan0: disconnect from AP 10:50:72:2f:39:35 for new auth to 10:50:72:2f:39:31
[74873.942998] rtw_8821au 1-8:1.0: failed to get tx report from firmware
[74874.027127] wlan0: authenticate with 10:50:72:2f:39:31 (local address=98:48:27:20:4c:84)
[74874.136227] wlan0: send auth to 10:50:72:2f:39:31 (try 1/3)
[74874.139109] wlan0: authenticated
[74874.140025] wlan0: associate with 10:50:72:2f:39:31 (try 1/3)
[74874.144450] wlan0: RX ReassocResp from 10:50:72:2f:39:31 (capab=0x411 status=0 aid=3)
[74874.149293] wlan0: associated
[85883.305347] wlan0: disconnect from AP 10:50:72:2f:39:31 for new auth to 10:50:72:2f:39:35
[85883.907053] wlan0: authenticate with 10:50:72:2f:39:35 (local address=98:48:27:20:4c:84)
[85884.081085] wlan0: send auth to 10:50:72:2f:39:35 (try 1/3)
[85884.081899] wlan0: authenticated
[85884.082065] wlan0: associate with 10:50:72:2f:39:35 (try 1/3)
[85884.084219] wlan0: RX ReassocResp from 10:50:72:2f:39:35 (capab=0x11 status=0 aid=4)
[85884.088983] wlan0: associated
[85884.144185] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 10:50:72:2f:39:35
```

Will report in a ~week or before if it crashes

dubhater, 2024-10-06:
Oh, sorry, I forgot to say that the problem is found: https://lore.kernel.org/linux-wireless/baca0d5d-072e-4003-ab4a-ecc524d6b89b@gmail.com/

I will come up with a solution, but first I want to send the second revision of the new drivers.

dubhater, 2024-10-25:
This is the fix I'm testing: [alloc_skb.diff.txt](https://github.com/user-attachments/files/17522683/alloc_skb.diff.txt)

I'm not 100% sure about the thread safety, but nothing exploded in 24 hours so it's probably fine.

I left the error message in for now, just to see how often it fails to allocate the memory and how long these episodes last.

stkw0, 2024-10-25:
Do you want me to test it too?

plumbeo, 2024-10-25:
I can reliably reproduce this bug in minutes (I just need to download a couple of gigabytes of data to an USB stick on a Raspberry Pi 3, I think it's because with just 1 GB of RAM writing to a slow device can quickly put the system into very bad memory pressure) and this patch definitely helps.

Without this patch after four `rtw_8821au 1-1.5:1.0: failed to allocate rx_skb` messages my DWA-171 adapter stops working, with this patch I still see a ton of errors in the logs but there are no other visible consequences:

```Oct 25 19:42:21 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 1
Oct 25 19:42:21 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 1
Oct 25 19:42:21 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 1
Oct 25 19:44:05 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 2
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 3
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 3
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 1
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 1
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 2
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 2
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 2
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
Oct 25 19:44:06 riverside kernel: rtw_8821au 1-1.5:1.0: failed to allocate rx_skb 0
```

stkw0, 2024-10-25:
It didn't worked for me. After a few minutes I got the next errors. The h2c command errors didn't happened before. Some duplicated entries are removed to make the log more readable
```
[  749.025782] wlan0: disconnect from AP .....
[  749.613214] wlan0: authenticate with ....
[  749.732858] wlan0: send auth to.... (try 1/3)
[  749.735042] wlan0: authenticated
[  749.735952] wlan0: associate with .... (try 1/3)
[  749.740156] wlan0: RX ReassocResp from .... (capab=0x411 status=0 aid=7)
[  749.744592] wlan0: associated
[  775.498247] wlan0: disconnect from AP.... for new auth to ....
[  776.089121] wlan0: authenticate with ...
[  776.272411] wlan0: send auth to .... (try 1/3)
[  776.273266] wlan0: authenticated
[  776.274196] wlan0: associate with.... (try 1/3)
[  776.275307] wlan0: RX ReassocResp from .... (capab=0x11 status=0 aid=3)
[  776.279978] wlan0: associated
[  776.361581] wlan0: Limiting TX power to 23 (23 - 0) dBm as advertised by .....
[ 1568.844019] wlan0: disconnect from AP ..... for new auth to....
[ 1569.416784] wlan0: authenticate with ... 
[ 1569.577352] wlan0: send auth to ... (try 1/3)
[ 1569.579568] wlan0: authenticated
[ 1569.580627] wlan0: associate with .... (try 1/3)
[ 1569.585840] wlan0: RX ReassocResp from ....(capab=0x411 status=0 aid=7)
[ 1569.590260] wlan0: associated
[ 1661.747600] rtw_8821au 1-2:1.0: failed to get tx report from firmware
[ 1662.563517] rtw_8821au 1-2:1.0: failed to get tx report from firmware
[ 1664.047399] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1664.051518] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1664.056115] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1664.547602] rtw_8821au 1-2:1.0: failed to get tx report from firmware
[ 1666.891584] rtw_8821au 1-2:1.0: failed to get tx report from firmware
.....
[ 1726.564157] rtw_8821au 1-2:1.0: failed to get tx report from firmware
[ 1728.048339] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1728.052845] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1730.861225] rtw_8821au 1-2:1.0: failed to get tx report from firmware
[ 1736.594180] usbcore: deregistering interface driver rtw_8821au
[ 1736.600379] wlan0: deauthenticating from .... by local choice (Reason: 3=DEAUTH_LEAVING)
[ 1736.605713] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1736.769477] usb 1-2: reset high-speed USB device number 2 using xhci_hcd
[ 1736.925704] Loading firmware: rtw88/rtw8821a_fw.bin
[ 1736.925818] rtw_8821au 1-2:1.0: Firmware version 42.4.0, H2C version 0
[ 1737.344206] usbcore: registered new interface driver rtw_8821au
...
[ 1739.493601] wlan0: RX AssocResp from ...(capab=0x411 status=0 aid=7)
[ 1739.498292] wlan0: associated
[ 1755.544767] rtw_8821au 1-2:1.0: read register 0xd04 failed with -71
[ 1755.544882] rtw_8821au 1-2:1.0: write register 0xc90 failed with -71
[ 1755.545038] rtw_8821au 1-2:1.0: read register 0x73 failed with -71
[ 1755.545155] rtw_8821au 1-2:1.0: write register 0x73 failed with -71
[ 1755.545277] rtw_8821au 1-2:1.0: read register 0x6c0 failed with -71
[ 1755.545394] rtw_8821au 1-2:1.0: write register 0x6c0 failed with -71
[ 1755.545517] rtw_8821au 1-2:1.0: write register 0x6c4 failed with -71
[ 1758.516477] rtw_8821au 1-2:1.0: failed to get tx report from firmware
[ 1759.589486] rtw_8821au 1-2:1.0: failed to get tx report from firmware
[ 1760.048981] rtw_8821au 1-2:1.0: failed to send h2c command
....
[ 1798.065230] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1798.070157] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1798.681863] rtw_8821au 1-2:1.0: failed to send h2c command
[ 1799.301887] rtw_8821au 1-2:1.0: failed to get tx report from firmware
```

belzerus, 2024-10-28:
> This is the fix I'm testing: [alloc_skb.diff.txt](https://github.com/user-attachments/files/17522683/alloc_skb.diff.txt)
>
> I'm not 100% sure about the thread safety, but nothing exploded in 24 hours [...]

@dubhater can you attach your entire usb.c file, or attach a patch based on latest wireless-next main branch? I tried to backport your patch to my 6.6.51 based kernel to give it a spin. It looks OK compared with your patch. But I get below crash on target: 

```
[  322.303514] ------------[ cut here ]------------
[  322.308160] WARNING: CPU: 0 PID: 0 at kernel/workqueue.c:1788 __queue_work+0x554/0x80c
[  322.316111] Modules linked in: rtw88_8822cu rtw88_8822c rtw88_usb rtw88_core mac80211 libarc4 cfg80211 btusb btrtl btintel btbcm btmtk imx_sdma ip_tables x_tables
[  322.330733] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W          6.6.51-00146-gbda4ce755409 #19
[  322.339968] Hardware name: Freescale i.MX6 SoloX (Device Tree)
[  322.345813]  unwind_backtrace from show_stack+0x10/0x14
[  322.351072]  show_stack from dump_stack_lvl+0x58/0x70
[  322.356148]  dump_stack_lvl from __warn+0x70/0x1c0
[  322.360971]  __warn from warn_slowpath_fmt+0x16c/0x294
[  322.366139]  warn_slowpath_fmt from __queue_work+0x554/0x80c
[  322.371828]  __queue_work from queue_work_on+0x8c/0x94
[  322.376986]  queue_work_on from rtw_usb_read_port_complete+0x60/0x148 [rtw88_usb]
[  322.384516]  rtw_usb_read_port_complete [rtw88_usb] from __usb_hcd_giveback_urb+0x90/0x124
[  322.392823]  __usb_hcd_giveback_urb from usb_giveback_urb_bh+0xa0/0x140
[  322.399464]  usb_giveback_urb_bh from tasklet_action_common+0x100/0x15c
[  322.406110]  tasklet_action_common from handle_softirqs+0xd4/0x49c
[  322.412318]  handle_softirqs from __irq_exit_rcu+0x154/0x194
[  322.418002]  __irq_exit_rcu from irq_exit+0x8/0x28
[  322.422817]  irq_exit from __irq_svc+0x90/0xd4
[  322.427283] Exception stack(0xc1401f40 to 0xc1401f88)
[  322.432349] 1f40: c0d39310 2e444000 00000001 00000000 c017c4f0 c1416840 c1416840 c141249c
[  322.440538] 1f60: ffffffff 00000000 c1371a80 00000000 00000000 c1401f90 c0d39310 c0d39314
[  322.448721] 1f80: 60010013 ffffffff
[  322.452217]  __irq_svc from default_idle_call+0x1c/0x1f4
[  322.457549]  default_idle_call from do_idle+0xac/0x124
[  322.462707]  do_idle from cpu_startup_entry+0x28/0x2c
[  322.467774]  cpu_startup_entry from rest_init+0xd0/0x184
[  322.473103]  rest_init from arch_post_acpi_subsys_init+0x0/0x8
[  322.478970] irq event stamp: 10860
[  322.482377] hardirqs last  enabled at (10859): [<c01f864c>] tick_nohz_idle_exit+0x64/0x178
[  322.490662] hardirqs last disabled at (10860): [<c0d3a81c>] __schedule+0x53c/0xbfc
[  322.498247] softirqs last  enabled at (10856): [<c012f3b4>] handle_softirqs+0x2ac/0x49c
[  322.506266] softirqs last disabled at (10843): [<c012f74c>] __irq_exit_rcu+0x154/0x194
[  322.514201] ---[ end trace 0000000000000000 ]---
```

Also I can see that your patch assumes that for example below part already exists in usb.c and I do not have that error message in my code. So might be something missing in the base for me:

```
       rxcb->rx_skb = alloc_skb(RTW_USB_MAX_RECVBUF_SZ, GFP_ATOMIC);
       if (!rxcb->rx_skb) {
               rtw_err(rtwdev, "failed to allocate rx_skb\n");
               return;
       }
```

dubhater, 2024-10-28:
@stkw0 Are you sure those errors are caused by my patch?

stkw0, 2024-10-28:
For the last 3 kernel releases I was having the ` failed to get tx report from firmware` error but not the h2c one nor the write/read register. It may also been a thing of the latest kernel release. I undone the patch with the same kernel version, so I could check if I see them also without that patch or not.

belzerus, 2024-10-29:
@dubhater it was just PEBKAC from my side. I managed to remove below line

```
INIT_WORK(&rtwusb->rx_work, rtw_usb_rx_handler);
```

from my code base when manually backporting your changes. So now it's up-and-running with your changes. Will keep you posted on my results. Will run it on both a i.MX6SoloX (armhf single cpu) and a i.MX8MP (arm64 multicore)

belzerus, 2024-10-30:
Have been running around 100 mender upgrades with this patch without stalling the driver. After that I also re-applied the "failed to allocate rx_skb " printout from the original patch to ensure that I really spotted the original issue, and I can see below printouts from time-to-time but the device is not stalling and I do not get any "failed to get tx report" so far. So looking really promising.

```
...
[ 8885.189994] rtw_8822cu 1-1:1.2: failed to allocate rx_skb 0
...
```

stkw0, 2024-11-04:
I pulled the latest master and it seems to work correctly now :D 
I can see some reconnections on dmesg but seems to be totally transparent by applications (eg: qbittorrent)

dubhater, 2024-11-06:
I was not happy with the solution from https://github.com/lwfinger/rtw88/issues/205#issuecomment-2437776828 so I did something else. Please pull and test again.

Let me know if you see any of these messages in the log:
* skipping RX URB, packet too big
* failed to allocate RX skb of size
* freeing excess RX skb
* left ____ frames in the rx queue for later
* allocating new RX skb
* no free RX skb and alloc_skb failed

I will get rid of them all after some testing.

henkv1, 2024-11-06:
@dubhater  The new RX code breaks AP mode. I get a kernel panic when I put my rtl8812cu device in AP mode.

dubhater, 2024-11-06:
@henkv1 Is your kernel older than 6.9? What's the panic about?

dubhater, 2024-11-07:
@henkv1 I fixed something related to RTL8812CU. Maybe it was the reason for your panic.

henkv1, 2024-11-07:
@dubhater Yes, it's fixed. Thank you.

plumbeo, 2024-11-07:
I've been testing the new changes, no issues and the speed seems to be a little better than before even with TCP transfers. I'm seeing a lot of `allocating new RX skb`/`freeing excess RX skb` messages in the logs though.

stkw0, 2024-11-13:
The current state is much better. Still, from time to time it disconnects and when it does dmesg shows a lot of `failed to allocate rx_skb` and `failed to get tx report from firmware`

dubhater, 2024-11-13:
@stkw0 If you are seeing that exact message, with lowercase rx followed by underscore, you are not running the latest code. The problem is fixed now. You should pull and recompile.

stkw0, 2024-11-16:
I updated to the latest and now I get  the "allocating new" and "freeing excess" messages
```
[ 1641.081770] rtw_8821au 1-8:1.0: allocating new RX skb
[ 1641.081809] rtw_8821au 1-8:1.0: freeing excess RX skb
[ 1736.708519] rtw_8821au 1-8:1.0: allocating new RX skb
[ 1736.708560] rtw_8821au 1-8:1.0: freeing excess RX skb
```

dubhater, 2024-11-17:
@stkw0 Do they show up a lot?

stkw0, 2024-11-17:
More or less, there are some bursts, then it keeps calm for a while. See attached log of `dmesg | grep rtw_8821au`.
[dmesg_grep_rtw_8821au.txt](https://github.com/user-attachments/files/17792914/dmesg_grep_rtw_8821au.txt)

dubhater, 2024-11-26:
@stkw0 Do you see a lot of "allocating new RX skb" if you apply the patch below?

```diff
diff --git a/usb.c b/usb.c
index 9dafc75..d1cce9c 100644
--- a/usb.c
+++ b/usb.c
@@ -623,13 +623,7 @@ static void rtw_usb_rx_handler(struct rtw_usb *rtwusb)
 			rx_desc += next_pkt;
 		} while (rx_skb->data + urb_len > rx_desc + pkt_desc_sz);
 
-		if (skb_queue_len(&rtwusb->rx_free_queue) >=
-		    RTW_USB_RX_SKB_NUM - RTW_USB_RXCB_NUM) {
-			rtw_err(rtwdev, "freeing excess RX skb\n");
-			dev_kfree_skb_any(rx_skb);
-		} else {
 			skb_queue_tail(&rtwusb->rx_free_queue, rx_skb);
-		}
 	}
 
 	if (limit == 200)
```

stkw0, 2024-11-27:
Been running for over 7h now, only saw three "allocating new RX skb" a few minutes after running qbittorrent but seems pretty stable. I will report back if there are any news, but looks very promising :) 
Thank you very much for your hard work!

PinkFreud, 2024-11-28:
I just started testing this driver yesterday after dusting off my old Edimax EW-7811UTC and connecting it to a sbc w/ a RISC-V cpu.

It seems to be fairly stable for the moment - scanning works and returns available networks, connecting to one of my networks w/ WPA3 works, etc.

Transferring a 1GB file over the wireless link caused a few messages to appear, though:
```
[18504.463591] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18504.468961] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18504.474251] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18504.479527] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18504.484771] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18504.511701] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18504.517037] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18504.522359] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18504.527633] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18505.103092] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18505.108443] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18505.113747] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18505.119020] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18505.124305] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18505.129569] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18505.134815] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18505.140075] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18505.202344] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18505.207654] rtw_8821au 1-1.3:1.0: allocating new RX skb
[18505.213003] rtw_8821au 1-1.3:1.0: freeing excess RX skb
[18505.218327] rtw_8821au 1-1.3:1.0: freeing excess RX skb
```

So far, they appear harmless - the connection is stable.

dubhater, 2024-11-28:
@PinkFreud You should try this too: https://github.com/lwfinger/rtw88/issues/205#issuecomment-2501890060

PinkFreud, 2024-11-28:
> @PinkFreud You should try this too: [#205 (comment)](https://github.com/lwfinger/rtw88/issues/205#issuecomment-2501890060)

I copied the same file after recompiling w/ the patch applied to usb.c.  No skb messages are appearing so far.

PinkFreud, 2024-11-28:
```
[ 8410.130350] rtw_8821au 1-1.3:1.0: allocating new RX skb
```

Well, I received one.  :)  Looks like these are becoming much more rare now - and no 'freeing excess...' message accompanying it, either.

[15 more replies]

closed 2025-02-13 by dubhater