lwfinger/rtw88 #207: Missing entries in dkms.conf

aikawayataro, 2024-06-26:
Hi there! I wanted to test this driver with 8812au, but there's no my kernel module in dkms.conf (I prefer to use dkms for drivers).
After looking at the Makefile, I've found that these are missing: 8703b r8723cs 8723x 8812au 8821a 8821ae 8821au. Is this intentional?
The sign step is also missing some, but I don't use it, just noticed.
After adding the necessary entry manually and loading drivers, I got
```
[ 2594.439232] rtw_8812au 3-4:1.0: Direct firmware load for rtw88/rtw8812a_fw.bin failed with error -2
[ 2594.439245] rtw_8812au 3-4:1.0: failed to request firmware
...
[ 2595.036974] rtw_8812au 4-4:1.0: failed to wait firmware completion
[ 2595.036988] rtw_8812au 4-4:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 2595.062601] rtw_8812au 4-4:1.0: failed to wait firmware completion
[ 2595.062610] rtw_8812au 4-4:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 2595.085873] rtw_8812au 4-4:1.0: failed to wait firmware completion
[ 2595.085878] rtw_8812au 4-4:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
```
My kernel is `6.6.35-2-lts` (Arch Linux).

dubhater, 2024-06-26:
It's not on purpose. Those modules are only missing because no one added them.

There is a pull request for rtw_8723x. I guess I can merge that. https://github.com/lwfinger/rtw88/pull/204

The firmware is in the repository. It's called rtw8812a_fw.bin. It looks like ``make install`` installs it. I'm not sure how it's supposed to be installed when you use dkms.

If you open a pull request, don't add rtw_8821ae. We're going to drop that module.

aikawayataro, 2024-06-26:
I think I will submit a pull request then.
As for the firmware, I guess it's on the packaging side of the driver. I will ship it separately, thanks for the insight.

closed 2024-06-27 by dubhater