lwfinger/rtw88 #212: Difference between rtl8821ce and alt_rtl8821ce

fahimscirex, 2024-07-02:
What's the use cases of these two drivers? How do I decide which one do I have to use?

dubhater, 2024-07-25:
If you mean this one: https://github.com/tomaspinho/rtl8821ce alt_rtl8821ce is the same, but newer and unmaintained.

fahimscirex, 2024-07-29:
I meant these two 
- https://github.com/lwfinger/rtw88/blob/master/rtw8821ce.c
- https://github.com/lwfinger/rtw88/tree/master/alt_rtl8821ce

dubhater, 2024-07-29:
alt_rtl8821ce is the traditional type of driver made by Realtek. You should use it only if you have to.

rtw8821ce.c is another driver made by Realtek, written from scratch in a way that is acceptable for inclusion in the Linux kernel. This is the one you should try first.