lwfinger/rtw88 #213: Firmware failed to leave lps state

asimovc, 2024-07-02:
I get this message when i connect my bluetooth headphones 

`rtw_8821ce 0000:02:00.0: firmware failed to leave lps state`

I don't know if this is related but also i get some sound errors when i use the bluetooth headphone, like a sound distortion, and only happens with this firmware. I also compile the firmware in my kernel and use `rtw8821c_fw.bin` and `rtl8821c_fw.bin` for bluetooth firmware

dubhater, 2024-07-25:
The place to report this is the linux-wireless@vger.kernel.org mailing list. Put "rtw88" in the subject to make sure the right people see it.

dubhater, 2025-08-21:
@asimovc @Benexl  What do you get from this command? ``grep RFE /sys/kernel/debug/ieee80211/phy*/rtw88/coex_info``

How many antennas does your card have?

asimovc, 2025-08-24:
> I had the exact same issues, solved using the solution from the thread linked above.
> 
> Which was, `/etc/modprobe.d/rtw88.conf`
> 
> ```
> options rtw88_core disable_lps_deep=y
> options rtw88_pci disable_aspm=y
> ```

Solved the issue with this.

asimovc, 2025-08-24:
Correction, it helps to not break for a time, prolongs the usability but at some point it jus collapse and disconnect