lwfinger/rtw88 #215: I've installed the drivers following the guide, but the USB WiFi adapter isn't detected or working. Please assist in resolving this issue.

arifrahmanumam, 2024-07-10:
I am using the USB WiFi adapter "CHANEVE RTL8812AU."
[CHANEVE RTL8812AU](https://www.aliexpress.com/item/4000762461362.html?spm=a2g0o.productlist.main.1.d15e701exWRdml&algo_pvid=24c83800-586b-4e64-bee2-56a98b9770b0&algo_exp_id=24c83800-586b-4e64-bee2-56a98b9770b0-0&pdp_npi=4%40dis%21IDR%21276270.00%21267964.00%21%21%2116.63%2116.13%21%40210133e017206219533578822e20ec%2110000007572029138%21sea%21MY%210%21AB&curPageLogUid=cL25p03dXrKR&utparam-url=scene%3Asearch%7Cquery_from%3A)
[image]
[image]

**uname -r :**
6.8.11-amd64

**mokutil --sb-state :**
SecureBoot disabled

**lsusb :**      
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 001 Device 004: ID 1bcf:2cb0 Sunplus Innovation Technology Inc. HD User Facing
Bus 001 Device 005: ID 1c7a:0570 LighTuning Technology Inc. EgisTec Touch Fingerprint Sensor
Bus 001 Device 010: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

**lsusb -t :**
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 001: Dev 010, If 0, Class=Vendor Specific Class, Driver=[none], 480M
    |__ Port 005: Dev 003, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 005: Dev 003, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 007: Dev 004, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 007: Dev 004, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 009: Dev 005, If 0, Class=Vendor Specific Class, Driver=[none], 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 5000M

**lsmod | grep rtw :**                                                                         
rtw_8812au             12288  0
rtw_8821a             114688  1 rtw_8812au
rtw_usb                28672  1 rtw_8812au
rtw_core              315392  2 rtw_usb,rtw_8821a
mac80211             1396736  3 rtw_usb,rtw_core,iwlmvm
cfg80211             1355776  4 rtw_core,iwlmvm,iwlwifi,mac80211
usbcore               409600  6 xhci_hcd,rtw_usb,uvcvideo,btusb,xhci_pci,rtw_8812au
[image]

**I have tried blacklisting :**  'rtw_8821a' but there has been no change at all.

**Disable/Enable a Kernel Module :**
[image]

**dmesg | grep rtw :**
[    5.431263] rtw_core: loading out-of-tree module taints kernel.
[    5.431269] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[    5.490206] rtw_8812au 1-1:1.0: firmware: direct-loading firmware rtw88/rtw8812a_fw.bin
[    5.490356] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[    5.534670] rtw_8812au 1-1:1.0: rfe 1 isn't supported
[    5.534718] rtw_8812au 1-1:1.0: failed to setup chip efuse info
[    5.534720] rtw_8812au 1-1:1.0: failed to setup chip information
[    5.535156] usbcore: registered new interface driver rtw_8812au
[    8.487031] Modules linked in: overlay qrtr sunrpc binfmt_misc nls_ascii nls_cp437 vfat fat snd_soc_skl_ssp_clk snd_sof_pci_intel_skl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation snd_sof_intel_hda_mlink soundwire_cadence snd_sof_intel_hda intel_uncore_frequency intel_uncore_frequency_common snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_soc_skl_hda_dsp snd_soc_hdac_hdmi btusb snd_soc_intel_hda_dsp_common btrtl snd_sof_utils rtw_8812au(OE) soundwire_bus snd_hda_codec_realtek btintel x86_pkg_temp_thermal btbcm rtw_8821a(OE) intel_powerclamp btmtk iwlmvm rtw_usb(OE) snd_soc_avs snd_hda_codec_generic rtw_core(OE) coretemp snd_soc_hda_codec snd_soc_dmic bluetooth kvm_intel snd_soc_skl snd_soc_hdac_hda mac80211 iwlwifi uvcvideo snd_hda_ext_core snd_soc_sst_ipc kvm snd_soc_sst_dsp snd_soc_acpi_intel_match videobuf2_vmalloc snd_soc_acpi libarc4 sha3_generic uvc jitterentropy_rng snd_soc_core mei_hdcp mei_pxp irqbypass intel_rapl_msr cfg80211 videobuf2_memops videobuf2_v4l2 snd_compress drbg videodev
[   18.043512] rtw_8812au 1-1:1.0: firmware: direct-loading firmware rtw88/rtw8812a_fw.bin
[   18.043867] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[   18.087239] rtw_8812au 1-1:1.0: rfe 1 isn't supported
[   18.087289] rtw_8812au 1-1:1.0: failed to setup chip efuse info
[   18.087296] rtw_8812au 1-1:1.0: failed to setup chip information
[   21.623821] rtw_8812au 1-2:1.0: firmware: direct-loading firmware rtw88/rtw8812a_fw.bin
[   21.624204] rtw_8812au 1-2:1.0: Firmware version 52.14.0, H2C version 0
[   21.668253] rtw_8812au 1-2:1.0: rfe 1 isn't supported
[   21.668302] rtw_8812au 1-2:1.0: failed to setup chip efuse info
[   21.668310] rtw_8812au 1-2:1.0: failed to setup chip information
[   36.860998] usbcore: deregistering interface driver rtw_8812au
[   45.428276] rtw_8812au 1-2:1.0: firmware: direct-loading firmware rtw88/rtw8812a_fw.bin
[   45.428446] rtw_8812au 1-2:1.0: Firmware version 52.14.0, H2C version 0
[   45.470214] rtw_8812au 1-2:1.0: rfe 1 isn't supported
[   45.470252] rtw_8812au 1-2:1.0: failed to setup chip efuse info
[   45.470254] rtw_8812au 1-2:1.0: failed to setup chip information
[   45.470815] usbcore: registered new interface driver rtw_8812au
[   93.557469] rtw_8812au 1-4:1.0: firmware: direct-loading firmware rtw88/rtw8812a_fw.bin
[   93.557877] rtw_8812au 1-4:1.0: Firmware version 52.14.0, H2C version 0
[   93.600221] rtw_8812au 1-4:1.0: rfe 1 isn't supported
[   93.600259] rtw_8812au 1-4:1.0: failed to setup chip efuse info
[   93.600262] rtw_8812au 1-4:1.0: failed to setup chip information
[  359.421830] rtw_8812au 1-2:1.0: firmware: direct-loading firmware rtw88/rtw8812a_fw.bin
[  359.421985] rtw_8812au 1-2:1.0: Firmware version 52.14.0, H2C version 0
[  359.464093] rtw_8812au 1-2:1.0: rfe 1 isn't supported
[  359.464131] rtw_8812au 1-2:1.0: failed to setup chip efuse info
[  359.464133] rtw_8812au 1-2:1.0: failed to setup chip information
[ 4747.774667] rtw_8812au 1-1:1.0: firmware: direct-loading firmware rtw88/rtw8812a_fw.bin
[ 4747.775255] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[ 4747.819401] rtw_8812au 1-1:1.0: rfe 1 isn't supported
[ 4747.819439] rtw_8812au 1-1:1.0: failed to setup chip efuse info
[ 4747.819441] rtw_8812au 1-1:1.0: failed to setup chip information

dubhater, 2024-07-10:
Thanks for your report. I pushed a fix.

arifrahmanumam, 2024-07-17:
How is the progress on this issue? Is there anyone who can provide a solution?

dubhater, 2024-07-17:
I pushed a fix. You can run ``git pull`` and ``make`` and try again.

arifrahmanumam, 2024-07-17:
Thank you! With your help, my USB Wi-Fi adapter "CHANEVE RTL8812AU" is now functioning properly and supports monitor mode.
[image]
[image]
[image]
[image]

The problem now is that I can’t switch it back to managed mode. When I try to revert to managed mode, it stays in monitor mode unless I unplug the USB Wi-Fi adapter and plug it back in, then it switches to managed mode. However, this isn't a major issue. Once again, thank you for your help!
[image]

arifrahmanumam, 2024-07-18:
There is another problem: the 'Available networks' on the Wi-Fi panel only appears for a few seconds and then disappears.
[image]

https://github.com/user-attachments/assets/a6b49c74-3717-4d49-9d53-47ae13227a28

closed 2024-07-25 by dubhater