lwfinger/rtw88 #218: WiFi scanning not working in AP mode

GCic8, 2024-07-16:
Hi, 
the driver for RTL8723DU works really well but when I'm in AP mode i cannot scan wifi networks. Below the output:

# iwlist wlan0 scan
wlan0     Interface doesn't support scanning : Operation not supported

# iw dev wlan0 scan
command failed: Operation not supported (-95)

# iw dev wlan0 scan ap-force
command failed: Operation not supported (-95)

Any suggestion?

morrownr, 2024-07-16:
Hi @GCic8 

Does anything work in AP mode?

Are you using hostapd?

GCic8, 2024-07-17:
Hi,I'm using hostapd and it works very well, both in client mode (using WPA supplicant) and AP mode (using hostapd). The only thing not working is ssid scanning while in AP mode.I'm on Debian bookworm with kernel 6.6

 

GC

 

Nota di
riservatezza: il contenuto di questo messaggio e gli eventuali documenti
allegati sono strettamente riservati al destinatario e hanno carattere
confidenziale e sono ad uso esclusivo del destinatario. Chiunque ricevesse
questo messaggio per errore o lo leggesse senza esserne legittimato è avvisato
che trattenerlo, copiarlo, divulgarlo, distribuirlo a persone diverse dal
destinatario è severamente proibito. In caso di erronea ricezione o
mancato recapito si prega di voler avvisare il mittente e cancellare il
messaggio. Grazie.

-------------------------

Disclaimer: ai
sensi del Regolamento UE 2016/679 del Consiglio e del Parlamento Europeo del 27
Aprile 2016 e Decreto legislativo n. 101 del 10 Agosto 2018, si precisa che le
informazioni contenute in questo messaggio sono riservate ed a uso esclusivo
del destinatario. Qualora il messaggio in parola Le fosse pervenuto per errore,
la preghiamo di eliminarlo senza copiarlo e di non inoltrarlo a terzi,
dandocene gentilmente comunicazione. Grazie.

 

Disclaimer: this message may contain confidential
and/or privileged information, according to the provisions of law UE 2016/679
and Dlgs n. 101/2018. If you are not the addressee or authorized to receive
this for the addressee, you must not use, copy, disclose this information to
any other person or take any action based on this message. If you have received
this message in error, please advise the sender immediately by reply e-mail and
delete this message. Thank
you for your cooperation.

 

Contact ***@***.***

Data: Tue, 16 Jul 2024 10:22:53 -0700

morrownr, 2024-07-17:
@GCic8 

Got it. The priority right now is upstreaming the new rtw8821au and rtw8812au drivers. You are welcome to use my out-of-kernel driver for now if you want:

https://github.com/morrownr/88x2bu-20210702

dubhater, 2024-07-25:
You get "Operation not supported" because rtw88 doesn't advertise support for scanning in AP mode. You should ask the linux-wireless@vger.kernel.org for the reason why.