lwfinger/rtw88 #219: Newb help getting monitor mode going on a RTL8812AU?

7MinSec, 2024-07-16:
Hello,

I've got a Kali Linux VM running 2024.2, 6.8.11-amd64.  I've got a RTL8812AU hooked up to this VM which is physically connected to a Proxmox server (latest version).

What I'd like to do is get monitor mode turned on for this RTL8812AU but I'm not 100% sure this is supported.  In places like [this](https://github.com/aircrack-ng/rtl8812au/issues/1173) it seems that maybe there are known versions with my kernel version and I have to use a different OS.

But other searches led me to this project and suggested I try installing the driver.  I did:

```
sudo apt install make gcc linux-headers-$(uname -r) build-essential git
git clone https://github.com/lwfinger/rtw88
cd rtw88
sudo make
sudo make install
```

This all completed without errors:

```
make -C /lib/modules/6.8.11-amd64/build M=/opt/rtw88 modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.11-amd64'
make[1]: Leaving directory '/usr/src/linux-headers-6.8.11-amd64'
Install rtw88 SUCCESS
```
From here I'm not sure exactly what I'm supposed to do.  When I run the `lsmod` command I get:

```
└─$ lsmod | grep rtw
rtw_8812au             12288  0
rtw_8821a             114688  1 rtw_8812au
rtw_usb                28672  1 rtw_8812au
rtw_core              311296  2 rtw_usb,rtw_8821a
mac80211             1396736  2 rtw_usb,rtw_core
cfg80211             1355776  4 rtw_core,8812au,mac80211,88XXau
usbcore               409600  9 xhci_hcd,rtw_usb,usbhid,ehci_hcd,8812au,xhci_pci,rtw_8812au,88XXau,uhci_hcd
```
So should I put the `mac80211` and `cfg80211` and `usbcore` on the blocklist, and then run `sudo modprobe -r rtw_core` to get the card going?

7MinSec, 2024-07-19:
I *think* I got it working by adding these to my `/etc/modprobe.d/blacklist.conf`:

```
blacklist 8812au
blacklist 88XXau
blacklist rtl8xxxu
blacklist r8188eu
```

closed 2024-07-25 by dubhater

hrentambil, 2024-11-16:
Hello guys, anything helped ?  kali linux 24.3 is here with kernel 6.11.2

any ideas how to make it work?

a5a5aa555oo, 2024-11-16:
@hrentambil 

doesn't it work for you? any error message shown in dmesg?