lwfinger/rtw88 #221: rtl8812au   amd64/build: No such file or directory

iamAika, 2024-07-19:
Good morning,

I am experiencing some issues. I am using a VirtualBox VM 7.0 with Kali Linux (Kali GNU/Linux Rolling release 2024.2)and a Realtek 8812au NIC, with the kernel version 6.8.11-amd64. I took a snapshot when everything was configured and working well with Wi-Fi connectivity. When I reopened the VM, it had a Wi-Fi signal, and the driver shown with lsmod was "88XXau". However, after restarting the VM, the Wi-Fi connection disappeared, and ip show addr shows that wlan0 is no longer present. Similarly, lsmod no longer shows the driver.

I followed the instructions in the README.txt file precisely, including the part about blacklisting the previous driver "88xxau" before installing the rtw88 driver. Everything seemed to go well, but after rebooting the VM, lsmod does not show the new rtw88 driver (at this point, I have no Wi-Fi).

When attempting the kernel update steps:

cd ~/rtw88
git pull (i have no internet so i skip this part)
make (here is the error)
sudo make install (same if i try this)

I encounter the following error:
└─# make              
make -C /lib/modules/6.8.11-amd64/build M=/root/rtw88 modules
make[1]: *** /lib/modules/6.8.11-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:166: all] Error 2

a5a5aa555oo, 2024-07-19:
Before installing any out-of-tree driver, you have to install kernel headers first.
Use this command to install it and then try again.

`sudo apt install linux-headers-$(uname -r)`

iamAika, 2024-07-19:
It said that everything was updated. I resolved it by just deleting the instance and starting over again from scratch. I followed the README.txt steps, and it works fine. I really appreciate your help anyway. If this teaches me anything, it's that when you take a snapshot, do it with the PC off.

Thx!

closed 2024-07-25 by dubhater