lwfinger/rtw88 #224: Testing rtw_8822bu, feed back. (Slow bitrate)

WallyZambotti, 2024-07-25:
System information follows at end.

Testing the rtw_8822bu(Simplecom NW621 AC1200 USB2/3 Techkey Wifi Adapter) from rtw88 downstream.

Can confirm USB3 is now working.  

Having a bit rate issue on all systems testing exhibiting slow bit rate after initial connection.  The two systems:

1) Intel - Ubuntu 22.04.4 Kernel 6.5.0-44 - with rtw88 downstream driver - to 40mhz router - USB3 (system info included)
2) Intel - Ubuntu 22.04.4 Kernel 6.5.0-44 - with mainline rtw88 driver - 80mhz router - USB2

Both systems start with a high bite rate but after several seconds drop to around 135-180mbits and then never raise under net load.

Following is the system info for 1):

$ uname -a
Linux WZ-MS-7B51 6.5.0-44-generic #44~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Jun 18 14:36:16 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

$ lsusb 
Bus 002 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

$ lsusb  -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 10000M
    |__ Port 6: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 5000M

$ mokutil --sb-state
SecureBoot disabled
Platform is in Setup Mode

$ rfkill list all
1: phy3: Wireless LAN
	Soft blocked: no
	Hard blocked: no

$ dkms status
nvidia/555.42.06, 5.15.0-116-generic, x86_64: installed
nvidia/555.42.06, 6.5.0-44-generic, x86_64: installed
rtl8812au/4.3.8.12175.20140902+dfsg, 5.15.0-116-generic, x86_64: installed
rtl8812au/4.3.8.12175.20140902+dfsg, 6.5.0-44-generic, x86_64: installed
rtl8821ce/5.5.2.1, 5.15.0-116-generic, x86_64: installed

$ iwconfig
lo        no wireless extensions.

eno1      no wireless extensions.

wlxbcec4322ff20  IEEE 802.11  ESSID:"TCS"  
          Mode:Managed  Frequency:5.2 GHz  Access Point: 74:83:C2:B5:56:CA   
          Bit Rate=180 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=63/70  Signal level=-47 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:923   Missed beacon:0

$ iw reg get
global
country AU: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 36), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5600 @ 80), (N/A, 27), (0 ms), DFS
	(5650 - 5730 @ 80), (N/A, 27), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 36), (N/A)
	(5925 - 6425 @ 160), (N/A, 24), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 43), (N/A), NO-OUTDOOR

$ iw dev
phy#3
	Interface wlxbcec4322ff20
		ifindex 4
		wdev 0x300000001
		addr bc:ec:43:22:ff:20
		ssid TCS
		type managed
		channel 40 (5200 MHz), width: 40 MHz, center1: 5190 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00

$ iperf3 -c 192.168.10.152
Connecting to host 192.168.10.152, port 5201
[  5] local 192.168.10.104 port 52594 connected to 192.168.10.152 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  3.86 MBytes  32.3 Mbits/sec    1    284 KBytes       
[  5]   1.00-2.00   sec  9.64 MBytes  80.9 Mbits/sec    0    672 KBytes       
[  5]   2.00-3.00   sec  7.50 MBytes  62.9 Mbits/sec    0    942 KBytes       
[  5]   3.00-4.00   sec  8.75 MBytes  73.4 Mbits/sec    0   1007 KBytes       
[  5]   4.00-5.00   sec  6.25 MBytes  52.4 Mbits/sec    0   1.03 MBytes       
[  5]   5.00-6.00   sec  7.50 MBytes  62.9 Mbits/sec    0   1.16 MBytes       
[  5]   6.00-7.00   sec  7.50 MBytes  62.9 Mbits/sec    0   1.23 MBytes       
[  5]   7.00-8.00   sec  7.50 MBytes  62.9 Mbits/sec    0   1.23 MBytes       
[  5]   8.00-9.00   sec  7.50 MBytes  62.9 Mbits/sec    2    980 KBytes       
[  5]   9.00-10.00  sec  6.25 MBytes  52.4 Mbits/sec    0   1.04 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  72.2 MBytes  60.6 Mbits/sec    3             sender
[  5]   0.00-10.67  sec  69.4 MBytes  54.5 Mbits/sec                  receiver

$ cat /etc/modprobe.d/88x2bu.conf
blacklist rtw88_8822bu 
blacklist rtw88_usb
blacklist rtw88_8822b
blacklist rtw88_core 
options 88x2bu  rtw_switch_usb_mode=1 rtw_led_ctrl=1

$ lsmod | grep rtw
rtw_8822bu             12288  0
rtw_usb                28672  1 rtw_8822bu
rtw_8822b             233472  1 rtw_8822bu
rtw_core              356352  2 rtw_usb,rtw_8822b
mac80211             1724416  2 rtw_usb,rtw_core
cfg80211             1323008  2 rtw_core,mac80211

dubhater, 2024-07-25:
Thank you for your detailed report. Something is still wrong with this chip. Unfortunately I don't have a device to test.

morrownr, 2024-07-26:
@dubhater 

I have several usb wifi adapters based on the rtl8812bu chip. We can worry about how to put one in your hands as we have time but for now, tell me what you want me to do.

FYI: I've been sick this week. Am starting to feel a little better now.

morrownr, 2024-07-26:
@dubhater 

Afterthought: The rtl8812bu chip is likely the most popular Realtek chip in the hands of Linux users so it is a priority...once rtw_8821au and rtw_8812au have gone in.

morrownr, 2024-07-26:
@dubhater 

Test system:
x86_64
Debian 12
kernel 6.6
AP: 80 MHz channel width and supports WiFi 5 (actuall it supports WiFi 6 but this adapter in question can't do WiFi 6).
Channel 100 (DFS) clean air
signal level of -25 dBm, through 2 walls and about 6 meters distance
Adapter: rtl8812bu based running as client, managed mode
Driver: rtw_8822bu

Here are two runs with iperf3:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.135 port 35716 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  69.6 MBytes   584 Mbits/sec    0    976 KBytes       
[  5]   1.00-2.00   sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
[  5]   2.00-3.00   sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
[  5]   3.00-4.00   sec  66.2 MBytes   555 Mbits/sec    0    976 KBytes       
[  5]   4.00-5.00   sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
[  5]   5.00-6.00   sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
[  5]   6.00-7.00   sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
[  5]   7.00-8.00   sec  68.8 MBytes   577 Mbits/sec    0    976 KBytes       
[  5]   8.00-9.00   sec  67.5 MBytes   566 Mbits/sec    0    976 KBytes       
[  5]   9.00-10.00  sec  66.2 MBytes   556 Mbits/sec    0    976 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   673 MBytes   565 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   671 MBytes   563 Mbits/sec                  receiver

```

```
$ iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.135 port 39732 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  35.0 MBytes   293 Mbits/sec                  
[  5]   1.00-2.00   sec  7.04 MBytes  59.0 Mbits/sec                  
[  5]   2.00-3.00   sec  7.61 MBytes  63.9 Mbits/sec                  
[  5]   3.00-4.00   sec  7.53 MBytes  63.2 Mbits/sec                  
[  5]   4.00-5.00   sec  7.56 MBytes  63.5 Mbits/sec                  
[  5]   5.00-6.00   sec  6.06 MBytes  50.8 Mbits/sec                  
[  5]   6.00-7.00   sec  7.50 MBytes  62.9 Mbits/sec                  
[  5]   7.00-8.00   sec  7.48 MBytes  62.7 Mbits/sec                  
[  5]   8.00-9.00   sec  7.39 MBytes  62.0 Mbits/sec                  
[  5]   9.00-10.00  sec  7.59 MBytes  63.7 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.11  sec   103 MBytes  85.1 Mbits/sec   24             sender
[  5]   0.00-10.00  sec   101 MBytes  84.5 Mbits/sec                  receiver

```

Quick conclusion: Slow bitrate is only in one direction. Hopefully that is enough to help you know where to start looking. Let me know what specific tests to run.

@WallyZambotti 

> $ iwconfig

That is a very old unsupported command. Recommend you discontinue use. Tell us what you want to do and we can recommend modern supported commands. You can start with `iw`.

dubhater, 2024-07-26:
@morrownr Something to check is the TX and RX rates in wavemon when you see the low speed with iperf3.

morrownr, 2024-07-26:
@dubhater 

$ iperf3 -c 192.168.1.1

wavemon shows:  rx rate: 86 Mbps, tx rate: 866 Mbps while iperf3 is running.

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.135 port 59808 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  71.1 MBytes   597 Mbits/sec    0    974 KBytes       
[  5]   1.00-2.00   sec  66.2 MBytes   556 Mbits/sec    0    974 KBytes       
[  5]   2.00-3.00   sec  68.8 MBytes   577 Mbits/sec    0    974 KBytes       
[  5]   3.00-4.00   sec  66.2 MBytes   556 Mbits/sec    0    974 KBytes       
[  5]   4.00-5.00   sec  67.5 MBytes   566 Mbits/sec    0    974 KBytes       
[  5]   5.00-6.00   sec  66.2 MBytes   556 Mbits/sec    0    974 KBytes       
[  5]   6.00-7.00   sec  67.5 MBytes   566 Mbits/sec    0    974 KBytes       
[  5]   7.00-8.00   sec  67.5 MBytes   566 Mbits/sec    0    974 KBytes       
[  5]   8.00-9.00   sec  66.2 MBytes   556 Mbits/sec    0    974 KBytes       
[  5]   9.00-10.00  sec  67.5 MBytes   566 Mbits/sec    0    974 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   675 MBytes   566 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   672 MBytes   563 Mbits/sec                  receiver

```

-----

$ iperf3 -c 192.168.1.1 -R

wavemon shows:  rx rate: 19 Mbps, tx rate: 866 Mbps while iperf3 is running.

```
$ iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.135 port 36470 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  28.9 MBytes   243 Mbits/sec                  
[  5]   1.00-2.00   sec  4.91 MBytes  41.2 Mbits/sec                  
[  5]   2.00-3.00   sec  7.07 MBytes  59.3 Mbits/sec                  
[  5]   3.00-4.00   sec  7.51 MBytes  63.0 Mbits/sec                  
[  5]   4.00-5.00   sec  7.49 MBytes  62.8 Mbits/sec                  
[  5]   5.00-6.00   sec  7.52 MBytes  63.1 Mbits/sec                  
[  5]   6.00-7.00   sec  7.66 MBytes  64.3 Mbits/sec                  
[  5]   7.00-8.00   sec  7.40 MBytes  62.1 Mbits/sec                  
[  5]   8.00-9.00   sec  7.63 MBytes  64.0 Mbits/sec                  
[  5]   9.00-10.00  sec  7.59 MBytes  63.6 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  95.0 MBytes  79.7 Mbits/sec  134             sender
[  5]   0.00-10.00  sec  93.7 MBytes  78.6 Mbits/sec                  receiver

```

Samt43, 2024-07-27:
@morrownr @dubhater  : i'm also noticing slow bandwith using 8822bu.
Connecting to host 10.0.0.1, port 5201
[  5] local 100.115.92.198 port 60874 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  6.27 MBytes  52.6 Mbits/sec    3    211 KBytes       
[  5]   1.00-2.00   sec  9.13 MBytes  76.6 Mbits/sec    0    242 KBytes       
[  5]   2.00-3.00   sec  8.82 MBytes  74.0 Mbits/sec    1    266 KBytes       
[  5]   3.00-4.00   sec  8.70 MBytes  73.0 Mbits/sec    0    290 KBytes       
[  5]   4.00-5.00   sec  5.78 MBytes  48.5 Mbits/sec    2    235 KBytes       
[  5]   5.00-6.00   sec  9.13 MBytes  76.6 Mbits/sec    0    264 KBytes       
[  5]   6.00-7.00   sec  9.45 MBytes  79.2 Mbits/sec    0    280 KBytes       
[  5]   7.00-8.03   sec  11.8 MBytes  95.6 Mbits/sec    0    307 KBytes       
[  5]   8.03-9.00   sec  10.9 MBytes  94.3 Mbits/sec    0    334 KBytes       
[  5]   9.00-10.00  sec  10.4 MBytes  87.0 Mbits/sec    2    355 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  90.3 MBytes  75.8 Mbits/sec    8             sender
[  5]   0.00-10.04  sec  89.7 MBytes  75.0 Mbits/sec                  receiver

iperf Done.

WallyZambotti, 2024-07-29:
Bitrate continues to degrade over time:

$ iw wlxbcec4322ff20 link
Connected to 74:83:c2:b5:56:ca (on wlxbcec4322ff20)
	SSID: TCS
	freq: 5200
	RX: 12596516 bytes (52511 packets)
	TX: 558933 bytes (3372 packets)
	signal: -48 dBm
	rx bitrate: 65.0 MBit/s VHT-MCS 7 VHT-NSS 1
	tx bitrate: 135.0 MBit/s VHT-MCS 7 40MHz VHT-NSS 1

	bss flags:	short-slot-time
	dtim period:	3
	beacon int:	100

misha4gps, 2024-07-29:
Drivers comparison:
```
88x2bu (morrownr)
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-60.00  sec  1.21 GBytes   173 Mbits/sec                  receiver
iperf3: the client has terminated
-----------------------------------------------------------

rtw_8822bu (lwfinger):
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-30.00  sec   538 MBytes   150 Mbits/sec                  receiver
iperf3: the client has terminated
-----------------------------------------------------------

rtw88_8822bu (kernel)
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-35.00  sec   121 MBytes  29.0 Mbits/sec                  receiver
iperf3: the client has terminated
-----------------------------------------------------------
````
Hardware: 
```
Bus 001 Device 004: ID 0bda:b82c Realtek Semiconductor Corp. 802.11ac NIC (rtl8822bu)
```
Kernel:
```
Linux Workstation 6.10.0-rc7+ #5 SMP PREEMPT_DYNAMIC Fri Jul 19 20:42:15 EEST 2024 x86_64 x86_64 x86_64 GNU/Linux
```
Successful test of 802.11r:
```
88x2bu  - no
rtw_8822bu  - yes
rtw88_8822bu  - yes
```

**Question:**
1. What is the main reason for the rtw88_8822bu driver being so slow?
2. Why is the rtw_8822bu driver slower than 88x2bu ?

morrownr, 2024-07-29:
@misha4gps 

> What is the main reason for the rtw88_8822bu driver being so slow?

Probably the following is largely the cause:

[1/4] wifi: rtw88: Init RX burst length for 8822cu/8822bu/8821cu

You can find that patch series at:

https://patchwork.kernel.org/project/linux-wireless/list/

It was sent in yesterday by Dubhater. My opinion is that Realtek did not do a very good job with rtw88 and they included no support for USB. That has all come from the community. By the time we see kernel 6.12 released, you should see better performance with rtw88_8822bu and the other drivers in rtw88.

> Why is the rtw_8822bu driver slower than 88x2bu ?

Dubhater can certainly answer this better than me but I'd rather keep him busy prepping patches for upstreaming the numerous patches that need to go upstream such as for the new rtl8821au and rtl8812au drivers. It is a lot of work.

While performance for rtw_8822bu is currently much better than rtw88_8822bu, performance is not symmetrical as shown in the last iperf3 testing I posted above. Why that is is not known to me at this time but if anyone can get to the bottom of it, Dubhater can we need to allow him time to get all of the patches that need to go upstream on their way right now and then we can make a plan and start gathering information. This project covers a lot of drivers and it is complicated work. It will be very nice to see the many patches that are going into 6.12 in place so we can begin the process if depreciating the Reaktek out-of-kernel drivers.

misha4gps, 2024-07-29:
@morrownr 
Thanks for the link and explanation.
and many thanks to Dubhater for his efforts in improving the driver

morrownr, 2024-07-29:
@misha4gps 

Let me make sure I did not mislead you. Any patches that have gone in to go into the kernel are already in this repo. You can look at that patch series right here.

misha4gps, 2024-07-29:
@morrownr,
I understand this and that is why i did the test.

WallyZambotti, 2024-08-01:
Not sure if the following information is at all relevant.

The laptop that I was testing the rtl8822bu just happen to have a PCI rtl8822ce for its Wifi adapter.

When I use iw <dev> link it shows the RX bit rate is 175 but the TX bit rate is 600.  I would rather that be the other way around but there may be a wider issue with the rtl driver.

henkv1, 2024-08-01:
When I use rtw88 as an AP, my client reports a rx bitrate of 6.0MBit/s, tx bitrate is 390.0 MBit/s.
It makes no difference if I use my 8811cu or my 8812bu adapter as AP. My client uses an Intel wireless adapter. The AP is a Raspberry Pi 3b running 64bit Arch.

WallyZambotti, 2024-08-02:
From my testing it appears to drop under load.  So the bit rate may start at 216 or higher but immediately after placing a load it will drop to 39.  And when the load stops the bit rate will rise back up.

likknb, 2024-08-24:
yes,help the question please 。rtw88_8822cs According to the same situation

a5a5aa555oo, 2024-09-18:
I bought a RTL8822BU adapter a few days ago and did some tests, I found this problem only occurred when the adapter was plugged into a USB 3.0 port.

OS: Arch Linux (kernel version: 6.6.51-1-lts)
Driver: latest rtw88 in this repo

Results:
1. RTL8822BU connected to a USB 2.0 port
```
$ iperf3 -c 192.168.0.34 -t 180 -i 60
Connecting to host 192.168.0.34, port 5201
[  5] local 192.168.0.31 port 53818 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-60.06  sec  1.85 GBytes   265 Mbits/sec  1964    758 KBytes       
[  5]  60.06-120.06 sec  1.82 GBytes   261 Mbits/sec  1333    208 KBytes       
[  5] 120.06-180.06 sec  1.79 GBytes   256 Mbits/sec  1172    417 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-180.06 sec  5.46 GBytes   261 Mbits/sec  4469             sender
[  5]   0.00-180.27 sec  5.46 GBytes   260 Mbits/sec                  receiver

iperf Done.
$ iperf3 -c 192.168.0.34 -t 180 -i 60 -R
Connecting to host 192.168.0.34, port 5201
Reverse mode, remote host 192.168.0.34 is sending
[  5] local 192.168.0.31 port 50712 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-60.06  sec  1.87 GBytes   267 Mbits/sec                  
[  5]  60.06-120.06 sec  1.94 GBytes   278 Mbits/sec                  
[  5] 120.06-180.06 sec  1.89 GBytes   270 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-180.08 sec  5.70 GBytes   272 Mbits/sec  4510             sender
[  5]   0.00-180.06 sec  5.70 GBytes   272 Mbits/sec                  receiver

iperf Done.
```

2. RTL8822BU connected to a USB 3.0 port.
```
$ iperf3 -c 192.168.0.34 -t 180 -i 60
Connecting to host 192.168.0.34, port 5201
[  5] local 192.168.0.31 port 50418 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-60.06  sec  2.63 GBytes   376 Mbits/sec  4172    373 KBytes       
[  5]  60.06-120.06 sec  2.41 GBytes   345 Mbits/sec  1918    383 KBytes       
[  5] 120.06-180.06 sec  2.51 GBytes   359 Mbits/sec  1870    403 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-180.06 sec  7.55 GBytes   360 Mbits/sec  7960             sender
[  5]   0.00-180.29 sec  7.55 GBytes   360 Mbits/sec                  receiver

iperf Done.
$ iperf3 -c 192.168.0.34 -t 180 -i 60 -R
Connecting to host 192.168.0.34, port 5201
Reverse mode, remote host 192.168.0.34 is sending
[  5] local 192.168.0.31 port 42602 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-60.06  sec   501 MBytes  69.9 Mbits/sec                  
[  5]  60.06-120.06 sec   477 MBytes  66.7 Mbits/sec                  
[  5] 120.06-180.06 sec   471 MBytes  65.8 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-180.08 sec  1.42 GBytes  67.7 Mbits/sec  109             sender
[  5]   0.00-180.06 sec  1.41 GBytes  67.5 Mbits/sec                  receiver

iperf Done.
```

dubhater, 2024-09-18:
@a5a5aa555oo Is it the same if you turn off RX aggregation?

```diff
diff --git a/rtw8822b.c b/rtw8822b.c
index a4ad554..5636343 100644
--- a/rtw8822b.c
+++ b/rtw8822b.c
@@ -1642,7 +1642,7 @@ static void rtw8822b_rx_aggregation(struct rtw_dev *rtwdev, bool enable)
 {
 	u8 size, timeout;
 	u16 val16;
-
+enable = false;
 	rtw_write32_set(rtwdev, REG_RXDMA_AGG_PG_TH, BIT_EN_PRE_CALC);
 	rtw_write8_set(rtwdev, REG_TXDMA_PQ_MAP, BIT_RXDMA_AGG_EN);
 	rtw_write8_clr(rtwdev, REG_RXDMA_AGG_PG_TH + 3, BIT(7));
```

a5a5aa555oo, 2024-09-18:
After turning off RX aggregation, the result is a little different.

RTL8822BU connected to a USB 2.0 port
```
$ iperf3 -c 192.168.0.34 -t 120 -i 30
Connecting to host 192.168.0.34, port 5201
[  5] local 192.168.0.31 port 34450 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.03  sec  1005 MBytes   281 Mbits/sec    0   3.15 MBytes       
[  5]  30.03-60.03  sec   670 MBytes   187 Mbits/sec  4664    331 KBytes       
[  5]  60.03-90.03  sec   802 MBytes   224 Mbits/sec  567    410 KBytes       
[  5]  90.03-120.03 sec   601 MBytes   168 Mbits/sec  411    387 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.03 sec  3.01 GBytes   215 Mbits/sec  5642             sender
[  5]   0.00-120.24 sec  3.01 GBytes   215 Mbits/sec                  receiver

iperf Done.
[alonso@BM5220 rtw88] $ iperf3 -c 192.168.0.34 -t 120 -i 30 -R
Connecting to host 192.168.0.34, port 5201
Reverse mode, remote host 192.168.0.34 is sending
[  5] local 192.168.0.31 port 33536 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-30.03  sec  96.1 MBytes  26.9 Mbits/sec                  
[  5]  30.03-60.03  sec  96.5 MBytes  27.0 Mbits/sec                  
[  5]  60.03-90.03  sec  97.1 MBytes  27.2 Mbits/sec                  
[  5]  90.03-120.03 sec  96.5 MBytes  27.0 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.10 sec   389 MBytes  27.2 Mbits/sec   19             sender
[  5]   0.00-120.03 sec   386 MBytes  27.0 Mbits/sec                  receiver

iperf Done.
```

RTL8822BU connected to a USB 3.0 port
```
$ iperf3 -c 192.168.0.34 -t 120 -i 30
Connecting to host 192.168.0.34, port 5201
[  5] local 192.168.0.31 port 32860 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.03  sec  1.59 GBytes   454 Mbits/sec  2322    909 KBytes       
[  5]  30.03-60.03  sec  1.29 GBytes   370 Mbits/sec  1885    379 KBytes       
[  5]  60.03-90.03  sec  1.24 GBytes   356 Mbits/sec  910    641 KBytes       
[  5]  90.03-120.03 sec  1.14 GBytes   326 Mbits/sec  1348    566 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.03 sec  5.26 GBytes   377 Mbits/sec  6465             sender
[  5]   0.00-120.03 sec  5.26 GBytes   376 Mbits/sec                  receiver

iperf Done.
[alonso@BM5220 rtw88] $ iperf3 -c 192.168.0.34 -t 120 -i 30 -R
Connecting to host 192.168.0.34, port 5201
Reverse mode, remote host 192.168.0.34 is sending
[  5] local 192.168.0.31 port 53140 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-30.03  sec   947 MBytes   265 Mbits/sec                  
[  5]  30.03-60.03  sec   890 MBytes   249 Mbits/sec                  
[  5]  60.03-90.03  sec   850 MBytes   238 Mbits/sec                  
[  5]  90.03-120.03 sec   894 MBytes   250 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.09 sec  3.50 GBytes   250 Mbits/sec  183             sender
[  5]   0.00-120.03 sec  3.50 GBytes   250 Mbits/sec                  receiver

iperf Done.
```

I will test it one more time later to ensure this result is correct.
By the way, do you need one 8822BU adapter? Please add it to wishlist if you need. :)

morrownr, 2024-09-19:
> I bought a RTL8822BU adapter a few days ago and did some tests, I found this problem only occurred when the adapter was plugged into a USB 3.0 port.

Some relevant information in case this is not known. The rtl8822bu is a USB2 chipset. The rtl8812bu is a USB3 chipset.

The rtl8822bu includes BT support so USB support was limited to USB2.

dubhater, 2024-09-19:
@a5a5aa555oo That's an interesting result. I looked at the code again and I still don't know what's wrong.

> By the way, do you need one 8822BU adapter? Please add it to wishlist if you need. :)

Thank you, I added one to my wishlist.

@morrownr I guess people see the module name rtw_8822bu and assume the chip is also 8822.

a5a5aa555oo, 2024-09-19:
> @a5a5aa555oo That's an interesting result. I looked at the code again and I still don't know what's wrong.

@dubhater RTL8822BU works good enough in USB 2.0 mode for me, so this is not an urgent problem.
Please keep focusing on 8811AU/8812AU/8821AU. :)

> 
> > By the way, do you need one 8822BU adapter? Please add it to wishlist if you need. :)
> 
> Thank you, I added one to my wishlist.

I only see a Type-C GBE adapter and a 8812BU adapter.

@morrownr 
https://wikidevi.wi-cat.ru/Realtek
The wiki page above says 8822BU supports USB 2.0 only, but lsusb shows its speed is 5000M,
so i think 8822BU does supports USB3.0?
```
$ lsusb
Bus 009 Device 003: ID 0bda:b82c Realtek Semiconductor Corp. 802.11ac NIC
$ lsusb -t
/:  Bus 009.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 001: Dev 002, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
    |__ Port 002: Dev 003, If 0, Class=Wireless, Driver=btusb, 5000M
    |__ Port 002: Dev 003, If 1, Class=Wireless, Driver=btusb, 5000M
    |__ Port 002: Dev 003, If 2, Class=Vendor Specific Class, Driver=rtw_8822bu, 5000M

```

morrownr, 2024-09-19:
@a5a5aa555oo 

The rtw_8822BU driver supports both the rtl8822bu and rtl8812bu chips. Until @dubhater added USB3 support to rtw_8822BU recently, the driver did not support USB3. USB3 is only needed for the rtl8812bu chip. If you see 5000M, indicating USB3, then the chip is a rtl8812bu.

dubhater, 2024-09-19:
> I only see a Type-C GBE adapter and a 8812BU adapter.

I assume RTL8812BU has the same problem because that's what @morrownr was testing. It is the same chip as RTL8822BU, just without Bluetooth.

a5a5aa555oo, 2024-09-19:
@dubhater I tried the vendor driver in [morrow's repo](https://github.com/morrownr/88x2bu-20210702) and the speed looks fine.

RTL8822BU connected to an USB 2.0 port
```
$ iperf3 -c 192.168.0.34 -t 120 -i 30
Connecting to host 192.168.0.34, port 5201
[  5] local 192.168.0.31 port 32952 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.03  sec   897 MBytes   251 Mbits/sec    0   1.68 MBytes       
[  5]  30.03-60.03  sec   805 MBytes   225 Mbits/sec  2643    395 KBytes       
[  5]  60.03-90.02  sec   796 MBytes   223 Mbits/sec  347    406 KBytes       
[  5]  90.02-120.03 sec   781 MBytes   218 Mbits/sec  269    396 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.03 sec  3.20 GBytes   229 Mbits/sec  3259             sender
[  5]   0.00-120.09 sec  3.20 GBytes   229 Mbits/sec                  receiver

iperf Done.
$ iperf3 -c 192.168.0.34 -t 120 -i 30 -R
Connecting to host 192.168.0.34, port 5201
Reverse mode, remote host 192.168.0.34 is sending
[  5] local 192.168.0.31 port 49716 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-30.03  sec   883 MBytes   247 Mbits/sec                  
[  5]  30.03-60.03  sec   949 MBytes   265 Mbits/sec                  
[  5]  60.03-90.03  sec   960 MBytes   269 Mbits/sec                  
[  5]  90.03-120.03 sec   958 MBytes   268 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.04 sec  3.67 GBytes   262 Mbits/sec  304             sender
[  5]   0.00-120.03 sec  3.66 GBytes   262 Mbits/sec                  receiver

iperf Done.
```

RTL8822BU connected to an USB 3.0 port (module parameter: rtw_switch_usb_mode=1)
```
$ iperf3 -c 192.168.0.34 -t 120 -i 30
Connecting to host 192.168.0.34, port 5201
[  5] local 192.168.0.31 port 55344 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.03  sec  1.37 GBytes   392 Mbits/sec  2864    414 KBytes       
[  5]  30.03-60.03  sec  1.27 GBytes   363 Mbits/sec  966    641 KBytes       
[  5]  60.03-90.03  sec  1.21 GBytes   347 Mbits/sec  1245    445 KBytes       
[  5]  90.03-120.03 sec  1.29 GBytes   369 Mbits/sec  1058    656 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.03 sec  5.14 GBytes   368 Mbits/sec  6133             sender
[  5]   0.00-120.04 sec  5.14 GBytes   368 Mbits/sec                  receiver

iperf Done.
$ iperf3 -c 192.168.0.34 -t 120 -i 30 -R
Connecting to host 192.168.0.34, port 5201
Reverse mode, remote host 192.168.0.34 is sending
[  5] local 192.168.0.31 port 46888 connected to 192.168.0.34 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-30.03  sec   930 MBytes   260 Mbits/sec                  
[  5]  30.03-60.03  sec  1.08 GBytes   310 Mbits/sec                  
[  5]  60.03-90.03  sec   912 MBytes   255 Mbits/sec                  
[  5]  90.03-120.03 sec   953 MBytes   266 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.04 sec  3.82 GBytes   273 Mbits/sec  292             sender
[  5]   0.00-120.03 sec  3.81 GBytes   273 Mbits/sec                  receiver

iperf Done.
```

@morrownr 
I'm pretty sure my adapter has a 8822BU chip because i saw these messages about bluetooth in dmesg.
[... 18 lines cut ...]
[   12.810385] Bluetooth: hci0: RTL: rom_version status=0 version=2
[   12.810398] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[   12.811804] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
[   12.811934] Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270
[   13.554204] Bluetooth: hci0: RTL: fw version 0xab6b705c
[   47.240750] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=ab6b lmp_ver=08 lmp_subver=705c
[   47.759007] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[   47.760003] Bluetooth: hci0: RTL: rom_version status=0 version=2
[   47.760016] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[   47.761475] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
[   47.761593] Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270
[   48.523781] Bluetooth: hci0: RTL: fw version 0xab6b705c
[  804.307631] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[  804.308629] Bluetooth: hci0: RTL: rom_version status=0 version=2
[  804.308633] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[  804.309646] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
[  804.309713] Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270
[  805.102797] Bluetooth: hci0: RTL: fw version 0xab6b705c
[ 1033.265971] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[ 1033.267863] Bluetooth: hci0: RTL: rom_version status=0 version=2
[ 1033.267878] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[ 1033.269247] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
[ 1033.269365] Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270
[ 1034.091869] Bluetooth: hci0: RTL: fw version 0xab6b705c
```

dubhater, 2024-09-19:
Of course, the vendor driver works. I was referring to this test @morrownr did with rtw88 and a RTL8812BU device: https://github.com/lwfinger/rtw88/issues/224#issuecomment-2253539166

WallyZambotti, 2024-09-20:
I can confirm the receive slowdown issue only occurs when the device is plugged into a USB3 port.  When plugged into a USB2 it works better!!!  It's more than 3 times faster in the USB2 port.  But if you monitor the receive transmit rates they constantly drop and raise when used with USB2 but on USB3 the rate drops and stays down.

dubhater, 2024-10-01:
This bug is actually my fault. I tried to be smarter than the original code and simplified it, changing the order of operations in the process. I didn't notice the original code is a little broken. These chips need the broken behaviour:

```C
enum halmac_ret_status
cfg_usb_rx_agg_88xx(struct halmac_adapter *adapter,
		    struct halmac_rxagg_cfg *cfg)
{
	u8 dma_usb_agg;
	u8 size;
	u8 timeout;
	u8 agg_enable;
	u32 value32;
	struct halmac_api *api = (struct halmac_api *)adapter->halmac_api;

	PLTFM_MSG_TRACE("[TRACE]%s ===>\n", __func__);

	dma_usb_agg = HALMAC_REG_R8(REG_RXDMA_AGG_PG_TH + 3);                   // read byte 3
	agg_enable = HALMAC_REG_R8(REG_TXDMA_PQ_MAP);

	// ..... modify dma_usb_agg

	value32 = HALMAC_REG_R32(REG_RXDMA_AGG_PG_TH);
	if (cfg->threshold.size_limit_en == 0)
		HALMAC_REG_W32(REG_RXDMA_AGG_PG_TH, value32 & ~BIT_EN_PRE_CALC);
	else
		HALMAC_REG_W32(REG_RXDMA_AGG_PG_TH, value32 | BIT_EN_PRE_CALC); // modify byte 3

	HALMAC_REG_W8(REG_TXDMA_PQ_MAP, agg_enable);
	HALMAC_REG_W8(REG_RXDMA_AGG_PG_TH + 3, dma_usb_agg);                    // discard previous modification
	HALMAC_REG_W16(REG_RXDMA_AGG_PG_TH,
		       (u16)(size | (timeout << BIT_SHIFT_DMA_AGG_TO_V1)));

	PLTFM_MSG_TRACE("[TRACE]%s <===\n", __func__);

	return HALMAC_RET_SUCCESS;
}
```

I pushed the fix.

a5a5aa555oo, 2024-10-04:
@dubhater Thanks, download speed is much faster now!

8822BU (USB 2.0)
```
$ iperf3 -c 192.168.0.52 -t 120 -i 60 
Connecting to host 192.168.0.52, port 5201
[  5] local 192.168.0.109 port 48752 connected to 192.168.0.52 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-60.06  sec  1.88 GBytes   269 Mbits/sec    0   1.65 MBytes       
[  5]  60.06-120.06 sec  1.72 GBytes   246 Mbits/sec    0   1.65 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.06 sec  3.60 GBytes   258 Mbits/sec    0             sender
[  5]   0.00-120.08 sec  3.60 GBytes   257 Mbits/sec                  receiver

iperf Done.
$ iperf3 -c 192.168.0.52 -t 120 -i 60 -R
Connecting to host 192.168.0.52, port 5201
Reverse mode, remote host 192.168.0.52 is sending
[  5] local 192.168.0.109 port 39156 connected to 192.168.0.52 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-60.06  sec  1.78 GBytes   254 Mbits/sec                  
[  5]  60.06-120.06 sec  1.85 GBytes   265 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.06 sec  3.63 GBytes   260 Mbits/sec   17             sender
[  5]   0.00-120.06 sec  3.63 GBytes   260 Mbits/sec                  receiver

iperf Done.
```

8822BU (USB 3.0)
```
$ iperf3 -c 192.168.0.52 -t 120 -i 60 
Connecting to host 192.168.0.52, port 5201
[  5] local 192.168.0.110 port 59562 connected to 192.168.0.52 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-60.06  sec  1.54 GBytes   220 Mbits/sec    0   1.43 MBytes       
[  5]  60.06-120.06 sec  1.20 GBytes   171 Mbits/sec    0   1.43 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.06 sec  2.74 GBytes   196 Mbits/sec    0             sender
[  5]   0.00-120.09 sec  2.73 GBytes   196 Mbits/sec                  receiver

iperf Done.
$ iperf3 -c 192.168.0.52 -t 120 -i 60 -R
Connecting to host 192.168.0.52, port 5201
Reverse mode, remote host 192.168.0.52 is sending
[  5] local 192.168.0.110 port 53792 connected to 192.168.0.52 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-60.06  sec  2.86 GBytes   409 Mbits/sec                  
[  5]  60.06-120.06 sec  2.92 GBytes   418 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.07 sec  5.78 GBytes   413 Mbits/sec   16             sender
[  5]   0.00-120.06 sec  5.77 GBytes   413 Mbits/sec                  receiver

iperf Done.
```

a5a5aa555oo, 2024-10-04:
rx/tx rate in wavemon are still very low, does the driver report the wrong value?

[image]

dubhater, 2024-10-04:
I think the RX channel width is wrong sometimes (often).

a5a5aa555oo, 2024-10-07:
@dubhater I think this isn't a big problem and most users won't notice it.

WallyZambotti, 2024-10-11:
I can report that after the last updates the performance problem on the receiving channel appears to be fixed and my RX rates now much my TX rates.  114.5 Mbps download & 109.1 Mbps upload.  The router is fixed to 2.4 GHZ for range reasons so I can't test the full speed of this adapter.

WallyZambotti, 2024-10-22:
Sorry to inform but since the last update the adapter does not function at all any more.  It will connect for a few seconds (at the slowest possible rate) and then disconnect.  

```Connected to 74:83:c2:b4:56:ca (on wlxbcec4322ff20)
	SSID: TCS
	freq: 2462
	RX: 123560 bytes (875 packets)
	TX: 2986 bytes (18 packets)
	signal: -47 dBm
	rx bitrate: 11.0 MBit/s
	tx bitrate: 130.0 MBit/s MCS 15

	bss flags:	CTS-protection short-preamble short-slot-time
	dtim period:	1
	beacon int:	100
Connected to 74:83:c2:b4:56:ca (on wlxbcec4322ff20)
	SSID: TCS
	freq: 2462
command failed: No such file or directory (-2)
Not connected.
Not connected.
Not connected.
Not connected.
```

dubhater, 2024-10-22:
I noticed this too. It seems to happen only in the 2.4 GHz band, possibly only in USB 3 mode.

Was 2.4 GHz working okay before, other than having low speed?

WallyZambotti, 2024-10-23:
> I noticed this too. It seems to happen only in the 2.4 GHz band, possibly only in USB 3 mode.
> 
> Was 2.4 GHz working okay before, other than having low speed?

Yes. Perfectly!!!  Our dual band router is restricted to 2.4GHZ for coverage reasons.  I happen to be just a few meters from it, hence the excellent signal quality.

a5a5aa555oo, 2024-10-23:
> I noticed this too. It seems to happen only in the 2.4 GHz band, possibly only in USB 3 mode.
> 
> Was 2.4 GHz working okay before, other than having low speed?

I cannot reproduce this with 8822BU on my pc, it seems only 8812BU has this problem!?

morrownr, 2024-10-23:
> I cannot reproduce this with 8822BU on my pc, it seems only 8812BU has this problem!?

This actually makes sense in that adapters with a 8822bu chip are usb2 adapters. With bt turned on, keep in mind that I am not fully aware of the exact logic, usb will be limited to usb2. On the other hand, 8812bu chips can go into usb3 mode... so there is a difference.

dubhater, 2024-10-23:
@WallyZambotti Do you get the same problem with the 88x2bu driver with rtw_switch_usb_mode=1?

WallyZambotti, 2024-10-24:
Ok so after rechecking everything and going through another git pull (no changes); make; sudo make install; reboot. I'm not sure.

The modules loaded are:

```lsmod | grep rtw
rtw_8822bu             12288  0
rtw_usb                28672  1 rtw_8822bu
rtw_8822b             229376  1 rtw_8822bu
rtw_core              356352  2 rtw_usb,rtw_8822b
mac80211             1724416  2 rtw_usb,rtw_core
cfg80211             1323008  2 rtw_core,mac80211
```

If this is correct everything is working again.  Both USB2 and USB3.

However looking in the /etc/modprobe.d/88x2bu.conf there are the following lines:
```# The following line blacklists (deactivates) the above in-kernel driver.
blacklist rtw88_8822bu 
blacklist rtw88_usb
blacklist rtw88_8822b
blacklist rtw88_core 
#
```

Which appear to match the module s listed by lsmod.

So I'm confused!

a5a5aa555oo, 2024-10-24:
> However looking in the /etc/modprobe.d/88x2bu.conf there are the following lines:
> 
> ```
> blacklist rtw88_8822bu 
> blacklist rtw88_usb
> blacklist rtw88_8822b
> blacklist rtw88_core 
>```
> Which appear to match the module s listed by lsmod.
>
> So I'm confused!

@WallyZambotti 

Please notice the differences:  
rtw**88**_8822bu, rtw**88**_8822b, rtw**88**_usb and rtw**88**_core are the drivers provided by your kernel,
rtw_8822bu, rtw_8822b, rtw_usb and rtw_core are provided by this repo. Their names are a little different.

btw, you don't need these lines in your /etc/modprobe.d/88x2bu.conf because in-kernel rtw88 drivers will be blacklisted by
/etc/modprobe.d/blacklist-rtw88.conf which is installed during installation.

morrownr, 2024-10-24:
@WallyZambotti 

@a5a5aa555oo is correct. Now the question is why are the driver names different here in this repo than in the kernel? Right?

The change in names mostly has to do with the need not to overwrite the in-kernel drivers so that this repo can be installed and removed without modifying status quo. You will see a missing `88` in the drivers. rtw88_8822bu in the kernel is rtw_8822bu here in this repo. Since both drivers will be installed if you install this repo, there is a need to blacklist the in-kernel drivers while this repo is installed. If you uninstall this repo, the blacklist file is removed activating the in-kernel drivers.

Keep in mind that this repo is based on the in-kernel drivers... but with all of the work that it gets, it is more advanced than the in-kernel drivers. Basically this repo feeds the kernel. Tested here and then upstream things go.