lwfinger/rtw88 #226: Access Point stopped working with rtw_8822cu

hjmallon, 2024-07-31:
I am using rtw_8822cu (build from this repo on 610e04fc38343dcdcef95475c1579efc07572f1f)

I set up an access point with NetworkManager and connected from another machine and left iperf3 running. After about 4hrs35min the access point stopped responding. `dmesg` is full up with (repeating after the error for hours).

```
[  +3.968461] ------------[ cut here ]------------
[  +0.000004] WARNING: CPU: 5 PID: 36764 at [SNIP]/rtw88/rtw8822c.c:4425 rtw8822c_phy_cck_pd_set
+0x310/0x350 [rtw_8822c]
[  +0.000012] Modules linked in: nft_reject_ipv4 nft_masq nf_nat_h323 nf_conntrack_h323 nf_nat_pptp nf_conntrack_pptp nf_nat_tftp nf_conntrack_tftp nf_
nat_sip nf_conntrack_sip nf_nat_irc nf_conntrack_irc nf_nat_ftp nf_conntrack_ftp nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet
 nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_
defrag_ipv4 nf_tables nfnetlink algif_hash algif_skcipher af_alg exfat snd_soc_tegra186_asrc snd_soc_tegra210_mixer snd_soc_tegra210_dmic snd_soc_tegra
186_dspk snd_soc_tegra210_ope snd_soc_tegra210_admaif snd_soc_tegra_pcm snd_soc_tegra210_mvc snd_soc_tegra210_adx snd_soc_tegra210_amx snd_soc_tegra210
_sfc snd_soc_tegra210_i2s snd_soc_tegra210_ahub tegra210_adma snd_soc_tegra_audio_graph_card snd_soc_audio_graph_card snd_soc_simple_card_utils snd_hda
_codec_hdmi btusb tegra_aconnect btrtl btintel btbcm at24 bluetooth snd_hda_tegra snd_hda_codec
[  +0.000033]  tegra_xudc ecdh_generic pwm_tegra snd_hda_core ecc tegra_bpmp_thermal joydev fuse xfs libcrc32c dm_multipath r8169 sg rtw_8822cu(OE) rtw
_8822c(OE) rtw_usb(OE) rtw_core(OE) mac80211 8821cu(OE) libarc4 cfg80211 rfkill uas nvme governor_userspace atlantic tegra_drm(OE) st7789v(OE) nvidia(O
E) cec drm_mipi_dbi host1x_nvhost(OE) drm_kms_helper spidev tsecriscv(OE) nvgpu(OE) drm host1x_fence(OE) mc_utils(OE) nvmap(OE) ina3221 nvsciipc(OE) cr
ct10dif_ce pwm_fan tegra_dce(OE) nvhwpm(OE) nvme_tcp nvme_fabrics phy_tegra194_p2u nvme_core spi_tegra114 pcie_tegra194 nvidia_vrs_pseq(OE) host1x(OE) 
dm_mirror dm_region_hash dm_log iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi ipv6
[  +0.000033] CPU: 5 PID: 36764 Comm: kworker/u16:5 Tainted: G        W  OE     5.15.136-7.smartdock.el9.aarch64 #1
[  +0.000002] Hardware name: NVIDIA NVIDIA Jetson Orin NX Engineering Reference Developer Kit/Jetson, BIOS 36.3.0 07/17/2024
[  +0.000002] Workqueue: phy0 rtw_watch_dog_work [rtw_core]
[  +0.000013] pstate: 20400009 (nzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  +0.000002] pc : rtw8822c_phy_cck_pd_set+0x310/0x350 [rtw_8822c]
[  +0.000004] lr : rtw8822c_phy_cck_pd_set+0x100/0x350 [rtw_8822c]
[  +0.000003] sp : ffff80000dfbbc30
[  +0.000001] x29: ffff80000dfbbc80 x28: 0000000000000008 x27: ffff80000dfbbc73
[  +0.000003] x26: 0000000000000002 x25: 0000000000000076 x24: 0000000000000003
[  +0.000003] x23: 0000000000000001 x22: ffffaeedd1f9b010 x21: ffff00008d8c20a0
[  +0.000002] x20: 0000000000000002 x19: 0000000000000003 x18: 0000000000000000
[  +0.000002] x17: 000000040044ffff x16: ffffaeee16331c00 x15: 0000000000000000
[  +0.000003] x14: ffff000080249f00 x13: ffff5115d7a57000 x12: 00000000b464d91d
[  +0.000002] x11: 0000000000000000 x10: ffff80000dfbbc6e x9 : b75a936ca9b5028b
[  +0.000002] x8 : ffff0003d4894d58 x7 : 000000002fff9154 x6 : 0000000000000003
[  +0.000002] x5 : 0000000000000002 x4 : 0000000000000001 x3 : 0000000000000000
[  +0.000003] x2 : 00000000ffffffff x1 : 000000000000008c x0 : ffffaeed0002001a
[  +0.000002] Call trace:
[  +0.000001]  rtw8822c_phy_cck_pd_set+0x310/0x350 [rtw_8822c]
[  +0.000003]  rtw_phy_dynamic_mechanism+0x450/0x5b0 [rtw_core]
[  +0.000010]  rtw_watch_dog_work+0x204/0x340 [rtw_core]
[  +0.000009]  process_one_work+0x20c/0x4d0
[  +0.000009]  worker_thread+0x144/0x490
[  +0.000002]  kthread+0x184/0x190
[  +0.000003]  ret_from_fork+0x10/0x20
[  +0.000004] ---[ end trace c65e759c8ea43424 ]---
```

Other possibly interesting messages from dmesg and journalctl

```
Jul 30 23:05:44  NetworkManager[1187]: <info>  [1722377144.8850] device (wlan0): set-hw-addr: set MAC address to [SNIP] (scanning)
Jul 30 23:05:44  NetworkManager[1187]: <info>  [1722377144.9263] device (wlan0): supplicant interface state: inactive -> disconnected
Jul 30 23:05:44  NetworkManager[1187]: <info>  [1722377144.9264] device (p2p-dev-wlan0): supplicant management interface state: inactive -> disconnected
Jul 30 23:05:48  wpa_supplicant[1363]: wlan0: Reject scan trigger since one is already pending
Jul 30 23:05:48  NetworkManager[1187]: <info>  [1722377148.6928] device (wlan0): supplicant interface state: disconnected -> inactive
Jul 30 23:05:48  NetworkManager[1187]: <info>  [1722377148.6929] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> inactive
Jul 30 23:05:54  kernel: rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware
Jul 30 23:05:58  wpa_supplicant[1363]: wlan1: AP-STA-DISCONNECTED [SNIP]
Jul 30 23:05:59  kernel: rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware
Jul 30 23:05:59 wpa_supplicant[1363]: wlan0: Reject scan trigger since one is already pending
Jul 30 23:06:00  kernel: rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware
Jul 30 23:06:16 wpa_supplicant[1363]: wlan0: Reject scan trigger since one is already pending
```

The bandwith drops to 0 at 23:02, but the warn messages only start at 04:31

dubhater, 2024-07-31:
Can I see the full journal, please?

hjmallon, 2024-08-01:
This was with iperf3 -s running on the AP 8822cu, so with data going from the client to the AP (which seems much faster than vice versa).

I will try to reproduce and grab the full logs, I think they have gone off the top.

hjmallon, 2024-08-02:
OK I was able to reproduce it. SSID is 'CCAP2'

* Test started at 16:27:07 1st Aug (stable bandwidth ~40Mbit/s
* Bandwith went to 0 bit/s between 23:57:08 and 23:58:08 1st Aug
* Network connection became disconnected at 00:12:44 2nd August

[client-side-log.txt](https://github.com/user-attachments/files/16467655/client-side-log.txt)

This time there seems to be hardly anything in dmesg on the server side! This may be due to ntp / clock shenanigans (there is not real time clock on this board so it seems to boot widly out).

```
[Aug 1 21:34] ------------[ cut here ]------------
[  +0.000007] Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0
[  +0.000051] WARNING: CPU: 7 PID: 50 at net/mac80211/rx.c:4937 ieee80211_rx_list+0x41c/0xa00 [mac80211]
[  +0.000047] Modules linked in: nft_reject_ipv4 nft_masq nf_nat_h323 nf_conntrack_h323 nf_nat_pptp nf_conntrack_pptp nf_nat_tftp nf_conntrack_tftp nf_nat_sip nf_conntrack_sip nf_nat_irc nf_conntrack_irc nf_nat_ftp nf_conntrack_ftp nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nf_log_syslog nft_log nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink algif_hash algif_skcipher af_alg exfat snd_soc_tegra186_asrc snd_soc_tegra210_admaif snd_soc_tegra_pcm snd_soc_tegra210_ope snd_soc_tegra210_mixer snd_soc_tegra210_mvc snd_soc_tegra210_sfc snd_soc_tegra210_adx snd_soc_tegra186_dspk snd_soc_tegra210_dmic snd_soc_tegra210_amx snd_soc_tegra210_i2s snd_soc_tegra210_ahub tegra210_adma snd_hda_codec_hdmi btusb snd_soc_tegra_audio_graph_card btrtl btintel snd_soc_audio_graph_card snd_soc_simple_card_utils btbcm bluetooth tegra_aconnect
[  +0.000041]  snd_hda_tegra ecdh_generic ecc snd_hda_codec at24 snd_hda_core tegra_xudc pwm_tegra tegra_bpmp_thermal joydev fuse xfs libcrc32c dm_multipath sg rtw_8822cu(OE) rtw_8822c(OE) rtw_usb(OE) rtw_core(OE) mac80211 libarc4 cfg80211 rfkill uas r8169 nvme governor_userspace tegra_drm(OE) cec st7789v(OE) nvidia(OE) drm_mipi_dbi nvgpu(OE) host1x_nvhost(OE) tsecriscv(OE) host1x_fence(OE) mc_utils(OE) drm_kms_helper atlantic drm spidev nvmap(OE) crct10dif_ce pwm_fan nvsciipc(OE) nvhwpm(OE) tegra_dce(OE) ina3221 nvme_tcp nvme_fabrics phy_tegra194_p2u nvme_core spi_tegra114 pcie_tegra194 nvidia_vrs_pseq(OE) host1x(OE) dm_mirror dm_region_hash dm_log iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi ipv6
[  +0.000050] CPU: 7 PID: 50 Comm: ksoftirqd/7 Tainted: G           OE     5.15.136-7.smartdock.el9.aarch64 #1
[  +0.000003] Hardware name: NVIDIA NVIDIA Jetson Orin NX Engineering Reference Developer Kit/Jetson, BIOS 36.3.0-9d299ecd 07/17/2024
[  +0.000001] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  +0.000003] pc : ieee80211_rx_list+0x41c/0xa00 [mac80211]
[  +0.000027] lr : ieee80211_rx_list+0x41c/0xa00 [mac80211]
[  +0.000024] sp : ffff8000084bbbb0
[  +0.000001] x29: ffff8000084bbc20 x28: 0000000000000001 x27: 0000000000000000
[  +0.000003] x26: ffffc3c3ec55f858 x25: ffff0000c17d0039 x24: 0000000000000002
[  +0.000002] x23: ffff000092558820 x22: ffff00008a253f28 x21: ffff8000084bbc88
[  +0.000002] x20: 0000000000000000 x19: ffff00008a253f00 x18: ffffffffffffffff
[  +0.000002] x17: 000000040044ffff x16: ffffc3c3eb2d562c x15: ffffc3c3ed03d9e7
[  +0.000002] x14: 0000000000000000 x13: ffffc3c3ed03d9e4 x12: 696c61766e692073
[  +0.000002] x11: 6920617461642074 x10: 000000000000000a x9 : 6164207475622065
[  +0.000002] x8 : 000000000000000a x7 : 612073612064656b x6 : 000000000000000a
[  +0.000002] x5 : ffff0003f02be9f0 x4 : 00000000fffff56a x3 : ffffc3c3ecc6c028
[  +0.000002] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff0000803c4d80
[  +0.000002] Call trace:
[  +0.000001]  ieee80211_rx_list+0x41c/0xa00 [mac80211]
[  +0.000025]  ieee80211_rx_napi+0x68/0xf0 [mac80211]
[  +0.000024]  ieee80211_tasklet_handler+0xf0/0xf4 [mac80211]
[  +0.000027]  tasklet_action_common.constprop.0+0x110/0x140
[  +0.000008]  tasklet_action+0x34/0x40
[  +0.000002]  __do_softirq+0x128/0x3c0
[  +0.000002]  run_ksoftirqd+0x58/0x70
[  +0.000003]  smpboot_thread_fn+0x23c/0x264
[  +0.000004]  kthread+0x184/0x190
[  +0.000003]  ret_from_fork+0x10/0x20
[  +0.000002] ---[ end trace 370da9fae80cb196 ]---
[Aug 2 00:02] rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware
[  +3.007930] rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware
[  +0.991975] rtw_8822cu 1-2.4:1.2: failed to get tx report from firmware

```

dubhater, 2024-08-02:
> [  +0.000007] Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0

Interesting. I saw that once, with RTL8812AU. I wonder if this is another bug that affects all the chips.

hjmallon, 2024-08-07:
I am running a test with the `rtl88x2cu` (`v5.15.8-52`) driver to compare, immediately I notice that I am getting ~70Mbit/s rather than ~40Mbit/s with `rtw88`. I will test the reliability over the next 24hrs.

dubhater, 2025-05-03:
It may be worth testing again. Things have changed since August.