lwfinger/rtw88 #227: Realtek 8822cu device no longer supports dual interface after installing rtw88

murrellr, 2024-07-31:
Our system is using an LM Technologies LM842 WiFi USB dongle for WiFi.  It uses the Realtek 8822CU chipset.  Ever since we updated the Linux kernel from 4.14 to 6.2, dual STA+AP has not worked.  In desperation, I built and installed rtw88 on our device, but it did not help.  I removed rtw88 and went back to the standard kernel drivers and firmware.  Before I installed rtw88, the command "iw phy phy0 info" would show something like this:
`        valid interface combinations:
                 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
                   total <= 3, #channels <= 2
`
This is output from a different device.  Note the "#channels <=2" line.  Now this command shows this:
`        interface combinations are not supported`
I have tried everything I could think of to restore this capability, even reflashing the device with an older image.  Is there any way to recover the interface options?

dubhater, 2024-07-31:
Indeed, rtw88 did not support interface combinations until kernel 6.4. To restore functionality quickly and reliably, you should keep using the driver you were using with kernel 4.14. You can find the various compilation fixes here: https://github.com/morrownr/88x2bu-20210702/commits/main/ (That's for RTL8822BU, very similar codebase to RTL8822CU.)

rtw88 still has problems in AP mode with all USB devices. These are being fixed now. It also has some unique problems with RTL8822CU. These may get fixed soon.

murrellr, 2024-08-02:
How do I continue using the old driver?  It won't load under Linux 6.2 (Invalid module format) and it won't build with the Linux 6.2 headers.

dubhater, 2024-08-02:
Of course, you can't use the same binary, so you have to fix the compilation errors. I'm pretty sure you can find all the changes you need to make in the commit history I linked above.