lwfinger/rtw88 #229: rtl8812au crashed: Rate marked as a VHT rate but data is invalid

darkblaze69, 2024-08-10:
```

[  968.228325] rtw_8812au 4-4:1.0: failed to get urb length:32768
[  992.191913] rtw_8812au 4-4:1.0: failed to get urb length:32768
[  992.192005] ------------[ cut here ]------------
[  992.192011] Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0
[  992.192057] WARNING: CPU: 3 PID: 40 at net/mac80211/rx.c:5405 ieee80211_rx_list+0x2b5/0xe90 [mac80211]
[  992.192188] Modules linked in: xt_conntrack xt_MASQUERADE xt_tcpudp xt_mark snd_seq_dummy snd_seq_midi snd_hrtimer snd_seq_midi_event snd_seq tun nf_tables ip6table_nat ip6table_filter ip6_tables iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter bridge stp llc ccm algif_aead crypto_null des3_ede_x86_64 cbc des_generic libdes algif_skcipher cmac md4 algif_hash af_alg vfat fat intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm crct10dif_pclmul crc32_pclmul snd_ice1712 snd_hda_codec_hdmi polyval_clmulni snd_cs8427 polyval_generic snd_i2c snd_hda_intel snd_ice17xx_ak4xxx ghash_clmulni_intel snd_ak4xxx_adda sha512_ssse3 rtw_8812au(OE) rtw_8821a(OE) snd_ac97_codec eeepc_wmi snd_intel_dspcfg sha1_ssse3 asus_wmi aesni_intel rtw_usb(OE) snd_intel_sdw_acpi iTCO_wdt gf128mul platform_profile crypto_simd snd_mpu401_uart rtw_core(OE) snd_hda_codec intel_pmc_bxt at24 mei_pxp mei_hdcp iTCO_vendor_support snd_rawmidi cryptd mac80211 rapl snd_seq_device snd_hda_core
[  992.192271]  i8042 r8169 ac97_bus i2c_i801 intel_cstate snd_hwdep sparse_keymap realtek libarc4 i2c_smbus snd_pcm serio wmi_bmof intel_uncore pcspkr mei_me mdio_devres cfg80211 i2c_mux lpc_ich snd_timer snd mei libphy rfkill soundcore joydev mousedev mac_hid pkcs8_key_parser sg crypto_user loop dm_mod nfnetlink zram ip_tables x_tables uas usb_storage hid_logitech_hidpp hid_logitech_dj hid_generic btrfs nouveau blake2b_generic libcrc32c crc32c_generic xor raid6_pq drm_ttm_helper gpu_sched nvme drm_gpuvm drm_exec crc32c_intel nvme_core mxm_wmi sha256_ssse3 i2c_algo_bit usbhid drm_buddy ttm intel_gtt sr_mod nvme_auth drm_display_helper cdrom xhci_pci xhci_pci_renesas video cec wmi
[  992.192347] CPU: 3 UID: 0 PID: 40 Comm: ksoftirqd/3 Tainted: G           OE      6.11.0-rc2-1-mainline #1 bc5b0bde4860f996927096ab1dc6dfbcf0f3d55d
[  992.192354] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[  992.192356] Hardware name: System manufacturer System Product Name/P8H77-V LE, BIOS 1307 03/17/2014
[  992.192358] RIP: 0010:ieee80211_rx_list+0x2b5/0xe90 [mac80211]
[  992.192465] Code: 0f 86 45 06 00 00 80 3d ff bc 08 00 00 0f 85 2f ff ff ff 0f b6 53 4a 48 c7 c7 70 5e 43 c1 c6 05 e7 bc 08 00 01 e8 6b 44 39 d7 <0f> 0b e9 11 ff ff ff 40 80 fe 0b 0f 86 8f 05 00 00 80 3d c9 bc 08
[  992.192468] RSP: 0018:ffffa3bc80183cc8 EFLAGS: 00010282
[  992.192471] RAX: 0000000000000000 RBX: ffff94cac439e400 RCX: 0000000000000027
[  992.192474] RDX: ffff94cfd7ba1a48 RSI: 0000000000000001 RDI: ffff94cfd7ba1a40
[  992.192476] RBP: ffff94cac7928900 R08: 0000000000000000 R09: ffffa3bc80183b48
[  992.192478] R10: ffffffff9a6b3fc8 R11: 0000000000000003 R12: 0000000000000000
[  992.192480] R13: 0000000000000001 R14: ffff94cac439e400 R15: ffffa3bc80183dc0
[  992.192482] FS:  0000000000000000(0000) GS:ffff94cfd7b80000(0000) knlGS:0000000000000000
[  992.192485] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  992.192488] CR2: 000055c7222ba010 CR3: 0000000200622006 CR4: 00000000001706f0
[  992.192491] Call Trace:
[  992.192495]  <TASK>
[  992.192498]  ? ieee80211_rx_list+0x2b5/0xe90 [mac80211 a6aeafbf640093ee187d3e10e33cb425f3c62837]
[  992.192601]  ? __warn.cold+0x8e/0xe8
[  992.192607]  ? ieee80211_rx_list+0x2b5/0xe90 [mac80211 a6aeafbf640093ee187d3e10e33cb425f3c62837]
[  992.192709]  ? report_bug+0xff/0x140
[  992.192715]  ? console_unlock+0x84/0x130
[  992.192722]  ? handle_bug+0x3c/0x80
[  992.192727]  ? exc_invalid_op+0x17/0x70
[  992.192731]  ? asm_exc_invalid_op+0x1a/0x20
[  992.192737]  ? ieee80211_rx_list+0x2b5/0xe90 [mac80211 a6aeafbf640093ee187d3e10e33cb425f3c62837]
[  992.192834]  ? ieee80211_rx_list+0x2b5/0xe90 [mac80211 a6aeafbf640093ee187d3e10e33cb425f3c62837]
[  992.192931]  ? sched_clock+0x10/0x30
[  992.192935]  ? sched_clock_cpu+0xf/0x1d0
[  992.192940]  ? raw_spin_rq_lock_nested+0x1c/0x80
[  992.192944]  ? psi_group_change+0x12a/0x300
[  992.192951]  ieee80211_rx_napi+0x54/0xe0 [mac80211 a6aeafbf640093ee187d3e10e33cb425f3c62837]
[  992.193048]  ? finish_task_switch.isra.0+0x99/0x2e0
[  992.193053]  ieee80211_handle_queued_frames+0xbe/0xd0 [mac80211 a6aeafbf640093ee187d3e10e33cb425f3c62837]
[  992.193138]  tasklet_action_common.isra.0+0xc0/0x230
[  992.193144]  handle_softirqs+0xe4/0x2a0
[  992.193148]  ? __pfx_smpboot_thread_fn+0x10/0x10
[  992.193153]  run_ksoftirqd+0x2f/0x40
[  992.193156]  smpboot_thread_fn+0xda/0x1d0
[  992.193162]  kthread+0xcf/0x100
[  992.193168]  ? __pfx_kthread+0x10/0x10
[  992.193174]  ret_from_fork+0x31/0x50
[  992.193178]  ? __pfx_kthread+0x10/0x10
[  992.193183]  ret_from_fork_asm+0x1a/0x30
[  992.193190]  </TASK>
[  992.193192] ---[ end trace 0000000000000000 ]---
[  993.520952] rtw_8812au 4-4:1.0: failed to get urb length:32768
[  994.241539] rtw_8812au 4-4:1.0: failed to get urb length:32768
[  999.995747] rtw_8812au 4-4:1.0: failed to get urb length:32768
[ 1021.027746] rtw_8812au 4-4:1.0: failed to get urb length:32768
[ 1025.469715] rtw_8812au 4-4:1.0: failed to get urb length:32768
[ 1945.519769] rtw_8812au 4-4:1.0: failed to get urb length:32768
[ 2990.919228] rtw_8812au 4-4:1.0: failed to get urb length:32768
[ 2991.249746] rtw_8812au 4-4:1.0: failed to get urb length:32768
```

dubhater, 2024-08-10:
Do you often see "Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0"? I have only seen it once.

What about "failed to get urb length:32768"? I have never seen that on my computer.

If you don't have to wait weeks for them to appear again, I would ask you to apply some patches in order to gather more information.

Does the wifi connection still work when these messages appear?

LuisMitaHL, 2024-08-19:
Hi, I have the same problem on latest rtw88 git with Arch Linux + linux-lts.

I have two APs:
- Wi-Fi 6, ch 149, 40MHz bw -- No problems
- Wi-Fi 6, ch 52, 80MHz bw -- Bug appears, I get dmesg spammed with `rtw_8812au 2-1:1.0: failed to get urb length:32768`

I can reliably reproduce the bug starting `iperf3 -s` somewhere on the network, and executing `iperf3 -c server.ip -t 30 -R` on the machine with the rtl8812au adapter.

LuisMitaHL, 2024-08-19:
Full report:

```
[  187.494405] wlan0: associated
[  187.563562] wlan0: Limiting TX power to 20 (30 - 10) dBm as advertised by 80:2a:a8:xx:xx:xx
[  192.549552] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  192.621840] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  192.641487] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  192.659191] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  193.031068] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  193.198498] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  193.332539] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  193.496463] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  193.639115] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  193.815311] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  194.017287] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  194.169972] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  194.338895] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  194.493368] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  194.493432] ------------[ cut here ]------------
[  194.493436] Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0
[  194.493487] WARNING: CPU: 1 PID: 26 at net/mac80211/rx.c:5344 ieee80211_rx_list+0x2b5/0xea0 [mac80211]
[  194.493636] Modules linked in: sctp ip6_udp_tunnel udp_tunnel cmac rtw_8812au(OE) rtw_8821a(OE) rtw_usb(OE) rtw_core(OE) ccm bnep overlay snd_seq_dummy snd_hrtimer snd_seq snd_seq_device intel_soc_dts_thermal intel_soc_dts_iosf intel_powerclamp coretemp kvm_intel ath3k btusb kvm spi_nor uvcvideo btrtl snd_hda_codec_hdmi mtd videobuf2_vmalloc btintel ath9k_common irqbypass crct10dif_pclmul snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio crc32_pclmul btbcm ath9k_hw uvc joydev btmtk spi_intel_platform mousedev polyval_generic videobuf2_memops at24 iTCO_wdt gf128mul videobuf2_v4l2 ath ghash_clmulni_intel intel_pmc_bxt acer_wmi spi_intel iTCO_vendor_support mac80211 mei_hdcp hid_multitouch mei_pxp sparse_keymap wmi_bmof intel_rapl_msr videodev cryptd bluetooth sha512_ssse3 snd_hda_intel sha1_ssse3 i915 snd_intel_dspcfg intel_cstate libarc4 videobuf2_common snd_intel_sdw_acpi pcspkr drm_buddy i2c_i801 r8169 i2c_algo_bit ecdh_generic mc realtek i2c_smbus ttm snd_hda_codec cfg80211 crc16 snd_hda_core
[  194.493753]  drm_display_helper rfkill mdio_devres lpc_ich snd_hwdep cec snd_pcm snd_timer libphy intel_gtt snd mei_txe mei soundcore int3401_thermal i2c_hid_acpi i2c_hid processor_thermal_device vfat dw_dmac processor_thermal_rfim fat video processor_thermal_mbox wmi processor_thermal_rapl int3403_thermal pwm_lpss_platform intel_rapl_common int3400_thermal acpi_thermal_rel int340x_thermal_zone pwm_lpss mac_hid pkcs8_key_parser i2c_dev crypto_user loop fuse dm_mod nfnetlink bpf_preload ip_tables x_tables serio_raw atkbd libps2 vivaldi_fmap sha256_ssse3 i8042 xhci_pci xhci_pci_renesas serio sdhci_acpi sdhci mmc_core btrfs blake2b_generic libcrc32c crc32c_generic crc32c_intel xor raid6_pq [last unloaded: ath9k]
[  194.493839] CPU: 1 PID: 26 Comm: ksoftirqd/1 Tainted: G           OE      6.6.46-1-lts #1 180e79e37528e3108a25b78d18bc12bb7965e9b7
[  194.493846] Hardware name: Acer Aspire ES1-111M/R2, BIOS V1.16 10/20/2015
[  194.493848] RIP: 0010:ieee80211_rx_list+0x2b5/0xea0 [mac80211]
[  194.493970] Code: 0f 86 4d 06 00 00 80 3d ac 95 08 00 00 0f 85 31 ff ff ff 0f b6 53 4a 48 c7 c7 78 b2 2c c1 c6 05 94 95 08 00 01 e8 fb a5 6e ee <0f> 0b e9 13 ff ff ff 40 80 fe 0b 0f 86 97 05 00 00 80 3d 76 95 08
[  194.493973] RSP: 0018:ffffc9000011fcd8 EFLAGS: 00010282
[  194.493978] RAX: 0000000000000000 RBX: ffff8881017f8700 RCX: 0000000000000027
[  194.493981] RDX: ffff888277ca16c8 RSI: 0000000000000001 RDI: ffff888277ca16c0
[  194.493983] RBP: ffff8881d09c8900 R08: 0000000000000000 R09: ffffc9000011fb48
[  194.493986] R10: ffffffffb16b23e8 R11: 0000000000000003 R12: 0000000000000000
[  194.493988] R13: 0000000000000001 R14: ffff8881017f8700 R15: ffffc9000011fdd0
[  194.493992] FS:  0000000000000000(0000) GS:ffff888277c80000(0000) knlGS:0000000000000000
[  194.493995] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  194.493998] CR2: 00007fad6a0b2740 CR3: 00000001799b2000 CR4: 00000000001006e0
[  194.494001] Call Trace:
[  194.494005]  <TASK>
[  194.494008]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 f6a0049740f21ea8fef6adac16fa3f2582a87e99]
[  194.494130]  ? __warn+0x81/0x130
[  194.494141]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 f6a0049740f21ea8fef6adac16fa3f2582a87e99]
[  194.494264]  ? report_bug+0x16f/0x1a0
[  194.494271]  ? handle_bug+0x3c/0x80
[  194.494275]  ? exc_invalid_op+0x17/0x70
[  194.494280]  ? asm_exc_invalid_op+0x1a/0x20
[  194.494286]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 f6a0049740f21ea8fef6adac16fa3f2582a87e99]
[  194.494413]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 f6a0049740f21ea8fef6adac16fa3f2582a87e99]
[  194.494543]  ? update_load_avg+0x7e/0x760
[  194.494550]  ? psi_group_change+0x21e/0x3d0
[  194.494555]  ? dequeue_entity+0x138/0x4e0
[  194.494562]  ieee80211_rx_napi+0x54/0xf0 [mac80211 f6a0049740f21ea8fef6adac16fa3f2582a87e99]
[  194.494688]  ? __switch_to_asm+0x3e/0x70
[  194.494695]  ieee80211_handle_queued_frames+0xbe/0xd0 [mac80211 f6a0049740f21ea8fef6adac16fa3f2582a87e99]
[  194.494809]  tasklet_action_common.isra.0+0xc3/0x230
[  194.494817]  handle_softirqs+0xe5/0x2e0
[  194.494824]  ? __pfx_smpboot_thread_fn+0x10/0x10
[  194.494829]  run_ksoftirqd+0x31/0x40
[... cut ...]
[  605.052094] R10: 0000000000000011 R11: 0000000000000011 R12: ffff888110be9800
[  605.052097] R13: 0000000000000048 R14: 0000000000000000 R15: ffff8881d09ca080
[  605.052101] FS:  0000000000000000(0000) GS:ffff888277c80000(0000) knlGS:0000000000000000
[  605.052105] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  605.052109] CR2: 0000000000000010 CR3: 00000001fe820000 CR4: 00000000001006e0
[  605.052112] note: kworker/u8:2[34] exited with irqs disabled
[  606.937777] rtw_8812au 2-1:1.0: failed to get tx report from firmware
```

dubhater, 2024-08-19:
@LuisMitaHL Hi! I have some questions and a patch for you.

First, let's make sure you are using only the right modules. ``lsmod | grep rtw88`` should return nothing.

What speed is iperf3 showing when you get these errors?

What is the brand and model of your wifi dongle?

Are you plugging it into a USB 2 or USB 3 port?

No matter what port you used, please run this command and then plug it again and run iperf3: ``echo 0 > /sys/module/rtw_usb/parameters/switch_usb_mode``. Do you still get the errors?

And now the patch. Make sure to reload all the modules after you recompile: rtw_8812au, rtw_8821a, rtw_usb, and rtw_core.

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/usb.c b/drivers/net/wireless/realtek/rtw88/usb.c
index bf55360f9daf..2ade04c6414c 100644
--- a/usb.c
+++ b/usb.c
@@ -639,7 +639,7 @@ static void rtw_usb_read_port_complete(struct urb *urb)
 	struct sk_buff *skb = rxcb->rx_skb;
 
 	if (urb->status == 0) {
-		if (urb->actual_length >= RTW_USB_MAX_RECVBUF_SZ ||
+		if (urb->actual_length > RTW_USB_MAX_RECVBUF_SZ ||
 		    urb->actual_length < 24) {
 			rtw_err(rtwdev, "failed to get urb length:%d\n",
 				urb->actual_length);
```

Hopefully this will get rid of the "failed to get urb length:32768" errors without causing other problems.

LuisMitaHL, 2024-08-19:
> First, let's make sure you are using only the right modules. `lsmod | grep rtw88` should return nothing.

Yes, the command does return nothing.

> What speed is iperf3 showing when you get these errors?

```
 ~ % iperf3 -c 192.168.20.20 -t 90 -R
Connecting to host 192.168.20.20, port 5201
Reverse mode, remote host 192.168.20.20 is sending
[  5] local 192.168.20.153 port 39132 connected to 192.168.20.20 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  38.6 MBytes   324 Mbits/sec                  
[  5]   1.00-2.00   sec  32.1 MBytes   270 Mbits/sec                  
[  5]   2.00-3.00   sec  20.0 MBytes   168 Mbits/sec                  
[  5]   3.00-4.00   sec  9.50 MBytes  79.7 Mbits/sec                  
[  5]   4.00-5.00   sec  5.75 MBytes  48.2 Mbits/sec                  
[  5]   5.00-6.00   sec  4.38 MBytes  36.7 Mbits/sec                  
[  5]   6.00-7.00   sec  3.88 MBytes  32.5 Mbits/sec                  
[  5]   7.00-8.00   sec  2.88 MBytes  24.1 Mbits/sec                  
[  5]   8.00-9.00   sec  2.50 MBytes  21.0 Mbits/sec                  
[  5]   9.00-10.00  sec  7.12 MBytes  59.7 Mbits/sec                  
[  5]  10.00-11.00  sec  3.00 MBytes  25.2 Mbits/sec                  
[  5]  11.00-12.00  sec  2.50 MBytes  21.0 Mbits/sec                  
[  5]  12.00-13.00  sec  3.38 MBytes  28.3 Mbits/sec                  
[  5]  13.00-14.00  sec  4.25 MBytes  35.6 Mbits/sec                  
[  5]  14.00-15.00  sec  4.88 MBytes  40.9 Mbits/sec                  
[  5]  15.00-16.00  sec  3.50 MBytes  29.4 Mbits/sec                  
[  5]  16.00-17.00  sec  4.25 MBytes  35.7 Mbits/sec                  
[  5]  17.00-18.00  sec  3.62 MBytes  30.4 Mbits/sec                  
[  5]  18.00-19.00  sec  3.38 MBytes  28.3 Mbits/sec                  
[  5]  19.00-20.00  sec  3.75 MBytes  31.5 Mbits/sec                  
[  5]  20.00-21.00  sec  5.75 MBytes  48.2 Mbits/sec                  
[  5]  21.00-22.00  sec  3.50 MBytes  29.4 Mbits/sec                  
[  5]  22.00-23.00  sec  3.38 MBytes  28.3 Mbits/sec                  
[  5]  23.00-24.00  sec  4.00 MBytes  33.6 Mbits/sec                  
[  5]  24.00-25.00  sec  14.9 MBytes   125 Mbits/sec                  
[  5]  25.00-26.00  sec  7.88 MBytes  66.1 Mbits/sec                  
[  5]  26.00-27.00  sec  25.9 MBytes   217 Mbits/sec                  
[  5]  27.00-28.00  sec  14.5 MBytes   122 Mbits/sec                  
[  5]  28.00-29.00  sec  2.62 MBytes  22.0 Mbits/sec                  
[  5]  29.00-30.00  sec  4.12 MBytes  34.6 Mbits/sec                  
[  5]  30.00-31.00  sec  5.12 MBytes  43.0 Mbits/sec                  
[  5]  31.00-32.00  sec  5.38 MBytes  45.1 Mbits/sec                  
[  5]  32.00-33.00  sec  2.88 MBytes  24.1 Mbits/sec                  
[  5]  33.00-34.00  sec  10.8 MBytes  90.2 Mbits/sec                  
[  5]  34.00-35.00  sec  7.62 MBytes  63.9 Mbits/sec                  
[  5]  35.00-36.00  sec  10.8 MBytes  90.2 Mbits/sec                  
[  5]  36.00-37.00  sec  11.5 MBytes  96.5 Mbits/sec                  
[  5]  37.00-38.00  sec  5.25 MBytes  44.0 Mbits/sec                  
[  5]  38.00-39.00  sec  7.50 MBytes  62.9 Mbits/sec                  
[  5]  39.00-40.00  sec  7.75 MBytes  65.0 Mbits/sec                  
[  5]  40.00-41.00  sec  19.8 MBytes   166 Mbits/sec                  
[  5]  41.00-42.00  sec  14.2 MBytes   120 Mbits/sec                  
[  5]  42.00-43.00  sec  8.88 MBytes  74.4 Mbits/sec                  
[  5]  43.00-44.00  sec   768 KBytes  6.30 Mbits/sec                  
[  5]  44.00-45.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  45.00-46.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  46.00-47.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  47.00-48.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  48.00-49.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  49.00-50.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  50.00-51.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  51.00-52.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  52.00-53.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  53.00-54.00  sec  99.8 KBytes   817 Kbits/sec                  
[  5]  54.00-55.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  55.00-56.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  56.00-57.00  sec  0.00 Bytes  0.00 bits/sec                  
[  5]  57.00-58.00  sec  0.00 Bytes  0.00 bits/sec                  
[... 170 lines cut ...]
[  120.200265]  ? __x64_sys_ioctl+0xaf/0xd0
[  120.200269]  ? syscall_exit_to_user_mode+0x22/0x40
[  120.200275]  ? do_syscall_64+0x66/0x80
[  120.200279]  ? exit_to_user_mode_prepare+0x132/0x1f0
[  120.200285]  entry_SYSCALL_64_after_hwframe+0x78/0xe2
[  120.200291] RIP: 0033:0x7abc85834ced
[  120.200340] Code: 04 25 28 00 00 00 48 89 45 c8 31 c0 48 8d 45 10 c7 45 b0 10 00 00 00 48 89 45 b8 48 8d 45 d0 48 89 45 c0 b8 10 00 00 00 0f 05 <89> c2 3d 00 f0 ff ff 77 1a 48 8b 45 c8 64 48 2b 04 25 28 00 00 00
[  120.200343] RSP: 002b:00007abc6b2bba60 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  120.200349] RAX: ffffffffffffffda RBX: 00007abc6b3945d8 RCX: 00007abc85834ced
[  120.200352] RDX: 00007abc6b2bbaf0 RSI: 0000000040406469 RDI: 0000000000000032
[  120.200354] RBP: 00007abc6b2bbab0 R08: 0000000000000000 R09: 0000000000000004
[  120.200357] R10: 0000000000000001 R11: 0000000000000246 R12: 0000000000000000
[... cut ...]
[  122.088892] rtw_8812au 2-1:1.0: failed to get urb length:16
[  126.085665] rtw_8812au 2-1:1.0: failed to get urb length:16
[  154.732665] rtw_8812au 2-1:1.0: failed to get urb length:12
[  155.037621] rtw_8812au 2-1:1.0: failed to get urb length:12
[  155.285109] rtw_8812au 2-1:1.0: failed to get urb length:12
[  156.486541] rtw_8812au 2-1:1.0: failed to get urb length:16
[  173.595379] rtw_8812au 2-1:1.0: failed to get urb length:16
[  179.410824] rtw_8812au 2-1:1.0: failed to get urb length:16
```

dubhater, 2024-08-20:
>> No matter what port you used, please run this command and then plug it again and run iperf3: echo 0 > /sys/module/rtw_usb/parameters/switch_usb_mode. Do you still get the errors?
>
> No. This command fixes all the urb & kernel warnings. The iperf3 speed is just fine (~200Mb/s)

That's good, that's a clue.

Now, please reset switch_usb_mode to the default value, which is 1, and plug the device in a USB 2 port. Do you get the errors?

And here is another patch to try:

```diff
diff --git a/rtw8821a.c b/rtw8821a.c
index 7031ca1..db4b731 100644
--- a/rtw8821a.c
+++ b/rtw8821a.c
@@ -572,6 +572,7 @@ static void rtw8821au_rx_aggregation(struct rtw_dev *rtwdev, bool enable)
 
 	val16 = (rxagg_usb_timeout << 8) | rxagg_usb_size;
 	rtw_write16(rtwdev, REG_RXDMA_AGG_PG_TH, val16);
+	rtw_write32_set(rtwdev, REG_RXDMA_AGG_PG_TH, BIT_EN_PRE_CALC);
 
 	rtw_write8_set(rtwdev, REG_TXDMA_PQ_MAP, BIT_RXDMA_AGG_EN);
 }
```

LuisMitaHL, 2024-08-21:
> Now, please reset switch_usb_mode to the default value, which is 1, and plug the device in a USB 2 port. Do you get the errors?

No, I can't reproduce the error. Everything is normal (with usb2 speeds)

> And here is another patch to try:
>
> ```diff
> diff --git a/rtw8821a.c b/rtw8821a.c
> index 7031ca1..db4b731 100644
> --- a/rtw8821a.c
> +++ b/rtw8821a.c
> @@ -572,6 +572,7 @@ static void [...]

I am obtaining the same problems (urb and kernel warnings)

dubhater, 2024-08-21:
Another experiment:

```diff
diff --git a/usb.c b/usb.c
index 3dd1bc5..193893c 100644
--- a/usb.c
+++ b/usb.c
@@ -331,7 +331,7 @@ static int rtw_usb_write_port(struct rtw_dev *rtwdev, u8 qsel, struct sk_buff *s
 		return -ENOMEM;
 
 	usb_fill_bulk_urb(urb, usbd, pipe, skb->data, skb->len, cb, context);
-	urb->transfer_flags |= URB_ZERO_PACKET;
+	// urb->transfer_flags |= URB_ZERO_PACKET;
 	ret = usb_submit_urb(urb, GFP_ATOMIC);
 
 	usb_free_urb(urb);
```

This is going to cause other problems, but I'm curious if "failed to get urb length:..." errors go away.

LuisMitaHL, 2024-08-21:
Hi,
The error continues, nothing has changed.

dubhater, 2024-10-30:
@LuisMitaHL Do you still have this problem?

LuisMitaHL, 2024-10-30:
Yes, with the latest code the error continues.
```
[  303.456556] wlan1: Limiting TX power to 23 (30 - 7) dBm as advertised by xx:xx:xx:xx:xx:xx
[  353.752604] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  353.918557] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  354.029746] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  354.250890] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  354.534210] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  354.683478] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  354.800034] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  355.101466] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  355.339014] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  355.445166] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  355.748265] rtw_8812au 2-1:1.0: failed to get urb length:32768
[  355.748383] ------------[ cut here ]------------
[  355.748387] Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0
[  355.748451] WARNING: CPU: 0 PID: 17 at net/mac80211/rx.c:5344 ieee80211_rx_list+0x2b5/0xea0 [mac80211]
[  355.748607] Modules linked in: cmac rtw_8812au(OE) rtw_8812a(OE) rtw_88xxa(OE) rtw_usb(OE) rtw_core(OE) ccm bnep overlay snd_seq_dummy snd_hrtimer snd_seq snd_seq_device snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio intel_soc_dts_thermal intel_soc_dts_iosf uvcvideo intel_powerclamp coretemp kvm_intel spi_nor videobuf2_vmalloc uvc kvm ath3k mtd ath9k videobuf2_memops i915 joydev videobuf2_v4l2 irqbypass btusb ath9k_common iTCO_wdt crct10dif_pclmul mei_pxp mousedev at24 spi_intel_platform snd_hda_intel intel_pmc_bxt mei_hdcp crc32_pclmul btrtl spi_intel iTCO_vendor_support snd_intel_dspcfg videodev ath9k_hw snd_intel_sdw_acpi btintel hid_multitouch polyval_generic snd_hda_codec btbcm acer_wmi videobuf2_common btmtk gf128mul wmi_bmof ath snd_hda_core sparse_keymap ghash_clmulni_intel bluetooth intel_rapl_msr r8169 mc cryptd snd_hwdep ecdh_generic sha512_ssse3 drm_buddy realtek crc16 sha1_ssse3 mdio_devres mac80211 snd_pcm i2c_algo_bit libphy libarc4 ttm intel_cstate cfg80211 snd_timer
[  355.748726]  drm_display_helper pcspkr cec snd lpc_ich i2c_i801 mei_txe i2c_smbus rfkill soundcore intel_gtt mei i2c_hid_acpi i2c_hid int3401_thermal processor_thermal_device processor_thermal_rfim int3403_thermal vfat video processor_thermal_mbox fat pwm_lpss_platform processor_thermal_rapl wmi pwm_lpss intel_rapl_common int3400_thermal int340x_thermal_zone acpi_thermal_rel dw_dmac mac_hid pkcs8_key_parser i2c_dev crypto_user loop fuse dm_mod nfnetlink bpf_preload ip_tables x_tables serio_raw atkbd libps2 vivaldi_fmap sha256_ssse3 xhci_pci xhci_pci_renesas i8042 sdhci_acpi serio sdhci mmc_core btrfs blake2b_generic libcrc32c crc32c_generic crc32c_intel xor raid6_pq
[  355.748810] CPU: 0 PID: 17 Comm: ksoftirqd/0 Tainted: G           OE      6.6.58-1-lts #1 1400000003000000474e5500ee53b845eb376bed
[  355.748817] Hardware name: Acer Aspire ES1-111M/R2, BIOS V1.16 10/20/2015
[  355.748820] RIP: 0010:ieee80211_rx_list+0x2b5/0xea0 [mac80211]
[  355.748969] Code: 0f 86 4d 06 00 00 80 3d 0c 94 08 00 00 0f 85 31 ff ff ff 0f b6 53 4a 48 c7 c7 78 22 af c0 c6 05 f4 93 08 00 01 e8 bb 43 8c da <0f> 0b e9 13 ff ff ff 40 80 fe 0b 0f 86 97 05 00 00 80 3d d6 93 08
[  355.748974] RSP: 0018:ffffc900000bfcd8 EFLAGS: 00010282
[  355.748978] RAX: 0000000000000000 RBX: ffff8881a90bd500 RCX: 0000000000000027
[  355.748981] RDX: ffff888277c216c8 RSI: 0000000000000001 RDI: ffff888277c216c0
[  355.748984] RBP: ffff888114300900 R08: 0000000000000000 R09: ffffc900000bfb48
[  355.748986] R10: ffffffff9d0b2748 R11: 0000000000000003 R12: 0000000000000000
[  355.748989] R13: 0000000000000001 R14: ffff8881a90bd500 R15: ffffc900000bfdd0
[  355.748992] FS:  0000000000000000(0000) GS:ffff888277c00000(0000) knlGS:0000000000000000
[  355.748995] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  355.748998] CR2: 000078e928002000 CR3: 00000001f7420000 CR4: 00000000001006f0
[  355.749001] Call Trace:
[  355.749006]  <TASK>
[  355.749009]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 1400000003000000474e55005a70c732631132ac]
[  355.749147]  ? __warn+0x81/0x130
[  355.749160]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 1400000003000000474e55005a70c732631132ac]
[  355.749284]  ? report_bug+0x16f/0x1a0
[  355.749291]  ? handle_bug+0x3c/0x80
[  355.749296]  ? exc_invalid_op+0x17/0x70
[  355.749300]  ? asm_exc_invalid_op+0x1a/0x20
[  355.749307]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 1400000003000000474e55005a70c732631132ac]
[  355.749429]  ? ieee80211_rx_list+0x2b5/0xea0 [mac80211 1400000003000000474e55005a70c732631132ac]
[  355.749553]  ? sched_clock+0x10/0x30
[  355.749558]  ? sched_clock_cpu+0xf/0x1d0
[  355.749564]  ? psi_group_change+0x227/0x3f0
[  355.749570]  ieee80211_rx_napi+0x54/0xf0 [mac80211 1400000003000000474e55005a70c732631132ac]
[  355.749703]  ? __switch_to_asm+0x3e/0x70
[  355.749711]  ieee80211_handle_queued_frames+0xbe/0xd0 [mac80211 1400000003000000474e55005a70c732631132ac]
[  355.749824]  tasklet_action_common.isra.0+0xc3/0x230
[  355.749830]  handle_softirqs+0xe5/0x2e0
[  355.749837]  ? __pfx_smpboot_thread_fn+0x10/0x10
[  355.749842]  run_ksoftirqd+0x31/0x40
[  355.749846]  smpboot_thread_fn+0xdc/0x1d0
[  355.749851]  kthread+0xe8/0x120
[  355.749858]  ? __pfx_kthread+0x10/0x10
[  355.749864]  ret_from_fork+0x34/0x50
[  355.749869]  ? __pfx_kthread+0x10/0x10
[  355.749874]  ret_from_fork_asm+0x1b/0x30
[... cut ...]
[  384.517973] R13: 0000000000000060 R14: 0000000000000000 R15: 0000000000000060
[  384.517977] FS:  0000000000000000(0000) GS:ffff888277c80000(0000) knlGS:0000000000000000
[  384.517982] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  384.517987] CR2: 0000000000000010 CR3: 00000001f7420000 CR4: 00000000001006e0
[  384.517992] note: kworker/u8:4[59] exited with irqs disabled
[  385.785867] rtw_8812au 2-1:1.0: failed to get tx report from firmware
[  387.783082] rtw_8812au 2-1:1.0: failed to get tx report from firmware

```

dubhater, 2024-10-30:
Does this fix it?

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/usb.c b/drivers/net/wireless/realtek/rtw88/usb.c
index d6908a93d5ef..07477083f607 100644
--- a/drivers/net/wireless/realtek/rtw88/usb.c
+++ b/drivers/net/wireless/realtek/rtw88/usb.c
@@ -815,7 +815,7 @@ static void rtw_usb_dynamic_rx_agg_v2(struct rtw_dev *rtwdev, bool enable)
 	u16 val16;
 
 	if (rtwusb->udev->speed == USB_SPEED_SUPER) {
-		size = 0x7;
+		size = 0x6;
 		timeout = 0x1a;
 	} else {
 		size = 0x5;
```

LuisMitaHL, 2024-10-31:
Yes, this fix the problem.
There is no more kernel errors and the connection speed is high

```
-----------------------------------------------------------
Server listening on 5201 (test #2)
-----------------------------------------------------------
Accepted connection from x, port 34096
[  5] local x port 5201 connected to x port 34108
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  39.9 MBytes   334 Mbits/sec   81    676 KBytes       
[  5]   1.00-2.00   sec  36.2 MBytes   304 Mbits/sec  538    474 KBytes       
[  5]   2.00-3.00   sec  43.9 MBytes   368 Mbits/sec    0    396 KBytes       
[  5]   3.00-4.00   sec  43.0 MBytes   361 Mbits/sec    0    700 KBytes       
[  5]   4.00-5.00   sec  47.5 MBytes   398 Mbits/sec    0    482 KBytes       
[  5]   5.00-6.00   sec  43.8 MBytes   367 Mbits/sec    0    467 KBytes       
[  5]   6.00-7.00   sec  55.8 MBytes   468 Mbits/sec    0    547 KBytes       
[  5]   7.00-8.00   sec  53.2 MBytes   446 Mbits/sec    0    646 KBytes       
[  5]   8.00-9.00   sec  44.1 MBytes   370 Mbits/sec    0    577 KBytes       
[  5]   9.00-10.00  sec  35.9 MBytes   301 Mbits/sec    0    462 KBytes       
[  5]  10.00-11.00  sec  28.6 MBytes   240 Mbits/sec    0    700 KBytes       
[  5]  11.00-12.00  sec  45.0 MBytes   377 Mbits/sec   36    484 KBytes       
[  5]  12.00-13.00  sec  33.8 MBytes   283 Mbits/sec    0    748 KBytes       
[  5]  13.00-14.00  sec  48.6 MBytes   408 Mbits/sec    1    523 KBytes       
[  5]  14.00-15.00  sec  63.2 MBytes   531 Mbits/sec    0    863 KBytes       
[  5]  15.00-16.00  sec  52.0 MBytes   436 Mbits/sec    0    581 KBytes       
[  5]  16.00-17.00  sec  49.5 MBytes   415 Mbits/sec    0    474 KBytes       
[  5]  17.00-18.00  sec  61.0 MBytes   512 Mbits/sec    0    769 KBytes       
[  5]  18.00-19.00  sec  61.5 MBytes   516 Mbits/sec    0    546 KBytes       
[  5]  19.00-20.00  sec  56.6 MBytes   475 Mbits/sec    0    537 KBytes       
[  5]  20.00-21.00  sec  55.1 MBytes   462 Mbits/sec    0    658 KBytes       
[  5]  21.00-22.00  sec  49.1 MBytes   412 Mbits/sec    0    519 KBytes       
[  5]  22.00-23.00  sec  54.6 MBytes   458 Mbits/sec    0    406 KBytes       
[  5]  23.00-24.00  sec  56.4 MBytes   473 Mbits/sec    0    567 KBytes       
[  5]  24.00-25.00  sec  53.8 MBytes   451 Mbits/sec    0    491 KBytes       
[  5]  25.00-26.00  sec  55.9 MBytes   469 Mbits/sec    0    452 KBytes       
[  5]  26.00-27.00  sec  54.1 MBytes   454 Mbits/sec    0   46.7 KBytes       
[  5]  27.00-28.00  sec  57.1 MBytes   479 Mbits/sec    0    646 KBytes       
[  5]  28.00-29.00  sec  66.1 MBytes   555 Mbits/sec    0    553 KBytes       
[  5]  29.00-30.00  sec  65.1 MBytes   546 Mbits/sec    0    566 KBytes       
[  5]  30.00-31.00  sec  50.4 MBytes   423 Mbits/sec    0    882 KBytes       
[  5]  31.00-32.00  sec  34.0 MBytes   285 Mbits/sec    0    663 KBytes       
[  5]  32.00-33.00  sec  27.5 MBytes   231 Mbits/sec    0    605 KBytes       
[  5]  33.00-34.00  sec  38.2 MBytes   321 Mbits/sec    0    573 KBytes       
[  5]  34.00-35.00  sec  56.0 MBytes   470 Mbits/sec    0    492 KBytes       
[  5]  35.00-36.00  sec  45.0 MBytes   377 Mbits/sec    0    764 KBytes       
[  5]  36.00-37.00  sec  33.4 MBytes   280 Mbits/sec    0    479 KBytes       
[  5]  37.00-38.00  sec  30.1 MBytes   253 Mbits/sec    0    392 KBytes       
[  5]  38.00-39.00  sec  37.9 MBytes   318 Mbits/sec    0    370 KBytes       
[  5]  39.00-40.00  sec  49.6 MBytes   416 Mbits/sec    0    496 KBytes       
[  5]  40.00-41.00  sec  45.0 MBytes   377 Mbits/sec    0    348 KBytes       
[  5]  41.00-42.00  sec  34.1 MBytes   286 Mbits/sec    0    346 KBytes       
[  5]  42.00-43.00  sec  29.6 MBytes   248 Mbits/sec    0    274 KBytes       
[  5]  43.00-44.00  sec  32.6 MBytes   274 Mbits/sec    0    301 KBytes       
[  5]  44.00-45.00  sec  35.4 MBytes   297 Mbits/sec    0    284 KBytes       
[  5]  45.00-46.00  sec  34.0 MBytes   285 Mbits/sec    0    296 KBytes       
[  5]  46.00-47.00  sec  34.5 MBytes   289 Mbits/sec    0    478 KBytes       
[  5]  47.00-48.00  sec  32.8 MBytes   275 Mbits/sec    0   38.2 KBytes       
[  5]  48.00-49.00  sec  32.8 MBytes   275 Mbits/sec    0    386 KBytes       
[  5]  49.00-50.00  sec  35.9 MBytes   301 Mbits/sec    0    433 KBytes       
[  5]  50.00-51.00  sec  42.2 MBytes   354 Mbits/sec    0    501 KBytes       
[  5]  51.00-52.00  sec  32.8 MBytes   275 Mbits/sec    0    313 KBytes       
[  5]  52.00-53.00  sec  34.1 MBytes   286 Mbits/sec    0    300 KBytes       
[  5]  53.00-54.00  sec  27.0 MBytes   226 Mbits/sec    0    277 KBytes       
[  5]  54.00-55.00  sec  36.9 MBytes   309 Mbits/sec    0    272 KBytes       
[  5]  55.00-56.00  sec  44.8 MBytes   376 Mbits/sec    0    502 KBytes       
[  5]  56.00-57.00  sec  34.4 MBytes   288 Mbits/sec    0    294 KBytes       
[  5]  57.00-58.00  sec  36.1 MBytes   303 Mbits/sec    0    444 KBytes       
[  5]  58.00-59.00  sec  32.1 MBytes   269 Mbits/sec    0    478 KBytes       
[  5]  59.00-60.00  sec  30.5 MBytes   256 Mbits/sec    0    443 KBytes       
[... 13 lines cut ...]
[  5]  73.00-74.00  sec  35.5 MBytes   298 Mbits/sec    0   50.9 KBytes       
[  5]  74.00-75.00  sec  38.4 MBytes   322 Mbits/sec    0    706 KBytes       
[  5]  75.00-76.00  sec  32.1 MBytes   269 Mbits/sec    1    898 KBytes       
[  5]  76.00-77.00  sec  46.5 MBytes   390 Mbits/sec    0    622 KBytes       
[... cut ...]
[  5]  88.00-89.00  sec  42.1 MBytes   353 Mbits/sec    0    693 KBytes       
[  5]  89.00-90.00  sec  39.4 MBytes   332 Mbits/sec    0    489 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-90.00  sec  3.63 GBytes   347 Mbits/sec  738             sender
-----------------------------------------------------------
Server listening on 5201 (test #3)
-----------------------------------------------------------

```

closed 2024-10-31 by dubhater

dubhater, 2024-10-31:
> Do you often see "Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0"? I have only seen it once.
>
> What about "failed to get urb length:32768"? I have never seen that on my computer.

I found out why I was seeing these errors only rarely. They appear when the AP sends larger frames, like 4000-5000 bytes, and that happens only when I have excellent reception. Most of the time I use my RTL8812AU in a place where I have only good reception, not excellent. Mystery solved!