lwfinger/rtw88 #232: Hard resets

vfiksdal, 2024-08-24:
Hi!

I previously used the the aircrack driver (https://github.com/aircrack-ng/rtl8812au.git) for my "TP-Link 802.11ac WLAN Adapter" (Archer T2U among friends, using the 8821au chipset), but it hard reset my system every ~24 hours. So I've been very happy with your version of the driver for the last week, but it suddenly hard crashed again.. An hard crash per week is obviously vastly superior to the daily crashes, but I would very much like to avoid any crashing whatsoever.

Problem is, neither this or the old driver logs any kind of problem at all. Journalctl simply shows " --- reboot ---" messages without any further information on what went wrong.. 

Is there any way of getting more verbose debugging for the wlan card without going full on debug-kernel for a week?

dubhater, 2024-08-24:
Yes, rtw_core has the debug_mask option, and the other driver has the rtw_drv_log_level option. But I don't think they will help you.

What exactly happens with the computer? Does it reboot by itself? Or does it just lock up and you have to press a button to reboot it? If it reboots by itself and it's a desktop computer, I would try a different PSU.

vfiksdal, 2024-08-25:
The computer simply reboots. It always happens when I'm using the computer actively, writing a document, browsing the web, etc. It simply goes black for two seconds before displaying the boot screen, and then entering grub.

And this is a laptop with a fully charged battery, so I doubt it's a PSU issue?

Thanks for the input though. Ill check out the debug options you pointed out.

dubhater, 2024-08-25:
If you can connect the laptop to another computer using a network cable, you can try netconsole: https://www.kernel.org/doc/Documentation/networking/netconsole.txt

vfiksdal, 2024-08-30:
Sorry for the lack of input. I'm away on a business trip and won't be able to follow up anytime soon.

Thanks for the tip though. The laptop don't have an ethernet socket, or even an available USB port (It's a cheap laptop, and the IO atrocious). I suppose I could approach this with an USB hub though. Worth considering :-)