lwfinger/rtw88 #235: Only getting same error logs, any fixes?

cemunal, 2024-09-16:
I am testing build in kernel drivers and getting same error logs only. Here is some info:

**sudo lspci -v:**

02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
	Subsystem: AzureWave Device 3040
	Flags: bus master, fast devsel, latency 0, IRQ 130
	I/O ports at e000 [size=256]
	Memory at ef000000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, IntMsgNum 0
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-c8-21-01
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Capabilities: [170] Precision Time Measurement
	Capabilities: [17c] Vendor Specific Information: ID=0003 Rev=1 Len=054 <?>
	Kernel driver in use: rtw_8821ce
	Kernel modules: rtw88_8821ce
__________

**uname -r:**

6.10.9-200.fc40.x86_64
__________

**sudo dmesg | grep rtw:**

[    7.785903] rtw_8821ce 0000:02:00.0: enabling device (0000 -> 0003)
[    7.799541] rtw_8821ce 0000:02:00.0: Firmware version 24.11.0, H2C version 12
[    7.852418] rtw_8821ce 0000:02:00.0 wlp2s0: renamed from wlan0
[  143.722023] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[  143.722043] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[  143.722058] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[  161.626544] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[  161.626558] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[  161.626573] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[  270.552854] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[  270.552863] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[  270.552873] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[  395.381350] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[  395.381354] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[  395.381359] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[  658.591238] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[  658.591248] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[  658.591260] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[  898.787339] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[  898.787347] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[  898.787355] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[  929.845965] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[  929.845969] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[  929.845974] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[ 1125.098219] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[ 1125.098230] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[ 1125.098244] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[ 1243.387084] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[ 1243.387092] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[ 1243.387103] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[ 1393.896592] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[ 1393.896605] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[ 1393.896619] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[ 1401.173918] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[ 1401.173928] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[ 1401.173939] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
[ 1415.097037] rtw_8821ce 0000:02:00.0: PCIe Bus Error: severity=Correctable, type=Physical Layer, (Receiver ID)
[ 1415.097046] rtw_8821ce 0000:02:00.0:   device [10ec:c821] error status/mask=00000001/0000e000
[ 1415.097058] rtw_8821ce 0000:02:00.0:    [ 0] RxErr                  (First)
__________

Any fixes?

dubhater, 2024-09-16:
You could try a BIOS update. If that doesn't help, email linux-wireless@vger.kernel.org.

cemunal, 2024-09-17:
The BIOS is up-to-date so I will send an e-mail to linux-wireless@vger.kernel.org. Thanks.