lwfinger/rtw88 #237: Commit fa950890a05365499c5dea30dd2b8b5de63c6ed8 breaks compatibility with kernels <5.16

plumbeo, 2024-10-14:
Hello, recent commit fa950890a05365499c5dea30dd2b8b5de63c6ed8 introduces a call to `cfg80211_get_ies_channel_number()` that breaks compatibility with kernel releases older than 5.16, in which that function was introduced, including the still supported LTS releases 5.4, 5.10 and 5.15. Will this be fixed?

Thanks in any case for all your work, I'm using this driver on a RTL8811AU dongle and it works very well, excluding a single issue which is already being discussed in the linux-wireless mailing list anyway (
https://lore.kernel.org/linux-wireless/6e7ecb47-7ea0-433a-a19f-05f88a2edf6b@gmail.com/T/#t).

dubhater, 2024-10-15:
Hi. Do you get any compile warnings with this?

```diff
diff --git a/rx.c b/rx.c
index 7e67dec..690dfb6 100644
--- a/rx.c
+++ b/rx.c
@@ -176,10 +176,12 @@ void rtw_update_rx_freq_from_ie(struct rtw_dev *rtwdev, struct sk_buff *skb,
 	else
 		goto fill_rx_status;
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 16, 0)
 	channel_number = cfg80211_get_ies_channel_number(variable, ielen,
 							 NL80211_BAND_2GHZ);
 	if (channel_number != -1)
 		channel = channel_number;
+#endif
 
 fill_rx_status:
 	rtw_set_rx_freq_band(pkt_stat, channel);
```

Or with this?

```diff
diff --git a/rx.c b/rx.c
index 7e67dec..1153613 100644
--- a/rx.c
+++ b/rx.c
@@ -176,8 +176,12 @@ void rtw_update_rx_freq_from_ie(struct rtw_dev *rtwdev, struct sk_buff *skb,
 	else
 		goto fill_rx_status;
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 16, 0)
 	channel_number = cfg80211_get_ies_channel_number(variable, ielen,
 							 NL80211_BAND_2GHZ);
+#else
+	channel_number = -1;
+#endif
 	if (channel_number != -1)
 		channel = channel_number;
 
```

a5a5aa555oo, 2024-10-15:
I got this when building with kernel 6.5.13. 

kernels < 6.6 have no definition of abs_diff()
https://elixir.bootlin.com/linux/v6.6-rc1/source/include/linux/math.h#L158

```
$ KSRC=/tmp/linux-6.5.13/ make clean all
make -C /tmp/linux-6.5.13/ M=$PWD clean
make[1]: Entering directory '/tmp/linux-6.5.13'
make[1]: Leaving directory '/tmp/linux-6.5.13'
make -j`nproc` -C /tmp/linux-6.5.13/ M=$PWD modules
make[1]: Entering directory '/tmp/linux-6.5.13'
  CC [M]  /tmp/rtw88/main.o

................

  CC [M]  /tmp/rtw88/rtw88xxa.o
  CC [M]  /tmp/rtw88/rtw8821au.o
  CC [M]  /tmp/rtw88/rtw8821c.o
  CC [M]  /tmp/rtw88/rtw8821c_table.o
/tmp/rtw88/rtw88xxa.c: In function ‘rtw88xxa_iqk_finish’:
/tmp/rtw88/rtw88xxa.c:1796:30: error: implicit declaration of function ‘abs_diff’ [-Wimplicit-function-declaration]
 1796 |                         dx = abs_diff(x_temp[i] >> 21, x_temp[ii] >> 21);
      |                              ^~~~~~~~
make[3]: *** [scripts/Makefile.build:243: /tmp/rtw88/rtw88xxa.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/tmp/linux-6.5.13/Makefile:2034: /tmp/rtw88] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/tmp/linux-6.5.13'
make: *** [Makefile:177: all] Error 2
```

a5a5aa555oo, 2024-10-15:
@dubhater 

cfg80211_get_ies_channel_number() first appeared in kernel 5.16 and it needs only three parameters in kernel 5.16 and 6.4+
but in kernel 5.17 to 6.3, it needs four parameter.
what do you think about this patch?

````diff
diff --git a/rx.c b/rx.c
index 7e67dec..06a9415 100644
--- a/rx.c
+++ b/rx.c
@@ -176,8 +176,16 @@ void rtw_update_rx_freq_from_ie(struct rtw_dev *rtwdev, struct sk_buff *skb,
        else
                goto fill_rx_status;
 
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 4, 0))
        channel_number = cfg80211_get_ies_channel_number(variable, ielen,
                                                         NL80211_BAND_2GHZ);
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 17, 0))
+       channel_number = cfg80211_get_ies_channel_number(variable, ielen,
+                                                        NL80211_BAND_2GHZ, 0);
+#else
+       channel_number = -1;
+#endif
+
        if (channel_number != -1)
                channel = channel_number;
 

````

a5a5aa555oo, 2024-10-15:
I did a compilation test with kernel 5.4.284 ~ 6.12-rc3 and the problems above are the only ones I found.

This patch fixes https://github.com/lwfinger/rtw88/issues/237#issuecomment-2413815416 but im not sure this is the best solution.
````diff
diff --git a/rtw88xxa.h b/rtw88xxa.h
index 09a45c1..809a1bb 100644
--- a/rtw88xxa.h
+++ b/rtw88xxa.h
@@ -8,6 +8,15 @@
 #include <asm/byteorder.h>
 #include "reg.h"
 
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(6, 6, 0))
+#define abs_diff(a, b) ({                      \
+       typeof(a) __a = (a);                    \
+       typeof(b) __b = (b);                    \
+       (void)(&__a == &__b);                   \
+       __a > __b ? (__a - __b) : (__b - __a);  \
+})
+#endif
+
 struct rtw8821au_efuse {
        u8 res4[48];                    /* 0xd0 */
        u8 vid[2];                      /* 0x100 */
````

dubhater, 2024-10-15:
@a5a5aa555oo abs_diff should go in main.h in case other drivers need it in the future.

Your fix for cfg80211_get_ies_channel_number looks fine to me.

a5a5aa555oo, 2024-10-15:
@dubhater I moved abs_diff to main.h and sent the PR, thanks for the suggestions!

plumbeo, 2024-10-16:
Hello, only compile-tested on 5.15 because I don't have physical access to the device now but this version open codes `cfg80211_get_ies_channel_number()`:

```
--- a/rx.c
+++ b/rx.c
@@ -155,6 +155,9 @@ void rtw_update_rx_freq_from_ie(struct rtw_dev *rtwdev, struct sk_buff *skb,
        size_t hdr_len, ielen;
        int channel_number;
        u8 *variable;
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 17, 0))
+       const struct element *tmp;
+#endif
 
        if (!test_bit(RTW_FLAG_SCANNING, rtwdev->flags))
                goto fill_rx_status;
@@ -176,8 +179,29 @@ void rtw_update_rx_freq_from_ie(struct rtw_dev *rtwdev, struct sk_buff *skb,
        else
                goto fill_rx_status;
 
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 4, 0))
        channel_number = cfg80211_get_ies_channel_number(variable, ielen,
                                                         NL80211_BAND_2GHZ);
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 17, 0))
+       channel_number = cfg80211_get_ies_channel_number(variable, ielen,
+                                                        NL80211_BAND_2GHZ, 0);
+#else
+       channel_number = -1;
+
+       tmp = cfg80211_find_elem(WLAN_EID_DS_PARAMS, variable, ielen);
+
+       if (tmp && tmp->datalen == 1) {
+               channel_number = tmp->data[0];
+       } else {
+               tmp = cfg80211_find_elem(WLAN_EID_HT_OPERATION, variable, ielen);
+
+               if (tmp && tmp->datalen >= sizeof(struct ieee80211_ht_operation)) {
+                       struct ieee80211_ht_operation *htop = (void *)tmp->data;
+                       channel_number = htop->primary_chan;
+               }
+       }
+#endif
+
        if (channel_number != -1)
                channel = channel_number;
 
```

I don't know if what the original commit did is worth doing at the cost of a little more complexity.

closed 2024-10-16 by dubhater