lwfinger/rtw88 #238: rtw88 r512.247209b and 6.11.3-zen1 Unknown symbols (err -2)

alium, 2024-10-15:
I wanted just  to try the latest changes, but the driver is broken.
rtw88 r512.247209b and 6.11.3-zen1, dmesg:
[    9.081894] rtw_core: loading out-of-tree module taints kernel.
[    9.081900] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[    9.096919] rtw_8821a: Unknown symbol rtw88xxa_query_phy_status (err -2)
[    9.096935] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_afe (err -2)
[    9.096948] rtw_8821a: Unknown symbol rtw88xxa_iqk_configure_mac (err -2)
[    9.096959] rtw_8821a: Unknown symbol rtw88xxa_set_channel (err -2)
[    9.096973] rtw_8821a: Unknown symbol rtw88xxa_power_on (err -2)
[    9.096983] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_mac_bb (err -2)
[    9.096993] rtw_8821a: Unknown symbol rtw88xxa_phy_read_rf (err -2)
[    9.097007] rtw_8821a: Unknown symbol rtw88xxa_phy_pwrtrack (err -2)
[    9.097017] rtw_8821a: Unknown symbol rtw88xxa_set_tx_power_index (err -2)
[    9.097027] rtw_8821a: Unknown symbol rtw88xxa_power_off (err -2)
[    9.097041] rtw_8821a: Unknown symbol rtw88xxa_read_efuse (err -2)
[    9.097051] rtw_8821a: Unknown symbol rtw88xxa_iqk_finish (err -2)
[    9.097065] rtw_8821a: Unknown symbol rtw88xxa_iqk_restore_mac_bb (err -2)
[    9.097075] rtw_8821a: Unknown symbol rtw88xxa_phy_cck_pd_set (err -2)
[    9.097085] rtw_8821a: Unknown symbol rtw88xxa_efuse_grant (err -2)
[    9.097096] rtw_8821a: Unknown symbol rtw88xxa_false_alarm_statistics (err -2)

dubhater, 2024-10-15:
Maybe you didn't load rtw_88xxa? It's a new module.

a5a5aa555oo, 2024-10-15:
I can reproduce this issue, and im trying to see if it is caused by [rtw88-dkms-git](https://aur.archlinux.org/packages/rtw88-dkms-git)

kernel: [6.11.3-zen1-1-zen](https://archlinux.org/packages/extra/x86_64/linux-zen/)

```
[  972.473687] rtw_8821a: Unknown symbol rtw88xxa_query_phy_status (err -2)
[  972.473705] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_afe (err -2)
[  972.473722] rtw_8821a: Unknown symbol rtw88xxa_iqk_configure_mac (err -2)
[  972.473734] rtw_8821a: Unknown symbol rtw88xxa_set_channel (err -2)
[  972.473751] rtw_8821a: Unknown symbol rtw88xxa_power_on (err -2)
[  972.473762] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_mac_bb (err -2)
[  972.473772] rtw_8821a: Unknown symbol rtw88xxa_phy_read_rf (err -2)
[  972.473787] rtw_8821a: Unknown symbol rtw88xxa_phy_pwrtrack (err -2)
[  972.473798] rtw_8821a: Unknown symbol rtw88xxa_set_tx_power_index (err -2)
[  972.473808] rtw_8821a: Unknown symbol rtw88xxa_power_off (err -2)
[  972.473821] rtw_8821a: Unknown symbol rtw88xxa_read_efuse (err -2)
[  972.473832] rtw_8821a: Unknown symbol rtw88xxa_iqk_finish (err -2)
[  972.473846] rtw_8821a: Unknown symbol rtw88xxa_iqk_restore_mac_bb (err -2)
[  972.473857] rtw_8821a: Unknown symbol rtw88xxa_phy_cck_pd_set (err -2)
[  972.473868] rtw_8821a: Unknown symbol rtw88xxa_efuse_grant (err -2)
[  972.473880] rtw_8821a: Unknown symbol rtw88xxa_false_alarm_statistics (err -2)
[  978.337868] rtw_8821a: Unknown symbol rtw88xxa_query_phy_status (err -2)
[  978.337887] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_afe (err -2)
[  978.337901] rtw_8821a: Unknown symbol rtw88xxa_iqk_configure_mac (err -2)
[  978.337912] rtw_8821a: Unknown symbol rtw88xxa_set_channel (err -2)
[  978.337928] rtw_8821a: Unknown symbol rtw88xxa_power_on (err -2)
[  978.337939] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_mac_bb (err -2)
[  978.337950] rtw_8821a: Unknown symbol rtw88xxa_phy_read_rf (err -2)
[  978.337964] rtw_8821a: Unknown symbol rtw88xxa_phy_pwrtrack (err -2)
[  978.337975] rtw_8821a: Unknown symbol rtw88xxa_set_tx_power_index (err -2)
[  978.337985] rtw_8821a: Unknown symbol rtw88xxa_power_off (err -2)
[  978.337999] rtw_8821a: Unknown symbol rtw88xxa_read_efuse (err -2)
[  978.338009] rtw_8821a: Unknown symbol rtw88xxa_iqk_finish (err -2)
[  978.338024] rtw_8821a: Unknown symbol rtw88xxa_iqk_restore_mac_bb (err -2)
[  978.338034] rtw_8821a: Unknown symbol rtw88xxa_phy_cck_pd_set (err -2)
[  978.338044] rtw_8821a: Unknown symbol rtw88xxa_efuse_grant (err -2)
[  978.338056] rtw_8821a: Unknown symbol rtw88xxa_false_alarm_statistics (err -2)
[ 1018.884577] rtw_8821a: Unknown symbol rtw88xxa_query_phy_status (err -2)
[ 1018.884605] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_afe (err -2)
[ 1018.884627] rtw_8821a: Unknown symbol rtw88xxa_iqk_configure_mac (err -2)
[ 1018.884643] rtw_8821a: Unknown symbol rtw88xxa_set_channel (err -2)
[ 1018.884667] rtw_8821a: Unknown symbol rtw88xxa_power_on (err -2)
[ 1018.884682] rtw_8821a: Unknown symbol rtw88xxa_iqk_backup_mac_bb (err -2)
[ 1018.884698] rtw_8821a: Unknown symbol rtw88xxa_phy_read_rf (err -2)
[ 1018.884719] rtw_8821a: Unknown symbol rtw88xxa_phy_pwrtrack (err -2)
[ 1018.884734] rtw_8821a: Unknown symbol rtw88xxa_set_tx_power_index (err -2)
[ 1018.884749] rtw_8821a: Unknown symbol rtw88xxa_power_off (err -2)
[ 1018.884769] rtw_8821a: Unknown symbol rtw88xxa_read_efuse (err -2)
[ 1018.884786] rtw_8821a: Unknown symbol rtw88xxa_iqk_finish (err -2)
[ 1018.884807] rtw_8821a: Unknown symbol rtw88xxa_iqk_restore_mac_bb (err -2)
[ 1018.884823] rtw_8821a: Unknown symbol rtw88xxa_phy_cck_pd_set (err -2)
[ 1018.884839] rtw_8821a: Unknown symbol rtw88xxa_efuse_grant (err -2)
[ 1018.884856] rtw_8821a: Unknown symbol rtw88xxa_false_alarm_statistics (err -2)
```

a5a5aa555oo, 2024-10-15:
I think i know what is wrong. I will update dkms.conf to fix this. :)

alium, 2024-10-15:
> Maybe you didn't load rtw_88xxa? It's a new module.
because missing in my system. As he mentioned @a5a5aa555oo maybe problem with dkms.conf?

dubhater, 2024-10-15:
Oh, yes, I forgot to update that.

a5a5aa555oo, 2024-10-15:
@alium I updated [rtw88-dkms-git](https://aur.archlinux.org/packages/rtw88-dkms-git), please reinstall it to see if it works for you.

closed 2024-10-16 by dubhater