lwfinger/rtw88 #241: ´error beacon valid´ error messages in AP mode

henkv1, 2024-10-19:
AP mode is really stable now and the connection speed is also fine, but I still observe a lot of the following error messages in the kernel log:

[533480.635435] rtw_8821cu 1-1.2.1:1.0: error beacon valid
[533480.635632] rtw_8821cu 1-1.2.1:1.0: failed to download drv rsvd page
[533480.635643] rtw_8821cu 1-1.2.1:1.0: failed to download beacon
[533714.828043] rtw_8821cu 1-1.2.1:1.0: error beacon valid
[533714.828237] rtw_8821cu 1-1.2.1:1.0: failed to download drv rsvd page
[533714.828247] rtw_8821cu 1-1.2.1:1.0: failed to download beacon
[534543.260394] rtw_8821cu 1-1.2.1:1.0: error beacon valid
[534543.260586] rtw_8821cu 1-1.2.1:1.0: failed to download drv rsvd page
[534699.420892] rtw_8821cu 1-1.2.1:1.0: error beacon valid
[534699.421090] rtw_8821cu 1-1.2.1:1.0: failed to download drv rsvd page
[534855.584947] rtw_8821cu 1-1.2.1:1.0: error beacon valid
[534855.585145] rtw_8821cu 1-1.2.1:1.0: failed to download drv rsvd page
[534855.585155] rtw_8821cu 1-1.2.1:1.0: failed to download beacon
[535371.892880] rtw_8821cu 1-1.2.1:1.0: error beacon valid
[535371.893079] rtw_8821cu 1-1.2.1:1.0: failed to download drv rsvd page
[535371.893090] rtw_8821cu 1-1.2.1:1.0: failed to download beacon

The messages also appear when there is (almost) no traffic and it does not seem to affect stability and/or performance.

morrownr, 2024-10-20:
Hi @henkv1 

dubhater is likely busy continuing to get the new drivers for rtl8812au and rtl8821/11au in the kernel.

dubhater, 2024-10-20:
Indeed. I can tell you what we already knew: the beacon has to be updated every now and then (rtw_ops_set_tim) and sometimes this fails. Maybe it's a timing problem?

henkv1, 2024-10-21:
I opened this ticket, because it's the last remaining issue I have with this driver. My AP is running very stable for some time now. This is a great achievement, so thank you all who worked hard to make this happen!
I don't think these messages have a negative impact on the performance and stability, but if someone knows what might cause this, maybe we can fix it. If not, maybe it's better to suppress the warnings, so they do not flood the kernel log.

morrownr, 2024-10-21:
@henkv1 

You mentioned this happens with AP mode. Do you know if it happens in any other mode and which one of the rtw88 drivers is the one in use in this case?

I ask just to document the issue for future reference.

henkv1, 2024-10-21:
I've seen this on rtw_8822bu, rtw_8821cu and rtw_8822cu on both 5GHz and 2GHz in AP mode. I don't use other modes, so I don't know if this happens in other modes as well. But I will try to see if I can reproduce this in other modes.

morrownr, 2024-10-21:
@henkv1 

FYI: dubhater merged a patch this morning that appears to shut down the messages you are seeing but it will provide a message that may be more meaningful for now. You might want to pull it and see.

Overall, it is great to see you report how well AP mode working these days. Does your report include all 3 of the chips you listed above?

henkv1, 2024-10-22:
Yes, all 3 run pretty well with the latest code. I've had some performance issues with rtw_8822bu in the past, but it looks like that is solved too.

morrownr, 2024-10-22:
That is really good news. Having solid drivers in kernel 6.12, the probable next LTS kernel, is good news. I was hoping the new drivers for rtl8812au and rtl8821/11au would be in kernel 6.12 but that did not work out but they are coming soon. I wonder if Linus would authorize a late merge? Does hurt to ask and if Ping-Ke supports it, I can see it happening.

Side note: A company sent me an adapter based on the rtl8852/32cu chip along with the most up to date out-of-kernel driver so I am evaluating that. I was hoping we could get beyond the out-of-kernel drivers but for WiFi 6 Realtek usb chips, it seems to be out-of-kernel or nothing for a LONG time. In the words of Larry Finger "those drivers are a pile of crap."

However, thanks to the work of dubhater and some others, USB WiFi is in better shape on Linux than it has ever been and it is continuing to get better. I sure wish someone would release an adapter based on the mt7925 chip. It sucks having a driver but no toy to play with.

Simon566, 2024-10-29:
my 8812au crashes regularly in AP mode , every 2 days or so 

Okt 29 16:51:48 odroid-h3 kernel: rtw_8812au 2-1:1.0: error beacon valid
Okt 29 16:51:48 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to download drv rsvd page
Okt 29 16:53:20 odroid-h3 kernel: rtw_8812au 2-1:1.0: error beacon valid
Okt 29 16:53:20 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to download drv rsvd page
Okt 29 16:53:20 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to download beacon
Okt 29 16:57:07 odroid-h3 kernel: rtw_8812au 2-1:1.0: error beacon valid
Okt 29 16:57:07 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to download drv rsvd page
Okt 29 17:00:01 odroid-h3 CRON[326044]: pam_unix(cron:session): session opened for user root(uid=0)>
Okt 29 17:00:01 odroid-h3 CRON[326045]: (root) CMD (/etc/dns_monitor.sh)
Okt 29 17:00:03 odroid-h3 CRON[326044]: pam_unix(cron:session): session closed for user root
Okt 29 17:02:01 odroid-h3 su[320535]: pam_unix(su:session): session closed for user simon
Okt 29 17:02:39 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to allocate rx_skb
Okt 29 17:02:40 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to allocate rx_skb
Okt 29 17:02:49 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to send h2c command
Okt 29 17:02:49 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to send h2c command
Okt 29 17:02:49 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to send h2c command
Okt 29 17:02:57 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to send h2c command
Okt 29 17:02:57 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to send h2c command
Okt 29 17:03:05 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to send h2c command
Okt 29 17:03:05 odroid-h3 kernel: rtw_8812au 2-1:1.0: failed to send h2c command

dubhater, 2024-10-29:
@Simon566 I assume it stops working? Clients can't connect to it anymore? If that's the case, try the patch here: https://github.com/lwfinger/rtw88/issues/205#issuecomment-2437776828

Simon566, 2024-10-29:
yes it stops working ..... 

should the patch normally fix it ? i will give it a shot then. Whats wrong in the code ?

Simon566, 2024-10-29:
how to apply the patch to a out of tree rtw88 repo ? patch -p1 doesnt seem right or are you really trying to apply to kernel sources ? then Hunk #1 #2 are failing for 6.11.5

dubhater, 2024-10-29:
The patch should fix it. The problem is described here: https://lore.kernel.org/linux-wireless/baca0d5d-072e-4003-ab4a-ecc524d6b89b@gmail.com/

For me, ``patch -i alloc_skb.diff.txt`` applies it, no -p needed. It's meant for this repository.

morrownr, 2024-10-29:
@dubhater 

Idea: Let me apply the patch to this repo so that @Simon566 and others can just use `git pull` to update. What do you think? This repo is for testing.

dubhater, 2024-10-30:
@morrownr Okay, yes.

Simon566, 2024-10-30:
i was able to apply the patch , but wifi broke up this night. without any special error message or so .... see here:

Okt 30 03:06:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:06:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:06:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:07:36 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:07:36 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:07:36 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:07:36 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:07:54 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:07:54 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:07:54 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:07:54 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:09:17 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:09:17 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:09:17 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:09:17 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:09:22 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:09:22 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:09:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:09:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:09:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:09:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:09:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 03:09:23 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 03:09:48 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:49 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.157: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:49 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.238: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:51 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.13.188: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:52 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.157: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:52 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.238: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:55 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.157: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:55 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.238: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:56 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.138: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:58 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.157: Die Ressource ist zur Zeit nicht verfügbar
Okt 30 03:09:58 odroid-h3 dnsmasq-dhcp[1741]: Fehler beim Senden des DHCP-Pakets an 10.10.15.238: Die Ressource ist zur Zeit nicht verfügbar

the first indicator was that dhcp requests were not delivered or so , but the devices continued working when i look at my home assistant logfiles. but roughly 30mins later the data logging via wlan stopped completely ( maybe the devices lost their dhcp leases)

i wasnt able to log my handy into the wlan this morning.

could recover by:
 ifdown wlx24050f9d9b4b; rmmod rtw_8812au; rmmod rtw_usb; modprobe rtw_8812au; ifup wlx24050f9d9b4b

a simple restart of hostapd before didnt solve the issue

Simon566, 2024-10-30:
here is the hostapd failed restart log , maybe it helps ( please also read post above)

Okt 30 06:42:42 odroid-h3 python[1666]: 2024-10-30 06:42:42,737 [web_server.INFO] Calling script restart wlan. User {'ip': '10.10.202.3'}
Okt 30 06:42:42 odroid-h3 python[1666]: 2024-10-30 06:42:42,737 [script_server.execution_service.INFO] Calling script #802: /root/script-server/restart-wlan.sh
Okt 30 06:42:42 odroid-h3 python[1666]: 2024-10-30 06:42:42,741 [tornado.access.INFO] 200 POST /executions/start (10.10.202.3) 40.87ms
Okt 30 06:42:42 odroid-h3 python[1666]: 2024-10-30 06:42:42,840 [tornado.access.INFO] 101 GET /executions/io/802 (10.10.202.3) 1.13ms
Okt 30 06:42:42 odroid-h3 python[1666]: 2024-10-30 06:42:42,853 [script_server.audit_utils.WARNING] Could not get hostname for 10.10.202.3
Okt 30 06:42:42 odroid-h3 python[1666]: 2024-10-30 06:42:42,865 [script_server.audit_utils.WARNING] Could not get hostname for 10.10.202.3
Okt 30 06:42:42 odroid-h3 systemd[1]: Stopping hostapd_8812.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Okt 30 06:42:43 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 06:42:43 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 06:42:43 odroid-h3 kernel: rtw_8812au 1-2:1.0: error beacon valid
Okt 30 06:42:43 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to download drv rsvd page
Okt 30 06:42:43 odroid-h3 kernel: rtw_8812au 1-2:1.0 wlx24050f9d9b4b: left allmulticast mode
Okt 30 06:42:43 odroid-h3 kernel: rtw_8812au 1-2:1.0 wlx24050f9d9b4b: left promiscuous mode
Okt 30 06:42:43 odroid-h3 kernel: br-guest: port 2(wlx24050f9d9b4b) entered disabled state
Okt 30 06:42:44 odroid-h3 systemd[1]: hostapd_8812.service: Deactivated successfully.
Okt 30 06:42:44 odroid-h3 systemd[1]: Stopped hostapd_8812.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Okt 30 06:42:44 odroid-h3 systemd[1]: hostapd_8812.service: Consumed 2min 38.128s CPU time.
Okt 30 06:42:45 odroid-h3 kernel: rtw_8812au 1-2:1.0: failed to get tx report from firmware
Okt 30 06:42:45 odroid-h3 systemd[1]: Starting hostapd_8812.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
Okt 30 06:42:45 odroid-h3 kernel: br-guest: port 2(wlx24050f9d9b4b) entered blocking state

dubhater, 2024-11-01:
@morrownr On second thought, let's not apply that patch. There's another problem in the RX code. You can see it by running the iperf3 client with ``--udp -b 0``.

logan001, 2025-02-15:
same problem. tones of these messages.
i also use the adapter in AP mode only.
version e32c796 on arch linux - k 6.13 and asus usb ac68 (8814au )

dubhater, 2025-05-03:
"error beacon valid" may be fixed now. Please test it.

henkv1, 2025-05-04:
Yes, the latest update fixed the issue. Thank you.
Is this fix only cosmetic, or does it also improve reliability/stability/performance?

dubhater, 2025-05-04:
It's not just cosmetic. It's necessary to make power saving work for the clients. (But there is another problem with that.)

henkv1, 2025-05-20:
The messages reappeared after two days on my Pi5 with rtw_8821cu.
I also observed the messages on my Pi4 with rtw_8822cu for a short period of time (about one minute), two days ago.

i486, 2025-06-11:
I'm fairly certain I was able to reproduce this error consistently on the latest commit. I’ve since forgotten the exact steps, but I can try to retrace them if needed.

dubhater, 2025-06-11:
@i486 Yes, please. A reliable way to reproduce a problem is always welcome.

berenddeschouwer, 2025-06-20:
I'm very certain I can reproduce the error about 5 minutes after boot, with the last commit.  All I need to do is connect or scan for it with more than one device.  A few beacon errors, and it stops.

That's on a
```
rtw_8822bu 1-1:1.0: Firmware version 30.20.0, H2C version 14
rtw_8822bu 1-1:1.0: switching to USB 3 mode
```
commit 461b696b51317ba4ca585a4ddb32f2e72cd4efc9

I can up debugging.  What am I looking for?  What info do you need?

berenddeschouwer, 2026-08-16:
@dubhater the easiest way is to start an AP, connect from a client, and transfer a large file from the AP the the client.

I've been debugging a bit, and here's what I can determine:

1. there's data in the queue: len(tx_queue[0..3]) > 0.  Most common, tx_queue[2] = [2..5].  At least one queue, but sometimes more.
2. usb.c/rtw_usb_write_port_tx_complete() receives a few URB where urb.status == -EPROTO
3. usb.c/rtw_usb_write_port_complete() also receives a URB where urb.status == -EPROTO

After that, every time a beacon is sent:

1. usb.c/rtw_usb_write_port_complete() gets urb.status == -EPROTO
2. usb.c/rtw_usb_write_port_tx_complete() *also* gets urb.status == -EPROTO if it happens at the same time.  If no beacon is being sent, urb.status == 0.

Also at the same time as the *first* failed beacon, if a there's a tx_report running, tx.c/rtw_tx_report_purge_timer() will trigger.  It looks like this only happens with the first failed beacon.

The only way I know to trigger tx_report, is by running an AP probe.

I've tried a few simple hacks, like re-submitting the beacon URB once, if urb.status == -EPROTO, but that doesn't seem to help.

I hope this is helpful.

dubhater, 2026-08-16:
Please try these new AP mode fixes:

https://github.com/lwfinger/rtw88/pull/455

https://lore.kernel.org/linux-wireless/20260813131845.1330003-1-mehmet.fide@gmail.com/

berenddeschouwer, 2026-08-19:
@dubhater 48 hours, still looking solved