lwfinger/rtw88 #242: [RTW8822CE] failed to connect 5ghz wifi

vtolstov, 2024-10-20:
i have latest stable fedora, and rtl8822ce network card. Why i cant see any wifi 5ghz ap? if i'm create manually connection to my 5ghz hotspot it fails to connect.
my macbook air succeseful connect to the same 5ghz hotspot without any issues (hotspot on mikrotik cap ax)

dubhater, 2024-10-20:
Maybe your 5 GHz channels are all disabled. What is the output of ``iw reg get`` and ``iw phy``?

vtolstov, 2024-10-20:
this is from linux with realtek 
```
global
country RU: DFS-UNSET
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5350 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(5650 - 5850 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(5925 - 6425 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A), NO-OUTDOOR

phy#0 (self-managed)
country US: DFS-FCC
	(2402 - 2472 @ 40), (6, 30), (N/A)
	(5170 - 5250 @ 80), (6, 24), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (6, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (6, 24), (0 ms), DFS, AUTO-BW
	(5735 - 5855 @ 80), (6, 30), (N/A), AUTO-BW
	(5855 - 5925 @ 40), (6, 30), (N/A), NO-OUTDOOR, AUTO-BW
	(5925 - 7125 @ 160), (N/A, 24), (N/A), NO-OUTDOOR, AUTO-BW
```

[image]
this is screenshot from output from macos

this is from another server with linux and intel wifi 
```
iw reg get
global
country RU: DFS-UNSET
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5350 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(5650 - 5850 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(5925 - 6425 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A), NO-OUTDOOR

phy#0 (self-managed)
country RU: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(5170 - 5190 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5190 - 5210 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5210 - 5230 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5230 - 5250 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5250 - 5270 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5270 - 5290 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5290 - 5310 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5310 - 5330 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5490 - 5510 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5510 - 5530 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5530 - 5550 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5550 - 5570 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5570 - 5590 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5590 - 5610 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5610 - 5630 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-320MHZ, PASSIVE-SCAN
	(5630 - 5650 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-320MHZ, PASSIVE-SCAN
	(5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
	(5815 - 5835 @ 40), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-80MHZ, NO-160MHZ, NO-320MHZ, PASSIVE-SCAN
```

dubhater, 2024-10-20:
I'm not sure why it says "phy#0 (self-managed)". I don't get that with RTL8822CE.

What is the output of ``iw phy``?

vtolstov, 2024-10-20:
```
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 323 bytes
	max # sched scan SSIDs: 4
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x19ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412.0 MHz [1] (20.0 dBm)
			* 2417.0 MHz [2] (20.0 dBm)
			* 2422.0 MHz [3] (20.0 dBm)
			* 2427.0 MHz [4] (20.0 dBm)
			* 2432.0 MHz [5] (20.0 dBm)
			* 2437.0 MHz [6] (20.0 dBm)
			* 2442.0 MHz [7] (20.0 dBm)
			* 2447.0 MHz [8] (20.0 dBm)
[... 160 lines cut ...]
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ POWERED_ADDR_CHANGE ]: can change MAC address while up
```

vtolstov, 2024-10-20:
> I'm not sure why it says "phy#0 (self-managed)". I don't get that with RTL8822CE.
> 
> What is the output of `iw phy`?

i'm provide few outputs on various devices that have connection to 5ghz. for realtek that have issues i have only
```
global
country RU: DFS-UNSET
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5350 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(5650 - 5850 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(5925 - 6425 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A), NO-OUTDOOR

```

dubhater, 2024-10-20:
``* 5500.0 MHz [100] (disabled)`` <- This is why you don't see your AP. It should work if you move the network to one of the channels which aren't disabled.

dubhater, 2024-10-20:
According to Wikipedia, channels 96-128 are not allowed in Russia: https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/n/ac/ax/be)

vtolstov, 2024-10-20:
hmm thanks, so i need to change country in ap and on linux ?

dubhater, 2024-10-20:
You can change the channel or the country, but changing the country is probably not legal.

vtolstov, 2024-10-20:
Big thanks

closed 2024-10-20 by vtolstov

PD-Kerman, 2025-01-19:
Hi there

is there any progress on this issue yet~?

encountering pretty much exactly the same issue here

RTL8821CU, but couldnt connect to / see a single SSID in 5ghz, not even any from my neighbors

changed all my router/ap(s) to channel 149 to avoid dfs, crda/wireless-regdb installed, using the latest driver from this repo, got headers installed first, then followed the guide and installed the driver under this very repository, but still no luck after all.

the no-ir mark in iwlist / iw phy constanly changes tho, it goes between completely without no-ir mark to all channels marked as no-ir mainly related to sleep hibernate reboot things like that, and as i remembered that it did connect to a 5ghz wifi once and the bitrate was promising, and even with no-ir marked at that time so i guess the accuracy of the result being shown in iw should be up for discission

and further more, just discovered some *weird* detection these days

its like, my ap(s) takes over several SSIDs, and there are one for 2.4band and one for 5ghz band

and as i discovered these days as i thought it would be no longer be able to detect any 5ghz signals,
weirdly when i check wavemon for scanning outputs, some 5ghz signals just *sneaked out* for like tens of seconds or so, and disappeared, and they were usually like without SSIDs or the SSIDs were for the signals on 2.4ghz band which should be wrong/impossible for such to be used

and one more thing is that
i went and borrowed a RTL8188GU dongle yesterday and tested with it
everything seemed to be working fine, its got all SSIDs on 5GHz scanned out, and most importanly 
what i found was like
Those 5GHz signals which could not be found/recognized by RTL8821CU earlier all shown in wavemon with a **Spectrum Mgmt** mark which RTL8821CU happened to not having it/those signal detected, and all tests above were done with at least 1 nearby 5ghz ap set to channel 149 and/or channel 36

P.S.  even though that it could mysteriously work with that specific 5GHz AP that all other AP/Devices cannot, not even with the exact bandwith & channel being set, it still had got weird/teribble non-comprehensive scanning results for all the time being

Nevertheless, the 8821CU ive got is still not being able to detect any 5ghz signals

Wish there would be some solutions

Any kinds of help will be appreciated !

Thanks in advanced,
And best wishes to all contributors,
Best Regards