lwfinger/rtw88 #245: rtw_8822cs  : no AP mode ?

chris4git, 2024-10-28:
Hi,
Context : Armbian bookworm 24.11 6.6.58-current-rockchip64
After make/make install of current version of the driver (and reboot)
iw list gives only managed and  monitor modes available.
Under Android all modes are available.

Logs :
Oct 28 15:08:39 station-m2 kernel: rtw_8822cs mmc1:0001:1: WOW Firmware version 9.9.4, H2C version 15
Oct 28 15:08:39 station-m2 kernel: rtw_8822cs mmc1:0001:1: Firmware version 9.9.15, H2C version 15

Something necessary at compilation time to activate AP mode ?

Chris

dubhater, 2024-10-28:
Yes, you need to make this change:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/main.c b/drivers/net/wireless/realtek/rtw88/main.c
index e91530ed05a0..4ba821dcc7eb 100644
--- a/drivers/net/wireless/realtek/rtw88/main.c
+++ b/drivers/net/wireless/realtek/rtw88/main.c
@@ -2228,7 +2228,7 @@ EXPORT_SYMBOL(rtw_core_deinit);
 
 int rtw_register_hw(struct rtw_dev *rtwdev, struct ieee80211_hw *hw)
 {
-	bool sta_mode_only = rtwdev->hci.type == RTW_HCI_TYPE_SDIO;
+	bool sta_mode_only = false;
 	struct rtw_hal *hal = &rtwdev->hal;
 	int max_tx_headroom = 0;
 	int ret;
```

SDIO (and USB) devices had some problems with AP mode, so someone decided that AP mode should be disabled until the problems are fixed. It's very possible that the problems are [all](https://github.com/lwfinger/rtw88/commit/8d8578a2e16b7732ab7fb4c4fece1642f7406302) [fixed](https://github.com/lwfinger/rtw88/commit/557078c4c9f9a3c5511101b0f7a6b312d8143ece) now. People with USB devices reported that it's working better. I don't know if anyone tested the SDIO side.

chris4git, 2024-10-29:
Thanks @dubhater ! Now all modes are present.
Next step : to get a working setup with armbian-config...

chris4git, 2024-11-01:
Hi,
no success ! 
I tried directly with hostapd with this config :

```
interface=wlan0
driver=nl80211
ssid=test
hw_mode=g
channel=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=mustbechanged
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```
the ssid "test" is seen but the connections attempts failed.
wpa_supplicant side one can see :
```
wpa_supplicant[1202]: wlp2s0: SME: Trying to authenticate with 4c:24:ce:cd:dd:38 (SSID='test' freq=2412 MHz)
wpa_supplicant[1202]: wlp2s0: Trying to associate with 4c:24:ce:cd:dd:38 (SSID='test' freq=2412 MHz)
wpa_supplicant[1202]: wlp2s0: Associated with 4c:24:ce:cd:dd:38
wpa_supplicant[1202]: wlp2s0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wpa_supplicant[1202]: wlp2s0: Authentication with 4c:24:ce:cd:dd:38 timed out.
```
and hostapd side :
```
nl80211: RX frame da=4c:24:ce:cd:dd:38 sa=5c:87:9c:fb:c3:ef bssid=4c:24:ce:cd:dd:38 freq=2412 ssi_signal=-55 fc=0x0 seq_ctrl=0x60 stype=0 (WLAN_FC_STYPE_ASSOC_REQ) len=104
wlan0: Event RX_MGMT (18) received
mgmt::assoc_req
association request: STA=5c:87:9c:fb:c3:ef capab_info=0x431 listen_interval=10 seq_ctrl=0x60
  new AID 1
wlan0: STA 5c:87:9c:fb:c3:ef IEEE 802.11: association OK (aid 1)
Add associated STA 5c:87:9c:fb:c3:ef (added_unassoc=1 auth_alg=0 ft_over_ds=0 reassoc=0 authorized=0 ft_tk=0 fils_tk=0)
nl80211: Set STA 5c:87:9c:fb:c3:ef
  * supported rates - hexdump(len=12): 02 04 0b 16 0c 12 18 24 30 48 60 6c
  * capability=0x431
  * aid=1
  * listen_interval=10
  * flags set=0xa4 mask=0xa4
nl80211: send_mlme - da=5c:87:9c:fb:c3:ef noack=0 freq=0 no_cck=0 offchanok=0 wait_time=0 no_encrypt=0 fc=0x10 (WLAN_FC_STYPE_ASSOC_RESP) nlmode=3
nl80211: send_mlme - Use bss->freq=2412
nl80211: send_mlme -> send_frame_cmd
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan0
nl80211: RX frame da=4c:24:ce:cd:dd:38 sa=5c:87:9c:fb:c3:ef bssid=4c:24:ce:cd:dd:38 freq=2412 ssi_signal=-56 fc=0xc0 seq_ctrl=0x70 stype=12 (WLAN_FC_STYPE_DEAUTH) len=26
wlan0: Event RX_MGMT (18) received
```
It seems wpa_supplicant didn't received something expected.
Any idea of the problem ?

Chris

dubhater, 2024-11-01:
I don't know. Maybe there is something in dmesg on the AP side.

chris4git, 2024-11-01:
Nothing in dmesg.
I tried the same hostapd configuration with a USB adapter (8192eu/rtl8xxxu) and it worked :

```
wlx7cc2c609e437: interface state UNINITIALIZED->COUNTRY_UPDATE
wlx7cc2c609e437: interface state COUNTRY_UPDATE->ENABLED
wlx7cc2c609e437: AP-ENABLED 
wlx7cc2c609e437: STA 5c:87:9c:fb:c3:ef IEEE 802.11: authenticated
wlx7cc2c609e437: STA 5c:87:9c:fb:c3:ef IEEE 802.11: associated (aid 1)
wlx7cc2c609e437: AP-STA-CONNECTED 5c:87:9c:fb:c3:ef
wlx7cc2c609e437: STA 5c:87:9c:fb:c3:ef RADIUS: starting accounting session D217E0BED48CC56D
wlx7cc2c609e437: STA 5c:87:9c:fb:c3:ef WPA: pairwise key handshake completed (RSN)
wlx7cc2c609e437: EAPOL-4WAY-HS-COMPLETED 5c:87:9c:fb:c3:ef
wlx7cc2c609e437: AP-STA-DISCONNECTED 5c:87:9c:fb:c3:ef
wlx7cc2c609e437: interface state ENABLED->DISABLED
wlx7cc2c609e437: AP-DISABLED 
wlx7cc2c609e437: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlx7cc2c609e437 disabled_11b_rates=0
```
With the rtw_8822cs :

```
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0

```
The authentication/association don't go to the end.
Something to do with mac80211.c  ?

dubhater, 2024-11-01:
The problem could be anywhere.

lone0, 2025-03-20:
I encountered the same problem on rtl8723cs in AP mode. And I think I have found the problem.

After hostapd sends out association ack to STA in step "associated (aid 1)", it expect an TX status reports to trigger function handle_assoc_cb(), where it either set the STA authorized or conducts further 4-way handshake.

TX status reports never come up.

Look at the place where rtw88-sdio sends tx status, you will see the following:

```
static void rtw_sdio_process_tx_queue()
{
    ...

	if (queue <= RTW_TX_QUEUE_VO)
		rtw_sdio_indicate_tx_status(rtwdev, skb);
	else
		dev_kfree_skb_any(skb);
}
```

That means packets from queue whose ID is larger than RTW_TX_QUEUE_VO (value 0x3) will not have tx status handling. The association ack goes through queue RTW_TX_QUEUE_MGMT (value 0x5) and gets freed right after transmitted without tx status reporting.

I did a quick fix:

 		rtw_sdio_indicate_tx_status(rtwdev, skb);
-	else
+	else if (queue == RTW_TX_QUEUE_MGMT) {
+		IEEE80211_SKB_CB(skb)->flags |= IEEE80211_TX_STAT_ACK;
+		ieee80211_tx_status_irqsafe(rtwdev->hw, skb);
+	} else
 		dev_kfree_skb_any(skb);

And I can see that STAs get authorized and start to tx/rx.

dubhater, 2025-03-20:
@lone0 Good catch! I think rtw_sdio_indicate_tx_status() should be called unconditionally, no matter what queue.

I think it's called only for the AC queues in order to hide some "failed to get tx report" messages... because the management frames are sent to the high queue and the chip is not configured to provide TX reports for the high queue.

@xdarklight @jernejsk Do y'all remember why it's like this? The old RTL8723DS driver seems to use the management queue for management frames, not the high queue.

```C
static u8 rtw_sdio_get_tx_qsel(struct rtw_dev *rtwdev, struct sk_buff *skb,
			       u8 queue)
{
	switch (queue) {
	case RTW_TX_QUEUE_BCN:
		return TX_DESC_QSEL_BEACON;
	case RTW_TX_QUEUE_H2C:
		return TX_DESC_QSEL_H2C;
	case RTW_TX_QUEUE_MGMT:
		if (rtw_chip_wcpu_11n(rtwdev))           // ????
			return TX_DESC_QSEL_HIGH;
		else
			return TX_DESC_QSEL_MGMT;
```

jernejsk, 2025-03-21:
@dubhater I don't remember anymore. 8822bs and 8822cs drivers were used as a base for writing SDIO support.

AndreySV, 2025-03-25:
@lone0 Thanks for the change.! I've tried your change on 6.12 on PinePhone with rtl8723cs and can confirm AP mode is working and sharing Internet using it is working okay.

It would be nice to get this fix merge upstream. If you need testing, I'm ready to help.

xdarklight, 2025-03-30:
@lone0 thanks for your patch!
I gave it a quick spin on my RTW8822CS in `sta` ("client") mode and there aren't any new/unexpected kernel logs.
Even calling `rtw_sdio_indicate_tx_status(rtwdev, skb);` unconditionally does not result in any extra log messages.
Are you going to send the patch upstream? If you want to do it but are not sure how: I'm happy to help.

@dubhater First of all: thanks for your work on the rtw88 driver!
As @jernejsk said we mainly focused on RTL8822BS and RTL8822CS - with an occasional glance at the RTL8723DS and RTL8723BS driver code. I also don't recall where the highlighted condition came from.
There was approx. one year gap between getting RTL8822BS/CS working and first testing of RTL8723DS - which then resulted in https://github.com/torvalds/linux/commit/e967229ead0e6c5047a1cfd5a0db58ceb930800b
I'm sure there's more hidden issues (for both, 802.11ac and 802.11n chipsets) - so any suggestions are welcome!

lone0, 2025-03-31:
> [@lone0](https://github.com/lone0) thanks for your patch! I gave it a quick spin on my RTW8822CS in `sta` ("client") mode and there aren't any new/unexpected kernel logs. Even calling [...]

I will be very glad to see my favorite SBC (Licheepi 4a, riscv core + 8723ds) get fully supported by upstream. This tiny patch is experimental just to highlight the tx report issue. There are questions like: Shall we wait for hardware TX report before handing it to ieee80111? Shall the rtw_sdio_indicate_tx_status() be generalized and cover all packets, rather than just management frames? I would like to see reviews from the realtek/wifi driver experts. What are your thoughts?

xdarklight, 2025-03-31:
> I will be very glad to see my favorite SBC (Licheepi 4a, riscv core + 8723ds) get fully supported by upstream.

nice riscv board!

> This tiny patch is experimental just to highlight the tx report issue. There are questions like: Shall we wait for hardware TX report before handing it to ieee80111? Shall the rtw_sdio_indicate_tx_status() be generalized and cover all packets, rather than just management frames?

Those are excellent questions

> I would like to see reviews from the realtek/wifi driver experts. What are your thoughts?

I'm not sure if @pkshih typically works with github.
However, you can start by sending a patch to the mailing list:
- clone http://github.com/torvalds/linux
- checkout the `v6.14` tag
- apply your change
- `git commit -s drivers/net/wireless/realtek/rtw88/`
  - in the first line use: `wifi: rtw88: sdio: ...your description here...`
  - keep a blank line
  - describe the change
- then run: `git format-patch HEAD~1 --subject-prefix="RFC -v1` (this assumes you have one patch, if you have more than one adjust the `~1` and append `--cover-letter`)
  - RFC stands for "request for comments" (meaning it's not a final patch, and you're looking for feedback from the maintainers)
  - if you send a second version of the patch, change `-v1` to `-v2`, ...
- send the patch with `git send-email --to="linux-wireless@vger.kernel.org" --cc="pkshih@realtek.com" --cc="linux-kernel@vger.kernel.org"`
  - Ping-Ke Shih is the upstream rtw88 driver maintainer and he works for Realtek. When I started sending rtw88 sdio patches he was super friendly and helpful!
  - you can also Cc me if you want
- wait for an answer

sbwml, 2025-07-22:
I enabled AP for rtl8822cs on OpenWrt and got a lot of logs. What should I do? 😭

```
[ 4033.015391] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4033.020718] br-lan: port 2(phy0-ap0) entered blocking state
[ 4033.020735] br-lan: port 2(phy0-ap0) entered disabled state
[ 4033.020785] rtw_8822cs mmc1:0001:1 phy0-ap0: entered allmulticast mode
[ 4033.021034] rtw_8822cs mmc1:0001:1 phy0-ap0: entered promiscuous mode
[ 4033.021142] br-lan: port 2(phy0-ap0) entered blocking state
[ 4033.021154] br-lan: port 2(phy0-ap0) entered forwarding state
[ 4033.023085] br-lan: port 2(phy0-ap0) entered disabled state
[ 4033.122776] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4033.122797] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4033.122803] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4033.122808] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
...
[ 4033.741607] rtw_8822cs mmc1:0001:1: failed to do dpk calibration
[ 4033.741635] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4033.741647] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4033.741655] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
...
[ 4033.988278] rtw_8822cs mmc1:0001:1: failed to do dpk calibration
[ 4034.049910] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4034.049995] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4034.050065] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4034.050133] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4034.050201] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4034.050269] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4034.050337] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4034.052933] br-lan: port 2(phy0-ap0) entered blocking state
[ 4034.052945] br-lan: port 2(phy0-ap0) entered forwarding state
[ 4036.035146] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
[ 4036.036323] rtw_8822cs mmc1:0001:1: unsupported rf path (1)
```

```
[root@OpenWrt ~]# iw dev phy0-ap0 info
Interface phy0-ap0
	ifindex 22
	wdev 0x9
	addr 70:68:71:a2:a8:0f
	ssid OpenWrt
	type AP
	wiphy 0
	channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
	txpower 23.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
[root@OpenWrt ~]#
```

sbwml, 2025-07-23:
> The following has only been reported today on the linux-wireless mailing list: https://lore.kernel.org/linux-wireless/7407FAA3-F4A4-48AA-91E5-A78DA711447D@gmail.com/T/#u
> 
> > rtl8822cs, mainline 6.16-rc7: kernel reports ' unsupported rf path'
> 
> Your log additionally has `failed to do dpk calibration`. It's not clear to me if that's related or not.
> 
> Suggestion: watch the thread on the linux-wireless mailing list. Once the cause of `unsupported rf path` has been fixed: test again and see if `failed to do dpk calibration` is still present.

Thanks, it's been resolved now