lwfinger/rtw88 #246: fail

briansune, 2024-10-31:
@lwfinger 

We are using RTL8723DU USB dongle on UBUNTU LTS 20.04.2 LTS
There are a lot of report on failed to get tx report from firmware.
How to solve this issue and we do see there are case that the driver will crash.
Attaching dmesg report below.
Thank you

>[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000000] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008198] Console: colour dummy device 80x25
[    0.012667] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022986] pid_max: default: 32768 minimum: 301
[    0.027761] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035057] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044141] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051376] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057515] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064725] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.070953] rcu: Hierarchical SRCU implementation.
[    0.075743] rcu:     Max phase no-delay instances is 1000.
[    0.081236] EFI services will not be available.
[    0.086013] smp: Bringing up secondary CPUs ...
[... 302 lines cut ...]
[ 3309.068523] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 3343.659618] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 3609.060794] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 3728.929580] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 3908.988718] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 3936.187982] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4015.545827] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4030.489414] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4047.672947] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4103.479423] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4208.021498] rtw_8723du 1-1:1.2: failed to allocate rx_skb
[ 4208.095236] rtw_8723du 1-1:1.2: failed to allocate rx_skb
[ 4208.123961] rtw_8723du 1-1:1.2: failed to allocate rx_skb
[ 4208.156729] rtw_8723du 1-1:1.2: failed to allocate rx_skb
[ 4209.076586] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4210.388549] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4213.204760] wlx90de80882cf1: authenticate with f8:6f:b0:0e:ae:e5
[ 4213.204825] wlx90de80882cf1: 80 MHz not supported, disabling VHT
[ 4214.783981] wlx90de80882cf1: send auth to f8:6f:b0:0e:ae:e5 (try 1/3)
[ 4215.316386] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4215.864583] wlx90de80882cf1: send auth to f8:6f:b0:0e:ae:e5 (try 2/3)
[ 4216.372322] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4216.889058] wlx90de80882cf1: send auth to f8:6f:b0:0e:ae:e5 (try 3/3)
[ 4217.396288] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 4217.876326] wlx90de80882cf1: authentication with f8:6f:b0:0e:ae:e5 timed out

dubhater, 2024-11-01:
Sadly Larry died this summer.

> [ 4208.021498] rtw_8723du 1-1:1.2: failed to allocate rx_skb

This is an old problem that only recently got an error message. A patch is available here: https://github.com/lwfinger/rtw88/issues/205#issuecomment-2437776828

For the random "failed to get tx report from firmware", try to load rtw_core with debug_mask=0x10. This may add some useful information to the log.

briansune, 2024-11-01:
@dubhater 

I don't even notice Larry referring to lwfinger had passed away.
Sad, although we were not able to resolve the blank-chip firmware issue at the pass, yet he is still a very nice guy on assisting.

I will give it a try and pass out some logs.

Thanks again.

briansune, 2024-11-01:
@dubhater 

No luck, the memory stress test immediately trigger the issue

>[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000001] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008197] Console: colour dummy device 80x25
[    0.012667] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022987] pid_max: default: 32768 minimum: 301
[    0.027760] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035057] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044147] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051382] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057521] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064732] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.070960] rcu: Hierarchical SRCU implementation.
[    0.075751] rcu:     Max phase no-delay instances is 1000.
[    0.081247] EFI services will not be available.
[    0.086022] smp: Bringing up secondary CPUs ...
[    0.091029] Detected VIPT I-cache on CPU1
[    0.091122] GICv3: CPU1: found redistributor 1 region 0:0x00000000dd060000
[    0.091167] CPU1: Booted secondary processor 0x0000000001 [0x411fd040]
[    0.091276] smp: Brought up 1 node, 2 CPUs
[... 656 lines cut ...]
[  301.962510] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  301.990705] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  301.991574] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  301.992133] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x3c, len=7
[  301.992328] rtw_8723du 1-1:1.2: send H2C content 2b000042 00000000
[  301.998209] rtw_8723du 1-1:1.2: send H2C content 00000058 00000212
[  302.000710] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  302.001077] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  303.978466] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  304.006792] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  304.007662] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[... cut ...]
[  305.962458] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  305.990759] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  305.991625] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  305.992182] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x3e, len=7
[  305.992374] rtw_8723du 1-1:1.2: send H2C content 2d000042 00000000
[  305.998631] rtw_8723du 1-1:1.2: send H2C content 00000058 00000212
[  306.001140] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  306.001498] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000

briansune, 2024-11-01:
@dubhater 

Good news is that the driver didn't crash and ping can keep run after those fail reports.
I had undergo another short test dmesg as follows:

>ubuntu@briansune:~$ dmesg
[ 8809.861469] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8809.873487] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8809.878731] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8809.879344] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8811.847009] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8811.875341] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8811.876320] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8811.876955] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x06, len=7
[ 8811.877298] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8811.887693] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8811.892283] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8811.895430] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8813.830932] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8813.859284] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8813.860273] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8813.860904] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x07, len=7
[ 8813.861516] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8813.867663] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8813.870012] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8813.870506] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8815.847396] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8815.875236] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8815.876238] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8815.876867] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x08, len=7
[ 8815.877494] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8815.883474] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8815.883967] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[ 8815.886216] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8815.886780] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8815.902496] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x09, len=7
[ 8816.004643] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x0a, len=7
[ 8817.830784] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8817.859220] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8817.860197] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8817.860780] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x0b, len=7
[ 8817.861182] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8817.867179] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8817.869449] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8817.869930] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8819.846786] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8819.875155] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8819.876154] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8819.876790] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x0c, len=7
[ 8819.877135] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8819.890091] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8819.892527] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8819.893021] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8821.830774] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8821.859111] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8821.860114] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8821.860747] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x0d, len=7
[ 8821.861350] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8821.871488] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8821.873747] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8821.874226] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8823.846753] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8823.875073] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8823.876071] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8823.876650] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x0e, len=7
[ 8823.877056] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[ 8823.882934] rtw_8723du 1-1:1.2: send H2C content 00000058 00000213
[ 8823.883430] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[ 8823.885695] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8823.886187] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[... 510 lines cut ...]
[ 8941.864551] rtw_8723du 1-1:1.2: send H2C content 00000058 00000212
[ 8941.866919] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8941.867288] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8943.812211] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8943.840532] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8943.841507] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8943.842050] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x69, len=7
[ 8943.842254] rtw_8723du 1-1:1.2: send H2C content 27000042 00000000
[ 8943.854527] rtw_8723du 1-1:1.2: send H2C content 00000058 00000212
[ 8943.859265] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8943.859746] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[... cut ...]
[ 8945.864479] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[ 8945.866845] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 8945.867335] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 8946.050347] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x6b, len=7
[ 8947.812140] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 8947.840306] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 8947.841294] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 8947.841976] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x6c, len=7

dubhater, 2024-11-01:
The timing here is interesting:
```
[ 8856.551252] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x24, len=8
[ 8857.061876] rtw_8723du 1-1:1.2: failed to get tx report from firmware
```
```
[ 8911.866562] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x44, len=8
...
[ 8912.388721] rtw_8723du 1-1:1.2: failed to get tx report from firmware
```

C2H with id=0x03 is the TX report. "failed to get tx report from firmware" is a bit misleading. It's possible the firmware did send the TX report but the driver received it earlier than expected. I think this can happen with rtw_usb. We need more information:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/tx.c b/drivers/net/wireless/realtek/rtw88/tx.c
index 215bfd731235..8994f316f367 100644
--- a/drivers/net/wireless/realtek/rtw88/tx.c
+++ b/drivers/net/wireless/realtek/rtw88/tx.c
@@ -181,11 +181,12 @@ void rtw_tx_report_purge_timer(struct timer_list *t)
 	struct rtw_dev *rtwdev = from_timer(rtwdev, t, tx_report.purge_timer);
 	struct rtw_tx_report *tx_report = &rtwdev->tx_report;
 	unsigned long flags;
+	u32 len = skb_queue_len(&tx_report->queue);
 
-	if (skb_queue_len(&tx_report->queue) == 0)
+	if (len == 0)
 		return;
 
-	rtw_warn(rtwdev, "failed to get tx report from firmware\n");
+	rtw_warn(rtwdev, "still have %u unprocessed tx report requests\n", len);
 
 	spin_lock_irqsave(&tx_report->q_lock, flags);
 	skb_queue_purge(&tx_report->queue);
@@ -214,6 +215,7 @@ void rtw_tx_report_enqueue(struct rtw_dev *rtwdev, struct sk_buff *skb, u8 sn)
 	/* pass sn to tx report handler through driver data */
 	drv_data = (u8 *)IEEE80211_SKB_CB(skb)->status.status_driver_data;
 	*drv_data = sn;
+pr_err("queueing tx report request with sn %d\n", sn);
 
 	spin_lock_irqsave(&tx_report->q_lock, flags);
 	__skb_queue_tail(&tx_report->queue, skb);
@@ -256,6 +258,7 @@ void rtw_tx_report_handle(struct rtw_dev *rtwdev, struct sk_buff *skb, int src)
 		sn = GET_CCX_REPORT_SEQNUM_V1(c2h->payload);
 		st = GET_CCX_REPORT_STATUS_V1(c2h->payload);
 	}
+pr_err("received tx report with sn %d\n", sn);
 
 	spin_lock_irqsave(&tx_report->q_lock, flags);
 	skb_queue_walk_safe(&tx_report->queue, cur, tmp) {

```

You don't need debug_mask anymore.

briansune, 2024-11-01:
@dubhater 

It feels like the TX issue had gone, if I am understanding correctly the message is just a timing issue.

After a very short test it feels like the "failed to allocate rx_skb #" still can trigger easily.

How I am triggering this is undergo a system stress test on main-memory.
Always can trigger this on running stressapptest

```
ubuntu@briansune:~$ cat ./tests.sh
#!/bin/bash

echo "start tests"

while true
do
        ping -c 4 8.8.8.8
        free -h
        stressapptest -s 60
        free -h
        sudo memtester 200M 1
done
```

The main-memory is:

```
              total        used        free      shared  buff/cache   available
Mem:          487Mi        70Mi       256Mi       0.0Ki       159Mi       404Mi
Swap:         1.0Gi          0B       1.0Gi
```

<details>

<summary>dmesg</summary>

ubuntu@briansune:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[... 1158 lines cut ...]
[  259.975844] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  259.976400] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x7d, len=7
[  259.976595] rtw_8723du 1-1:1.2: send H2C content 25000042 00000000
[  259.982972] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020e
[  259.983347] rtw_8723du 1-1:1.2: send H2C content 49810040 000fe000
[  259.985591] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  259.985965] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  261.962761] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  261.991057] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  261.991929] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  261.992491] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x7e, len=7
[  261.992680] rtw_8723du 1-1:1.2: send H2C content 26000042 00000000
[  261.998936] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020e
[  262.001309] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  262.001679] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  263.978737] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  264.007021] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  264.007888] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  264.008442] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x7f, len=7
[  264.008640] rtw_8723du 1-1:1.2: send H2C content 27000042 00000000
[  264.014895] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020e
[  264.017141] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  264.017512] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000

</details>

briansune, 2024-11-01:
@dubhater 

Bad news, I just discover a couples of TX issue.

<details>

<summary>dmesg</summary>

ubuntu@briansune:~$ dmesg
[  107.984739] rtw_8723du 1-1:1.2: send H2C content 49810040 000fe000
[  107.987116] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  107.987483] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  108.106614] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x23, len=7
[  109.962836] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  109.963700] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  109.965468] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  109.966038] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x24, len=7
[  109.966464] rtw_8723du 1-1:1.2: send H2C content 25000042 00000000
[  109.980469] rtw_8723du 1-1:1.2: send H2C content 00000058 00001010
[  109.982969] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  109.983462] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  110.359674] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x25, len=7
[  111.946836] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  111.975303] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  111.976303] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  111.976943] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x26, len=7
[  111.977285] rtw_8723du 1-1:1.2: send H2C content 26000042 00000000
[  111.983663] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[  111.985912] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  111.986283] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  113.962807] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  113.991267] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  113.992265] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  113.992890] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x27, len=7
[  113.993244] rtw_8723du 1-1:1.2: send H2C content 24000042 00000000
[  113.999624] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[  114.002005] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  114.002491] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  115.946804] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  115.975349] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  115.976352] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  115.976955] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x28, len=7
[  115.977331] rtw_8723du 1-1:1.2: send H2C content 25000042 00000000
[  115.983710] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[  115.984086] rtw_8723du 1-1:1.2: send H2C content 49810040 000fe000
[  115.986336] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  115.986757] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  117.962875] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  117.991312] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  117.992309] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  117.992922] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x29, len=7
[  117.993297] rtw_8723du 1-1:1.2: send H2C content 26000042 00000000
[  117.999668] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[  118.002051] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  118.002542] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  119.978944] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  120.007275] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  120.008266] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  120.008860] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x2a, len=7
[  120.009387] rtw_8723du 1-1:1.2: send H2C content 25000042 00000000
[  120.022013] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[  120.024511] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  120.025016] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  121.962818] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  121.991362] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  121.992353] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  121.992940] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x2b, len=7
[  121.993346] rtw_8723du 1-1:1.2: send H2C content 25000042 00000000
[  122.005615] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[  123.979448] rtw_8723du 1-1:1.2: send H2C content 25000042 00000000
[... 2730 lines cut ...]
[  838.964265] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[  839.948203] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x20, len=8
[  840.971228] rtw_8723du 1-1:1.2: send H2C content 28000042 00000000
[  840.977721] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020e
[  842.955311] rtw_8723du 1-1:1.2: send H2C content 29000042 00000000
[  842.961808] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020e
[  844.971144] rtw_8723du 1-1:1.2: send H2C content 29000042 00000000
[  844.979017] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020e
[  844.981265] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  844.981631] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  845.069496] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x21, len=7
[  846.955233] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  846.983247] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  846.984231] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  846.984829] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x22, len=7
[... cut ...]
[  846.992108] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[  846.994488] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[  846.995105] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[  847.117502] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x23, len=7
[  848.970804] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[  848.999108] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[  849.000055] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[  849.000745] rtw_8723du 1-1:1.2

dubhater, 2024-11-01:
> After a very short test it feels like the "failed to allocate rx_skb #" still can trigger easily.

Yes, that's normal. The memory allocation sometimes fails, but it's retried. I will remove the message eventually.

This latest log makes me think you're seeing the same bug I ran into and fixed recently. I haven't published the fix yet. But to confirm, you should apply that patch I gave you above and let it run for a while.

briansune, 2024-11-01:
@dubhater 

Those two dmesg had already applied the patch you gave me from
It is recompiled and replaced accordingly.

> The timing here is interesting:
>
> ```
> [ 8856.551252] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x24, len=8
> [ 8857.061876] rtw_8723du 1-1:1.2: failed to get tx report from firmware
> ```
>
> ```
> [ [...]

briansune, 2024-11-01:
@dubhater 

I had run a longer test the log is attached here:

<details>

<summary>dmesg</summary>

ubuntu@briansune:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000001] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008197] Console: colour dummy device 80x25
[    0.012667] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022987] pid_max: default: 32768 minimum: 301
[    0.027761] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035059] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044144] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051379] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057517] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064728] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.070956] rcu: Hierarchical SRCU implementation.
[    0.075746] rcu:     Max phase no-delay instances is 1000.
[    0.081238] EFI services will not be available.
[... 898 lines cut ...]
[ 3584.038262] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.047852] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.181239] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.181738] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.181996] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.189656] rtw_8723du 1-1:1.2: failed to allocate rx_skb 3
[ 3584.254693] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[ 3584.255276] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.255600] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.255840] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.439155] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[ 3584.505823] rtw_8723du 1-1:1.2: failed to allocate rx_skb 3
[ 3584.609954] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[ 3584.610294] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.612851] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[ 3584.719251] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[ 3584.719568] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.846351] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[ 3584.947504] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[ 3584.948451] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[ 3584.948708] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3584.948944] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 3585.706666] rtw_8723du 1-1:1.2: failed to get tx report from firmware

<\details>

dubhater, 2024-11-01:
But you didn't apply the patch with the extra logging. It changes "failed to get tx report from firmware" into a different message.

briansune, 2024-11-01:
@dubhater 

Strange I do have modified the tx.c
Maybe clean is not properly done.
[image]

I will do it again.

briansune, 2024-11-01:
@dubhater 

Real sorry didn't notice the clean is missing at the beginning of the build and result in such situation.

The new log is attached.
I didn't run any scripts or action and had suddenly burst a lot of TX reports

<details>

<summary>dmesg</summary>

ubuntu@briansune:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000001] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008198] Console: colour dummy device 80x25
[    0.012669] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022989] pid_max: default: 32768 minimum: 301
[    0.027763] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035060] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044142] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051376] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057516] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064727] cblist_init_generic: Setting shift to 1 and lim to 1.
[... 468 lines cut ...]
[  305.377805] received tx report with sn 48
[  305.563788] queueing tx report request with sn 52
[  305.605584] received tx report with sn 52
[  305.681150] queueing tx report request with sn 56
[  305.684455] received tx report with sn 56
[  305.883776] queueing tx report request with sn 60
[  305.925580] received tx report with sn 60
[  306.001146] queueing tx report request with sn 64
[  306.003421] received tx report with sn 64
[  306.203778] queueing tx report request with sn 68
[  306.219697] received tx report with sn 68
[  306.297143] queueing tx report request with sn 72
[  306.299499] received tx report with sn 72
[  306.499766] queueing tx report request with sn 76
[  306.541567] received tx report with sn 76
[  306.617135] queueing tx report request with sn 80
[  306.619420] received tx report with sn 80
[  306.819760] queueing tx report request with sn 84
[  306.899186] received tx report with sn 84
[  307.005704] queueing tx report request with sn 88
[  307.005733] queueing tx report request with sn 92
[  307.011518] received tx report with sn 88
[  307.011554] received tx report with sn 92

<\details>

briansune, 2024-11-01:
@dubhater 

Attaching a longer dmesg with stressapptest and memtester and ping running.

<details>

<summary>dmesg</summary>

ubuntu@briansune:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000001] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008198] Console: colour dummy device 80x25
[    0.012669] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022989] pid_max: default: 32768 minimum: 301
[    0.027763] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035060] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044142] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051376] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057516] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064727] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.070955] rcu: Hierarchical SRCU implementation.
[    0.075745] rcu:     Max phase no-delay instances is 1000.
[    0.081241] EFI services will not be available.
[... 1257 lines cut ...]
[ 1359.810551] received tx report with sn 140
[ 1360.012011] queueing tx report request with sn 144
[ 1360.032820] received tx report with sn 144
[ 1360.157378] queueing tx report request with sn 148
[ 1360.158402] received tx report with sn 148
[ 1360.360006] queueing tx report request with sn 152
[ 1360.380687] received tx report with sn 152
[ 1360.505498] queueing tx report request with sn 156
[ 1360.507455] received tx report with sn 156
[ 1360.708000] queueing tx report request with sn 160
[ 1360.728808] received tx report with sn 160
[ 1360.853490] queueing tx report request with sn 164
[ 1360.857279] received tx report with sn 164
[ 1361.055993] queueing tx report request with sn 168
[ 1361.076800] received tx report with sn 168
[ 1361.201610] queueing tx report request with sn 172
[ 1361.203468] received tx report with sn 172
[ 1361.403988] queueing tx report request with sn 176
[ 1361.445670] received tx report with sn 176
[ 1361.601977] queueing tx report request with sn 180
[ 1361.605731] queueing tx report request with sn 184
[ 1361.611737] queueing tx report request with sn 188
[ 1361.612000] received tx report with sn 180
[ 1361.612090] received tx report with sn 184
[ 1361.615321] received tx report with sn 188

briansune, 2024-11-01:
@dubhater 

Had successfully trigger the "still have error" message.
I run ping /w 500ms to 8.8.8.8 and concurrently run the stressapptest or memtester background.

<details>

<summary>dmesg</summary>

ubuntu@briansune:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000001] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008198] Console: colour dummy device 80x25
[    0.012669] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022989] pid_max: default: 32768 minimum: 301
[    0.027763] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035060] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044142] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051376] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057516] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064727] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.070955] rcu: Hierarchical SRCU implementation.
[    0.075745] rcu:     Max phase no-delay instances is 1000.
[... 1432 lines cut ...]
[ 2261.051083] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.052728] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.054218] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.065565] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.068212] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.068750] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.071970] queueing tx report request with sn 128
[ 2261.078625] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.087401] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.115275] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.132271] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.132587] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.132783] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.213583] queueing tx report request with sn 132
[ 2261.331074] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[... cut ...]
[ 2261.331420] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[ 2261.416138] queueing tx report request with sn 136
[ 2261.576217] queueing tx report request with sn 140
[ 2261.576323] queueing tx report request with sn 144
[ 2261.828937] received tx report with sn 124
[ 2262.090722] rtw_8723du 1-1:1.2: still have 6 unprocessed tx report requests
[ 2262.470953] queueing tx report request with sn 148
[ 2262.472454] received tx report with sn 144
[ 2262.986738] rtw_8723du 1-1:1.2: still have 1 unprocessed tx report requests

dubhater, 2024-11-01:
As I suspected, that's the bug I fixed recently. The firmware can't always handle many TX report requests in a short time. I pushed the fix.

briansune, 2024-11-01:
@dubhater 

This is the log after modifying those area you had committed.
Looks like no more missing requests.

But the  "tw_8723du 1-1:1.2: failed to allocate rx_skb #" is still problematic.
What do rx_skb really related to?

<details>

<summary>dmesg</summary>

ubuntu@briansune:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366092K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27124K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000001] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008198] Console: colour dummy device 80x25
[    0.012667] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022987] pid_max: default: 32768 minimum: 301
[    0.027759] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035055] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044140] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051374] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057513] cblist_init_generic: Setting adjustable number of callback queues.
[... 361 lines cut ...]
[  707.977297] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  707.982304] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  707.988434] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  707.988872] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  707.996947] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  708.005008] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  708.008781] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  708.017434] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  708.026142] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  708.032174] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  708.040131] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  708.048441] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  708.052427] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  708.061017] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  708.068725] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  708.080488] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  708.095359] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  708.097138] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  708.411204] rtw_8723du 1-1:1.2: failed to allocate rx_skb 3
[  708.534188] rtw_8723du 1-1:1.2: failed to allocate rx_skb 3
[  708.650003] rtw_8723du 1-1:1.2: failed to allocate rx_skb 3
[  708.807407] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  709.067761] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  709.067956] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  709.108918] rtw_8723du 1-1:1.2: failed to allocate rx_skb 3

dubhater, 2024-11-01:
It's a buffer for the RX data. The problem seems to be that it's kinda large (32 KiB) and the system doesn't always have that much contiguous memory. So when the allocation fails it tries again a little later.

briansune, 2024-11-01:
@dubhater 

Do you suggesting I need to increase the kernel cma settings?

briansune, 2024-11-01:
@dubhater 

I am not too good at system programming
But what I can observed is 

[  138.426441] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  138.451158] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  138.465533] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  138.465926] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  138.466285] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  138.466750] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  138.467090] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  138.579248] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  138.681979] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  138.690904] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  138.691984] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  139.336383] rtw_8723du 1-1:1.2: failed to allocate rx_skb 7
[  139.336530] rtw_8723du 1-1:1.2: failed to allocate rx_skb 7
[  139.336655] rtw_8723du 1-1:1.2: failed to allocate rx_skb 7
ubuntu@briansune:~$ free
              total        used        free      shared  buff/cache   available
Mem:         498700      466772        3940         792       27988       19852
Swap:       1048572       37120     1011452
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:          110344 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB

briansune, 2024-11-01:
@dubhater 

If my guessing is correct the retry is only trying on the same No. of buffer space?
I am not sure what I am suggestion is correct nor even happening, just let me do my theory.

I guess there are 4 allocation spaces from beginning as the define states?
>RTW_USB_RXCB_NUM

From the previous log I do see increasing the number also stuck at 0 for many times.

And when 4 allocation spaces are all using it need to retry and the action only check on the same No. of allocation space?
rather than there is a dirty bit to quickly seek from the free memory space No. ?
Which this will result in a circular buffer not being fully used?

No idea this could help, I don't know much on driver coding. =[

briansune, 2024-11-01:
@dubhater 

I guess this log explain a lot why it is happening

Before driver load:

>ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:          130688 kB

Once I boot the system and loaded the driver.

>ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:          123928 kB

Do stressapptest

>ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:               0 kB

>[  187.249028] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.282255] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.282676] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.283374] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.283998] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.284413] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.462890] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.522025] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.522637] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.523249] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.528736] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.529277] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.627282] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.638228] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.640132] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.640322] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.640488] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.640746] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.641155] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  187.659589] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.660307] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.661244] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.662094] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.662990] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.663722] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.664481] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.665299] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.666069] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  187.668687] rtw_8723du 1-1:1.2: failed to allocate rx_skb 0
[  188.274820] rtw_8723du 1-1:1.2: failed to allocate rx_skb 3
[  188.477803] rtw_8723du 1-1:1.2: failed to allocate rx_skb 1
[  188.618862] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2
[  188.619015] rtw_8723du 1-1:1.2: failed to allocate rx_skb 2

So the question now back to why the CMA memory is not locked by the driver and allowing other application to use more than it allows?

Or at the state after the driver load it only use i.e. X size of memory and after other application is loaded Y size of memory is used.
Turns out X+Y=MAX memory, however at that moment the driver request more than X and X+Y<=MAX so the rx_skb shows up?

dubhater, 2024-11-01:
I don't think you need to do anything at this point. It's a problem in the driver. There is another problem which requires a small redesign of the RX code, so I will try to fix both at the same time.

dubhater, 2024-11-06:
I pushed a fix for "failed to allocate rx_skb". I'm curious how it works with your stress tests.

briansune, 2024-11-06:
@dubhater 

Appreciate the great works. See the comments.
I will schedule a test plane ASAP.
Handout the testresult once it is completed.

Again speed and great work. =]

briansune, 2024-11-06:
@dubhater 

A quick compile shows:

>/home/tg5/drr/rtw88/usb.c: In function 'rtw_usb_init_rx':
./include/linux/workqueue.h:241:17: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
   (_work)->func = (_func);    \
                 ^
./include/linux/workqueue.h:246:2: note: in expansion of macro '__INIT_WORK'
  __INIT_WORK((_work), (_func), 0)
  ^~~~~~~~~~~
/home/tg5/drr/rtw88/usb.c:956:2: note: in expansion of macro 'INIT_WORK'
  INIT_WORK(&rtwusb->rx_work, rtw_usb_rx_handler);
  ^~~~~~~~~
cc1: some warnings being treated as errors

dubhater, 2024-11-07:
Oh, I forgot to disable that INIT_WORK. I'll fix it tomorrow.

briansune, 2024-11-07:
@dubhater 

I guess only need to comment those lines 

>	skb_queue_head_init(&rtwusb->rx_queue);
	skb_queue_head_init(&rtwusb->rx_free_queue);

	//INIT_WORK(&rtwusb->rx_work, rtw_usb_rx_handler);
	//INIT_WORK(&rtwusb->rx_urb_work, rtw_usb_rx_resubmit_work);

Test result:
No error found and running multiple stress test also cannot trigger memory allocation fault like before.
Looks very good.
As for the CMA usage looks good before the usage can completely use up and 0 remains.

>ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:            5840 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:            5736 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:            5960 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:            5960 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:          128348 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:          128424 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:            5860 kB
ubuntu@briansune:~$ cat /proc/meminfo | grep -i cma
CmaTotal:         131072 kB
CmaFree:            5736 kB

briansune, 2024-11-07:
@dubhater 

Waiting on the instruction of the INIT_WORK. Thank you.

dubhater, 2024-11-07:
I fixed it.

briansune, 2024-11-07:
@dubhater 

Had successfully build t he driver.
Sounds like it is just a kernel version switch case for those lines.
Pending on the hardware full stress test.

briansune, 2024-11-07:
@dubhater 

According to the previous repeatable fail skb behavior.
It will always trigger on boot up with stressapptest immediately running and holds up all CMA memory.

As for the new push, Good result!
Running both ping and memory stressapptest concurrently backgroundly.
Doesn't trigger any RX skb fail and TX fail also no longer showing with a short test.
Enclosing dmesg for reference.

Appreciate all the fixes. =]

<details>
  <summary>dmesg</summary>

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #8 SMP PREEMPT Thu Oct 31 07:04:49 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 366080K/524288K available (10048K kernel code, 1384K rwdata, 3168K rodata, 1536K init, 581K bss, 27136K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000001] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008269] Console: colour dummy device 80x25
[    0.012736] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.023064] pid_max: default: 32768 minimum: 301
[    0.027888] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035193] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044612] cblist_init_generic: Setting adjustable number of callback queues.
[... 282 lines cut ...]
[  510.660811] usb 1-1: New USB device found, idVendor=0bda, idProduct=d723, bcdDevice= 2.00
[  510.660833] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  510.660845] usb 1-1: Product: 802.11n WLAN Adapter
[  510.660854] usb 1-1: Manufacturer: Realtek
[  510.660863] usb 1-1: SerialNumber: 00e04c000001
[  510.677206] Bluetooth: hci0: RTL: examining hci_ver=08 hci_rev=000d lmp_ver=08 lmp_subver=8723
[  510.678794] Bluetooth: hci0: RTL: rom_version status=0 version=2
[  510.678821] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723d_fw.bin
[  510.678874] rtw_8723du 1-1:1.2: Firmware version 48.0.0, H2C version 0
[  510.683177] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723d_config.bin
[  510.684658] Bluetooth: hci0: RTL: cfg_sz 10, total sz 33266
[  510.947783] Bluetooth: hci0: RTL: fw version 0x828a96f1
[  511.011527] Bluetooth: MGMT ver 1.22
[  511.947432] rtw_8723du 1-1:1.2 wlx90de80882cf1: renamed from wlan0
[  516.390201] wlx90de80882cf1: authenticate with f8:6f:b0:0e:ae:e5
[  516.390281] wlx90de80882cf1: 80 MHz not supported, disabling VHT
[... cut ...] send auth to f8:6f:b0:0e:ae:e5 (try 1/3)
[  517.954569] wlx90de80882cf1: authenticated
[  517.959573] wlx90de80882cf1: associate with f8:6f:b0:0e:ae:e5 (try 1/3)
[  518.043212] wlx90de80882cf1: RX AssocResp from f8:6f:b0:0e:ae:e5 (capab=0x1411 status=0 aid=3)
[  518.061916] wlx90de80882cf1: associated
[  518.124892] wlx90de80882cf1: Limiting TX power to 35 (35 - 0) dBm as advertised by f8:6f:b0:0e:ae:e5
[  518.270742] IPv6: ADDRCONF(NETDEV_CHANGE): wlx90de80882cf1: link becomes ready

</details>

briansune, 2024-11-10:
@dubhater 

Do you think the issue is settled?
If so you can close this issue with a happy ending =]

dubhater, 2024-11-13:
Yes, I think it's fixed.

briansune, 2024-11-14:
@dubhater 

I have a small following up question not sure why this is the case.
Why the wlan name always replace as the mac?

dmesg 

>[   14.872193] wlx90de80882cf1: authenticate with f8:6f:b0:0e:ae:e5
[   14.872251] wlx90de80882cf1: 80 MHz not supported, disabling VHT
[   16.408568] wlx90de80882cf1: send auth to f8:6f:b0:0e:ae:e5 (try 1/3)
[   16.414224] wlx90de80882cf1: authenticated
[   16.415931] wlx90de80882cf1: associate with f8:6f:b0:0e:ae:e5 (try 1/3)
[   16.487030] wlx90de80882cf1: RX AssocResp from f8:6f:b0:0e:ae:e5 (capab=0x1411 status=0 aid=5)
[   16.505341] wlx90de80882cf1: associated
[   16.617380] wlx90de80882cf1: Limiting TX power to 35 (35 - 0) dBm as advertised by f8:6f:b0:0e:ae:e5

I have no idea is the related to the driver setting or not?

briansune, 2024-11-14:
@dubhater 

I am sorry that to reopen this issue due to today testing shows a puzzling TX dmesg
"rtw_8723du 1-1:1.2: failed to get tx report from firmware"
This issue is back and bayme excited on Christmas is coming soon =]

<details>

<summary>dmesg</summary>

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #10 SMP PREEMPT Thu Nov 14 12:18:45 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s45864 r8192 d31960 u86016
[    0.000000] pcpu-alloc: s45864 r8192 d31960 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 365172K/524288K available (10752K kernel code, 1410K rwdata, 3156K rodata, 1664K init, 577K bss, 28044K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000000] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008208] Console: colour dummy device 80x25
[    0.012681] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.023001] pid_max: default: 32768 minimum: 301
[    0.027773] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035069] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044165] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051396] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057538] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064750] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.070984] rcu: Hierarchical SRCU implementation.
[    0.075776] rcu:     Max phase no-delay instances is 1000.
[... 244 lines cut ...]
[    3.651403] systemd[1]: Starting Remount Root and Kernel File Systems...
[    3.668345] systemd[1]: Starting udev Coldplug all Devices...
[    3.685081] systemd[1]: Started Journal Service.
[    3.706285] EXT4-fs (mmcblk0p2): re-mounted. Quota mode: disabled.
[    3.975005] Adding 1048572k swap on /swapfile.  Priority:-2 extents:8 across:5857276k SS
[    4.040636] systemd-journald[146]: Received client request to flush runtime journal.
[    4.064923] systemd-journald[146]: File /var/log/journal/9855905ab2944be891538ff748241542/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    5.814539] rtw_8723du 1-1:1.2 wlx90de80882cf1: renamed from wlan0
[    6.657385] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    9.522186] Bluetooth: MGMT ver 1.22
[   14.798237] wlx90de80882cf1: authenticate with f8:6f:b0:0e:ae:e5
[... cut ...]
[   16.533927] wlx90de80882cf1: Limiting TX power to 35 (35 - 0) dBm as advertised by f8:6f:b0:0e:ae:e5
[   16.550164] IPv6: ADDRCONF(NETDEV_CHANGE): wlx90de80882cf1: link becomes ready
[   25.128132] systemd-journald[146]: Failed to set ACL on /var/log/journal/9855905ab2944be891538ff748241542/user-1000.journal, ignoring: Operation not supported
[  919.802223] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1219.761761] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1519.763631] rtw_8723du 1-1:1.2: failed to get tx report from firmware

</details>

dubhater, 2024-11-14:
The interface name is not related to the driver. It's renamed by udev, I think. Some distros configure it to use the MAC address, others the hardware location.

I wonder if you are somehow using an older version of the driver?

When you use debug_mask=0x10 how many c2h with ID 0x03 appear in a row?

briansune, 2024-11-14:
@dubhater 

Q1: I wonder if you are somehow using an older version of the driver?
A1: No, I had clone the latest repository and compile (CLEAN).

T1: When you use debug_mask=0x10 how many c2h with ID 0x03 appear in a row?
Will conduct and post the result ASAP.

dubhater, 2024-11-14:
You can add it to the kernel's command line:

```
rtw_core.debug_mask=0x10 rtw88_core.debug_mask=0x10
```

Add both versions because I'm not sure which is the right one when compiled in.

Since this is not a PC, I'm not sure exactly how to do it.

briansune, 2024-11-14:
@dubhater 

May I ask these flag is putting in the inside the RTW88 Makefile?
Deep sorry only know very little kernel driver stuffs.

dubhater, 2024-11-14:
The easier solution is to change the source code:
```diff
diff --git a/drivers/net/wireless/realtek/rtw88/main.c b/drivers/net/wireless/realtek/rtw88/main.c
index 5e8baac2dfba..6d6038815c5a 100644
--- a/drivers/net/wireless/realtek/rtw88/main.c
+++ b/drivers/net/wireless/realtek/rtw88/main.c
@@ -23,7 +23,7 @@
 bool rtw_disable_lps_deep_mode;
 EXPORT_SYMBOL(rtw_disable_lps_deep_mode);
 bool rtw_bf_support = true;
-unsigned int rtw_debug_mask;
+unsigned int rtw_debug_mask = 0x10;
 EXPORT_SYMBOL(rtw_debug_mask);
 /* EDCCA is enabled during normal behavior. For debugging purpose in
  * a noisy environment, it can be disabled via edcca debugfs. Because
```

dubhater, 2024-11-14:
Sorry, I don't know.

briansune, 2024-11-14:
@dubhater 

I will try to make those message out.
Not sure I can make that happen or not but try as much as I can.
If I got those debug message out I will handout the dmesg.
Again appreciate the help

briansune, 2024-11-14:
@dubhater 

I had successfully compile the kernel with the debug message.
Modify the below is only part of the debug need to include those debug flags inside the makefile.

>unsigned int rtw_debug_mask = 0x10;

I am reading those debug message and waiting on the fail message.

briansune, 2024-11-14:
@dubhater 

When the debug message is turned on and background is running ping the "failed to get tx report from firmware" not showing up at short test.

It appears that 3 repeated id 0x03.

<details>

<summary>dmesg</summary>

ubuntu@briansune:~/scripts$ dmesg | grep "id=0x03"
[   16.031919] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x02, len=8
[   16.044124] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x03, len=8
[  319.079682] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x3b, len=8
[  319.080067] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x3c, len=8
[  319.116326] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x3d, len=8
[  619.147887] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x84, len=8
[  619.148253] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x85, len=8
[  619.148633] rtw_8723du 1-1:1.2: recv C2H, id=0x03, seq=0x86, len=8

<details>

briansune, 2024-11-14:
@dubhater 

Repeatable:
I will describe as such.
The system do not run any user task and just let it sit there after bootup.
The after a while the TX fail message comes out.

<details>

<summary>dmesg</summary>

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]
[    0.000000] Linux version 6.1.54-rt15 (tg5@tg5) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12) 7.5.0, GNU ld (Linaro_Binutils-2019.12) 2.28.2.20170706) #35 SMP PREEMPT Thu Nov 14 22:27:41 HKT 2024
[    0.000000] Machine model: Anlogic, DR1M90 FPSoc
[    0.000000] earlycon: uart0 at MMIO32 0x00000000f8401000 (options '')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016400000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu s45864 r8192 d31960 u86016
[    0.000000] pcpu-alloc: s45864 r8192 d31960 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyS0,115200n8 earlycon=uart,mmio32,0xf8401000 loglevel=8 root=/dev/mmcblk0p2 rootwait rootfstype=ext4 rw
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 365172K/524288K available (10752K kernel code, 1410K rwdata, 3156K rodata, 1664K init, 577K bss, 28044K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: 160 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000dd040000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 33.33MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x7b00c4bad, max_idle_ns: 440795202744 ns
[    0.000000] sched_clock: 56 bits at 33MHz, resolution 30ns, wraps every 2199023255551ns
[    0.008207] Console: colour dummy device 80x25
[    0.012679] Calibrating delay loop (skipped), value calculated using timer frequency.. 66.66 BogoMIPS (lpj=133333)
[    0.022999] pid_max: default: 32768 minimum: 301
[    0.027772] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.035069] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.044170] cblist_init_generic: Setting adjustable number of callback queues.
[    0.051402] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.057544] cblist_init_generic: Setting adjustable number of callback queues.
[    0.064756] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.070990] rcu: Hierarchical SRCU implementation.
[... 251 lines cut ...]
[    3.783736] systemd-journald[148]: File /var/log/journal/9855905ab2944be891538ff748241542/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    5.632026] rtw_8723du 1-1:1.2 wlx90de80882cf1: renamed from wlan0
[    6.496376] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    9.334539] Bluetooth: MGMT ver 1.22
[   14.301180] wlx90de80882cf1: authenticate with f8:6f:b0:0e:ae:e5
[   14.301240] wlx90de80882cf1: 80 MHz not supported, disabling VHT
[   15.839490] wlx90de80882cf1: send auth to f8:6f:b0:0e:ae:e5 (try 1/3)
[   15.848363] wlx90de80882cf1: authenticated
[   15.851968] wlx90de80882cf1: associate with f8:6f:b0:0e:ae:e5 (try 1/3)
[   15.957668] wlx90de80882cf1: RX AssocResp from f8:6f:b0:0e:ae:e5 (capab=0x1411 status=0 aid=4)
[   15.975633] wlx90de80882cf1: associated
[   16.016854] wlx90de80882cf1: Limiting TX power to 35 (35 - 0) dBm as advertised by f8:6f:b0:0e:ae:e5
[   16.170166] IPv6: ADDRCONF(NETDEV_CHANGE): wlx90de80882cf1: link becomes ready
[... cut ...]
[  964.741549] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1219.491196] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1349.213025] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1461.595953] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1519.484086] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1617.906876] rtw_8723du 1-1:1.2: failed to get tx report from firmware
[ 1622.226404] rtw_8723du 1-1:1.2: failed to get tx report from firmware

</details>

briansune, 2024-11-14:
@dubhater 

OK, I finally can grep it with a debug mask is turned on.

<details>

<summary>dmesg</summary>

[ 6305.815233] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6305.843498] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6305.844365] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6305.844928] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x20, len=7
[ 6305.845116] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6305.851990] rtw_8723du 1-1:1.2: send H2C content 00000058 00000211
[ 6305.854116] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6305.854491] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 6307.831166] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6307.859459] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6307.860328] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6307.860875] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x21, len=7
[ 6307.861084] rtw_8723du 1-1:1.2: send H2C content 2b000042 00000000
[ 6307.867953] rtw_8723du 1-1:1.2: send H2C content 00000058 00000211
[ 6307.868328] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[ 6307.870575] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6307.870949] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 6308.031884] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x22, len=7
[ 6309.815149] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6309.843421] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6309.844291] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6309.844865] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x23, len=7
[ 6309.845168] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6309.852041] rtw_8723du 1-1:1.2: send H2C content 00000058 00000210
[ 6309.854163] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6309.854537] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 6311.799062] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6311.827379] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6311.828252] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6311.828822] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x24, len=7
[ 6311.829003] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6311.834754] rtw_8723du 1-1:1.2: send H2C content 00000058 00000210
[ 6311.837006] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6311.837375] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 6313.815004] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6313.843219] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6313.844089] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6313.844687] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x25, len=7
[ 6313.844968] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6313.850720] rtw_8723du 1-1:1.2: send H2C content 00000058 00000210
[ 6313.852839] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6313.853211] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 6315.798960] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6315.827179] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6315.828051] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6315.828611] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x26, len=7
[ 6315.828806] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6315.834680] rtw_8723du 1-1:1.2: send H2C content 00000058 00000210
[ 6315.835183] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[ 6315.837425] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6315.837797] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 6316.019017] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x27, len=7
[ 6316.400841] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6316.427166] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6316.428038] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6316.428656] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x28, len=7
[ 6316.867550] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x29, len=7
[ 6317.139533] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x2a, len=7
[ 6317.446394] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x2b, len=7
[ 6317.891551] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x2c, len=7
[ 6318.061112] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x2d, len=7
[ 6318.778108] rtw_8723du 1-1:1.2: recv C2H, id=0x0c, seq=0x2e, len=7
[... 20 lines cut ...]
[ 6323.859896] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6323.860493] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x35, len=7
[ 6323.860778] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6323.866527] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[ 6323.868647] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6323.869018] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000
[ 6325.814688] rtw_8723du 1-1:1.2: send H2C content 01000020 00000c00
[ 6325.842987] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6325.843856] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6325.844455] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x36, len=7
[ 6325.844739] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6325.850489] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[ 6325.850992] rtw_8723du 1-1:1.2: send H2C content 49810040 000fc000
[ 6325.853233] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[... cut ...]
[ 6327.858822] rtw_8723du 1-1:1.2: send H2C content 00000161 00000000
[ 6327.859697] rtw_8723du 1-1:1.2: send H2C content 03306160 00001111
[ 6327.860289] rtw_8723du 1-1:1.2: recv C2H, id=0x09, seq=0x37, len=7
[ 6327.860576] rtw_8723du 1-1:1.2: send H2C content 2a000042 00000000
[ 6327.866323] rtw_8723du 1-1:1.2: send H2C content 00000058 0000020f
[ 6327.868449] rtw_8723du 1-1:1.2: send H2C content 00000060 00000000
[ 6327.868815] rtw_8723du 1-1:1.2: send H2C content 01210120 00000000

<\details>

dubhater, 2024-11-15:
Okay, so my fix is not enough. I will report the problem and we'll see what happens.

If you don't need to support roaming, you can disable the "bgscan" performed by wpa_supplicant. That will do a better job avoiding the problem than my fix did.

briansune, 2024-11-15:
@dubhater 

Would it possible to explain what is happening behind this issue?
Sound like this is not what you can control.
Because the system is completely sitting there doing nothing and turning out this would show up from time to time.
Thank you

dubhater, 2024-11-15:
Every five minutes wpa_supplicant triggers a scan, I assume to see if there are better APs available. During these scans mac80211 transmits many QoS Null frames in quick succession. Because these frames are marked with IEEE80211_TX_CTL_REQ_TX_STATUS, rtw88 asks the firmware to report the TX ACK status for each of these frames. Sometimes the firmware can't process the TX status requests quickly enough, they add up, it only processes some of them, and it permanently messes up the internal sequence number assigned to each TX status report. Then the firmware marks every subsequent TX status report with the wrong number, so rtw88 ignores them and prints a warning instead of reporting the TX status back to mac80211.

If the device is going to be connected to just one AP, you can disable the "bgscan" in wpa_supplicant.

briansune, 2024-11-15:
@dubhater 

Thank you for the explanation learn a lot =]
 
I may not have good enough knowledge to understand all the theory behind.

If my understand correctly:
If the device using this driver need to move around such as in a office or some large area.
The WIFI setup must undergo a numbers of AP on top of my head.
Every time I move around the zone AP(1) ->AP(2) but the WIFI is still using the same SSID PASSWORD etc.
So that what roaming you are meaning?

However, if the device using this driver is stationary and the only WIFI AP is always the same one then removing the

>bgscan 

is allowed and not even needed.

Hope I am getting this correctly =]

dubhater, 2024-11-16:
> I will report the problem and we'll see what happens.

I changed my mind. The obvious solution was easier to implement than I first thought, so I just did it.

You will still see "failed to get tx report from firmware" occasionally, but not every five minutes. You can safely ignore it.

briansune, 2024-11-16:
@dubhater 

May I know the first thought about the firmware is that you are reporting back to realtek?
But you do find a simpler solution and eventually settle most of the case but not a complete resolving solution.

Am I getting this correctly?

dubhater, 2024-11-16:
Yes.

briansune, 2024-11-17:
@dubhater 

Any appreciate all the works and assisting users and developers.
I think we all got to live with a unperfect world.
As much as this repository is open-source and free based.

But just a simple question to closeup the issue.
Do the TX from beginning could introduce any driver crash, driver stuck, disconnect issue?
If not then this is just a message to warn user rather than problematic from beginning?

dubhater, 2024-11-17:
> Do the TX from beginning could introduce any driver crash, driver stuck, disconnect issue?

I think the problem with the sequence numbers can cause unnecessary disconnections, for example during GTK rekeying. Nothing worse.

briansune, 2024-11-18:
@dubhater 

Got it~
Will undergo a disable the "bgscan" in wpa_supplicant test to verify TX message issue still problematic or not.

LuisMitaHL, 2024-11-22:
Hi,
I am following this same bug on a 8812au and OpenWrt (Cudy TR1200, a portable MIPS router). I'm using the router with Travelmate, a software that needs to do a bg scan every x seconds (60 on my case). I need to roam and don't want to disable the bg scan option.

I'm testing with the "Scan" button on OpenWrt (triggering a scan every ~10 seconds), iperf3 -R for speed testing and using 8812au as a client on a far away AP (-70, -75 dbm).

The old fix (140da497c5890f874427df45ec4be636e41fb39b) solves the problem. Of course iperf3 slows down a little, but stopping the aggressive scan ramps up the speed.
The latest code (ce76fa323d563f047f65aad75362e0c40dae1a6d) does not show anything interesting on dmesg, but iperf3 stalls at 0 b/s for 1-3 seconds and the wifi link get stuck to speeds < 9 Mbps. Stopping the aggressive scan and restarting iperf3 does nothing.

My thought is that the internal firmware corruption affects more than just the TX status on 8812au. Will it be possible to retain the old patch, to prevent the firmware from becoming corrupted?

---

With the old patch
```
[  5]  37.00-38.01  sec  9.38 MBytes  77.7 Mbits/sec                  
[  5]  38.01-39.00  sec  8.75 MBytes  74.2 Mbits/sec                  
[  5]  39.00-40.00  sec  6.50 MBytes  54.5 Mbits/sec                  
[  5]  40.00-41.00  sec  6.25 MBytes  52.4 Mbits/sec                  
[  5]  41.00-42.00  sec  6.62 MBytes  55.5 Mbits/sec                  
[  5]  42.00-43.00  sec  6.50 MBytes  54.6 Mbits/sec                  
[  5]  43.00-44.00  sec  5.38 MBytes  45.0 Mbits/sec                  
[  5]  44.00-45.00  sec  6.88 MBytes  57.8 Mbits/sec                  
[  5]  45.00-46.00  sec  6.75 MBytes  56.7 Mbits/sec                  
[  5]  46.00-47.00  sec  6.75 MBytes  56.6 Mbits/sec                  
[  5]  47.00-48.00  sec  6.50 MBytes  54.5 Mbits/sec                  
[  5]  48.00-49.00  sec  5.62 MBytes  47.2 Mbits/sec                  
[  5]  49.00-50.00  sec  5.50 MBytes  46.1 Mbits/sec                  
[  5]  50.00-51.00  sec  5.62 MBytes  47.2 Mbits/sec                  
[  5]  51.00-52.01  sec  4.25 MBytes  35.5 Mbits/sec                  
[  5]  52.01-53.00  sec  7.62 MBytes  64.0 Mbits/sec                  
[  5]  53.00-54.00  sec  7.75 MBytes  65.1 Mbits/sec                  
[  5]  54.00-55.00  sec  5.88 MBytes  49.4 Mbits/sec                  
[  5]  55.00-56.00  sec  4.62 MBytes  38.7 Mbits/sec                  
[  5]  56.00-57.00  sec  5.25 MBytes  44.1 Mbits/sec                  
[  5]  57.00-58.00  sec  5.00 MBytes  41.9 Mbits/sec                  
[  5]  58.00-59.00  sec  5.88 MBytes  49.3 Mbits/sec                  
[  5]  59.00-60.00  sec  5.50 MBytes  46.2 Mbits/sec                  
[  5]  60.00-61.00  sec  6.50 MBytes  54.6 Mbits/sec                  
[  5]  61.00-62.00  sec  5.88 MBytes  49.3 Mbits/sec                  
[  5]  62.00-63.00  sec  6.75 MBytes  56.6 Mbits/sec                  
[  5]  63.00-64.00  sec  6.88 MBytes  57.7 Mbits/sec                  
[  5]  64.00-65.00  sec  6.25 MBytes  52.4 Mbits/sec                  
[  5]  65.00-66.00  sec  6.50 MBytes  54.6 Mbits/sec                  
[  5]  66.00-67.01  sec  6.50 MBytes  54.2 Mbits/sec                  
[  5]  67.01-68.00  sec  9.88 MBytes  83.4 Mbits/sec                  
[  5]  68.00-69.00  sec  8.88 MBytes  74.4 Mbits/sec                  
[  5]  69.00-70.01  sec  6.38 MBytes  53.1 Mbits/sec                  
[  5]  70.01-71.02  sec  6.88 MBytes  57.0 Mbits/sec                  
[  5]  71.02-72.00  sec  6.75 MBytes  57.7 Mbits/sec                  
[  5]  72.00-73.00  sec  6.12 MBytes  51.4 Mbits/sec                  
[  5]  73.00-74.00  sec  6.50 MBytes  54.6 Mbits/sec                  
[  5]  74.00-75.00  sec  7.00 MBytes  58.7 Mbits/sec                  
[  5]  75.00-76.00  sec  6.50 MBytes  54.4 Mbits/sec                  
[  5]  76.00-77.01  sec  5.25 MBytes  43.6 Mbits/sec                  
[  5]  77.01-78.00  sec  5.50 MBytes  46.8 Mbits/sec                  
[  5]  78.00-79.00  sec  4.88 MBytes  40.9 Mbits/sec                  
[  5]  79.00-80.00  sec  5.38 MBytes  45.1 Mbits/sec                  
[  5]  80.00-81.00  sec  3.50 MBytes  29.4 Mbits/sec                  
[  5]  81.00-82.00  sec  6.25 MBytes  52.4 Mbits/sec                  
[  5]  82.00-83.00  sec  8.38 MBytes  70.3 Mbits/sec                  
[  5]  83.00-84.00  sec  8.00 MBytes  67.1 Mbits/sec                  
[  5]  84.00-85.00  sec  9.50 MBytes  79.7 Mbits/sec                  
[  5]  85.00-86.00  sec  6.50 MBytes  54.5 Mbits/sec                  
[  5]  86.00-87.01  sec  6.88 MBytes  57.5 Mbits/sec                  
[  5]  87.01-88.01  sec  7.12 MBytes  59.4 Mbits/sec                  
[  5]  88.01-89.00  sec  5.75 MBytes  48.6 Mbits/sec                  
[  5]  89.00-90.00  sec  5.88 MBytes  49.4 Mbits/sec                  
[  5]  90.00-91.00  sec  6.38 MBytes  53.5 Mbits/sec                  
[  5]  91.00-92.00  sec  5.62 MBytes  47.2 Mbits/sec                  
[  5]  92.00-93.00  sec  5.62 MBytes  47.2 Mbits/sec                  
[... 295 lines cut ...]
[  5]  62.00-63.00  sec  1.00 MBytes  8.39 Mbits/sec                  
^C[  5]  63.00-63.66  sec   384 KBytes  4.80 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-63.66  sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-63.66  sec   173 MBytes  22.8 Mbits/sec                  receiver
[... cut ...]
[  5]   2.00-3.00   sec  1.00 MBytes  8.39 Mbits/sec                  
^C[  5]   3.00-3.17   sec   256 KBytes  12.1 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-3.17   sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-3.17   sec  2.00 MBytes  5.29 Mbits/sec                  receiver
iperf3: interrupt - the client has terminated

```

I can't see anything on dmesg. I definitely did not see the “failed to get tx report from firmware” error.

briansune, 2024-11-22:
@LuisMitaHL 

Big question before people start debugging.
How the driver is loaded w or w/o the debug flag is turned on?
As you do mentioned the dmesg had nothing to show, this sounds like you don't even turned on the debug flag from beginning.

dubhater, 2024-11-22:
@LuisMitaHL Let me see if I understood correctly:

The RTL8812AU is plugged into the Cudy TR1200?

This Travelmate software is also running on the Cudy?

You are using the Cudy's internal wifi in AP mode, and the RTL8812AU in client mode, connected to another AP?

> The old fix (https://github.com/lwfinger/rtw88/commit/140da497c5890f874427df45ec4be636e41fb39b) solves the problem.

Did you have the problem before the old fix? With this commit: https://github.com/lwfinger/rtw88/commit/e9a6539a81a4ac1a539f510ded6a91c42efb7e2e

[19 more replies]

closed 2025-01-21 by briansune