lwfinger/rtw88 #251: Development Discussion

morrownr, 2024-11-13:
This issue is being established as a location for developers to discuss issues related to the future direction of this repo.

A milestone was recently reached as the new rtl8812au and rtl8821/11au driver were accepted for inclusion in kernel 6.13. A lot of progress has been made and upstreamed over the last year due to this repo and some outstanding talent that has taking on the job of improving and expanding rtw88. One more new driver is on the agenda: rtl8814au. 

Hopefully this issue can serve as a fixed location to discuss development issues of interest.

@morrownr

dubhater, 2024-11-29:
Getting there:
```
Nov 29 03:05:16 ideapad2 kernel: rtw_8814au 1-2:1.0: Firmware version 33.6.0, H2C version 6
```

morrownr, 2024-11-29:
@dubhater 

Glad to see you making progress. So that you know, the vendor driver for rtl8814au that I have up here is the latest that I am aware of yes it is more than 5 years old. It is not clear to me what happened but here is what I think happened:

Realtek released the rtl8814au chip. It did not sell well and Linux support was dropped. I think for it to be considered a commercial success by Reaktek, it would have needed to sell well in the USB Module form for industrial use. One of the problems with this chip is that it gets hot. I've measure power use and seen upwards of 900 mA use when used at full speed.

The chip has, however, been used a lot for USB Adapters and many Linux users have adapters with this chip. It will be a great thing to have good in-kernel support. When you are ready for testing, let me know and I will establish an issue in my vendor driver and start channeling testers to this repo. Keep us posted.

@morrownr

dubhater, 2024-12-09:
A carefully selected speed test:
```
Connecting to host 192.168.0.1, port 5201
Reverse mode, remote host 192.168.0.1 is sending
[  5] local 192.168.0.50 port 52502 connected to 192.168.0.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  73.0 MBytes   612 Mbits/sec                  
[  5]   1.00-2.00   sec  98.2 MBytes   824 Mbits/sec                  
[  5]   2.00-3.00   sec  99.4 MBytes   834 Mbits/sec                  
[  5]   3.00-4.00   sec   100 MBytes   841 Mbits/sec                  
[  5]   4.00-5.00   sec  99.1 MBytes   832 Mbits/sec                  
[  5]   5.00-6.00   sec  99.8 MBytes   837 Mbits/sec                  
[  5]   6.00-7.00   sec   100 MBytes   841 Mbits/sec                  
[  5]   7.00-8.00   sec  99.2 MBytes   833 Mbits/sec                  
[  5]   8.00-9.00   sec  99.2 MBytes   833 Mbits/sec                  
[  5]   9.00-10.00  sec  99.8 MBytes   837 Mbits/sec                  
[  5]  10.00-11.00  sec  99.4 MBytes   834 Mbits/sec                  
[  5]  11.00-12.00  sec  98.2 MBytes   824 Mbits/sec                  
[  5]  12.00-13.00  sec  97.0 MBytes   814 Mbits/sec                  
```

a5a5aa555oo, 2024-12-09:
[image]

morrownr, 2024-12-09:
> A carefully selected speed test:

Is this the new 8814au driver?

On a separate subject: I was testing kernel 6.13 this weekend. The rtl8822bu driver was not going into USB3 mode. I know you sent in a patch to get the USB3 adapters switching modes and it is here in this repo but I lost track of where the upstream patch is. Do have any idea what the status is?

a5a5aa555oo, 2024-12-09:
@morrownr 

Do you mean [this one](https://github.com/torvalds/linux/commit/315c23a64e99552502dd4d18d6ddc073fad9a7c3)? it is in kernel 6.12 and later.

Maybe mode switching for 8822BU/8822CU is disabled on your system?
```
# Mode switching is disabled if this command prints "N".
$ sudo cat /sys/module/rtw88_usb/parameters/switch_usb_mode
Y
```

morrownr, 2024-12-09:
@a5a5aa555oo 

> Do you mean this one? 

That is it. I was testing 6.13 rc1 so I'll test rc2 sometime this week. Glad to see the USB3 patch for 8812au going into 6.14... just in time for Ubuntu 25.04 and all the little Buntu's. Gotta get all these distros to start compiling it and installing the firmware and cleaning out the various versions of the vendor drivers that they have hacked in.

morrownr, 2024-12-09:
@dubhater 

Right now I have the latest versions of the rtl8852bu and rtl8852cu vendor drivers in private repos trying to get them in good enough shape to go public. They are better than the earlier WiFi 6 drivers but,.. and I won't repeat what Larry used to say. I have a compilation problem with the rtl8852bu with kernel 6.12 that I am having a hard time figuring out. You don't have to have an adapter, just kernel 6.12. Do you want to take a crack at it?

@a5a5aa555oo 

You are invited to the party also if interested.

dubhater, 2024-12-09:
I can have a look.

morrownr, 2024-12-09:
> I can have a look.

Invite is on the way. I am in no hurry.

The compilation works fine through kernel 6.11 and then blows up with 6.12. I can usually sort out the problem but it is not obvious to me what the problem is.

morrownr, 2024-12-10:
@a5a5aa555oo 

> I am not a skilled programmer...

I'm not a skilled wifi programmer but I can often figure things out given time and it is fun to learn. If you are interested, you could get an adapter with the rtl8832cu chip and I could put you in that repo so you could test. That chip is interesting in that it has 160 MHz channel width as well as 6 Ghz capability. I can easily do over 1 Gbps with iperf3 on 5 GHz.

The only single-state adapter with that chip is the Edup 1671 as far as I know. Do not get the Edup 1671S as it has onboard storage and multi-state adapters that have this chip are proving to be really problematic.

https://github.com/morrownr/USB-WiFi/discussions/299

My reason for bringing the Realtek WiFi 6 vendor drivers up for public release is to lay the groundwork in case dubhater is still interesting in continuing on after the rtl8814au. It would be a big challenge as rtw89 has no usb support as far as I know but if we don't prepare then it will be more difficult.

morrownr, 2024-12-17:
@a5a5aa555oo

I have a really busy schedule over the next two days and I have a patch that needs to go in. Any chance of getting you to sent it in if you have time. I have prepared the patch minus the Report-by: and Tested-by: info. I am waiting on that info and can pass it to you when available:

```
From 647960fed3629e0ae112a6c76c8ed2d5dcd3a727 Mon Sep 17 00:00:00 2001
From: Nick Morrow <usbwifi2024@gmail.com>
Date: Tue, 17 Dec 2024 06:40:50 -0600
Subject: [PATCH] wifi:mt76:mt7921au:Add new VID/PID 35bc/0107 for TP-Link TXE50UH
Add new VID/PID 35bc/0107 for TP-Link TXE50UH

Reported-by:
Tested-by:

Cc: stable@vger.kernel.org
---
 drivers/net/wireless/mediatek/mt76/mt7921/usb.c | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
index 8aa4f0203208..e3459295ad88 100644
--- a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
@@ -21,6 +21,9 @@ static const struct usb_device_id mt7921u_device_table[] = {
 	/* Netgear, Inc. [A8000,AXE3000] */
 	{ USB_DEVICE_AND_INTERFACE_INFO(0x0846, 0x9060, 0xff, 0xff, 0xff),
 		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
+	/* TP-Link TXE50UH */
+	{ USB_DEVICE_AND_INTERFACE_INFO(0x35bc, 0x0107, 0xff, 0xff, 0xff),
+		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
 	{ },
 };
 
-- 
2.39.5

```

morrownr, 2024-12-19:
@a5a5aa555oo 

Nevermind on the patch. I was able to send it in this morning.

briansune, 2024-12-19:
@dubhater 
@a5a5aa555oo 
@morrownr 

I am planning on all unique USB and SDIO devices testing on ARM 32bit and ARM 64bit CPU.
I will create an additional GitHub repository on the result.

Enjoy~

dubhater, 2024-12-20:
rtw_8814au is ready for testing.

You will see "failed to get tx report from firmware" right after connecting. This is normal for this chip, at least for now.

> I am planning on all unique USB and SDIO devices testing on ARM 32bit and ARM 64bit CPU.

@briansune That will come in handy.

briansune, 2024-12-20:
@dubhater 

Will added to the test list. The previous module still pending on test.
Shipping and logistic take time.
Hardware modification also take time.

morrownr, 2024-12-20:
@dubhater 

> rtw_8814au is ready for testing

Great! I'll start posting information in places that 8814au users hang out.

briansune, 2024-12-21:
> @dubhater
>
> Glad to see you making progress. So that you know, the vendor driver for rtl8814au that I have up here is the latest that I am aware of yes it is more than 5 years old. It is not clear to [...]

@morrownr 

You had pointed out 900mA, do the dongle have any external LNA implemented? This sounds like wrong configuration on LDO or DCDC. 4.5W is a lot !!!!!

dubhater, 2024-12-21:
According to the driver, all RTL8814AU have external LNA and PA for the 5 GHz band.

briansune, 2024-12-21:
@dubhater 

Then this explain why it consume so much power.
But this doesn't means hardware design would necessary need LNA and PA implemented.
Because LNA and PA are to enhance the signal strength and have nothing to do with the inherent chip maximum speed etc.
So some low cost design might still remove it from beginning.

morrownr, 2024-12-22:
@briansune 

> No wonder RTL8814AU product is not common at all and module it self is also not common to find.

I suspect the rtl8814au never sold in the module format very well at all. I was never able to find many sources of modules. That might account for Realtek dropping support for their vendor driver around 6 years ago. It is a bad driver. It can't even do WPA3. I think my repo has been the only rtl8814au repo that has been actively maintained for a long time.

There were a lot of Linux users that bought adapters with this chipset as my repo still gets over 100 views per day.

morrownr, 2024-12-22:
@dubhater 

I have added notices about the new 8814au driver being available for testing in several locations that should get attention. It is the holiday season so we may not see much of a pickup in testing until after the first of the year. Of note, unlike what I did for previous drivers, this time I am pushing users to post reports in this repo. It might be a good idea if I post a notice close to the top of the README. Is there anything specific you want in the notice? Also, do you have in mind any specific timeline for upstreaming this driver?

briansune, 2024-12-22:
> @briansune
>
>
> I suspect the rtl8814au never sold in the module format very well at all. I was never able to find many sources of modules. That might account for Realtek dropping support for their [...]

According to my research why people view the repo is because USB dongle is common but PCBA and PCB-Module is completely dead.

You have zero chance to find it like stamp PCBA nor a PCBA with connector.

The one I can find closest is a TV replacement parts in my area (Shipping phase).

So from hardware point of view the driver as you also described in the README said that it is expect to use on standalone ARM, RISC based platform etc. as there are still lot of people play with mini-PC these days to build projects.

But do HW designer really can do custom board design, the answer is NO and BIG NO.

So small-scale to large scale productions are completely dead.

Just like RTL8723DS most HW designer can transfer design w/o any major issues.
 
So support his driver is just like a bonus but do it necessary need on PC-based LINUX I guess you already answer (NO) as there are default general driver to support dongle like RTL8814-series.

dubhater, 2024-12-23:
> I have added notices about the new 8814au driver being available for testing in several locations that should get attention. It is the holiday season so we may not see much of a pickup in testing [...]

Thanks for letting everyone know. Whatever you can think of is fine.

> Also, do you have in mind any specific timeline for upstreaming this driver?

I was thinking to let people test it for a month, so around the end of January.

briansune, 2024-12-27:
@dubhater @morrownr 

This is the repository I plan to provide most test report.
If there are any test or checking is needed please test me.

https://github.com/briansune/rtw88-hw

a5a5aa555oo, 2025-01-20:
@dubhater 

I found that once 8822BU enters USB3 mode, it can no longer be switched back to USB2 mode. The only solution is replugging it.
is it normal?🤔

```
# At the beginning, 8822BU is in USB 3 mode.

$ cat /sys/module/rtw_usb/parameters/switch_usb_mode
Y
$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=ehci-pci/5p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=ehci-pci/5p, 480M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=ohci-pci/5p, 12M
    |__ Port 003: Dev 002, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 004: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 004: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 005: Dev 004, If 0, Class=Wireless, Driver=[none], 12M
    |__ Port 005: Dev 004, If 1, Class=Wireless, Driver=[none], 12M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=ohci-pci/5p, 12M
/:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=ohci-pci/2p, 12M
/:  Bus 006.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 007.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 008.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 009.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 002: Dev 002, If 0, Class=Wireless, Driver=[none], 5000M
    |__ Port 002: Dev 002, If 1, Class=Wireless, Driver=[none], 5000M
    |__ Port 002: Dev 002, If 2, Class=Vendor Specific Class, Driver=rtw_8822bu, 5000M
$ cat /etc/modprobe.d/rtw88.conf 
options rtw_usb switch_usb_mode=Y

# switch back to USB 2 mode

$ sudo sh -c 'echo options rtw_usb switch_usb_mode=N > /etc/modprobe.d/rtw88.conf'
$ cat /etc/modprobe.d/rtw88.conf
options rtw_usb switch_usb_mode=N
$ sudo rmmod rtw_8822bu rtw_8822b rtw_usb rtw_core
$ sudo modprobe rtw_8822bu
$ cat /sys/module/rtw_usb/parameters/switch_usb_mode
N

# lsusb shows 8822BU is still in USB 3 mode.

$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=ehci-pci/5p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=ehci-pci/5p, 480M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=ohci-pci/5p, 12M
    |__ Port 003: Dev 002, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 004: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 004: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 005: Dev 004, If 0, Class=Wireless, Driver=[none], 12M
    |__ Port 005: Dev 004, If 1, Class=Wireless, Driver=[none], 12M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=ohci-pci/5p, 12M
/:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=ohci-pci/2p, 12M
/:  Bus 006.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 007.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 008.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 009.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 002: Dev 002, If 0, Class=Wireless, Driver=[none], 5000M
    |__ Port 002: Dev 002, If 1, Class=Wireless, Driver=[none], 5000M
    |__ Port 002: Dev 002, If 2, Class=Vendor Specific Class, Driver=rtw_8822bu, 5000M

# speed test result shows 8822BU is in USB 3 mode indeed.

$ iperf3 -c 192.168.0.13 -t 10 -i 5 -R
Connecting to host 192.168.0.13, port 5201
Reverse mode, remote host 192.168.0.13 is sending
[  5] local 192.168.0.45 port 56950 connected to 192.168.0.13 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-5.00   sec   239 MBytes   401 Mbits/sec                  
[  5]   5.00-10.01  sec   312 MBytes   524 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.08  sec   556 MBytes   462 Mbits/sec   35             sender
[  5]   0.00-10.01  sec   552 MBytes   463 Mbits/sec                  receiver

iperf Done.
```

briansune, 2025-01-20:
@a5a5aa555oo 

Haha you got a great point again.

But did you make a conf file like here mentioned?

https://github.com/lwfinger/rtw88#4-driver-parameter-configuration-options-

If you play around on the rmmod and insmod.

You will enter some case stuck or report huge sets of cannot find firmware sometimes got stuck and cannot connect WIFI.

But who will do this so rapidly unless you just like to plug and unplug thing HEHE

briansune, 2025-01-20:
@a5a5aa555oo 

Just want to learn.

https://github.com/lwfinger/rtw88#4-driver-parameter-configuration-options-

Do the naming matters about the conf file? rtw88.conf

a5a5aa555oo, 2025-01-20:
@dubhater 

Tried the vendor driver and it has the same problem, so it is normal maybe.

@briansune  

No.

briansune, 2025-01-20:
@a5a5aa555oo 

With little software idea, will this be more simpler to force the USB root to switch speed rather than the WIFI chip?

Complete no idea what I am talking about, but forcing USB to kill the 5G P/N pair under the USB driver?

dubhater, 2025-01-20:
I didn't think there was any need for switching from USB 3 to USB 2, so I didn't add that code.

a5a5aa555oo, 2025-01-20:
> I didn't think there was any need for switching from USB 3 to USB 2, so I didn't add that code.

It's okay. I have added some instructions about how to switch the adapter to USB2 mode in README.md, 
so the users shoud know how to do it now.

briansune, 2025-01-20:
@a5a5aa555oo 

I might be wrong. but if it is forced to USB 2.0 do it still getting the USB 3.0 900 mA or cap 500mA?

Then it will introduce insufficient power during heavy loading at 5G band.

Some HUB chip will trigger current protection?

briansune, 2025-01-20:
@a5a5aa555oo 

Because I recall he had mentioned here. So this is why it is not needed to force USB 3.0->2.0 from first place?

https://github.com/lwfinger/rtw88/issues/261#issuecomment-2542540447

morrownr, 2025-01-23:
@briansune 

> I might be wrong. but if it is forced to USB 2.0 do it still getting the USB 3.0 900 mA or cap 500mA? Then it will introduce insufficient power during heavy loading at 5G band.

I remember checking the power consumption on my rtl8814au based adapter a few years ago. If I recall correctly, if it is plugged into a USB2 port, power consumption would back off to 500 mA. That makes sense because the spec max power for USB2 is 500 mA while spec max power for USB 3 is 900 mA. I do not recall if I tested changing the mod parm to change to USB2 while still in a USB3 port.

I have the equipment do some testing. I would just need to know what tests are needed.

briansune, 2025-01-24:
@morrownr 

### Intro.

First appreciate but You don't need to test it. I had made all necessary observation.

And why I will save the time from your side because I am not sure you got a down-costed USB dongle to compare from first place?

Meantime I had already run through most devices on RTW88 supported I got all entry-class test gear in the lab.

If you want actual numbers aka power current and voltage I can do a python drop. But quick and dirty fast check.

And you are completely correct that the 3.0 PLUG-A is 900mA which result in a 4.5W profile (don't care IR-drop) vs 2.5W profile.

### TODO: Of cause if you really like to cross-check on both party the easiest way is use two USB sponging dongle.

[this kind](https://www.google.com/search?sca_esv=2f83c0f5b204b677&rlz=1C1ONGR_enHK1143HK1143&q=usb+power+check&udm=2&fbs=ABzOT_AGBMogrnfXHu6GxeqSvos9YNXXoqT4M1zPLW-rz66fPS6nWbUndoCz-Bup9JcwNgaA8se1iUEusHQCFhBholU8rZqkBtpVZH8LpMKLv8NukpTnMCdEwe1uPpGtaJTq_vWeIGDaEDbZ3zhPwko1bOwwf10RvJ-ncz1fM2FPvDlSReZEui78tDJlLSFI3D-R8ev04kJY&sa=X&ved=2ahUKEwiYw676jo2LAxVibfUHHRDOBYMQtKgLegQIEhAB&biw=1745&bih=828&dpr=1.1#vhid=6lLW5fgQ4_UmUM&vssid=mosaic)

Why two or one? Because you the one on the USB dongle you are working with.

The other one can be PSU or the USB checker depends on the gear and DUT setup.

So the peak number on supply source and USB port can cross-check the actual final power /w all IR-drop and gear-loss etc.

### Possible hazard?

According to my limited software knowledge and just stupid observations.

If recalled from the issue I had asked during my test on USB 3.0 supported device and return efuse: thermal=0xff? Which EFUSE sanity is dead.

No more clear memory,  I am not sure it will run too hard and thermal overrun?

### On reports:

Case 1: See USB C is turned to black and directly passing through external Linear-PSU. Why?

Because during test most other USB devices can run cont'd w/o power recycling which immediate test me power over drawn.

https://github.com/briansune/rtw88-hw/tree/main/rtl8812au_arm_a35_usb2_5g

Case 2: That module had a onboard DC/DC and both 5V and 3.3V had run through and even a 3.3V under 5V 2.5W profile still result power over drain.

2.0-based powered devices are still problematic. Power cycling still happen and additional PSU is needed.

Because you are more familiar with 8814AU I had quick run some measurements:

1) peak ideal: 0.3A (power had subtracted other components / usages)

2) heavy run on 5G iperf3 -s and quick peak shows 0.95 - [0.2-0.3](CPU and other device) >= 0.5 and this is just possible stable current not transient current. If possible fast enough it depends on the hub or USB HW already lockout and possible current-protection. I didn't use additional 4-wire sensing aka Kelvin probing. Too many hands on work to test already.

https://github.com/briansune/rtw88-hw/tree/main/rtl8814au_arm_a35

Case 3: USB 3.0 can properly run on USB3.0 but same as CASE1+2.

8822BU is even drawing more power than I thought.

My PSU always got to run up to around 1.5A to prevent some puzzling behavior.

Even the onboard ARM process is running hard it is most like just running less than 2W but those number is very coarse. 

https://github.com/briansune/rtw88-hw/tree/main/rtl8822bu_arm_a35

### Quick Findings

It is too hard to tell what USB dongle (NOT MODULE) on 8812AU 8814AU 8822BU have what result on USB 2.0 but a very high-change to lockout on current-limited under heavy load.

### Not sure these helps

briansune, 2025-01-24:
@morrownr 

After second thought, now I see why you mentioned:

> I do not recall if I tested changing the mod parm to change to USB2 while still in a USB3 port.

No what my inherent question is about without using USB3.0 port from beginning to test current will over drawn.

Q1:
I might be wrong but do you mean even the USB dongle is plug into a USB 2.0 when it is supported 3.0 also requires switch_usb_mode=Y/N flag to cap the current to 500mA or it does not?

morrownr, 2025-01-24:
@briansune 

> No what my inherent question is about without using USB3.0 port from beginning to test current will over drawn.

My statement was not linked specially to what you were saying. It was more that I was thinking out loud. 

Let me backup and say what was really on my mind, that I did not say:

Would it be cool if an adapter could switch to USB2 mode, if not already there, if it uses band 1 (2.4 GHz) and then change to USB3 if the radio is switched to band 2 (5 GHz), if the conditions are correct for USB3 mode? I say this because using USB3 when on band 1 can result in interference AND USB3 is simply not needed for band 1. Now, this would be cool for all of the WiFi 5 adapter that are capable of USB3 here in rtw88. Whether this can actually be done in a reliable way, I do not know.

briansune, 2025-01-24:
@morrownr 

Okay, maybe we are deeply misunderstand each other from first place. 
(BUT we do wanna help from first place at least I rephrase myself again)

### 1)

> If I recall correctly, if it is plugged into a USB2 port, power consumption would back off to 500 mA. That makes sense because the spec max power for USB2 is 500 mA while spec max power for USB 3 is 900 mA

On the above result that you had mentioned, I got a different result from first place, this is why I emphasize. And I am not sure it is caused by those flags (switch_usb_mode=Y/N) or any software related mistakes.

So that is why I ask "Do you mean even the USB dongle is plug into a USB 2.0 when it is supported 3.0 also requires  flag to cap the current from 900mA to 500mA? W/o the flag will introduce wrong power profile"

And I am not sure what method you had done and get those result from first place. This is why I am confused.

### 2)
What I trying to tells are there are possible cases on RTL8822BU, RTL8812AU (5G+2G+USB3.0 supported) devices and RTL8814AU (I don't have USB 3.0 on hand) are going to trigger USB 2.0 current limiting on the hardware as the current state during those device measurements.

What I observe is that if you run USB 3.0 aka RTL8822BU and RTL8812AU on USB 3.0 port I measured it never over 550mA even it is 900mA profile as USB 3.0 states.

And when it is plugged into a USB 2.0 port it will runs under 500mA at cold state and after a while the peaks will reach above it.

### 3)
So based on 1. 2 this already tells me one important idea is that even a USB 2.0 give you 500mA maximum current profile.
Do you really stable and hazard free to run it a to the limit current from first place (when it is hot as well).
Additional USB protection IC could also trigger a lockout from first place.
So how to define the "Margin of safety" from first place?
A poor USB 2.0 port with high voltage drop will introduce more current draw from first place.
A better USB 2.0 voltage compensated port could reduce current draw from first place.

### READ HERE <-
This is why even your hardware is telling us it is stable, do other USB protection IC setup on other USB 2.0 ports are good to tell this for sure the answer is NO.
So this is why I don't want you to waste resource aka time and gear cost from first place.

Of cause I cannot stop you to give it a try from first place, but what my hardware experience tells me this clearly.

### TODO

> TODO: Of cause if you really like to cross-check on both party the easiest way is use ONE USB sponging dongle.

briansune, 2025-01-24:
@morrownr 

I had reviewed the history of RTW88 and I do understand THE WHYS:

There are reports telling us that USB 3.0 5G signals are affecting WIFI 5G.

> So the needs for changing USB 3.0 back to USB 2.0 speed on those USB dongle exist.

And the needs for confirming the USB power stability is a must before the above feature is enable or launch.

Again I am just trying to help

morrownr, 2025-01-24:
@dubhater 

Regarding your plan to upstream rtl8814au support:

You mentioned late January as a target. There was a discussion about AP mode but I had to backoff from testing that for a while due to multiple issues. I've now starting testing AP mode again. I am of the opinion that a faulty extension cable could have been the cause of the problems that I was seeing and have retired the cable. I am seeing solid AP mode performance currently and am not aware of other issues that would constitute a reason not to upstream the driver at this time.

My 2 cents

morrownr, 2025-01-24:
@briansune 

> Okay, maybe we are deeply misunderstand each other from first place.

Misunderstandings in this medium are common because there is far more than words on a screen that are required for effective communications. If we have a desire to overcome, we can. 

> "Do you mean even the USB dongle is plug into a USB 2.0 when it is supported 3.0 also requires flag to cap the current from 900mA to 500mA? W/o the flag will introduce wrong power profile"

Let me make it clear: I am not saying that I have done extensive testing of all possibilities. A few years ago I did power usage testing at the request of some users. What I remember for sure is that in USB3 mode, with iperf3 testing, I would see between 800 and 900 mA of power use with the rtl8814au chip. That was information that was useful to users as they did not understand why they were having problems. Beyond that, I would have to do new testing.

> What I observe is that if you run USB 3.0 aka RTL8822BU and RTL8812AU on USB 3.0 port I measured it never over 550mA even it is 900mA profile as USB 3.0 states.

I have not seen rtl8812au or rtl8812bu based adapters use more than about 550 mA as well. The only USB adapters that I have found to be power hogs are based on the rtl8814au. The versions of the rtl8822bu and rtl8822cu that have onboard bluetooth are limited to USB2. That is not an issue with the rtl8812au and rtl8814au as there are no versions of those chips that support bluetooth as far as I know.

> So this is why I don't want you to waste resource aka time and gear cost from first place. Of cause I cannot stop you to give it a try from first place, but what my hardware experience tells me this clearly.

You lost me on 3). I do not have any plans to do additional testing right now. I simply offered to do more testing based on specific requests if there is a reason to do so.

briansune, 2025-01-24:
@morrownr 

I am completely lost .... 

> I have the equipment do some testing. I would just need to know what tests are needed.

Okay first deeply appreciated testing offer from your side, but I don't have good plans to resolve the problem. But in section 3 I had tired to mention what is possible outcomes and it is not worth as it is almost not possible to resolve such problem from first place.

> I have not seen rtl8812au or rtl8812bu based adapters use more than about 550 mA as well.

But do you read my response completely? I had tried to rephase it again. When a USB 3,0 supported 8812AU 8822BU dongle applied onto a USB 2.0 PORT A, once it is running more heavily and get hot it have peaks that easily over 500mA profile which means already passed the USB 2.0 power envelope.

I deeply appreciate the cofounder patch peoples which one had passed away. And I also hope it can be more stronger than the pass.

dubhater, 2025-02-12:
I had a quick look at this old driver for the RTL8814AE PCIe cards: https://github.com/TsXor/rtl8814ae_comfast (check out that README.md)

It seems rtw88 could support it already so I added rtw8814ae.c. Let's see if anyone will test it.

briansune, 2025-02-12:
@dubhater 

I am still having difficulties on 8821CE and 8822CE and 8723DE under my test gear.

I even bought two different style mPCIe and NGFF and still the board cannot detect card normally.

Got to confirm those before have room to test that.

I do see GRIS 8814AE, looks like it is a 4 antenna card.

briansune, 2025-02-13:
@dubhater 

Okay I will test the RTL8814AE as well.

All PCIe devices are tested, reports are here, most PCIe devices are stable but TRX have some degrade on 5G.

AP shows no error or stuck or any issue.

https://github.com/briansune/rtw88-hw?tab=readme-ov-file#rtw-revision---e32c796572ee22ae172411fe5c1183921eca498e

Will test the basic connection and also need more guidance on deeper test on the RTL8814AE PCIe card,

morrownr, 2025-02-14:
@briansune 

Does that 8814ae card support bluetooth?

briansune, 2025-02-14:
@morrownr 

This is the card shipping out.

[image]

According to my understanding BT uses USB protocol on PCIe cards when PCIe as well as m-PCIe formfactor is used NGFF uses M.2 pins.

So if there are no USB pins cable I guess answer is NO <-

@dubhater 

I think there is very very very high chance the EFUSE will be wrong or even blank.

I had checked with the seller they don't even group with Realtek to do card design.

Will see how it goes.

morrownr, 2025-02-14:
@briansune 

> According to my understanding BT uses USB protocol...

Yes, USB2. I had suspected there were no 8814a chips that had BT support. One less thing to worry about.

dubhater, 2025-02-15:
@briansune Nice, I like the colour. You can use the module parameter ``debug_mask=0x20`` to see the efuse.

briansune, 2025-02-20:
Had complete a full report after the latest patch.

Added 5G AP test as well, very very good TRX rate.

5G STA have hard time to connect need to reconnect multiple times to success, 2.4G shows no such issue.

2.4G TRX rate even better than 5G on STA operations.

Same 2.4G AP issue on BT and other 2.4G devices.

Full Report: https://github.com/briansune/rtw88-hw/tree/main/rtl8814ae_arm_a53

dubhater, 2025-02-20:
@briansune That's amazing! I can't believe it works.

morrownr, 2025-02-20:
@dubhater 

Should I add RTL8814AE to the README?

dubhater, 2025-02-20:
@morrownr Sure, thank you.

briansune, 2025-02-22:
@dubhater 

Another brand 8814AE card had tested also.

Report: https://github.com/briansune/rtw88-hw/tree/main/rtl8814ae_arm_a53_comfast

5G compared with GRIS card is very fast connect on STA (no delay at all).

2.4G AP test shows very bad and easily interrupted by BT devices

5G AP the best peak I can get is 604mbps (not in report).

Final conclusion:

I will guess buy GRIS card if you have plans on using it on notebook. (x2 price compared with COMFAST)

Very good shield on COMFAST, 5G is good as well. (very cheap card)

dubhater, 2025-02-23:
> [@dubhater](https://github.com/dubhater)
> 
> Another brand 8814AE card had tested also.
> 

Oh, nice!

> Report: https://github.com/briansune/rtw88-hw/tree/main/rtl8814ae_arm_a53_comfast
> 
> 5G compared with GRIS card is very fast connect on STA (no delay at all).
> 
> 2.4G AP test shows very bad and easily interrupted by BT devices
> 
> 5G AP the best peak I can get is 604mbps (not in report).
> 

RTL8814AE has 3 spatial streams. It should be able to reach 800 Mbps. Did the client device also have 3 spatial streams when you tested the 5G AP performance?

> Final conclusion:
> 
> I will guess buy GRIS card if you have plans on using it on notebook. (x2 price compared with COMFAST)
> 
> Very good shield on COMFAST, 5G is good as well. (very cheap card)

Is the GRIS card actually a half size miniPCIe card inside? It's strange because I couldn't find any RTL8814AE for laptops, only desktop cards.

I added some more code for RTL8814AE. Could you test the 5 GHz signal strength again, please?

morrownr, 2025-02-24:
> It's strange because I couldn't find any RTL8814AE for laptops...

Given that you can cook breakfast on the rtl8814au chips that I have experience with, that might explain why putting one inside the closed spaces of a laptop might not work well. Put them through their paces and watch 900 mA on the gauge.

briansune, 2025-02-24:
@dubhater 

Q1:Is the GRIS card actually a half size miniPCIe card inside? It's strange because I couldn't find any RTL8814AE for laptops, only desktop cards.

A1: Yes GRIS design it as a miniPCIe card and convert it into standard PCIe.

So I proposing that user are allow to remove it from the conversion card and simply bring it to the notebook that have PCIe MIni slot.

** HOWEVER how to bring the heat out and the antennas number all those you got to DIY.

Q2: It should be able to reach 800 Mbps. Did the client device also have 3 spatial streams when you tested the 5G AP performance?

A2: Yes I also notice that as well. That's why I had two 8814AE cards, But before I plugin through the extension wrong side I got 604Mbps peak and after I toasted a bit the GRIS card (still alive after that) the peak no longer back to 600. HAHA

R3: I will try when time is free.

@morrownr 

The case on this is not correct.

Because it does not have any LNA or PA and it is a PCIe-based hardware there are almost no issue on minPCIe slot about current limitation.

Meantime, if you read the miniPCIe slot how many contact pins are on the power and you will see why USB have power issue. Contact losses are huge on USB. Ground returns are so so bad and no where to start the discussion.

See report image to understand more.

From hardware design SDIO / PCIe always have the most stability because they are fully populated onto the board or screwed onto the board.

morrownr, 2025-02-24:
@briansune 

Thanks for the explanation on the heating issue.

briansune, 2025-02-24:
@morrownr 

I didn't see are heating issue on RTL8814AE PCIe cards from first place.

Both COMFAST and GRIS card have huge heatsink but it is barely warm.

GRIS heatsink even uses a 2mm thick thermal pad to transfer the heat to the RED heatsink.

COMFAST use two layers design which is even worst. thermal pad -> shield -> thermal pad -> heatsink.

So I cannot see notebook casing cannot remove the heat.

DIY ability is the most concern.

[52 more replies]