lwfinger/rtw88 #253: warning: cast between incompatible function types

morrownr, 2024-11-17:
Kernel: 6.6 Debian
gcc: 12.2

Just a warning, not an error:

...
  CC [M]  /home/morrow/src/rtw88/usb.o
/home/morrow/src/rtw88/usb.c: In function ‘rtw_usb_init_rx’:
/home/morrow/src/rtw88/usb.c:943:22: warning: cast between incompatible function types from ‘void (*)(struct rtw_usb *)’ to ‘void (*)(long unsigned int)’ [-Wcast-function-type]
  943 |                      (void(*)(unsigned long))rtw_usb_rx_handler,

LuisMitaHL, 2024-12-10:
Same warning on OpenWrt 24.10-rc2 with kernel 6.6.63, but here is treated as a error

```
  CC [M]  /media/Standard/openwrt/openwrt-sdk-24.10.0-rc2-mediatek-filogic_gcc-13.3.0_musl.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/rtw88-usb-2024.11.27~611874c6/usb.o
/media/Standard/openwrt/openwrt-sdk-24.10.0-rc2-mediatek-filogic_gcc-13.3.0_musl.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/rtw88-usb-2024.11.27~611874c6/usb.c: In function 'rtw_usb_init_rx':
/media/Standard/openwrt/openwrt-sdk-24.10.0-rc2-mediatek-filogic_gcc-13.3.0_musl.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/rtw88-usb-2024.11.27~611874c6/usb.c:943:22: error: cast between incompatible function types from 'void (*)(struct rtw_usb *)' to 'void (*)(long unsigned int)' [-Werror=cast-function-type]
  943 |                      (void(*)(unsigned long))rtw_usb_rx_handler,
      |                      ^
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:243: /media/Standard/openwrt/openwrt-sdk-24.10.0-rc2-mediatek-filogic_gcc-13.3.0_musl.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/rtw88-usb-2024.11.27~611874c6/usb.o] Error 1
make[4]: *** [/media/Standard/openwrt/openwrt-sdk-24.10.0-rc2-mediatek-filogic_gcc-13.3.0_musl.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.6.63/Makefile:1921: /media/Standard/openwrt/openwrt-sdk-24.10.0-rc2-mediatek-filogic_gcc-13.3.0_musl.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/rtw88-usb-2024.11.27~611874c6] Error 2
make[3]: *** [Makefile:234: __sub-make] Error 2
```

dubhater, 2024-12-10:
Does it work now?

closed 2024-12-10 by dubhater

morrownr, 2024-12-10:
@dubhater 

It works now. Thanks.

I see that the firmware for the rtl8814au chip is now in the repo. I guess I need to get my adapter out and dust it off.