lwfinger/rtw88 #254: rtl8812au performance is poor

morrownr, 2024-11-17:
New pull and compile from 2024-11-16

I won't post the entire results from iperf3 testing to save space but will post them if needed.

iperf3 showing ~ 250 Mbps. Client mode. Single level -28 dBm and Link Quality 100%.

I then installed my vendor driver ( https://github.com/morrownr/8812au-20210820 ) and iperf3 showed a consistent 420 Mbps. Nothing was touched except for installing the vendor driver. The vendor driver automatically blacklisted this driver and will remove the backlist when the removal script is used.

An additional check to confirm: I tested my rtl8811au based adapter in the same system and it averaged around 220 Mbps. That is good performance for that chip and is almost as good as the rtl8812au.

I checked that the rtl8812au based adapter was running in USB3 mode and it was. This testing is on a Debian 12 installation with kernel 6.6 and gcc 12.2. There was nothing obvious to suggest what the problem is.

Overall conclusion: The rtl8812au driver seems to be suffering some performance problems. The rtl8811au driver is working well.

dubhater, 2024-11-17:
Is that the TX or RX rate?

morrownr, 2024-11-18:
```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.188 port 59216 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  36.5 MBytes   306 Mbits/sec    0    819 KBytes       
[  5]   1.00-2.00   sec  30.0 MBytes   252 Mbits/sec    0    860 KBytes       
[  5]   2.00-3.00   sec  31.2 MBytes   262 Mbits/sec    0    860 KBytes       
[  5]   3.00-4.00   sec  31.2 MBytes   262 Mbits/sec    0    860 KBytes       
[  5]   4.00-5.00   sec  28.8 MBytes   241 Mbits/sec    0    860 KBytes       
[  5]   5.00-6.00   sec  28.8 MBytes   241 Mbits/sec    0    860 KBytes       
[  5]   6.00-7.00   sec  30.0 MBytes   252 Mbits/sec    0    860 KBytes       
[  5]   7.00-8.00   sec  30.0 MBytes   252 Mbits/sec    0    860 KBytes       
[  5]   8.00-9.00   sec  30.0 MBytes   252 Mbits/sec    0    936 KBytes       
[  5]   9.00-10.00  sec  32.5 MBytes   273 Mbits/sec    0    936 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   309 MBytes   259 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   306 MBytes   256 Mbits/sec                  receiver

iperf Done.

```

```
$ iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.188 port 43860 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  52.1 MBytes   437 Mbits/sec                  
[  5]   1.00-2.00   sec  58.7 MBytes   492 Mbits/sec                  
[  5]   2.00-3.00   sec  54.5 MBytes   457 Mbits/sec                  
[  5]   3.00-4.00   sec  57.5 MBytes   482 Mbits/sec                  
[  5]   4.00-5.00   sec  58.3 MBytes   489 Mbits/sec                  
[  5]   5.00-6.00   sec  57.3 MBytes   480 Mbits/sec                  
[  5]   6.00-7.00   sec  55.7 MBytes   468 Mbits/sec                  
[  5]   7.00-8.00   sec  57.1 MBytes   479 Mbits/sec                  
[  5]   8.00-9.00   sec  57.2 MBytes   479 Mbits/sec                  
[  5]   9.00-10.00  sec  54.6 MBytes   458 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   565 MBytes   474 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   563 MBytes   472 Mbits/sec                  receiver

iperf Done.

```

Testing is on a totally clean air DFS channel. There is zero sharing going on.

dubhater, 2024-11-20:
What rates do you see in wavemon while iperf3 is running?

morrownr, 2024-11-21:
I need to put this issue on hold for now. I had changed the lone 8812au adapter that I have to another system for testing and I am seeing some issues that need to be investigated. I plug and unplug a lot so my adapters, cable and ports wear far more than what would happen with most people. I assume you were not seeing this problem.

dubhater, 2024-11-22:
I don't see a difference that big.

morrownr, 2024-12-10:
I'm closing this issue as the problem may have been here. Sorry for the noise.

closed 2024-12-10 by morrownr