lwfinger/rtw88 #255: Failed to find module 'rtw_8822ce' and connect to WPA/WPA2-Enterprise network

io-qar, 2024-11-18:
I was satisfied till this day with the driver, but when I opened logs after suddenly long booting on Fedora 41, I appeared on this thing:
[image]
However, I see the driver is in the kernel 6.11.6 and is in use (`lspci -k` command)
Here is what GNOME logs app shows, just in case:
[image]

a5a5aa555oo, 2024-11-18:
Did you upgrade your system recently? what are the outputs of these two commands?

```lsmod | grep rtw```
```sudo modinfo rtw_8822ce```

io-qar, 2024-11-18:
Yes, I did
`lsmod | grep rtw`:
`
rtw_8822ce             12288  0
rtw_8822c             507904  1 rtw_8822ce
rtw_pci                40960  1 rtw_8822ce
rtw_core              385024  2 rtw_8822c,rtw_pci
mac80211             1773568  2 rtw_core,rtw_pci
cfg80211             1445888  2 rtw_core,mac80211
`
`sudo modinfo rtw_8822ce`:
`
filename:       /lib/modules/6.11.7-300.fc41.x86_64/kernel/drivers/net/wireless/realtek/rtw88/rtw_8822ce.ko.xz
license:        Dual BSD/GPL
description:    Realtek 802.11ac wireless 8822ce driver
author:         Realtek Corporation
alias:          pci:v000010ECd0000C82Fsv*sd*bc*sc*i*
alias:          pci:v000010ECd0000C822sv*sd*bc*sc*i*
depends:        rtw_pci,rtw_8822c
retpoline:      Y
name:           rtw_8822ce
vermagic:       6.11.7-300.fc41.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Custom MOK
sig_key:        1E:1B:30:B8:4F:D0:56:FD:70:55:70:FA:9E:AF:86:56:BA:9B:D6:7E
sig_hashalgo:   sha256
signature:      B4:86:6D:F0:35:86:20:6D:B4:1E:6A:4C:CF:F0:7E:81:03:7D:4B:6B:
		30:68:B4:B6:E2:5E:73:4B:8F:2D:EE:0E:E5:3C:AC:99:06:0A:46:DD:
		D7:D7:6F:F8:6E:4C:3C:19:D3:C6:F2:61:18:CC:31:2F:69:9A:FF:03:
		27:5E:A5:56:51:5A:E1:D8:31:F6:F6:88:F8:FA:97:EC:8B:3E:31:94:
		2D:D7:A1:43:1C:03:18:FE:F5:25:66:76:63:1C:10:5E:C5:4F:57:C4:
		C6:7C:4B:54:8F:3A:F9:BA:F0:EA:A8:9B:D1:75:AD:9E:F7:63:CB:C0:
		98:71:52:DD:8F:C1:0C:E9:E9:51:8B:C1:5D:9F:62:06:7B:4A:99:22:
		15:4B:AF:11:BB:61:9F:3E:CB:85:B9:13:5E:AA:CE:09:8B:4D:14:5B:
		A1:FF:FF:91:2A:93:F0:9A:32:96:C1:6C:CA:BB:B5:02:21:BB:4D:DE:
		BF:2B:8C:62:0D:AD:1C:03:58:B0:12:8F:4E:3C:B1:A2:73:88:DB:32:
		75:DA:E4:CD:04:57:23:F7:A8:4A:97:E8:91:F8:71:6E:73:C2:BC:0E:
		59:74:C2:C3:BB:05:FF:D5:8A:C8:F7:F4:C6:09:77:C6:EE:44:1B:6C:
		02:3D:EB:B2:7A:17:12:D4:55:E2:5E:57:EB:43:20:30
`