lwfinger/rtw88 #256: Latest version cross-compile fail

briansune, 2024-11-22:
@dubhater 

The last commit had critical compile issue:

Kernel Version 5.4.0

rtw88/main.c:2350:25: error: parameter 2 ('brightness') has incomplete type
 2350 |     enum led_brightness brightness)
rtw88/main.c:2349:12: error: function declaration isn't a prototype [-Werror=strict-prototypes]
 2349 | static int rtw_led_set_blocking(struct led_classdev *led,
error: dereferencing pointer to incomplete type 'struct led_classdev'
  994 |  BUILD_BUG_ON_MSG(!__same_type(*(ptr), ((type *)0)->member) && \
error: 'struct rtw_dev' has no member named 'led_cdev'
  994 |  BUILD_BUG_ON_MSG(!__same_type(*(ptr), ((type *)0)->member) && \

dubhater, 2024-11-22:
Should be fixed by https://github.com/lwfinger/rtw88/commit/beee28a47d6e27ff2996e311852f6d480fcd8cfe.

briansune, 2024-11-22:
@dubhater 

How about those C99 looks like it will also return error due to

>for(int i 

Seems like old C don't allow such syntax?

briansune, 2024-11-22:
@dubhater 

After pull the repository I found there is a dependence issue related to

[#ifdef CONFIG_MAC80211_LEDS
	bool led_registered;
	char led_name[32];
	struct led_classdev led_cdev;
#endif](https://github.com/lwfinger/rtw88/blob/beee28a47d6e27ff2996e311852f6d480fcd8cfe/main.h#L2243-L2247)

It will return errors:

kernel.h:994:51: error: 'struct rtw_dev' has no member named 'led_cdev'
  994 |  BUILD_BUG_ON_MSG(!__same_type(*(ptr), ((type *)0)->member) && \
compiler_types.h:129:35: error: 'struct rtw_dev' has no member named 'led_cdev'
  129 | #define __compiler_offsetof(a, b) __builtin_offsetof(a, b)
main.c:2373:36: error: 'struct rtw_dev' has no member named 'led_cdev'
 2373 |  struct led_classdev *led = &rtwdev->led_cdev;
main.c:2384:17: error: 'struct rtw_dev' has no member named 'led_name'
 2384 |  snprintf(rtwdev->led_name, sizeof(rtwdev->led_name)

dubhater, 2024-11-22:
Fixed now.

briansune, 2024-11-22:
@dubhater 

Yes, tried linux kernel 5.4.0 and 6.1.0.
Both can compile properly.

Great Job~

closed 2024-11-22 by briansune