lwfinger/rtw88 #261: The 8812au shows in lsusb but not in nmtui or nmcli

ss8688, 2024-12-04:
```
$ lsusb
Bus 001 Device 004: ID 0bda:881a Realtek Semiconductor Corp. RTL8812AU-VS 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 006: ID 152d:a580 JMicron Technology Corp. / JMicron USA Technology Corp. SATA JMS576A Storage
Bus 001 Device 005: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 007: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC)
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
~ $ nmcli device show
GENERAL.DEVICE:                         wlan0
GENERAL.TYPE:                           wifi
GENERAL.HWADDR:                         B8:27:EB:80:29:B2
GENERAL.MTU:                            1500
GENERAL.STATE:                          100 (connected)
GENERAL.CONNECTION:                     Xiaomi_7183
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/ActiveConnection/2
IP4.ADDRESS[1]:                         192.168.31.248/24
IP4.GATEWAY:                            192.168.31.1
IP4.ROUTE[1]:                           dst = 192.168.31.0/24, nh = 0.0.0.0, mt = 600
IP4.ROUTE[2]:                           dst = 0.0.0.0/0, nh = 192.168.31.1, mt = 600
IP4.DNS[1]:                             192.168.31.1
IP6.ADDRESS[1]:                         fe80::ea9c:f07a:7225:3345/64
IP6.GATEWAY:                            --
IP6.ROUTE[1]:                           dst = fe80::/64, nh = ::, mt = 1024

GENERAL.DEVICE:                         lo
GENERAL.TYPE:                           loopback
GENERAL.HWADDR:                         00:00:00:00:00:00
GENERAL.MTU:                            65536
GENERAL.STATE:                          100 (connected (externally))
GENERAL.CONNECTION:                     lo
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/ActiveConnection/1
IP4.ADDRESS[1]:                         127.0.0.1/8
IP4.GATEWAY:                            --
IP6.ADDRESS[1]:                         ::1/128
IP6.GATEWAY:                            --

GENERAL.DEVICE:                         p2p-dev-wlan0
GENERAL.TYPE:                           wifi-p2p
GENERAL.HWADDR:                         (unknown)
GENERAL.MTU:                            0
GENERAL.STATE:                          30 (disconnected)
GENERAL.CONNECTION:                     --
GENERAL.CON-PATH:                       --

GENERAL.DEVICE:                         eth0
GENERAL.TYPE:                           ethernet
GENERAL.HWADDR:                         B8:27:EB:D5:7C:E7
GENERAL.MTU:                            1500
GENERAL.STATE:                          20 (unavailable)
GENERAL.CONNECTION:                     --
GENERAL.CON-PATH:                       --
WIRED-PROPERTIES.CARRIER:               off
IP4.GATEWAY:                            --
IP6.GATEWAY:                            --
```
```
$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.62+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001e400000..0x000000002e3fffff (262144 KiB) map reusable linux,cma
[... 453 lines cut ...]
[   10.111033] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
[   10.111227] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   10.111378] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   10.111517] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   10.111629] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
[   10.131697] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   10.132215] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[   10.133732] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[   10.542597] Bluetooth: hci0: BCM: features 0x2f
[   10.546023] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[   10.546056] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[   10.547288] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[   10.846003] Adding 1048572k swap on /var/swap.  Priority:-2 extents:7 across:1490940k
[   11.445410] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.445437] Bluetooth: BNEP filters: protocol multicast
[   11.445460] Bluetooth: BNEP socket layer initialized
[   11.463855] Bluetooth: MGMT ver 1.22
[   11.495497] NET: Registered PF_ALG protocol family
[   11.673065] alsactl[580]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   12.233989] lan78xx 1-1.1.1:1.0 eth0: Link is Down
[   12.262988] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   22.208704] Bluetooth: RFCOMM TTY layer initialized
[   22.208753] Bluetooth: RFCOMM socket layer initialized
[   22.208791] Bluetooth: RFCOMM ver 1.11
```

ss8688, 2024-12-04:
Is this an issue with nmtui or the driver？

dubhater, 2024-12-04:
It looks like the driver did not load. Try ``modprobe rtw_8812au".

ss8688, 2024-12-04:
> It looks like the driver did not load. Try `modprobe rtw_8812au`.

There hasn't been any change. I run `modprobe rtw_8812au` and reboot

dubhater, 2024-12-04:
After you run ``modprobe rtw_8812au`` what do you see in dmesg?

dubhater, 2024-12-04:
Without rebooting.

ss8688, 2024-12-04:
```
[   12.478774] rtw_8812au 1-1.3:1.0: failed to dump efuse logical map
[   12.487712] rtw_8812au 1-1.3:1.0: failed to setup chip efuse info
[   12.487742] rtw_8812au 1-1.3:1.0: failed to setup chip information
```

ss8688, 2024-12-04:
```
~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.62+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001e400000..0x000000002e3fffff (262144 KiB) map reusable linux,cma
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x37e0b3c0-0x37e0dfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] percpu: Embedded 30 pages/cpu s85928 r8192 d28760 u122880
[    0.000000] pcpu-alloc: s85928 r8192 d28760 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=fb33757d-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=CN
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 225792
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 599300K/917504K available (13504K kernel code, 2206K rwdata, 4292K rodata, 4864K init, 1086K bss, 56060K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43495 entries in 170 pages
[    0.000000] ftrace: allocated 170 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000402] Console: colour dummy device 80x25
[    0.000419] printk: console [tty1] enabled
[    0.001422] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001471] pid_max: default: 32768 minimum: 301
[    0.001591] LSM: initializing lsm=capability,integrity
[    0.001845] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001898] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004782] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.004963] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.005129] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.005479] rcu: Hierarchical SRCU implementation.
[    0.005506] rcu:     Max phase no-delay instances is 1000.
[... 1411 lines cut ...]
[  509.398812] usb 1-1.3: USB disconnect, device number 4
[  509.622460] usb 1-1.3: new high-speed USB device number 8 using dwc_otg
[  509.723189] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  509.723218] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  509.723225] usb 1-1.3: Product: 802.11n NIC
[  509.723230] usb 1-1.3: Manufacturer: Realtek
[... cut ...]
[  512.539173] usb 1-1.3: Product: 802.11n NIC
[  512.539179] usb 1-1.3: Manufacturer: Realtek
[  512.539184] usb 1-1.3: SerialNumber: 123456
[  512.540014] rtw_8812au 1-1.3:1.0: Firmware version 52.14.0, H2C version 0
[  513.097925] rtw_8812au 1-1.3:1.0: failed to dump efuse logical map
[  513.100914] rtw_8812au 1-1.3:1.0: failed to setup chip efuse info
[  513.100926] rtw_8812au 1-1.3:1.0: failed to setup chip information
[  513.101118] rtw_8812au: probe of 1-1.3:1.0 failed with error -22
```

ss8688, 2024-12-04:
I have another 8812au network adapter that has reported different errors：
```
[   12.643819] rtw_8812au 1-1.3:1.0: rfe 2 isn't supported
[   12.648150] rtw_8812au 1-1.3:1.0: read register 0x5 failed with -71
[   12.648629] rtw_8812au 1-1.3:1.0: read register 0x5 failed with -71
[   12.648759] rtw_8812au 1-1.3:1.0: read register 0x3 failed with -71
[   12.648899] rtw_8812au 1-1.3:1.0: write register 0x3 failed with -71
[   12.649066] rtw_8812au 1-1.3:1.0: read register 0x80 failed with -71
[   12.649218] rtw_8812au 1-1.3:1.0: write register 0x80 failed with -71
[   12.650952] rtw_8812au 1-1.3:1.0: write register 0x42 failed with -71
[   12.652552] rtw_8812au 1-1.3:1.0: write register 0x43 failed with -71
[   12.660758] rtw_8812au 1-1.3:1.0: failed to setup chip efuse info
[   12.660783] rtw_8812au 1-1.3:1.0: failed to setup chip information
```
```
[   13.619130] rtw_8812au 1-1.3:1.0: rfe 2 isn't supported
[   13.623893] rtw_8812au 1-1.3:1.0: failed to setup chip efuse info
[   13.623921] rtw_8812au 1-1.3:1.0: failed to setup chip information
```

ss8688, 2024-12-04:
full output:
```
 dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.62+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001e400000..0x000000002e3fffff (262144 KiB) map reusable linux,cma
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x37e0b3c0-0x37e0dfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] percpu: Embedded 30 pages/cpu s85928 r8192 d28760 u122880
[    0.000000] pcpu-alloc: s85928 r8192 d28760 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=fb33757d-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=CN
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 225792
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 599300K/917504K available (13504K kernel code, 2206K rwdata, 4292K rodata, 4864K init, 1086K bss, 56060K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43495 entries in 170 pages
[    0.000000] ftrace: allocated 170 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000402] Console: colour dummy device 80x25
[    0.000419] printk: console [tty1] enabled
[    0.001422] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.001471] pid_max: default: 32768 minimum: 301
[    0.001591] LSM: initializing lsm=capability,integrity
[    0.001845] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001898] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004782] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.004963] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.005129] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.005479] rcu: Hierarchical SRCU implementation.
[... 1311 lines cut ...]
[   13.623921] rtw_8812au 1-1.3:1.0: failed to setup chip information
[   13.662129] systemd-journald[242]: File /var/log/journal/91c6ddc4ba344490b95e3544916f9ed5/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   13.763668] usb 1-1.3: USB disconnect, device number 8
[   13.985693] usb 1-1.3: new high-speed USB device number 9 using dwc_otg
[   14.086805] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=881a, bcdDevice= 0.00
[... cut ...]
[   15.662106] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=881a, bcdDevice= 0.00
[   15.662124] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   15.662130] usb 1-1.3: Product: 802.11n NIC
[   15.662135] usb 1-1.3: Manufacturer: Realtek
[   15.662140] usb 1-1.3: SerialNumber: 123456
[   15.663031] rtw_8812au 1-1.3:1.0: Firmware version 52.14.0, H2C version 0
[   32.424537] Bluetooth: RFCOMM TTY layer initialized
[   32.424573] Bluetooth: RFCOMM socket layer initialized
[   32.424594] Bluetooth: RFCOMM ver 1.11
```

dubhater, 2024-12-04:
> [  513.097925] rtw_8812au 1-1.3:1.0: failed to dump efuse logical map
> [  513.100914] rtw_8812au 1-1.3:1.0: failed to setup chip efuse info
> [  513.100926] rtw_8812au 1-1.3:1.0: failed to setup chip information
> [  513.101118] rtw_8812au: probe of 1-1.3:1.0 failed with error -22

That explains it.

Just out of curiosity, what are these devices called on their packaging?

One other person reported this problem a while ago. (Or maybe it was you?) But I can't reproduce it. Please apply this patch and paste here the new output. Just the hex dump will do.

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/efuse.c b/drivers/net/wireless/realtek/rtw88/efuse.c
index b85075cd68d0..963712a57822 100644
--- a/efuse.c
+++ b/efuse.c
@@ -166,6 +166,10 @@ int rtw_parse_efuse_map(struct rtw_dev *rtwdev)
 		goto out_free;
 	}
 
+	pr_err("physical efuse map:\n");
+	print_hex_dump(KERN_INFO, "", DUMP_PREFIX_OFFSET, 16, 1,
+			phy_map, phy_size, true);
+
 	memset(log_map, 0xff, log_size);
 	ret = rtw_dump_logical_efuse_map(rtwdev, phy_map, log_map);
 	if (ret) {
```

dubhater, 2024-12-04:
For the RFE 2 error:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/rtw8812a.c b/drivers/net/wireless/realtek/rtw88/rtw8812a.c
index 2505bcd77f80..7cf335c3caed 100644
--- a/drivers/net/wireless/realtek/rtw88/rtw8812a.c
+++ b/drivers/net/wireless/realtek/rtw88/rtw8812a.c
@@ -1008,6 +1008,9 @@ static const struct rtw_rfe_def rtw8812a_rfe_defs[] = {
 	[1] = { .phy_pg_tbl	= &rtw8812a_bb_pg_tbl,
 		.txpwr_lmt_tbl	= &rtw8812a_txpwr_lmt_tbl,
 		.pwr_track_tbl	= &rtw8812a_rtw_pwr_track_tbl, },
+	[2] = { .phy_pg_tbl	= &rtw8812a_bb_pg_tbl,
+		.txpwr_lmt_tbl	= &rtw8812a_txpwr_lmt_tbl,
+		.pwr_track_tbl	= &rtw8812a_rtw_pwr_track_tbl, },
 	[3] = { .phy_pg_tbl	= &rtw8812a_bb_pg_rfe3_tbl,
 		.txpwr_lmt_tbl	= &rtw8812a_txpwr_lmt_tbl,
 		.pwr_track_tbl	= &rtw8812a_rtw_pwr_track_rfe3_tbl, },
```

ss8688, 2024-12-04:
The first one is inferior brand 
[image]

ss8688, 2024-12-04:
The second one was disassembled from Skyworth equipment 
[image]

ss8688, 2024-12-04:
> The first one is inferior brand [...]

This is called qin-fast 1200M

dubhater, 2024-12-04:
Does this help with the efuse errors?

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/rtw8812a.c b/drivers/net/wireless/realtek/rtw88/rtw8812a.c
index 7cf335c3caed..880b5b94a48f 100644
--- a/rtw8812a.c
+++ b/rtw8812a.c
@@ -1050,7 +1050,7 @@ const struct rtw_chip_info rtw8812a_hw_spec = {
 	.rx_buf_desc_sz = 8,
 	.phy_efuse_size = 512,
 	.log_efuse_size = 512,
-	.ptct_efuse_size = 96 + 1, /* TODO or just 18? */
+	.ptct_efuse_size = 0,
 	.txff_size = 131072,
 	.rxff_size = 16128,
 	.rsvd_drv_pg_num = 9,
```

ss8688, 2024-12-05:
> For the RFE 2 error:
> 
> ```diff
> diff --git a/drivers/net/wireless/realtek/rtw88/rtw8812a.c b/drivers/net/wireless/realtek/rtw88/rtw8812a.c
> index 2505bcd77f80..7cf335c3caed 100644
> --- a/rtw8812a.c
> +++ b/rtw8812a.c
> @@ -1008,6 +1008,9 @@ static const struct rtw_rfe_def rtw8812a_rfe_defs[] = {
>  	[1] = { .phy_pg_tbl	= &rtw8812a_bb_pg_tbl,
>  		.txpwr_lmt_tbl	= &rtw8812a_txpwr_lmt_tbl,
>  		.pwr_track_tbl	= &rtw8812a_rtw_pwr_track_tbl, },
> +	[2] = { .phy_pg_tbl	= &rtw8812a_bb_pg_tbl,
> +		.txpwr_lmt_tbl	= &rtw8812a_txpwr_lmt_tbl,
> +		.pwr_track_tbl	= &rtw8812a_rtw_pwr_track_tbl, },
>  	[3] = { .phy_pg_tbl	= &rtw8812a_bb_pg_rfe3_tbl,
>  		.txpwr_lmt_tbl	= &rtw8812a_txpwr_lmt_tbl,
>  		.pwr_track_tbl	= &rtw8812a_rtw_pwr_track_rfe3_tbl, },
> ```

I have tried this on my Skyworth adapter，rf2 error was fixed，but still other errors：

```
[   10.028900] rtw_8812au 1-1.3:1.0: Firmware version 52.14.0, H2C version 0
[   10.511683] Bluetooth: hci0: BCM: features 0x2f
[   10.513933] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[   10.513969] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[   10.514429] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[   10.719996] Adding 1048572k swap on /var/swap.  Priority:-2 extents:7 across:1490940k
[   11.696802] alsactl[581]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   11.699080] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.699101] Bluetooth: BNEP filters: protocol multicast
[   11.699121] Bluetooth: BNEP socket layer initialized
[   11.706861] Bluetooth: MGMT ver 1.22
[   11.733942] NET: Registered PF_ALG protocol family
[   12.320503] lan78xx 1-1.1.1:1.0 eth0: Link is Down
[   12.363944] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   15.010824] rtw_8812au 1-1.3:1.0: failed to poll offset=0x5 mask=0x1 value=0x0
[   15.010846] rtw_8812au 1-1.3:1.0: mac power on failed
[   15.010851] rtw_8812au 1-1.3:1.0: failed to power on mac
[   15.010857] rtw_8812au 1-1.3:1.0: failed to setup chip efuse info
[   15.010862] rtw_8812au 1-1.3:1.0: failed to setup chip information
[   15.011031] rtw_8812au: probe of 1-1.3:1.0 failed with error -16
[   15.011174] usbcore: registered new interface driver rtw_8812au
[   85.444326] Bluetooth: RFCOMM TTY layer initialized
[   85.444390] Bluetooth: RFCOMM socket layer initialized
[   85.444442] Bluetooth: RFCOMM ver 1.11
```

dubhater, 2024-12-05:
Does it work with another driver?

ss8688, 2024-12-05:
The other adapter does not have an RFE2 error. It has efuse dump error.

dubhater, 2024-12-05:
Does the Skyworth device work with another driver, like [this one](https://github.com/morrownr/8812au-20210820)?

ss8688, 2024-12-05:
Skyworth device can work with [PixelPilot](https://github.com/OpenIPC/PixelPilot) on android. They may use this driver https://github.com/OpenIPC/realtek-wlan
But I don't know exactly which driver they used.

dubhater, 2024-12-05:
Does this help with the power on error?

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/main.c b/drivers/net/wireless/realtek/rtw88/main.c
index 5e8baac2dfba..7674859f4635 100644
--- a/main.c
+++ b/main.c
@@ -1987,11 +1987,6 @@ static int rtw_chip_efuse_info_setup(struct rtw_dev *rtwdev)
 
 	mutex_lock(&rtwdev->mutex);
 
-	/* power on mac to read efuse */
-	ret = rtw_chip_efuse_enable(rtwdev);
-	if (ret)
-		goto out_unlock;
-
 	ret = rtw_parse_efuse_map(rtwdev);
 	if (ret)
 		goto out_disable;
@@ -2039,9 +2034,6 @@ static int rtw_chip_efuse_info_setup(struct rtw_dev *rtwdev)
 	}
 
 out_disable:
-	rtw_chip_efuse_disable(rtwdev);
-
-out_unlock:
 	mutex_unlock(&rtwdev->mutex);
 	return ret;
 }
```

ss8688, 2024-12-07:
I feel that these errors may be caused by a USB extension cable. Now I am tasting.All rtl adapters, including rtl8812bu with kernel driver, do not work with this USB extension cable, but MTK one can work, including mt7921 and mt7612.

ss8688, 2024-12-07:
I tested three different USB extension cabless and found that rtl devices cannot be used on all

a5a5aa555oo, 2024-12-07:
@ss8688 Do all your usb cables support USB3? My 8822bu works fine with an USB2 extension cable.

dubhater, 2024-12-10:
If it's the same with three different cables, the cables are probably good. But if you can plug the devices directly, try that.

dubhater, 2024-12-13:
@briansune I'm glad that patch fixes the efuse error.

> I notice the USB 3.0 signal link components are not placed aka capacitors and resistors.

What device is this? It's not all RTL8812AU, others really are USB 3.

> aka rtw_8812au 1-1.3:1.0: rfe 2 isn't supported

If you pulled today, that error is already fixed.

I don't think RFE 2 is newer, just a different combination of PA and LNA. "RFE" is "RF front end".

> The connection latency is very poor and unstable

The ping results should improve if you disable the power saving. I noticed this problem too, but I don't know what's wrong.

The driver disables the power saving when there is a bit of traffic. ping just doesn't generate enough traffic. So when you're browsing or downloading something it's not going to be a problem.

> [ 23.939271] efuse thermal meter is 0xff

Your device is probably missing some calibration info. If you use the debug_mask=0x20 module parameter we can see what else is missing in the efuse.

Apparently when the chip gets hot, the TX power can degrade. The driver compensates for this, but only if it finds the thermal meter calibration info in the efuse. In your case that byte is 0xff (undefined). It will still work.

> USB might require huge power > 500mA

The official driver has some code to reduce the TX power in USB 2 mode to avoid drawing more than 500 mA. rtw88 doesn't have that code.

briansune, 2024-12-14:
@dubhater 

As mentioned this is RTL8812AU USB dongle.
This is completely normal on the hardware design using super-speed or high-speed can be select via the connector itself or disable by no placing the components from first place.

So my dongle completely ignore all USB 3.0 possible hazard from first place.
No hardware modification is done just bought the dongle and use to test the driver.

As for RFE this is ture from the board you can clearly see there are no extra AMP chip or components are placed in the upper part.
So RFE is not an issue in my dongle.

Will try to use debug flag but after all RTW88 on USB, SDIO also show a poor performance compared to old RLT8732DS.

Meantime it is also not a complete fixed when the a chip is blank w/o firmware and no recover system is embedded in the RTW88 driver (possibly I am wrong)

briansune, 2024-12-14:
@dubhater 

BTW if you follow the hardware setup in this:
https://github.com/lwfinger/rtw88/issues/261#issuecomment-2517829605

If this is a RTL8812AU USB dongle then I am very sure that there is no chance this can work in super-speed.
This setup can only support up to USB high-speed due to only D+/D- pair.
So this also proven that RTL8812AU doesn't necessary need USB super speed to work properly.
And from theory USB itself all support lower speed grade when higher speed grade is not working.
So Super speed  -> High speed -> Full speed etc.
How the super speed work from beginning is rely on SSTX P/N and SSRX P/N to run 5Gbps.

briansune, 2024-12-14:
@dubhater 

For the debug the efuse I guess you are referring to this modification:
https://github.com/lwfinger/rtw88/issues/261#issuecomment-2517790993

I also rebuilt the kernel with the debug flag turned on with 0x20.

<details>
  <summary>EFUSE</summary>

[    1.779725] DEBUG FOR - physical efuse map:
[    1.779744] 00000000: 08 29 81 0b 86 0c 00 4f 31 1a 88 41 ff c2 ff 6f  .).....O1..A...o
[    1.779758] 00000010: 33 ff 03 03 01 8f 30 05 03 4e 49 43 03 03 01 07  3.....0..NIC....
[    1.779765] 00000020: 00 00 10 00 00 00 00 00 00 00 00 20 28 28 28 28  ........... ((((
[    1.779772] 00000030: 26 26 2a 2a 38 2a 2a 28 25 ee ee 41 2a 2a 27 27  &&**8**(%..A**''
[    1.779779] 00000040: 29 29 50 24 24 23 23 24 24 26 26 62 04 ee ee ff  ))P$$##$$&&b....
[    1.779786] 00000050: dd ee 71 24 24 23 23 27 27 80 26 26 25 25 29 25  ..q$$##''.&&%%)%
[    1.779793] 00000060: ee ee 93 31 31 2d 2d a0 2d 2d 25 25 23 23 25 25  ...11--.--%%##%%
[    1.779800] 00000070: b4 28 28 04 ee ee ff ce dd ee ef 20 29 1e 18 00  .((........ )...
[    1.779807] 00000080: 00 00 00 88 0f 30 00 01 00 10 00 00 00 55 2f 3c  .....0.......U/<
[    1.779814] 00000090: 00 ff 02 ff 4f 32 da 0b 41 66 c2 1c 6f 30 a7 70  ....O2..Af..o0.p
[    1.779821] 000000a0: aa a6 cd 09 03 52 8f 30 65 61 6c 74 65 6b 0d 03  .....R.0ealtek..
[    1.779828] 000000b0: af 30 38 30 32 2e 31 31 6e 20 cf 30 4e 49 43 08  .0802.11n .0NIC.
[    1.779835] 000000c0: 03 31 32 33 ef 30 34 35 36 00 00 00 00 00 0f 40  .123.0456......@
[    1.779842] 000000d0: 00 00 00 00 00 00 00 00 2f 40 00 00 00 00 00 00  ......../@......
[    1.779849] 000000e0: 00 00 4f 40 00 00 00 00 00 00 00 00 cf 40 81 ae  ..O@.........@..
[    1.779856] 000000f0: 98 2d 03 93 96 11 ef 48 fc 8c 00 11 9b ff 07 08  .-.....H........
[    1.779863] 00000100: 00 1e 02 00 20 2e 2e 2e 2f 2f 2f 32 32 3c 32 32  .... ...///22<22
[    1.779870] 00000110: 32 02 41 24 22 20 1e 22 20 50 1f 1f 1d 1c 1a 1a  2.A$" ." P......
[    1.779877] 00000120: 1a 1a 66 02 ee 20 ee 71 2b 2b 2b 2c 2c 2c 88 32  ..f.. .q+++,,,.2
[    1.779884] 00000130: 32 32 32 31 04 93 24 23 21 21 a0 24 21 1f 1d 1a  2221..$#!!.$!...
[    1.779891] 00000140: 17 19 18 bc 18 18 02 ee ce 20 ee ef 24 7f 27 ff  ......... ..$.'.
[    1.779897] 00000150: 00 00 00 2f 3d 01 ff 4f 35 12 88 42 48 6f 38 8f  .../=..O5..BHo8.
[    1.779904] 00000160: 4c 90 16 0a 09 ef 31 36 ff ff ff ff ff 0f 40 ff  L.....16......@.
[    1.779912] 00000170: ff ff ff ff ff ff ff 2f 40 ff ff ff ff ff ff ff  ......./@.......
[    1.779918] 00000180: ff 4f 40 ff ff ff ff ff ff ff 21 6f 48 03 03 0a  .O@.......!oH...
[    1.779925] 00000190: ff 00 ff ef 43 9b 78 4a ce 0f 50 e0 48 e0 70 27  ....C.xJ..P.H.p'
[    1.779932] 000001a0: 00 78 ea 2f 50 60 d0 e1 20 32 63 4f 3c 4f 50 12  .x./P`.. 2cO<OP.
[    1.779939] 000001b0: 40 92 15 8b 4a 4f 8a 6f 5c 15 d1 98 00 af 50 00  @...JO.o\.....P.
[    1.779946] 000001c0: 02 e1 63 12 29 03 00 cf 50 cb 3f a0 e0 c2 f0 f7  ..c.)...P.?.....
[    1.779953] 000001d0: cf ef 50 01 36 08 32 05 eb 94 d7 0f 60 14 6d c8  ..P.6.2.....`.m.
[    1.779960] 000001e0: 67 80 e0 f8 80 ff ff ff ff ff ff ff ff ff ff ff  g...............
[    1.779967] 000001f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff 0c 00  ................

[    2.520544] usb 1-1: new high-speed USB device number 3 using dwc2
[    2.729045] usb 1-1: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[    2.729072] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.729083] usb 1-1: Product: 802.11n NIC
[    2.729091] usb 1-1: Manufacturer: Realtek
[    2.729099] usb 1-1: SerialNumber: 123456
[    2.729894] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[    3.156752] DEBUG FOR - physical efuse map:
[    3.156770] 00000000: 08 29 81 0b 86 0c 00 4f 31 1a 88 41 ff c2 ff 6f  .).....O1..A...o
[    3.156785] 00000010: 33 ff 03 03 01 8f 30 05 03 4e 49 43 03 03 01 07  3.....0..NIC....
[    3.156792] 00000020: 00 00 10 00 00 00 00 00 00 00 00 20 28 28 28 28  ........... ((((
[    3.156799] 00000030: 26 26 2a 2a 38 2a 2a 28 25 ee ee 41 2a 2a 27 27  &&**8**(%..A**''
[    3.156806] 00000040: 29 29 50 24 24 23 23 24 24 26 26 62 04 ee ee ff  ))P$$##$$&&b....
[    3.156813] 00000050: dd ee 71 24 24 23 23 27 27 80 26 26 25 25 29 25  ..q$$##''.&&%%)%
[    3.156820] 00000060: ee ee 93 31 31 2d 2d a0 2d 2d 25 25 23 23 25 25  ...11--.--%%##%%
[    3.156827] 00000070: b4 28 28 04 ee ee ff ce dd ee ef 20 29 1e 18 00  .((........ )...
[    3.156834] 00000080: 00 00 00 88 0f 30 00 01 00 10 00 00 00 55 2f 3c  .....0.......U/<
[    3.156841] 00000090: 00 ff 02 ff 4f 32 da 0b 41 66 c2 1c 6f 30 a7 70  ....O2..Af..o0.p
[    3.156848] 000000a0: aa a6 cd 09 03 52 8f 30 65 61 6c 74 65 6b 0d 03  .....R.0ealtek..
[    3.156855] 000000b0: af 30 38 30 32 2e 31 31 6e 20 cf 30 4e 49 43 08  .0802.11n .0NIC.
[    3.156862] 000000c0: 03 31 32 33 ef 30 34 35 36 00 00 00 00 00 0f 40  .123.0456......@
[    3.156869] 000000d0: 00 00 00 00 00 00 00 00 2f 40 00 00 00 00 00 00  ......../@......
[    3.156876] 000000e0: 00 00 4f 40 00 00 00 00 00 00 00 00 cf 40 81 ae  ..O@.........@..
[    3.156883] 000000f0: 98 2d 03 93 96 11 ef 48 fc 8c 00 11 9b ff 07 08  .-.....H........
[    3.156889] 00000100: 00 1e 02 00 20 2e 2e 2e 2f 2f 2f 32 32 3c 32 32  .... ...///22<22
[    3.156896] 00000110: 32 02 41 24 22 20 1e 22 20 50 1f 1f 1d 1c 1a 1a  2.A$" ." P......
[... 33 lines cut ...]
[   32.214725] wlx488f4c90160a: send auth to f8:6f:b0:0e:ae:e6 (try 2/3)
[... cut ...]
[   80.467389] wlx488f4c90160a: authenticated
[   80.468611] wlx488f4c90160a: associate with f8:6f:b0:0e:ae:e5 (try 1/3)
[   80.519596] wlx488f4c90160a: RX AssocResp from f8:6f:b0:0e:ae:e5 (capab=0x1411 status=0 aid=4)
[   80.528333] wlx488f4c90160a: associated
[   80.602377] wlx488f4c90160a: Limiting TX power to 35 (35 - 0) dBm as advertised by f8:6f:b0:0e:ae:e5
[   80.718061] IPv6: ADDRCONF(NETDEV_CHANGE): wlx488f4c90160a: link becomes ready
[   84.963532] efuse thermal meter is 0xff

</details>

briansune, 2024-12-14:
@dubhater 

As for the power saving is triggering a poor ping result.
According to the inspection this is not the case.

I am showing iperf3 + ping concurrently running together with large packet setting.

<details>
  <summary>Ping Result + IPERF3</summary>

ubuntu@briansune:~$ ping 192.168.1.1 -s 10000
PING 192.168.1.1 (192.168.1.1) 10000(10028) bytes of data.
10008 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=120 ms
10008 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=142 ms
10008 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=162 ms
Accepted connection from 192.168.1.3, port 54665
[  5] local 192.168.1.19 port 5201 connected to 192.168.1.3 port 54666
10008 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=71.1 ms
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.34 MBytes  19.6 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=94.8 ms
[  5]   1.00-2.00   sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=81.7 ms
[  5]   2.00-3.00   sec  1.90 MBytes  15.9 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=64.6 ms
[  5]   3.00-4.00   sec  1.84 MBytes  15.4 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=61.5 ms
[  5]   4.00-5.00   sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=71.1 ms
[  5]   5.00-6.00   sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=79.5 ms
[  5]   6.00-7.00   sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=89.8 ms
[  5]   7.00-8.00   sec  1.84 MBytes  15.4 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=60.9 ms
[  5]   8.00-9.00   sec  1.84 MBytes  15.4 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=130 ms
[  5]   9.00-10.00  sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=73.4 ms
[  5]  10.00-11.00  sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=92.2 ms
[  5]  11.00-12.00  sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=76.2 ms
[  5]  12.00-13.00  sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=59.3 ms
[  5]  13.00-14.00  sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=82.2 ms
[  5]  14.00-15.00  sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=91.7 ms
[  5]  15.00-16.00  sec  1.23 MBytes  10.3 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=20 ttl=64 time=99.4 ms
[  5]  16.00-17.00  sec  1.84 MBytes  15.4 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=21 ttl=64 time=46.2 ms
[  5]  17.00-18.00  sec  1.53 MBytes  12.8 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=22 ttl=64 time=74.5 ms
[  5]  18.00-19.00  sec  1.84 MBytes  15.4 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=23 ttl=64 time=52.9 ms
[  5]  19.00-20.00  sec  2.51 MBytes  21.1 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=24 ttl=64 time=64.0 ms
[  5]  20.00-21.00  sec  2.45 MBytes  20.6 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=25 ttl=64 time=40.3 ms
[  5]  21.00-22.00  sec  2.45 MBytes  20.6 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=26 ttl=64 time=36.3 ms
[  5]  22.00-23.00  sec  2.45 MBytes  20.6 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=27 ttl=64 time=52.1 ms
[  5]  23.00-24.00  sec  2.76 MBytes  23.1 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=28 ttl=64 time=62.5 ms
[  5]  24.00-25.00  sec  2.14 MBytes  18.0 Mbits/sec    0    135 KBytes
10008 bytes from 192.168.1.1: icmp_seq=29 ttl=64 time=52.1 ms
[  5]  25.00-26.00  sec  2.21 MBytes  18.5 Mbits/sec    0    135 KBytes
[... 218 lines cut ...]
10008 bytes from 192.168.1.1: icmp_seq=139 ttl=64 time=44.9 ms
[  5] 135.00-136.00 sec   941 KBytes  7.71 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=140 ttl=64 time=67.6 ms
[  5] 136.00-137.00 sec  1.84 MBytes  15.4 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=141 ttl=64 time=70.4 ms
[  5] 137.00-138.00 sec  1.84 MBytes  15.4 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=142 ttl=64 time=64.3 ms
[  5] 138.00-139.00 sec  1.53 MBytes  12.8 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=143 ttl=64 time=64.6 ms
[  5] 139.00-140.00 sec  1.84 MBytes  15.4 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=144 ttl=64 time=102 ms
[  5] 140.00-141.00 sec  1.53 MBytes  12.8 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=145 ttl=64 time=75.4 ms
[  5] 141.00-142.00 sec  1.23 MBytes  10.3 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=146 ttl=64 time=113 ms
[  5] 142.00-143.00 sec  1.23 MBytes  10.3 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=147 ttl=64 time=64.9 ms
[  5] 143.00-144.00 sec  1.23 MBytes  10.3 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=148 ttl=64 time=115 ms
[  5] 144.00-145.00 sec  1.23 MBytes  10.3 Mbits/sec    0    130 KBytes
10008 bytes from 192.168.1.1: icmp_seq=149 ttl=64 time=57.2 ms
[  5] 145.00-146.00 sec  1.53 MBytes  12.8 Mbits/sec    0    130 KBytes
[  5] 145.00-146.00 sec  1.53 MBytes  12.8 Mbits/sec    0    130 KBytes

</details>

dubhater, 2024-12-14:
Now the pings are high because of iperf3. :) Try to turn off power saving with ``iw wlx.... set power_save off`` and test ping by itself.

> SDIO also show a poor performance compared to old RLT8732DS.

There is not much I can do about that because I don't have any SDIO devices.

briansune, 2024-12-14:
> Now the pings are high because of iperf3. :) Try to turn off power saving with `iw wlx.... set power_save off` and test ping by itself.
> 
> > SDIO also show a poor performance compared to old RLT8732DS.
> 
> There is not much I can do about that because I don't have any SDIO devices.

No I had tried all three situations all result is bad latency and unstable return.

Of cause I haven't manually turn off the power save because it should automatically exit power save when large packet.

A: Only ping large packet

B: iperf3 only 

C: Both concurrently running

When manually remove power save as you said indeed the latency is better a lot.
But pinging the main router is around 30ms which is ok but still there are stalls.

ping 192.168.1.1 -s 10000
PING 192.168.1.1 (192.168.1.1) 10000(10028) bytes of data.
10008 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=5.96 ms
10008 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=5.79 ms
10008 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=12.0 ms
10008 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=5.42 ms
10008 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=8.43 ms
10008 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=5.80 ms
10008 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=8.50 ms
10008 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=6.87 ms
10008 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=6.82 ms
10008 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=20.8 ms
10008 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=6.41 ms
10008 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=5.60 ms
10008 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=11.8 ms
10008 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=15.0 ms
10008 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=10.1 ms
10008 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=14.5 ms
10008 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=14.7 ms

BTW the thermal is still reporting issue:

[   24.967555] efuse thermal meter is 0xff

briansune, 2024-12-17:
@dubhater 

I got some unusage engineer board with RTL8723DS SDIO based WIFI do you need me to ship it to you?

dubhater, 2024-12-17:
@briansune Sure, that could work.

briansune, 2024-12-17:
@dubhater 

Lets doing of this line via EMAIL.

briansune@gmail.com

briansune, 2024-12-18:
@dubhater 

Email me or share your Email address to follow up the shipping etc.

dubhater, 2024-12-18:
@briansune Yes, I will email you today.

ackPeng, 2025-07-04:
@dubhater hi,I also encountered the error "failed to dump efuse logical map" when using another device, rtw8823du. However, this error does not occur every time the device is started. Sometimes it occurs once after restart. Do you have any ideas for troubleshooting?

ackPeng, 2025-07-04:
@dubhater 以下是我的print_hex_dump信息：[   11.970299] rtw_8723du 1-3:1.2: Firmware version 48.0.0, H2C version 0
[   13.168294] rtw_8723du 1-3:1.2: EFUSE raw logical map:
[   13.168314] rtw_8723du 1-3:1.2: Parsed rtw8723x EFUSE data:
[   13.168316] rtw_8723du 1-3:1.2: rtl_id=0x8129
[   13.168317] rtw_8723du 1-3:1.2: afe=0x01
[   13.168319] rtw_8723du 1-3:1.2: Power index table (2.4G):
[   13.168320] rtw_8723du 1-3:1.2: CCK base
[   13.168321] rtw_8723du 1-3:1.2: RF    G0  G1  G2  G3  G4  G5
[   13.168323] rtw_8723du 1-3:1.2: [A]:  43  43  42  42  42  42
[   13.168324] rtw_8723du 1-3:1.2: [B]:  43  43  42  42  42  42
[   13.168325] rtw_8723du 1-3:1.2: [C]: 255 255 255 255 255 255
[   13.168327] rtw_8723du 1-3:1.2: [D]: 255 255 255 255 255 255
[   13.168328] rtw_8723du 1-3:1.2: CCK diff
[   13.168329] rtw_8723du 1-3:1.2: RF   1S 2S 3S 4S
[   13.168331] rtw_8723du 1-3:1.2: [A]:  0 -1 -1 -1
[   13.168332] rtw_8723du 1-3:1.2: [B]:  0 -1 -1 -1
[   13.168333] rtw_8723du 1-3:1.2: [C]:  0 -1 -1 -1
[   13.168335] rtw_8723du 1-3:1.2: [D]:  0 -1 -1 -1
[   13.168336] rtw_8723du 1-3:1.2: BW40-1S base
[   13.168337] rtw_8723du 1-3:1.2: RF    G0  G1  G2  G3  G4
[   13.168338] rtw_8723du 1-3:1.2: [A]:  42  42  41  41  41
[   13.168340] rtw_8723du 1-3:1.2: [B]:  42  42  41  41  41
[   13.168342] rtw_8723du 1-3:1.2: [C]: 255 255 255 255 255
[   13.168343] rtw_8723du 1-3:1.2: [D]: 255 255 255 255 255
[   13.168344] rtw_8723du 1-3:1.2: OFDM diff
[   13.168346] rtw_8723du 1-3:1.2: RF   1S 2S 3S 4S
[   13.168347] rtw_8723du 1-3:1.2: [A]:  0 -1 -1 -1
[   13.168348] rtw_8723du 1-3:1.2: [B]:  0 -1 -1 -1
[   13.168349] rtw_8723du 1-3:1.2: [C]: -1 -1 -1 -1
[   13.168350] rtw_8723du 1-3:1.2: [D]: -1 -1 -1 -1
[   13.168352] rtw_8723du 1-3:1.2: BW20 diff
[   13.168353] rtw_8723du 1-3:1.2: RF   1S 2S 3S 4S
[   13.168354] rtw_8723du 1-3:1.2: [A]:  0 -1 -1 -1
[   13.168356] rtw_8723du 1-3:1.2: [B]:  0 -1 -1 -1
[   13.168357] rtw_8723du 1-3:1.2: [C]: -1 -1 -1 -1
[   13.168358] rtw_8723du 1-3:1.2: [D]: -1 -1 -1 -1
[   13.168359] rtw_8723du 1-3:1.2: BW40 diff
[   13.168360] rtw_8723du 1-3:1.2: RF   1S 2S 3S 4S
[   13.168361] rtw_8723du 1-3:1.2: [A]:  0 -1 -1 -1
[   13.168363] rtw_8723du 1-3:1.2: [B]:  0 -1 -1 -1
[   13.168364] rtw_8723du 1-3:1.2: [C]:  0 -1 -1 -1
[   13.168365] rtw_8723du 1-3:1.2: [D]:  0 -1 -1 -1
[   13.168366] rtw_8723du 1-3:1.2: channel_plan=0x20
[   13.168367] rtw_8723du 1-3:1.2: xtal_k=0x1e
[   13.168368] rtw_8723du 1-3:1.2: thermal_meter=0x1d
[   13.168369] rtw_8723du 1-3:1.2: iqk_lck=0x00
[   13.168370] rtw_8723du 1-3:1.2: pa_type=0x00
[   13.168372] rtw_8723du 1-3:1.2: lna_type_2g=0x00ff
[   13.168373] rtw_8723du 1-3:1.2: lna_type_5g=0xffff
[   13.168374] rtw_8723du 1-3:1.2: rf_board_option=0x29
[   13.168375] rtw_8723du 1-3:1.2: rf_feature_option=0x20
[   13.168376] rtw_8723du 1-3:1.2: rf_bt_setting=0x51
[   13.168377] rtw_8723du 1-3:1.2: eeprom_version=0x00
[   13.168378] rtw_8723du 1-3:1.2: eeprom_customer_id=0x00
[   13.168379] rtw_8723du 1-3:1.2: tx_bb_swing_setting_2g=0x00
[   13.168380] rtw_8723du 1-3:1.2: tx_pwr_calibrate_rate=0x00
[   13.168381] rtw_8723du 1-3:1.2: rf_antenna_option=0xff
[   13.168383] rtw_8723du 1-3:1.2: rfe_option=0x11
[   13.168384] rtw_8723du 1-3:1.2: country_code=0xffff
[   13.168386] rtw_8723du 1-3:1.2: u.vendor_id=0xda0b
[   13.168387] rtw_8723du 1-3:1.2: u.product_id=0x23d7
[   13.168388] rtw_8723du 1-3:1.2: u.usb_option=0xc7
[   13.168390] rtw_8723du 1-3:1.2: mac_addr=14:5d:34:3b:c8:ee
[   13.179479] rtw_8723du 1-3:1.2: hw cap: hci=0x05, bw=0x03, ptcl=0x02, ant_num=2, nss=1
[   13.246185] usbcore: registered new interface driver rtw_8723du