lwfinger/rtw88 #262: Connect / Disconnect loop

StephenAtty, 2024-12-06:
Git cloned the repository and went through the build and install steps.

Rebooted and it worked for a couple of minutes and then disconnected from the router ( :Archer VR600 v2 00000000 )

Wouldn't reconnect

Restarted  and it went into a connect/drop cycle

dmesg shows :

[   26.036502] usb 2-2: new SuperSpeed USB device number 3 using xhci_hcd
[   26.049096] usb 2-2: New USB device found, idVendor=2357, idProduct=012d, bcdDevice= 3.00
[   26.049105] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   26.049108] usb 2-2: Product: 802.11ac NIC
[   26.049111] usb 2-2: Manufacturer: Realtek
[   26.049113] usb 2-2: SerialNumber: 123456
[   26.051878] rtw_8822bu 2-2:1.0: Firmware version 30.20.0, H2C version 14
[   26.277125] rtw_8822bu 2-2:1.0 wlx30de4b2dcfd1: renamed from wlan0
[   27.465951] rtw_8822bu 2-2:1.0: failed to do USB write, ret=-19
[   27.465964] rtw_8822bu 2-2:1.0: failed to send h2c packet
[   27.465968] rtw_8822bu 2-2:1.0: failed to do USB write, ret=-19
[   27.465971] rtw_8822bu 2-2:1.0: failed to send h2c packet
[   27.576491] usb 2-2: USB disconnect, device number 3
[   28.621528] usb 2-2: new SuperSpeed USB device number 4 using xhci_hcd
[   28.634015] usb 2-2: New USB device found, idVendor=2357, idProduct=012d, bcdDevice= 3.00
[   28.634026] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   28.634029] usb 2-2: Product: 802.11ac NIC
[   28.634032] usb 2-2: Manufacturer: Realtek
[   28.634035] usb 2-2: SerialNumber: 123456
[   28.637588] rtw_8822bu 2-2:1.0: Firmware version 30.20.0, H2C version 14
[   28.861863] rtw_8822bu 2-2:1.0 wlx30de4b2dcfd1: renamed from wlan0
[   35.354929] wlx30de4b2dcfd1: authenticate with d8:47:32:8c:79:5a (local address=1e:1e:58:8c:4b:4e)
[   35.420617] wlx30de4b2dcfd1: send auth to d8:47:32:8c:79:5a (try 1/3)
[   35.421229] wlx30de4b2dcfd1: authenticated
[   35.422383] wlx30de4b2dcfd1: associate with d8:47:32:8c:79:5a (try 1/3)
[   35.423462] wlx30de4b2dcfd1: RX AssocResp from d8:47:32:8c:79:5a (capab=0x11 status=0 aid=7)
[   35.427813] wlx30de4b2dcfd1: associated
[   45.984521] usb 2-2: USB disconnect, device number 4
[   45.990436] wlx30de4b2dcfd1: deauthenticating from d8:47:32:8c:79:5a by local choice (Reason: 3=DEAUTH_LEAVING)
[   46.845585] usb 2-2: new SuperSpeed USB device number 5 using xhci_hcd
[   46.858095] usb 2-2: New USB device found, idVendor=2357, idProduct=012d, bcdDevice= 3.00
[   46.858107] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   46.858110] usb 2-2: Product: 802.11ac NIC
[   46.858113] usb 2-2: Manufacturer: Realtek
[   46.858116] usb 2-2: SerialNumber: 123456
[   46.861643] rtw_8822bu 2-2:1.0: Firmware version 30.20.0, H2C version 14
[   47.086175] rtw_8822bu 2-2:1.0 wlx30de4b2dcfd1: renamed from wlan0
[   53.575383] wlx30de4b2dcfd1: authenticate with d8:47:32:8c:79:5a (local address=a6:d5:70:5e:87:c2)
[   53.643033] wlx30de4b2dcfd1: send auth to d8:47:32:8c:79:5a (try 1/3)
[   53.643660] wlx30de4b2dcfd1: authenticated
[   53.644373] wlx30de4b2dcfd1: associate with d8:47:32:8c:79:5a (try 1/3)
[   53.645454] wlx30de4b2dcfd1: RX AssocResp from d8:47:32:8c:79:5a (capab=0x11 status=0 aid=8)
[   53.649709] wlx30de4b2dcfd1: associated
[   65.266856] show_signal_msg: 117 callbacks suppressed
[   65.266863] nm-applet[2130]: segfault at 32633a4a ip 000071d9922c5cd8 sp 00007ffe10d06de8 error 4 in libgobject-2.0.so.0.8000.0[71d992298000+37000] likely on CPU 1 (core 1, socket 0)
[   65.266877] Code: 8d 15 8c 54 02 00 48 c1 e9 02 4c 8b 0c ca 4d 85 c9 74 07 41 f6 41 16 20 75 36 48 8b 38 48 81 ff fc 03 00 00 76 3c 48 83 e7 fc <f6> 47 16 10 74 05 4d 85 c9 75 4d 31 c0 c3 66 2e 0f 1f 84 00 00 00
[   94.400380] usb 2-2: USB disconnect, device number 5
[   94.407454] wlx30de4b2dcfd1: deauthenticating from d8:47:32:8c:79:5a by local choice (Reason: 3=DEAUTH_LEAVING)
[   95.590421] usb 2-2: new SuperSpeed USB device number 6 using xhci_hcd
[   95.602927] usb 2-2: New USB device found, idVendor=2357, idProduct=012d, bcdDevice= 3.00
[   95.602936] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   95.602939] usb 2-2: Product: 802.11ac NIC
[   95.602942] usb 2-2: Manufacturer: Realtek
[   95.602944] usb 2-2: SerialNumber: 123456
[   95.605901] rtw_8822bu 2-2:1.0: Firmware version 30.20.0, H2C version 14
[   95.832974] rtw_8822bu 2-2:1.0 wlx30de4b2dcfd1: renamed from wlan0
[  102.320541] wlx30de4b2dcfd1: authenticate with d8:47:32:8c:79:5a (local address=c2:13:3f:1f:dc:bf)
[  102.386803] wlx30de4b2dcfd1: send auth to d8:47:32:8c:79:5a (try 1/3)
[  102.387506] wlx30de4b2dcfd1: authenticated
[  102.388368] wlx30de4b2dcfd1: associate with d8:47:32:8c:79:5a (try 1/3)
[... 69 lines cut ...]
[  149.996612] wlx30de4b2dcfd1: associated
[  160.704430] usb 2-2: USB disconnect, device number 10
[  160.714439] wlx30de4b2dcfd1: deauthenticating from d8:47:32:8c:79:5a by local choice (Reason: 3=DEAUTH_LEAVING)
[  161.657521] usb 2-2: new SuperSpeed USB device number 11 using xhci_hcd
[  161.670368] usb 2-2: New USB device found, idVendor=2357, idProduct=012d, bcdDevice= 3.00
[  161.670378] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  161.670382] usb 2-2: Product: 802.11ac NIC
[  161.670384] usb 2-2: Manufacturer: Realtek
[  161.670387] usb 2-2: SerialNumber: 123456
[  161.673737] rtw_8822bu 2-2:1.0: Firmware version 30.20.0, H2C version 14
[  161.899619] rtw_8822bu 2-2:1.0 wlx30de4b2dcfd1: renamed from wlan0
[  168.391508] wlx30de4b2dcfd1: authenticate with d8:47:32:8c:79:5a (local address=8e:46:9c:d2:cb:9d)
[  168.458038] wlx30de4b2dcfd1: send auth to d8:47:32:8c:79:5a (try 1/3)
[  168.458785] wlx30de4b2dcfd1: authenticated
[  168.459378] wlx30de4b2dcfd1: associate with d8:47:32:8c:79:5a (try 1/3)
[  168.460496] wlx30de4b2dcfd1: RX AssocResp from d8:47:32:8c:79:5a (capab=0x11 status=0 aid=9)
[  168.465044] wlx30de4b2dcfd1: associated
`
Card works fine at the lower speed using the drivers built into the kernel 6.8.0-49.49.   Running Linux Mint 22 XCFE 

  Port 002: Dev 003, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 480M

and works fine under windows 11 at the higher speed

I turned off Power Management but it didn't seem to make any difference

dubhater, 2024-12-07:
Does it work with this driver if you plug it into a USB 2 port? Or you can disable the switching with a module parameter.

StephenAtty, 2024-12-07:
Yes it's rock solid - with the little green led flashing away.

/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 002: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 002: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 003: Dev 006, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 480M
    |__ Port 006: Dev 004, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 006: Dev 004, If 1, Class=Video, Driver=uvcvideo, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 10000M

On the USB 3 Port
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 002: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 002: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 006: Dev 004, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 006: Dev 004, If 1, Class=Video, Driver=uvcvideo, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 10000M
    |__ Port 001: Dev 004, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 5000M

I have two two of these adaptors ( Archer T3U AC1300 Mini Wireless MU-MIMO USB WiFi Adapter ) and two Dell Inspiron 3793 Laptops

Both exhibit the same issues with the disconnect / reconnect loop. It seems to take a random time to happen the first time and doesn't seem to be related to network activity.

Both laptops have 2 USB3 ports and the adaptors exhibit the same behaviour in either port

a5a5aa555oo, 2024-12-07:
@StephenAtty 
Do you wanna try the old firmware v27.2.0? maybe your issue is caused by the [new firmware](https://github.com/lwfinger/rtw88/commit/4a0d5478a798f2867e6adcdb9710e55ee696f5b2) I uploaded last month?
```
# download firmware v27.2.0 from linux-firmware
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtw88/rtw8822b_fw.bin

# remove the firmware you are using currently
sudo rm -vf /lib/firmware/rtw88/*8822b*

# copy firmware v27.2.0 to /lib/firmware/rtw88/
sudo cp -v ./rtw8822b_fw.bin /lib/firmware/rtw88/

# reload the rtw88 driver
sudo rmmod  rtw_8822bu rtw_8822b rtw_usb rtw_core
sudo modprobe rtw_8822bu
```

StephenAtty, 2024-12-07:
Firmware downgrade didn't help - I thought it had because it lasted for about 10 minutes before giving up.

What commands do I need to put into which file to get it to run as USB2 in the USB3 port as I'd like to keep it in the USB3 port

a5a5aa555oo, 2024-12-07:
@StephenAtty 

This command will create a file named rtw88.conf in /etc/modprobe.d and it will keep your adapter in USB2 mode.

```
sudo sh -c 'echo options rtw_usb switch_usb_mode=0 > /etc/modprobe.d/rtw88.conf'
```

StephenAtty, 2024-12-07:
@a5a5aa555oo 

Thanks - that's done it and its now stable in the USB3 port running as USB2.

Should I upgrade the firmware back to what it was as it doesn't seem to be the problem

a5a5aa555oo, 2024-12-07:
@StephenAtty It is up to you, both version of firmware work fine with my 8822BU.

a5a5aa555oo, 2024-12-07:
dubhater is busy adding support for RTL8814AU in the rtw88 driver, so this issue may be fixed after that.

StephenAtty, 2024-12-07:
with the USB  setting forcing it down to USB2 on the USB3 port the new driver is rock solid - just moved a 30GB file down and not a single error message anywhere.

dubhater, 2024-12-07:
Does this patch help? [wifi rtw88 Add USB PHY configuration.diff.txt](https://github.com/user-attachments/files/18049616/wifi.rtw88.Add.USB.PHY.configuration.diff.txt)

StephenAtty, 2024-12-07:
I've applied that, turned off the USB2 config and rebooted. 

Pulled 30GB off my server
Pushed 30GB back to the server

Applied changes to second dell
Currently pushing 30GB over to that.

No failures or errors in dmesg output on either laptop

So it looks like it might have fixed it .... will keep on testing and let you know how things go

StephenAtty, 2024-12-08:
Both machines solid - moved more data around with no glitches. Throughput is about double what it was with the USB2 setting and the in kernel driver.

closed 2024-12-10 by dubhater