lwfinger/rtw88 #263: Cannot use AP to share wifi signal (rtw88_8822be)

AndreyAdamenco, 2024-12-09:
I cant share wifi using AP mode. I tried using built in hotspot feature and tried using "Linux-wifi-hotspot". It shared wifi on Windows but not on linux, I tried Pop os and now on Manjaro they both have this issue.
[image]
**Results of iw list command** 
[image]

dubhater, 2024-12-10:
wihotspot wants a wifi device which can work in station mode and AP mode at the same time. This is also known as concurrent mode. rtw88 in the kernel only supports concurrent mode with RTL8822C. I just enabled concurrent mode for all chips in this repository. If you install the driver from here you can test if that works for your RTL8822BE device.

morrownr, 2024-12-10:
@dubhater 

Please correct me if I am wrong but I think `Concurrent Mode` is Realtek only terminology. Other wireless hardware/driver providers call it Virtual Interface. When I run `iw list` on one of my Mediatek adapters, I get the following:

valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2

I have not run `iw list` on rtw88 based adapters but the vendor drivers, while they have Concurrent Mode capability, they report No Interface Combinations supported. This is just what I think that I know and maybe it adds to the conversation.

AndreyAdamenco, 2024-12-10:
when I try to build , it says that the file exists and  I suppose it is installed ?
[image]

a5a5aa555oo, 2024-12-10:
Hi @AndreyAdamenco,

I updated [rtw88-dkms-git](https://aur.archlinux.org/packages/rtw88-dkms-git), please reinstall it again to see if it works on Manjaro.

briansune, 2024-12-27:
> wihotspot wants a wifi device which can work in station mode and AP mode at the same time. This is also known as concurrent mode. rtw88 in the kernel only supports concurrent mode with RTL8822C. I [...]

How to do it on other USB and SDIO chip?
In the previous driver like RTL8723DS vendor driver Larry hosted I can see there are two WLAN0 and WLAN1 when the MakeFile turned on concurrent mode but for rtw88 driver I am not sure how to do so.
Need  a bit guidance. Thank you.

dubhater, 2024-12-27:
You add a second interface using iw. I never tried it, so I don't know the exact command.

briansune, 2024-12-28:
@dubhater 

Do the latest driver support concurrent mode like vendor RTL8723DS?
Then I will try to enable both STA & AP and test again the driver.

dubhater, 2024-12-28:
Yes, I enabled it for all chips in this repository, just for testing. In the kernel it's enabled only for RTL8822C.

briansune, 2024-12-28:
@dubhater 

I am not too familiar with the idea of "in the kernel" So if I just use make and compile the modules do it default enabled this feature? Where is the control flag to turn on/off this feature during build (MakeFile).

Thankyou

dubhater, 2024-12-28:
If you clone this repository and run make, you can test the feature with any supported device. It's enabled by default, there is no flag to turn it off.

briansune, 2024-12-28:
@dubhater 

Can it changed to a optional feature like previous vendor driver?
Btw If I read correctly on the above response of @morrownr this is just a software stuff to make use of the chip bandwidth.
I might be wrong on conceptually but this means it is allowed to make more than two WLAN0+1 ?

dubhater, 2024-12-28:
Maybe it could be made optional, but there is no need.

It's not only a software thing. It also depends on the firmware/hardware. RTL8822C, RTL8822B, RTL8821C, and RTL8814A can support five interfaces, theoretically. The others can support only two. But rtw88 allows only two, even if the chip can support five.

briansune, 2024-12-28:
@dubhater 

Okay then no flag to disable during build.

As for firmware, do this referring to EFUSE or? If firmware then it is about the xxx.bin during driver initial but do you have a source code on that to change?

For hardware then it is just about the MIMO hardware so there is no point to break the inherent design such as 2x2 etc.
I am not sure do people really need to use 5 interfaces on a small power device like the USB 1 STA and 1 AP should make good enough on most applications. Even ESP8/32 only uses 1 STA and 1 AP most cases.

dubhater, 2024-12-28:
Firmware is the rtw8xxxy_fw.bin files. They contain code which runs on the wifi chip's MCU. We don't have access to the source code.

The efuse is just a few bytes which contain TX power calibration data, the permanent MAC address, things like that.

briansune, 2024-12-29:
@dubhater 

Appreciate the clear explanations. =]

So the community driver on EFUSE recover phase might had hard time to resolve.
As mentioned there is blank chip in the market (not module pure chip) so people might experience trouble on wrong initialization.
i.e. RTL8812AU USB dongle I bought had EFUSE issue like the thermal setting not sure it is wrong loading or it is missing.