lwfinger/rtw88 #265: Kernel crash dump loading module rtw_8812au.

bac0n, 2024-12-12:
Last good git commit was 611874c62b1cf5ff487885284c369080857e8f9e.

Hardware:
Bus 001 Device 006: ID 0b05:17d2 ASUSTek Computer, Inc. USB-AC56 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU]
Negotiated speed: High Speed (480Mbps)

dmesg:
[Wed Dec 11 19:33:07 2024] BUG: kernel NULL pointer dereference, address: 0000000000000008
[Wed Dec 11 19:33:07 2024] #PF: supervisor read access in kernel mode
[Wed Dec 11 19:33:07 2024] #PF: error_code(0x0000) - not-present page
[Wed Dec 11 19:33:07 2024] PGD 0 P4D 0 
[Wed Dec 11 19:33:07 2024] Oops: Oops: 0000 [#1] PREEMPT SMP
[Wed Dec 11 19:33:07 2024] CPU: 4 UID: 0 PID: 419 Comm: (udev-worker) Tainted: G           OE      6.12.4-2-amd64-betazoid #1
[Wed Dec 11 19:33:07 2024] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[Wed Dec 11 19:33:07 2024] Hardware name: Gigabyte Technology Co., Ltd. GA-MA790X-UD4/GA-MA790X-UD4, BIOS F7d 07/14/2010
[Wed Dec 11 19:33:07 2024] RIP: 0010:rtw_usb_phy_cfg+0x1b/0xe0 [rtw_usb]
[Wed Dec 11 19:33:07 2024] Code: 0f 1e fa 0f 1f 44 00 00 c3 66 0f 1f 44 00 00 0f 1f 44 00 00 41 56 83 fe 05 41 55 41 54 55 53 48 8b 47 38 48 8b 80 98 00 00 00 <48> 8b 58 08 48 0f 45 18 48 85 db 74 1c 0f b6 4f 48 0f b7 2b 41 be
[Wed Dec 11 19:33:07 2024] RSP: 0018:ffffb77fc050fa80 EFLAGS: 00010293
[Wed Dec 11 19:33:07 2024] RAX: 0000000000000000 RBX: 0000000000000000 RCX: ffff983406915d27
[Wed Dec 11 19:33:07 2024] RDX: 0000000000000018 RSI: 0000000000000003 RDI: ffff983406912000
[Wed Dec 11 19:33:07 2024] RBP: ffff983406910900 R08: ffff983406912336 R09: 0000000000000003
[Wed Dec 11 19:33:07 2024] R10: ffff983406915d27 R11: ffff983406912d1a R12: ffff9834048c7000
[Wed Dec 11 19:33:07 2024] R13: ffff9834069179a8 R14: ffff9834048c7050 R15: ffff983406912000
[Wed Dec 11 19:33:07 2024] FS:  00007f15fc29d8c0(0000) GS:ffff98371fd00000(0000) knlGS:0000000000000000
[Wed Dec 11 19:33:07 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Wed Dec 11 19:33:07 2024] CR2: 0000000000000008 CR3: 0000000103ce8000 CR4: 00000000000006f0
[Wed Dec 11 19:33:07 2024] Call Trace:
[Wed Dec 11 19:33:07 2024]  <TASK>
[Wed Dec 11 19:33:07 2024]  ? __die_body.cold+0x19/0x2b
[Wed Dec 11 19:33:07 2024]  ? page_fault_oops+0x156/0x2d0
[Wed Dec 11 19:33:07 2024]  ? search_module_extables+0x19/0x50
[Wed Dec 11 19:33:07 2024]  ? search_bpf_extables+0x5f/0x80
[Wed Dec 11 19:33:07 2024]  ? exc_page_fault+0x7e/0x180
[Wed Dec 11 19:33:07 2024]  ? asm_exc_page_fault+0x26/0x30
[Wed Dec 11 19:33:07 2024]  ? rtw_usb_phy_cfg+0x1b/0xe0 [rtw_usb]
[Wed Dec 11 19:33:07 2024]  rtw_usb_probe+0x4b8/0x94f [rtw_usb]
[Wed Dec 11 19:33:07 2024]  usb_probe_interface+0xe3/0x2a0 [usbcore]
[Wed Dec 11 19:33:07 2024]  really_probe+0xd5/0x330
[Wed Dec 11 19:33:07 2024]  ? pm_runtime_barrier+0x54/0x90
[Wed Dec 11 19:33:07 2024]  ? __device_attach_driver+0x100/0x100
[Wed Dec 11 19:33:07 2024]  __driver_probe_device+0x78/0x110
[Wed Dec 11 19:33:07 2024]  driver_probe_device+0x1f/0xa0
[Wed Dec 11 19:33:07 2024]  __driver_attach+0xaf/0x1b0
[Wed Dec 11 19:33:07 2024]  bus_for_each_dev+0x88/0xe0
[Wed Dec 11 19:33:07 2024]  bus_add_driver+0x10e/0x1f0
[Wed Dec 11 19:33:07 2024]  driver_register+0x72/0xc0
[Wed Dec 11 19:33:07 2024]  usb_register_driver+0x8c/0x130 [usbcore]
[Wed Dec 11 19:33:07 2024]  ? 0xffffffffc1fb1000
[Wed Dec 11 19:33:07 2024]  do_one_initcall+0x58/0x2f0
[Wed Dec 11 19:33:07 2024]  do_init_module+0x60/0x210
[Wed Dec 11 19:33:07 2024]  __do_sys_init_module+0x176/0x1b0
[Wed Dec 11 19:33:07 2024]  do_syscall_64+0x82/0x190
[Wed Dec 11 19:33:07 2024]  ? do_syscall_64+0x8e/0x190
[Wed Dec 11 19:33:07 2024]  ? __count_memcg_events+0x4b/0xe0
[Wed Dec 11 19:33:07 2024]  ? count_memcg_events.constprop.0+0x1a/0x30
[Wed Dec 11 19:33:07 2024]  ? handle_mm_fault+0x1b7/0x2c0
[Wed Dec 11 19:33:07 2024]  ? do_user_addr_fault+0x368/0x620
[Wed Dec 11 19:33:07 2024]  ? exc_page_fault+0x7e/0x180
[Wed Dec 11 19:33:07 2024]  entry_SYSCALL_64_after_hwframe+0x4b/0x53
[Wed Dec 11 19:33:07 2024] RIP: 0033:0x7f15fc4557ca
[Wed Dec 11 19:33:07 2024] Code: 48 8b 0d 59 56 0d 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 26 56 0d 00 f7 d8 64 89 01 48
[Wed Dec 11 19:33:07 2024] RSP: 002b:00007fff82ccbba8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[Wed Dec 11 19:33:07 2024] RAX: ffffffffffffffda RBX: 0000564657a2dc00 RCX: 00007f15fc4557ca
[Wed Dec 11 19:33:07 2024] RDX: 00007f15fc295541 RSI: 000000000000310d RDI: 0000564657c10200
[Wed Dec 11 19:33:07 2024] RBP: 0000564657c10200 R08: 00000000000153e0 R09: 0000000000000000
[Wed Dec 11 19:33:07 2024] R10: 00007f15fc52bb20 R11: 0000000000000246 R12: 00007f15fc295541
[Wed Dec 11 19:33:07 2024] R13: 0000000000020000 R14: 0000564657a39e30 R15: 0000000000000000
[Wed Dec 11 19:33:07 2024]  </TASK>
[Wed Dec 11 19:33:07 2024] Modules linked in: uhid cmac algif_hash algif_skcipher af_alg bnep rtw_8812au(OE+) rtw_8812a(OE) rtw_88xxa(OE) rtw_usb(OE) rtw_core(OE) mac80211 btusb libarc4 binfmt_misc btrtl btintel cfg80211 btbcm btmtk bluetooth rfkill wmi_bmof amdgpu kvm_amd ccp joydev snd_hda_codec_hdmi kvm evdev i2c_algo_bit serio_raw drm_ttm_helper snd_hda_intel snd_intel_dspcfg ttm acpi_cpufreq snd_hda_codec video snd_hwdep snd_rme96 drm_exec drm_suballoc_helper amdxcp snd_hda_core drm_buddy dm_multipath snd_pcm snd_timer dm_mod gpu_sched drm_display_helper snd drm_kms_helper soundcore wmi button loop it87_wdt watchdog it87 hwmon_vid i2c_dev hid_playstation led_class_multicolor ff_memless sg drivetemp parport_pc ppdev lp drm parport efi_pstore configfs nfnetlink ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 raid10 raid456 libcrc32c crc32c_generic async_raid6_recov async_memcpy async_pq async_xor xor async_tx raid6_pq raid1 raid0 md_mod hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid sd_mod r8169 ohci_pci ata_generic
[Wed Dec 11 19:33:07 2024]  realtek ehci_pci ohci_hcd mdio_devres pata_atiixp ehci_hcd ahci i2c_piix4 libahci i2c_smbus libphy libata scsi_mod usbcore scsi_common usb_common
[Wed Dec 11 19:33:07 2024] CR2: 0000000000000008
[Wed Dec 11 19:33:07 2024] ---[ end trace 0000000000000000 ]---
[Wed Dec 11 19:33:07 2024] RIP: 0010:rtw_usb_phy_cfg+0x1b/0xe0 [rtw_usb]
[Wed Dec 11 19:33:07 2024] Code: 0f 1e fa 0f 1f 44 00 00 c3 66 0f 1f 44 00 00 0f 1f 44 00 00 41 56 83 fe 05 41 55 41 54 55 53 48 8b 47 38 48 8b 80 98 00 00 00 <48> 8b 58 08 48 0f 45 18 48 85 db 74 1c 0f b6 4f 48 0f b7 2b 41 be
[Wed Dec 11 19:33:07 2024] RSP: 0018:ffffb77fc050fa80 EFLAGS: 00010293
[Wed Dec 11 19:33:07 2024] RAX: 0000000000000000 RBX: 0000000000000000 RCX: ffff983406915d27
[Wed Dec 11 19:33:07 2024] RDX: 0000000000000018 RSI: 0000000000000003 RDI: ffff983406912000
[Wed Dec 11 19:33:07 2024] RBP: ffff983406910900 R08: ffff983406912336 R09: 0000000000000003
[Wed Dec 11 19:33:07 2024] R10: ffff983406915d27 R11: ffff983406912d1a R12: ffff9834048c7000
[Wed Dec 11 19:33:07 2024] R13: ffff9834069179a8 R14: ffff9834048c7050 R15: ffff983406912000
[Wed Dec 11 19:33:07 2024] FS:  00007f15fc29d8c0(0000) GS:ffff98371fd00000(0000) knlGS:0000000000000000
[Wed Dec 11 19:33:07 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Wed Dec 11 19:33:07 2024] CR2: 0000000000000008 CR3: 0000000103ce8000 CR4: 00000000000006f0

a5a5aa555oo, 2024-12-12:
@bac0n 

What distro do you use?
It seem only 8812AU users are affected, I cannot reproduce this issue with 8822BU on my Arch Linux (kernel: 6.12.4-arch1-1).

bac0n, 2024-12-12:
$ uname -a
Linux betazoid 6.12.4-2-amd64-betazoid #1 SMP PREEMPT_DYNAMIC Tue Dec 10 18:05:57 CET 2024 x86_64 GNU/Linux # (vanilla kernel.org)

$ cat /etc/debian_version
trixie/sid

dubhater, 2024-12-12:
Ah, sorry about that. It should be fixed now.

closed 2024-12-12 by dubhater

bac0n, 2024-12-12:
yep, fixed, thanks :)=