lwfinger/rtw88 #268: RTL8822BE random bluetooth drops

pschonmann, 2024-12-19:
Hi the bluetooth on the card in subject randomly drops. Nothing more in kernel than. Even when restarted bluetooth device is not displayed under hciconfig

I was trying to resolve my issue before i installed this driver on linux mint forum, but still drops bluetooth ocassionally
https://forums.linuxmint.com/viewtopic.php?t=430407 

Here is actual dmesg
```
[Thu Dec 19 14:35:05 2024] Bluetooth: hci0: command tx timeout
[Thu Dec 19 14:35:07 2024] Bluetooth: hci0: command tx timeout
[Thu Dec 19 14:35:09 2024] Bluetooth: hci0: command tx timeout
[Thu Dec 19 14:35:20 2024] Bluetooth: hci0: command tx timeout
[Thu Dec 19 14:35:22 2024] Bluetooth: hci0: command tx timeout
[Thu Dec 19 14:35:22 2024] Bluetooth: hci0: Resetting usb device.
[Thu Dec 19 14:35:22 2024] usb 3-1: reset full-speed USB device number 2 using xhci_hcd
[Thu Dec 19 14:35:22 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:22 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:22 2024] usb 3-1: reset full-speed USB device number 2 using xhci_hcd
[Thu Dec 19 14:35:22 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:23 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:23 2024] usb 3-1: reset full-speed USB device number 2 using xhci_hcd
[Thu Dec 19 14:35:23 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:23 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:23 2024] usb 3-1: device not accepting address 2, error -71
[Thu Dec 19 14:35:23 2024] usb 3-1: WARN: invalid context state for evaluate context command.
[Thu Dec 19 14:35:23 2024] usb 3-1: reset full-speed USB device number 2 using xhci_hcd
[Thu Dec 19 14:35:23 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:24 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:24 2024] usb 3-1: device not accepting address 2, error -71
[Thu Dec 19 14:35:24 2024] usb 3-1: USB disconnect, device number 2
[Thu Dec 19 14:35:24 2024] usb 3-1: new full-speed USB device number 4 using xhci_hcd
[Thu Dec 19 14:35:24 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:24 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:25 2024] usb 3-1: new full-speed USB device number 5 using xhci_hcd
[Thu Dec 19 14:35:25 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:25 2024] usb 3-1: device descriptor read/64, error -71
[Thu Dec 19 14:35:25 2024] usb usb3-port1: attempt power cycle
[Thu Dec 19 14:35:25 2024] usb 3-1: new full-speed USB device number 6 using xhci_hcd
[Thu Dec 19 14:35:25 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:26 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:26 2024] usb 3-1: device not accepting address 6, error -71
[Thu Dec 19 14:35:26 2024] usb 3-1: WARN: invalid context state for evaluate context command.
[Thu Dec 19 14:35:26 2024] usb 3-1: new full-speed USB device number 7 using xhci_hcd
[Thu Dec 19 14:35:26 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:26 2024] usb 3-1: Device not responding to setup address.
[Thu Dec 19 14:35:26 2024] usb 3-1: device not accepting address 7, error -71
[Thu Dec 19 14:35:26 2024] usb 3-1: WARN: invalid context state for evaluate context command.
[Thu Dec 19 14:35:26 2024] usb usb3-port1: unable to enumerate USB device
pschonmann@E495:~$ hciconfig 
pschonmann@E495:~$ systemctl restart bluetooth
pschonmann@E495:~$ hciconfig 
pschonmann@E495:~$ 
```

dubhater, 2024-12-19:
Has it ever worked? Does it work in Windows?

dubhater, 2024-12-19:
Does it work if you blacklist rtw_8822be and rtw88_8822be?