lwfinger/rtw88 #270: rtw_8814au report

morrownr, 2024-12-20:
Overall: Looking good.

```
$ sudo dmesg | grep 8814
[sudo] password for morrow:         
[    6.186835] rtw_8814au 6-4.3:1.0: Firmware version 33.6.0, H2C version 6
[    6.446440] usbcore: registered new interface driver rtw_8814au
[    6.469473] rtw_8814au 6-4.3:1.0 wlxbcec23c35225: renamed from wlan0
[   12.604171] rtw_8814au 6-4.3:1.0: failed to get tx report from firmware

```

The only thing I see is that last line about the tx report. 

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.217 port 57342 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  48.9 MBytes   410 Mbits/sec    0   1.27 MBytes       
[  5]   1.00-2.00   sec  46.2 MBytes   388 Mbits/sec    0   1.27 MBytes       
[  5]   2.00-3.00   sec  47.5 MBytes   398 Mbits/sec    0   1.27 MBytes       
[  5]   3.00-4.00   sec  46.2 MBytes   388 Mbits/sec    0   1.27 MBytes       
[  5]   4.00-5.00   sec  46.2 MBytes   388 Mbits/sec    0   1.27 MBytes       
[  5]   5.00-6.00   sec  46.2 MBytes   388 Mbits/sec    0   1.27 MBytes       
[  5]   6.00-7.00   sec  47.5 MBytes   398 Mbits/sec    0   1.27 MBytes       
[  5]   7.00-8.00   sec  47.5 MBytes   398 Mbits/sec    0   1.27 MBytes       
[  5]   8.00-9.00   sec  47.5 MBytes   398 Mbits/sec    0   1.27 MBytes       
[  5]   9.00-10.00  sec  46.2 MBytes   388 Mbits/sec    0   1.27 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   470 MBytes   394 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   468 MBytes   392 Mbits/sec                  receiver

```

That is not bad. It is not to full potential but I'll need to get my backup AC2600 wifi router out to test again and I'll test with the vendor driver as well.

```
$ lsusb -t
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 3: Dev 4, If 0, Class=Vendor Specific Class, Driver=rtw_8814au, 5000M

```

USB3 appears to be working.

Is there anything specific that you want me to test?

dubhater, 2024-12-21:
I can't think of anything specific right now.

D357R0Y3R, 2024-12-22:
I am getting sometimes minimum 780 MBPS and highest 866.7 MBPS from my AC1200 router where my distance is very close to the router on 5GHz channel.

morrownr, 2024-12-22:
@D357R0Y3R 

> I am getting sometimes minimum 780 MBPS and highest 866.7 MBPS from my AC1200 router...

What were you expecting?

morrownr, 2024-12-22:
@dubhater 

I need to do more testing but it appears that the vendor driver is using 3 Spatial Streams but the driver here is only using 2. I do understand that the wireless router has to be capable of 3 Spatial Streams so that is why I moved to my backup router for this testing. It is an AC2600 class router and, as you can see below, is using 3 Spatial Streams with the vendor driver. On my previous router, I could get about 400 Mbps which is about what we should expect with a router that is capable of only 2 Spatial Streams.

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.217 port 51546 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  77.6 MBytes   651 Mbits/sec    0   2.15 MBytes       
[  5]   1.00-2.00   sec  75.0 MBytes   629 Mbits/sec    0   2.15 MBytes       
[  5]   2.00-3.00   sec  76.2 MBytes   640 Mbits/sec    0   2.15 MBytes       
[  5]   3.00-4.00   sec  77.5 MBytes   650 Mbits/sec    0   2.15 MBytes       
[  5]   4.00-5.00   sec  75.0 MBytes   629 Mbits/sec    0   2.15 MBytes       
[  5]   5.00-6.00   sec  76.2 MBytes   640 Mbits/sec    0   2.15 MBytes       
[  5]   6.00-7.00   sec  76.2 MBytes   640 Mbits/sec    0   2.15 MBytes
[  5]   7.00-8.00   sec  76.2 MBytes   640 Mbits/sec    0   2.15 MBytes       
[  5]   8.00-9.00   sec  76.2 MBytes   640 Mbits/sec    0   2.15 MBytes       
[  5]   9.00-10.00  sec  76.2 MBytes   640 Mbits/sec    0   2.15 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   763 MBytes   640 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   760 MBytes   638 Mbits/sec                  receiver

```

I need to reconfigure back to this driver and continue testing but earlier tests showed this driver appearing to only use 2 Spatial Streams. Can you clarify if this driver should be capable of 3 Spatial Streams at this point?

D357R0Y3R, 2024-12-22:
> @D357R0Y3R
> 
> > I am getting sometimes minimum 780 MBPS and highest 866.7 MBPS from my AC1200 router...
> 
> What were you expecting?

I didn't tested with iperf3. I just noticed the speed from my router interface. Seems like you are have configured the `iperf3 -s` on your OpenWRT router?

D357R0Y3R, 2024-12-22:
Hey, @morrownr!

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapter_out-of-kernel_drivers_for_Linux.md#chipset---rtl8814au---ac1900---usb-3

For an example there are so many not supported options from the older driver, Did you managed to try to get them all working with this new driver?

dubhater, 2024-12-23:
@morrownr It can use 3 spatial streams in USB 3 mode. It can use only 2 spatial streams in USB 2 mode.

morrownr, 2024-12-23:
@D357R0Y3R 

> Seems like you are have configured the iperf3 -s on your OpenWRT router?

That is correct. Actually I use 3 different WiFi routers but only 1 at a time plus a RasPi4B as an AP. This gives me the flexibility to test many things.

> Did you managed to try to get them all working with this new driver?

We'll find out. The first thing to understand is @dubhater is the primary coder while I am the primary tester. Once in a while he throws a coding job my way. It is much much faster this way.  He has a black belt in wireless programming, I do not.

Based on my testing over the last year on the various drivers in this repo, I can say that many of the missing features have been added.

I think monitor mode is in much better shape but we need testers and reports on that. I have only done basic monitor mode testing. Virtual interface support is not where we want it. Realtek seems to have this thing about Concurrent mode. There is no Active Monitor Mode but most people that really need that have probably got a Mediatek adapter by now.

morrownr, 2024-12-23:
@dubhater 

> It can use 3 spatial streams in USB 3 mode. It can use only 2 spatial streams in USB 2 mode.

Yes. I have been checking to see if I am in USB3 mode. 

I am now seeing proper operation in managed mode with 3 spatial streams...

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.217 port 59058 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  73.6 MBytes   618 Mbits/sec    0   2.18 MBytes       
[  5]   1.00-2.00   sec  71.2 MBytes   598 Mbits/sec    0   2.18 MBytes       
[  5]   2.00-3.00   sec  72.5 MBytes   608 Mbits/sec    0   2.18 MBytes       
[  5]   3.00-4.00   sec  72.5 MBytes   608 Mbits/sec    0   2.18 MBytes       
[  5]   4.00-5.00   sec  72.5 MBytes   608 Mbits/sec    0   2.18 MBytes       
[  5]   5.00-6.00   sec  72.5 MBytes   608 Mbits/sec    0   2.18 MBytes       
[  5]   6.00-7.00   sec  71.2 MBytes   598 Mbits/sec    0   2.18 MBytes       
[  5]   7.00-8.00   sec  72.5 MBytes   608 Mbits/sec    0   2.18 MBytes       
[  5]   8.00-9.00   sec  72.5 MBytes   608 Mbits/sec    0   2.18 MBytes       
[  5]   9.00-10.00  sec  71.2 MBytes   598 Mbits/sec    0   2.18 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   722 MBytes   606 Mbits/sec    0             sender
[  5]   0.00-10.03  sec   721 MBytes   603 Mbits/sec                  receiver

```

Basically what we should see is around a max of 600 Mbps with 3 spatial streams and a max of around 400 Mbps with 2 spatial streams. In earlier tests I could not see 3  spatial streams working as it should but some of my equipment is well worn so I stayed at it. We need more tests from more people to see if operation is consistent.

Also, I am seeing automatic USB3 mode work correctly every time. I do need to test USB2 mode. Speaking of USB2 mode, I noticed we have not documented the new parameter to limit to USB2 mode yet. Throw me some information about it and I'll see about documenting it.

dubhater, 2024-12-23:
```
parm:           switch_usb_mode:Set to N to disable switching to USB 3 mode to avoid potential interference in the 2.4 GHz band (default: Y) (bool)
```

It currently has an effect for RTL8822CU, RTL8812CU, RTL8822BU, RTL8812BU, RTL8812AU, and RTL8814AU.

It's okay to leave the parameter on even when plugging the device into a USB 2 port because it will try to switch to USB 3 mode only once.

morrownr, 2024-12-24:
Thanks. I added the information to the existing section but I don't like that placement. I think more changes need to be made to make it as clear as possible. I ponder the issue an come up with something.

s1shed, 2024-12-30:
Since I was testing another driver for IBSS I thought I'd try this one too...  My Alfa AWUS1900 was able to join an encrypted IBSS network created with `wpa_supplicant`, but there was an error in the kernel logs.

Kali Linux
Kernel: 6.11.2-amd64
Alfa AWUS1900 in USB3 mode (if it matters)

`adhoc.conf`:

```conf
#  wpa_supplicant -B -i wlan1 -c adhoc.conf

ctrl_interface=/run/wpa_supplicant GROUP=root

network={
        ssid="testnet"
        mode=1
        frequency=2432
        proto=RSN
        key_mgmt=WPA-PSK
        pairwise=CCMP
        group=CCMP
        psk="testnet1234"
}
```

```
Dec 30 18:00:45 kali kernel: wlan1: No active IBSS STAs - trying to scan for other IBSS networks with same SSID (merge)
Dec 30 18:01:16 kali kernel: wlan1: No active IBSS STAs - trying to scan for other IBSS networks with same SSID (merge)
Dec 30 18:01:18 kali kernel: ------------[ cut here ]------------
Dec 30 18:01:18 kali kernel: WARNING: CPU: 1 PID: 82735 at net/mac80211/ibss.c:1062 ieee80211_ibss_rx_queued_mgmt+0xa98/0xc00 [mac80211]
Dec 30 18:01:18 kali kernel: Modules linked in: rtw_8814au(OE) rtw_8814a(OE) rtw_usb(OE) rtw_core(OE) 8852bu(OE) ccm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device overlay qrtr sunrpc btusb btrtl binfmt_misc btintel btbcm btmtk uvcvideo bluetooth videobuf2_vmalloc uvc videobuf2_memops videobuf2_v4l2 cdc_mbim cdc_wdm cdc_acm videodev videobuf2_common mc snd_sof_pci_intel_skl snd_sof_intel_hda_generic soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda_common snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils soundwire_bus nls_ascii nls_cp437 snd_soc_avs vfat fat intel_rapl_msr intel_rapl_common snd_soc_hda_codec intel_uncore_frequency intel_uncore_frequency_common snd_soc_skl intel_pmc_core_pltdrv intel_pmc_core ext4 iwlmvm snd_soc_hdac_hda intel_vsec crc16 mbcache snd_hda_ext_core pmt_telemetry jbd2 mac80211 snd_soc_sst_ipc pmt_class snd_hda_codec_hdmi snd_soc_sst_dsp x86_pkg_temp_thermal intel_powerclamp snd_soc_acpi_intel_match coretemp libarc4 snd_soc_acpi
Dec 30 18:01:18 kali kernel:  snd_hda_codec_realtek kvm_intel snd_hda_codec_generic snd_soc_core iwlwifi snd_hda_scodec_component snd_compress kvm snd_pcm_dmaengine tps6598x snd_hda_intel mei_pxp mei_hdcp typec snd_intel_dspcfg snd_intel_sdw_acpi rapl snd_ctl_led snd_hda_codec intel_cstate cfg80211 intel_uncore snd_hda_core thinkpad_acpi iTCO_wdt snd_hwdep intel_pmc_bxt think_lmi snd_pcm iTCO_vendor_support firmware_attributes_class pcspkr wmi_bmof intel_wmi_thunderbolt nvram mei_me sparse_keymap ee1004 watchdog snd_timer mei platform_profile intel_xhci_usb_role_switch snd intel_pch_thermal roles serial_multi_instantiate soundcore rfkill ac acpi_pad joydev evdev serio_raw vmwgfx drm_ttm_helper msr efi_pstore configfs nfnetlink efivarfs ip_tables x_tables autofs4 btrfs blake2b_generic xor raid6_pq libcrc32c crc32c_generic dm_crypt dm_mod hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid cdc_ncm cdc_ether usbnet mii i915 drm_buddy i2c_algo_bit crct10dif_pclmul drm_display_helper crc32_pclmul crc32c_intel xhci_pci cec xhci_hcd
Dec 30 18:01:18 kali kernel:  rc_core ghash_clmulni_intel ttm sha512_ssse3 drm_kms_helper rtsx_pci_sdmmc intel_lpss_pci usbcore nvme mmc_core i2c_i801 sha256_ssse3 psmouse sha1_ssse3 drm rtsx_pci nvme_core e1000e i2c_smbus intel_lpss usb_common idma64 video battery wmi button aesni_intel gf128mul crypto_simd cryptd
Dec 30 18:01:18 kali kernel: CPU: 1 UID: 0 PID: 82735 Comm: kworker/u16:2 Tainted: G        W  OE      6.11.2-amd64 #1  Kali 6.11.2-1kali1
Dec 30 18:01:18 kali kernel: Tainted: [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Dec 30 18:01:18 kali kernel: Hardware name: LENOVO 20HN0012MX/20HN0012MX, BIOS R0IET72W (1.50 ) 11/08/2023
Dec 30 18:01:18 kali kernel: Workqueue: events_unbound cfg80211_wiphy_work [cfg80211]
Dec 30 18:01:18 kali kernel: RIP: 0010:ieee80211_ibss_rx_queued_mgmt+0xa98/0xc00 [mac80211]
Dec 30 18:01:18 kali kernel: Code: 88 85 30 ff ff ff e8 c7 2f a7 ff 44 0f b6 85 30 ff ff ff 4c 8b 8d 28 ff ff ff 48 85 c0 4c 8b 9d 18 ff ff ff 0f 85 4a fb ff ff <0f> 0b e9 43 fb ff ff 0f 0b e9 cc fb ff ff 41 c6 81 92 0a 00 00 01
Dec 30 18:01:18 kali kernel: RSP: 0018:ffffc1d90df0fce8 EFLAGS: 00010246
Dec 30 18:01:18 kali kernel: RAX: 0000000000000000 RBX: ffffa0b98cfaca00 RCX: 0000000000000001
Dec 30 18:01:18 kali kernel: RDX: 0000000000000000 RSI: ffffc1d90df0fd60 RDI: ffffa0b98cfad470
Dec 30 18:01:18 kali kernel: RBP: ffffc1d90df0fdf0 R08: 0000000000000000 R09: ffffa0b957e68000
Dec 30 18:01:18 kali kernel: R10: 0000000000000000 R11: ffffffffc2a93740 R12: ffffa0ba0204e900
Dec 30 18:01:18 kali kernel: R13: ffffa0ba46549800 R14: ffffa0b95386cf38 R15: 00000000000000a4
Dec 30 18:01:18 kali kernel: FS:  0000000000000000(0000) GS:ffffa0ba6a680000(0000) knlGS:0000000000000000
Dec 30 18:01:18 kali kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Dec 30 18:01:18 kali kernel: CR2: 00005560783de588 CR3: 000000010fa22005 CR4: 00000000003706f0
Dec 30 18:01:18 kali kernel: Call Trace:
Dec 30 18:01:18 kali kernel:  <TASK>
Dec 30 18:01:18 kali kernel:  ? ieee80211_ibss_rx_queued_mgmt+0xa98/0xc00 [mac80211]
Dec 30 18:01:18 kali kernel:  ? __warn.cold+0x8e/0xe8
Dec 30 18:01:18 kali kernel:  ? ieee80211_ibss_rx_queued_mgmt+0xa98/0xc00 [mac80211]
Dec 30 18:01:18 kali kernel:  ? report_bug+0xff/0x140
Dec 30 18:01:18 kali kernel:  ? handle_bug+0x3c/0x80
Dec 30 18:01:18 kali kernel:  ? exc_invalid_op+0x17/0x70
[... cut ...]
Dec 30 18:01:18 kali kernel:  ? __pfx_worker_thread+0x10/0x10
Dec 30 18:01:18 kali kernel:  kthread+0xcf/0x100
Dec 30 18:01:18 kali kernel:  ? __pfx_kthread+0x10/0x10
Dec 30 18:01:18 kali kernel:  ret_from_fork+0x31/0x50
Dec 30 18:01:18 kali kernel:  ? __pfx_kthread+0x10/0x10
Dec 30 18:01:18 kali kernel:  ret_from_fork_asm+0x1a/0x30
Dec 30 18:01:18 kali kernel:  </TASK>
Dec 30 18:01:18 kali kernel: ---[ end trace 0000000000000000 ]---
Dec 30 18:01:19 kali kernel: rtw_8814au 2-6:1.0: failed to get tx report from firmware

```

morrownr, 2025-01-12:
@s1shed 

Let me suggest that you post the IBSS issue in a new separate issue so that others looking at IBSS can see it and add their information.

clseibold, 2025-02-13:
I had to switch back to @morrownr 's older rtl8814au driver because rtw88 halved the speed of my connection, and was very inconsistent even on the local network (ssh would be sluggish most of the time, with brief spurts where it was faster).

I get ~220 Mbps download with the older driver, and only ~118-150 Mbps download with the new driver.

dubhater, 2025-02-13:
@clseibold More information would be good. Let's start with the output of ``iw dev`` and then ``iw dev <interface> station dump``.

clseibold, 2025-02-14:
```bash
clseibold@auragem:~$ iw dev
phy#0
        Interface wlx0013ef4f02a1
                ifindex 3
                wdev 0x1
                addr 00:13:ef:4f:02:a1
                ssid myers
                type managed
                txpower 20.00 dBm

clseibold@auragem:~$ iw dev wlx0013ef4f02a1 station dump
clseibold@auragem:~$
```

The second command gave no output. Btw, these are being run while using the older rtl8814au driver.

morrownr, 2025-02-14:
@clseibold 

> Btw, these are being run while using the older rtl8814au driver.

Dubhater is asking regarding the new driver, not the old one.

clseibold, 2025-02-14:
@morrownr Ok, thanks. I will reinstall the new driver and post the output to those commands here.

clseibold, 2025-02-14:
@dubhater Ok, here's the commands' outputs using the new driver:
```
clseibold@auragem:~$ iw dev
phy#0
	Interface wlx0013ef4f02a1
		ifindex 3
		wdev 0x1
		addr 00:13:ef:4f:02:a1
		ssid myers
		type managed
		channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

and for the second command:
```
clseibold@auragem:~$ iw dev wlx0013ef4f02a1 station dump
Station dc:69:b5:fa:fb:47 (on wlx0013ef4f02a1)
	inactive time:	2 ms
	rx bytes:	1500156
	rx packets:	3681
	tx bytes:	970347
	tx packets:	1630
	tx retries:	0
	tx failed:	0
	beacon loss:	0
	beacon rx:	1073
	rx drop misc:	3
	signal:  	-62 [-68, -62, -58, -54] dBm
	signal avg:	-61 [-68, -60, -59, -54] dBm
	beacon signal avg:	-60 dBm
	tx bitrate:	585.0 MBit/s VHT-MCS 7 80MHz VHT-NSS 2
	tx duration:	0 us
	rx bitrate:	702.0 MBit/s VHT-MCS 8 80MHz VHT-NSS 2
	rx duration:	0 us
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	187 seconds
	associated at [boottime]:	34.084s
	associated at:	1739519922872 ms
	current time:	1739520108854 ms
```

dubhater, 2025-02-15:
@clseibold If you delete this line from main.c and recompile, does it get better?
```
	ieee80211_hw_set(hw, SUPPORTS_AMSDU_IN_AMPDU);
```

clseibold, 2025-02-15:
@dubhater It might have helped slightly. `speedtest` went up to ~150-173 Mbit/s now (it went up to 239 at one point, for one test). I still get a sluggish ssh connection over local network, however.

dubhater, 2025-02-15:
How is the ping on the local network?

clseibold, 2025-02-15:
Here's the result of the ping command from my laptop to the server with the rtw8814au driver:
```
clseibold@fedora:~$ ping 192.168.4.26
PING 192.168.4.26 (192.168.4.26) 56(84) bytes of data.
64 bytes from 192.168.4.26: icmp_seq=1 ttl=64 time=256 ms
64 bytes from 192.168.4.26: icmp_seq=2 ttl=64 time=279 ms
64 bytes from 192.168.4.26: icmp_seq=3 ttl=64 time=6.42 ms
64 bytes from 192.168.4.26: icmp_seq=4 ttl=64 time=16.5 ms
64 bytes from 192.168.4.26: icmp_seq=5 ttl=64 time=41.0 ms
64 bytes from 192.168.4.26: icmp_seq=6 ttl=64 time=41.5 ms
64 bytes from 192.168.4.26: icmp_seq=7 ttl=64 time=83.5 ms
64 bytes from 192.168.4.26: icmp_seq=8 ttl=64 time=207 ms
64 bytes from 192.168.4.26: icmp_seq=9 ttl=64 time=231 ms
64 bytes from 192.168.4.26: icmp_seq=10 ttl=64 time=254 ms
64 bytes from 192.168.4.26: icmp_seq=11 ttl=64 time=275 ms
64 bytes from 192.168.4.26: icmp_seq=12 ttl=64 time=3.59 ms
64 bytes from 192.168.4.26: icmp_seq=13 ttl=64 time=4.74 ms
64 bytes from 192.168.4.26: icmp_seq=14 ttl=64 time=26.5 ms
64 bytes from 192.168.4.26: icmp_seq=15 ttl=64 time=47.4 ms
64 bytes from 192.168.4.26: icmp_seq=16 ttl=64 time=4.11 ms
64 bytes from 192.168.4.26: icmp_seq=17 ttl=64 time=102 ms
64 bytes from 192.168.4.26: icmp_seq=18 ttl=64 time=225 ms
64 bytes from 192.168.4.26: icmp_seq=19 ttl=64 time=25.9 ms
64 bytes from 192.168.4.26: icmp_seq=20 ttl=64 time=274 ms
64 bytes from 192.168.4.26: icmp_seq=21 ttl=64 time=4.70 ms
64 bytes from 192.168.4.26: icmp_seq=22 ttl=64 time=11.1 ms
64 bytes from 192.168.4.26: icmp_seq=23 ttl=64 time=33.8 ms
64 bytes from 192.168.4.26: icmp_seq=24 ttl=64 time=55.7 ms
64 bytes from 192.168.4.26: icmp_seq=25 ttl=64 time=90.5 ms
64 bytes from 192.168.4.26: icmp_seq=26 ttl=64 time=99.8 ms
64 bytes from 192.168.4.26: icmp_seq=27 ttl=64 time=223 ms
64 bytes from 192.168.4.26: icmp_seq=28 ttl=64 time=144 ms
64 bytes from 192.168.4.26: icmp_seq=29 ttl=64 time=33.1 ms
64 bytes from 192.168.4.26: icmp_seq=30 ttl=64 time=290 ms
^C
--- 192.168.4.26 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29048ms
rtt min/avg/max/mdev = 3.593/112.976/289.511/104.138 ms
```

morrownr, 2025-02-15:
That is a lot of jitter.

@dubhater 

Would it help if I set my 8814au adapter up again and do iperf3 and ping tests? I thought it was working well and went on to something else.

dubhater, 2025-02-15:
@clseibold What if you turn off the power saving? It won't help with the speed, but maybe ssh will work better.

```
iw wlx0013ef4f02a1 set power_save off
```

@morrownr I don't know. It works well for me too.

clseibold, 2025-02-15:
@dubhater I'm getting much better ping timings after doing that. SSH is working well now, but I'll continue to test it to see how it works over time. Thanks!
```
clseibold@fedora:~/Dev/auragem_sis$ ping 192.168.4.26
PING 192.168.4.26 (192.168.4.26) 56(84) bytes of data.
64 bytes from 192.168.4.26: icmp_seq=1 ttl=64 time=4.36 ms
64 bytes from 192.168.4.26: icmp_seq=2 ttl=64 time=3.64 ms
64 bytes from 192.168.4.26: icmp_seq=3 ttl=64 time=4.33 ms
64 bytes from 192.168.4.26: icmp_seq=4 ttl=64 time=6.68 ms
64 bytes from 192.168.4.26: icmp_seq=5 ttl=64 time=4.02 ms
64 bytes from 192.168.4.26: icmp_seq=6 ttl=64 time=4.09 ms
64 bytes from 192.168.4.26: icmp_seq=7 ttl=64 time=3.82 ms
64 bytes from 192.168.4.26: icmp_seq=8 ttl=64 time=2.90 ms
64 bytes from 192.168.4.26: icmp_seq=9 ttl=64 time=2.71 ms
64 bytes from 192.168.4.26: icmp_seq=10 ttl=64 time=2.84 ms
64 bytes from 192.168.4.26: icmp_seq=11 ttl=64 time=2.94 ms
64 bytes from 192.168.4.26: icmp_seq=12 ttl=64 time=3.99 ms
64 bytes from 192.168.4.26: icmp_seq=13 ttl=64 time=3.94 ms
64 bytes from 192.168.4.26: icmp_seq=14 ttl=64 time=5.12 ms
64 bytes from 192.168.4.26: icmp_seq=15 ttl=64 time=4.89 ms
64 bytes from 192.168.4.26: icmp_seq=16 ttl=64 time=10.2 ms
64 bytes from 192.168.4.26: icmp_seq=17 ttl=64 time=4.72 ms
64 bytes from 192.168.4.26: icmp_seq=18 ttl=64 time=3.18 ms
64 bytes from 192.168.4.26: icmp_seq=19 ttl=64 time=4.84 ms
^C
--- 192.168.4.26 ping statistics ---
19 packets transmitted, 19 received, 0% packet loss, time 18027ms
rtt min/avg/max/mdev = 2.712/4.378/10.203/1.664 ms
```

morrownr, 2025-02-15:
@clseibold 

If you are in managed (client) mode and you use Network Manager, here is a persistent solution:

Go to:  https://github.com/morrownr/USB-WiFi

Scroll down. There is a menu item called Frequently Asked Questions. Select it and look for no. 5.

-----

Thinking out loud: Turning off power_save is not really a fix but then power_save has been problematic for a long time. That would be a heck of a rabbit hole.

clseibold, 2025-02-15:
@morrownr Will do! Thanks again!

morrownr, 2025-02-15:
@clseibold 

I had some notes waiting to go into the FAQ and one was on No. 5 that I pointed you to above. I have now added to No. 5. I added another option for those that do not use NM but do use systemd.

jagger235711, 2025-02-23:
ok,let me share my log. by the way,thx for @morrownr s old 8814au driver, that is work well in my ubuntu with AMD cpu and gpu.
in the new driver, i have a question

- how to config it? in old driver, it can use a file to control ,like the light ,the mode of usb3 or 2. i cant find in readMe in new driver repo

```
➜  ~ iw dev
phy#0
	Interface wlx803f5d123bbc
		ifindex 3
		wdev 0x1
		addr 80:3f:5d:12:3b:bc
		ssid CWRT
		type managed
		channel 40 (5200 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00

```

```
➜  ~ iw dev wlx803f5d123bbc station dump
Station ee:e7:c2:90:90:3a (on wlx803f5d123bbc)
	inactive time:	946 ms
	rx bytes:	126265272
	rx packets:	128805
	tx bytes:	13804096
	tx packets:	52441
	tx retries:	0
	tx failed:	10
	beacon loss:	0
	beacon rx:	32776
	rx drop misc:	489
	signal:  	-64 [-64, -64, -62, -46] dBm
	signal avg:	-64 [-64, -64, -60, -46] dBm
	beacon signal avg:	-63 dBm
	tx bitrate:	351.0 MBit/s VHT-MCS 4 80MHz VHT-NSS 2
	tx duration:	0 us
	rx bitrate:	526.6 MBit/s VHT-MCS 6 80MHz VHT-NSS 2
	rx duration:	0 us
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		yes
	TDLS peer:	no
	DTIM period:	1
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	3409 seconds
	associated at [boottime]:	14.531s
	associated at:	1740303722936 ms
	current time:	1740307132042 ms

```

morrownr, 2025-02-23:
Hi @jagger235711 

> how to config it? in old driver, it can use a file to control ,like the light ,the mode of usb3 or 2. i cant find in readMe in new driver repo

There do need to be some additions to the README but for now, let me see if I can shed some light on the questions.

The new driver is what is called a mac80211 driver or sometimes referred to as modern Linux Wireless Standards Compliant. It was design from the beginning to be a driver that would be allowed into the Linux mainline kernel. In fact, patches for the new rtl8814au driver are currently going in and hopefully the new driver will be in kernel 6.15... or maybe 6.16.

These modern drivers are designed to be much more user friendly. messing around with a text file is not user friendly for many users, especially new users. Therefore, what you see with the modern drivers is a situation where they detect what USB mode is available and adjust accordingly. Example: The Mediatek mt7612u and mt7921u drivers have been doing this for a long time now. If they do not detect USB3 capability, then they leave USB in USB2 mode. If they detect USB3, then USB is switched into USB3 mode.

You can check the mode with this command:

$ lsusb -t

Find the line for your adapter and if you see 5000m it is in USB3 mode or if 480M then it is in USB2 mode.

The LED should work unless @dubhater hasn't added the capability. I don't think there is a way to turn it off. I'll throw my rtl8814au based adapter back in and check both of these issues. It seems to me that both were working as designed when I stopped testing a couple of weeks ago.

Bottom line: The modern drivers are different than the old Realtek out-of-kernel drivers. Expect things to happen automatically. The capabilities of the new drivers will be somewhat different so checking capabilities is a good idea:

$ iw list

dubhater, 2025-02-24:
The LED is exposed in ``/sys/class/leds/`` and can be controlled using the standard Linux LED subsystems methods. It blinks when there is network activity. The tricky part is making it stop permanently. Maybe some udev rule can do it.

morrownr, 2025-02-24:
@jagger235711 

Did a git pull, recompile here to recheck the rtl8814au driver. The little blinking LED is blinking. 

$ lsusb -t

Driver=rtw_8814au, 5000M

So things are working with no driver parameters. Wait until the driver is in the kernel, then you won't even need to find and compile a driver. Kinda makes life simple.

jagger235711, 2025-02-24:
> [@jagger235711](https://github.com/jagger235711)
> 
> Did a git pull, recompile here to recheck the rtl8814au driver. The little blinking LED is blinking.进行了git拉，在此处重新编译以重新检查RTL8814AU驱动程序。闪烁的LED闪烁。
> 
> $ lsusb -t
> 
> Driver=rtw_8814au, 5000M驱动程序= RTW_8814AU，5000m
> 
> So things are working with no driver parameters. Wait until the driver is in the kernel, then you won't even need to find and compile a driver. Kinda makes life simple.因此，没有驱动程序参数可以使用。等到驾驶员进入内核之后，您甚至不需要查找和编译驾驶员。 Kinda使生活变得简单。

ok,cool. opensource community is great!! and thanks for all the coder😀

jagger235711, 2025-02-24:
> The LED is exposed in `/sys/class/leds/` and can be controlled using the standard Linux LED subsystems methods. It blinks when there is network activity. The tricky part is making it stop permanently. Maybe some udev rule can do it.LED在`/sys/class/leds/`中暴露，可以使用标准Linux LED子系统方法来控制。有网络活动时，它会闪烁。棘手的部分是使其永久停止。也许某些UDEV规则可以做到。

thanks for reply!
yes, i want to make the it stop permanently. since it is so dazzling. but it is not a big deal, i have musked it.

Mastiffen, 2025-03-10:
> [@jagger235711](https://github.com/jagger235711)
> Driver=rtw_8814au, 5000M

I have an USB-AC68 AC1900 USB WiFi Adapter that runs on 8814au, and I have been using the old driver until today. But with the new driver  (after the fixes above) the transfer speed is a lot slower, and when I do my regular image backups with DD I loose connetion after a while and need to reboot. I have turned off wifi (that was off with the old driver as well), and I have not changed anything else on the system than installing the driver (twice, the second time with the fix).

I see on lsusb -t that it's listed as 480M, and that surprises me, because I was pretty sure I had the old one running on USB3. The hub and cable up is most certainly USB3.

Any ideas to what I could try?

dubhater, 2025-03-10:
@Mastiffen We need more information, like the full output of dmesg.

Mastiffen, 2025-03-10:
@dubhater Here it is. I see at the end that it's loading a Bluetooth module, I can't understand it. I have it disabled both as a DT-Overlay and with the file `/etc/modprobe.d/local-dontload.conf`!

```
pi@MadMax:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.74+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ee66100..0x000000003ee664ff (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ee66540..0x000000003ee6658c (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xfb7f33c0-0xfb7f5fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b2fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19712 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85928 r8192 d28760 u122880
[    0.000000] pcpu-alloc: s85928 r8192 d28760 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:26:1A:3C vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=c7b68b10-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles video=HDMI-A-1:1920x1080@60D cfg80211.ieee80211_regdom=NO
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0'
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] Policy zone: DMA32
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037300000-0x000000003b300000] (64MB)
[    0.000000] Memory: 3335280K/4049920K available (13504K kernel code, 2208K rwdata, 4292K rodata, 4864K init, 1086K bss, 190352K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43546 entries in 171 pages
[    0.000000] ftrace: allocated 171 pages with 5 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[... 503 lines cut ...]
[   19.557727] wlx3497f6664c9d: associated
[   19.613099] wlx3497f6664c9d: Limiting TX power to 30 (30 - 0) dBm as advertised by 04:d9:f5:52:13:5c
[   20.790019] Key type cifs.spnego registered
[   20.790037] Key type cifs.idmap registered
[... cut ...]
[   25.209557] eth0: renamed from vethbdff510
[   25.242589] br-94bd3b98927d: port 1(vethce8abf6) entered blocking state
[   25.242610] br-94bd3b98927d: port 1(vethce8abf6) entered forwarding state
[   56.883178] Bluetooth: Core ver 2.22
[   56.883298] NET: Registered PF_BLUETOOTH protocol family
[   56.883304] Bluetooth: HCI device and connection manager initialized
[   56.883320] Bluetooth: HCI socket layer initialized
[   56.883326] Bluetooth: L2CAP socket layer initialized
[   56.883337] Bluetooth: SCO socket layer initialized
```

dubhater, 2025-03-10:
It's not trying to switch to USB 3 mode.

This should make it print the reason why it's not switching:

```
echo 0x80000 > /sys/module/rtw_core/parameters/debug_mask
```

You just need to plug it again after changing the debug_mask and look at dmesg.

morrownr, 2025-03-10:
@Mastiffen 

Don't let me take the attention away from what Dubhater is asking for but I do have a couple of questions:

How many usb devices do you have plugged into that RasPi4B?

Is one of the devices a powered hub?

Mastiffen, 2025-03-10:
@dubhater: "no access" (in Norwegian) for that line, even with sudo. Do you mean that I have to plug it physically again? That is kind of a problem, because everything is inside waterproof boxes with lots of hot glue.

@morrownr Lots of them (NMEA2000, NMEA0183, a couple of USB drives. Here's the list, and yes, I believe at least one or maybe two of the hubs are powered.

`lsusb
Bus 002 Device 004: ID 05e3:0749 Genesys Logic, Inc. SD Card Reader and Writer
Bus 002 Device 005: ID 04e8:61fb Samsung Electronics Co., Ltd PSSD T7 Shield
Bus 002 Device 003: ID 05e3:0612 Genesys Logic, Inc. Hub
Bus 002 Device 002: ID 05e3:0626 Genesys Logic, Inc. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 004: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Phone Data Cable
Bus 001 Device 008: ID 08bb:2902 Texas Instruments PCM2902 Audio Codec
Bus 001 Device 009: ID 0b05:1817 ASUSTek Computer, Inc. USB-AC68 802.11a/b/g/n/ac (4x4) Wireless Adapter [Realtek RTL8814AU]
Bus 001 Device 007: ID 0bda:5401 Realtek Semiconductor Corp. RTL 8153 USB 3.0 hub with gigabit ethernet
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

dubhater, 2025-03-10:
You don't have to unplug it.

You can reload the rtw88 modules:

```
rmmod rtw_8814au
rmmod rtw_8814a
rmmod rtw_usb
rmmod rtw_core

insmod /path/rtw_core.ko debug_mask=0x80000
insmod /path/rtw_usb.ko switch_usb_mode=1
insmod /path/rtw_8814a.ko
insmod /path/rtw_8814au.ko
```

Mastiffen, 2025-03-10:
I copied the code to a sudo prompt, and that didn't really turn out well... I connected my phone to the client side of the boat's network and  saw this:

[image]

morrownr, 2025-03-10:
@Mastiffen 

If you had ask me if this setup would work reliably, I would have said no, not a chance. But you seemed to indicate some success.

Why? USB 3 is not mankind's great invention and the RasPi4B USB subsystem is interesting. By spec, it should provide 2800 mA of power to USB devices. However, said subsystem is not in accordance with spec and only provides 1200 mA total power. Your rtl8814au based adapter can use up to 900 mA... leaving only 300 mA for other things.

Okay, maybe you have a powered hub in the mix and maybe that can help but, there is always a but with the RasPi4B USB subsystem. But, the RasPi's are notorious for taking backfeeding from powered hubs. Here is what Google's AI has to say about that:

`While technically a powered USB hub can backfeed power to a Raspberry Pi 4B, it is not recommended as it can potentially damage the device due to bypassing the internal fuse and potentially causing unstable power delivery; most powered hubs are designed to not send power back to the host system.`

I would argue that most powered hubs are not well designed to not backfeed on hardware designed like the RasPi4B. This backfeeding problem has been known for years. My advice to users that want to use a rtl8814au based adapter as an AP on a RasPi4B was to have one and only one device plugged into a USB port and that was the adapter itself. Control the Pi headlessly. You may get stable decent performance that way... but, then there is the issue of the USB3 hub chipset. Not a good choice was made when selecting the chipset. My guess after much work a few years ago was that there is a problem in silicone that could not fully be corrected in the driver. Scatter-gather could cause problems even with AC1200 class adapters. Not sure if this rtl8814au driver uses scatter-gather.

I'll be quiet now and wish you luck..

morrownr, 2025-03-10:
@Mastiffen 

> I connected my phone to the client side of the boat's network and saw this:

You have to put the real path to the modules.

Mastiffen, 2025-03-10:
@morrownr This has been running with very close to 100 % reliability in my boat with the same Pi 4 for about three and a half years. The only problems have been because of updates, usually caused by me forgetting to run the USB wifi update before I do a reboot after one of the Pi updates. They are the devil's invention! ;)

The trick about the power for USB stuff with the Pi 4 (I was told, and it has never given me any problems) is that there must always be a passive USB hub first, and then the active one, and preferably not a cheap active hub, they often mess things up. I checked over the system and I have the active hub as number two in the box where the Pi itself is, and then another one in the box on the roof where the Asus USB is. 

Btw it's not the AP, I have a LAN cable to a Pi 3B+ that has as it's main job to be the AP.and run GPSd from a secondary GPS on the roof, which works even if the main chart plotter goes down. This Pi 4 with the Asus USB wifi card is the router/connection to the cabin's internet when the boat is docked, and it runs media (J.River Media Center), SignalK (charting and boating software), Home Asisstant, Node-RED and a few other assorted things flawlessly.  Both the Pi's and the active hubs (and about eight ESP32's) get their power from Hobbywing 12V-5V 6 ampere power supplies that are often used for drones, radio controlled planes and stuff like that. They are quite precise in the voltage and have no problem powering the gear. I feed the Pi's on the GPIO I think the power is more stable than when the USB power input is used.

And I have no idea what the path for the modules is, but I did a reboot after having used the `echo 0x80000 > /sys/module/rtw_core/parameters/debug_mask` from a sudo prompt, not just with sudo. This time it didn't complain. Here is the output. Does it have the necessary output?

```
dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.74+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ee66100..0x000000003ee664ff (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ee66540..0x000000003ee6658c (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xfb7f33c0-0xfb7f5fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b2fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19712 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85928 r8192 d28760 u122880
[    0.000000] pcpu-alloc: s85928 r8192 d28760 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:26:1A:3C vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=c7b68b10-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles video=HDMI-A-1:1920x1080@60D cfg80211.ieee80211_regdom=NO
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0'
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[... cut ...]
[   23.866597] br-94bd3b98927d: port 1(veth847afa2) entered blocking state
[   23.866622] br-94bd3b98927d: port 1(veth847afa2) entered forwarding state
[   55.543349] Bluetooth: Core ver 2.22
[   55.543448] NET: Registered PF_BLUETOOTH protocol family
[   55.543452] Bluetooth: HCI device and connection manager initialized
[   55.543466] Bluetooth: HCI socket layer initialized
[   55.543473] Bluetooth: L2CAP socket layer initialized
[   55.543488] Bluetooth: SCO socket layer initialized
pi@MadMax:~ $ 
```

Mastiffen, 2025-03-11:
I think those two dmesg outputs are quite similar after a check, so I guess the line with `echo 0x80000 > /sys/module/rtw_core/parameters/debug_mask`  does not survive a reboot. I need to find the module paths and put into a shell script to disconnect and reconnect them. I'll get to that this afternoon.

a5a5aa555oo, 2025-03-11:
@Mastiffen 

You can create a file named `rtw88.conf` in `/etc/modprobe.d` and add these lines into it, then dmesg will show the info that dubhater needs after reboot.
```
options rtw_core debug_mask=0x80000
options rtw_usb switch_usb_mode=1
```

Mastiffen, 2025-03-11:
Thank you, @a5a5aa555oo ! Here is the output of a reboot after adding that:

```
dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.74+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ee66100..0x000000003ee664ff (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ee66540..0x000000003ee6658c (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xfb7f33c0-0xfb7f5fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b2fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19712 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85928 r8192 d28760 u122880
[    0.000000] pcpu-alloc: s85928 r8192 d28760 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory numa_policy=interleave snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:26:1A:3C vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=c7b68b10-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles video=HDMI-A-1:1920x1080@60D cfg80211.ieee80211_regdom=NO
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0'
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] Policy zone: DMA32
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037300000-0x000000003b300000] (64MB)
[    0.000000] Memory: 3335280K/4049920K available (13504K kernel code, 2208K rwdata, 4292K rodata, 4864K init, 1086K bss, 190352K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43546 entries in 171 pages
[    0.000000] ftrace: allocated 171 pages with 5 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[... 495 lines cut ...]
[   16.566071] wlx3497f6664c9d: send auth to 04:d9:f5:52:13:5c (try 1/3)
[   16.567043] wlx3497f6664c9d: authenticated
[   16.568714] wlx3497f6664c9d: associate with 04:d9:f5:52:13:5c (try 1/3)
[   16.570270] wlx3497f6664c9d: RX AssocResp from 04:d9:f5:52:13:5c (capab=0x1011 status=0 aid=1)
[   16.582628] wlx3497f6664c9d: associated
[... cut ...]
[   21.671288] veth679aa18: entered allmulticast mode
[   21.671445] veth679aa18: entered promiscuous mode
[   21.672076] br-94bd3b98927d: port 1(veth679aa18) entered blocking state
[   21.672095] br-94bd3b98927d: port 1(veth679aa18) entered forwarding state
[   21.672562] br-94bd3b98927d: port 1(veth679aa18) entered disabled state
[   21.768151] eth0: renamed from vethd1053d5
[   21.791631] br-94bd3b98927d: port 1(veth679aa18) entered blocking state
[   21.791651] br-94bd3b98927d: port 1(veth679aa18) entered forwarding state
pi@MadMax:~ $ 

```

a5a5aa555oo, 2025-03-11:
@Mastiffen 
It's strange that it shows me nothing, too.🤨

```
[alonso@BM5220 parameters] $ cat /sys/module/rtw_core/parameters/debug_mask 
524288
[alonso@BM5220 parameters] $ cat /sys/module/rtw_usb/parameters/switch_usb_mode 
Y
[alonso@BM5220 parameters] $ sudo dmesg | grep rtw
[ 2799.947678] rtw_8822bu 8-2:1.2: Firmware version 30.20.0, H2C version 14
[ 2801.910140] rtw_8822bu 8-2:1.2: write register 0xc4 failed with -71
[ 2801.910631] rtw_8822bu 8-2:1.2: rtw_usb_reg_sec: reg 0x4e0, usb write 1 fail, status: -71
[ 2801.910644] rtw_8822bu 8-2:1.2: switching to USB 3 mode
[ 2801.910886] usbcore: registered new interface driver rtw_8822bu
[ 2802.407486] rtw_8822bu 9-2:1.2: Firmware version 30.20.0, H2C version 14
[ 2804.494999] rtw_8822bu 9-2:1.2 wlp0s16f1u2i2: renamed from wlan0
```

@dubhater 
We need help, I am not quite sure how to make use of `debug_mask`.

Mastiffen, 2025-03-11:
@a5a5aa555oo Will you please enlighten me to how it works? So I can have the correct info ready when @dubhater has time to look at it?

a5a5aa555oo, 2025-03-11:
I am not good at programming (and English), so the following may not 100% correct.

[Here is a list of available values](https://github.com/lwfinger/rtw88/blob/6ca4faca0e5cfeeecd522eb07b605a2bb890cd88/debug.h#L8) for `debug_mask`, if we set `debug_mask` to [0x80000](https://github.com/lwfinger/rtw88/blob/6ca4faca0e5cfeeecd522eb07b605a2bb890cd88/debug.h#L28) ,  we will see some debug info about USB in dmesg.

but the driver won't show them in detail.

You can search `RTW_DBG_USB` in [usb.c](https://github.com/lwfinger/rtw88/blob/master/usb.c), and will see only 6 kind of error messages will be shown.

That's what I know so far, we need dubhater to solve the issue you encountered.

It would be good to post the output of these three commands to make sure the module parameters are set correctly.
`sudo cat /sys/module/rtw_core/parameters/debug_mask `
`sudo cat /sys/module/rtw_usb/parameters/switch_usb_mode `
`sudo dmesg | grep rtw`

Mastiffen, 2025-03-11:
Ok, good. Then I wait. 😁 Here are the outputs:
```
sudo cat /sys/module/rtw_core/parameters/debug_mask 
524288
```
```
 sudo cat /sys/module/rtw_usb/parameters/switch_usb_mode 
Y
```
```
sudo dmesg | grep rtw
[    7.110306] rtw_core: loading out-of-tree module taints kernel.
[    7.282182] rtw_8814au 1-1.1.1.3.4:1.0: Firmware version 33.6.0, H2C version 6
[    8.226846] usbcore: registered new interface driver rtw_8814au
[    8.352081] rtw_8814au 1-1.1.1.3.4:1.0 wlx3497f6664c9d: renamed from wlan0
```

a5a5aa555oo, 2025-03-11:
@Mastiffen 

You said it's a problem to replug the 8814AU adapter, right?
What about powering off your RasPi4B **completely** and then restart it?
is it impossible,too? I think a cold boot of the adapter may be helpful..

dubhater, 2025-03-11:
@Mastiffen Maybe you compiled rtw88 with CONFIG_RTW88_DEBUG disabled? You can check this easily by changing debug_mask to 0x20. If CONFIG_RTW88_DEBUG is enabled it will definitely print some more things in dmesg.

Mastiffen, 2025-03-11:
I managed to find the correct path to the files by using `sudo modinfo rtw_8814au -n` and similar on the module, so I made a shell script like this:

```
#!/bin/bash
rmmod rtw_8814au
rmmod rtw_8814a
rmmod rtw_usb
rmmod rtw_core

insmod /lib/modules/6.6.74+rpt-rpi-v8/kernel/drivers/net/wireless/realtek/rtw88/rtw_core.ko.xz debug_mask=0x80000
insmod /lib/modules/6.6.74+rpt-rpi-v8/kernel/drivers/net/wireless/realtek/rtw88/rtw_usb.ko.xz switch_usb_mode=1
insmod /lib/modules/6.6.74+rpt-rpi-v8/kernel/drivers/net/wireless/realtek/rtw88/rtw_8814a.ko.xz
insmod /lib/modules/6.6.74+rpt-rpi-v8/kernel/drivers/net/wireless/realtek/rtw88/rtw_8814au.ko.xz
```

And ran that from a sudo su prompt. After the run, I get this in dmesg:
```[  480.693324] rtw_8814au 1-1.1.1.3.4:1.0: failed to get tx report from firmware
[  485.874962] wlx3497f6664c9d: authenticate with 04:d9:f5:52:13:5c
[  486.755311] wlx3497f6664c9d: send auth to 04:d9:f5:52:13:5c (try 1/3)
[  486.756328] wlx3497f6664c9d: authenticated
[  486.765434] wlx3497f6664c9d: associate with 04:d9:f5:52:13:5c (try 1/3)
[  486.766554] wlx3497f6664c9d: RX AssocResp from 04:d9:f5:52:13:5c (capab=0x1011 status=0 aid=1)
[  486.786356] wlx3497f6664c9d: associated
[  486.864020] wlx3497f6664c9d: Limiting TX power to 30 (30 - 0) dBm as advertised by 04:d9:f5:52:13:5c
[  517.093862] usbcore: deregistering interface driver rtw_8814au
[  517.126112] wlx3497f6664c9d: deauthenticating from 04:d9:f5:52:13:5c by local choice (Reason: 3=DEAUTH_LEAVING)
[  517.406436] usb 1-1.1.1.3.4: reset high-speed USB device number 9 using xhci_hcd
[  517.734372] rtw_8814au 1-1.1.1.3.4:1.0: Firmware version 33.6.0, H2C version 6
[  518.030096] usbcore: registered new interface driver rtw_8814au
[  518.037689] rtw_8814au 1-1.1.1.3.4:1.0 wlx3497f6664c9d: renamed from wlan0
[  522.847996] wlx3497f6664c9d: authenticate with 04:d9:f5:52:13:5c
[  523.687558] wlx3497f6664c9d: send auth to 04:d9:f5:52:13:5c (try 1/3)
[  523.688261] wlx3497f6664c9d: authenticated
[  523.690070] wlx3497f6664c9d: associate with 04:d9:f5:52:13:5c (try 1/3)
[  523.691016] wlx3497f6664c9d: RX AssocResp from 04:d9:f5:52:13:5c (capab=0x1011 status=0 aid=1)
[  523.701196] wlx3497f6664c9d: associated
[  523.729005] wlx3497f6664c9d: Limiting TX power to 30 (30 - 0) dBm as advertised by 04:d9:f5:52:13:5c
```

By changing the mask to 0x20 I got this:
```
[  626.468748] usbcore: deregistering interface driver rtw_8814au
[  626.513002] wlx3497f6664c9d: deauthenticating from 04:d9:f5:52:13:5c by local choice (Reason: 3=DEAUTH_LEAVING)
[  626.737490] usb 1-1.1.1.3.4: reset high-speed USB device number 9 using xhci_hcd
[  627.061285] rtw_8814au 1-1.1.1.3.4:1.0: Firmware version 33.6.0, H2C version 6
[  627.366328] usbcore: registered new interface driver rtw_8814au
[  627.376640] rtw_8814au 1-1.1.1.3.4:1.0 wlx3497f6664c9d: renamed from wlan0
[  632.567458] wlx3497f6664c9d: authenticate with 04:d9:f5:52:13:5c
[  633.406751] wlx3497f6664c9d: send auth to 04:d9:f5:52:13:5c (try 1/3)
[  633.407494] wlx3497f6664c9d: authenticated
[  633.409206] wlx3497f6664c9d: associate with 04:d9:f5:52:13:5c (try 1/3)
[  633.410127] wlx3497f6664c9d: RX AssocResp from 04:d9:f5:52:13:5c (capab=0x1011 status=0 aid=1)
[  633.420455] wlx3497f6664c9d: associated
[  633.504957] wlx3497f6664c9d: Limiting TX power to 30 (30 - 0) dBm as advertised by 04:d9:f5:52:13:5c
```

Does that mean that I need to recompile?

a5a5aa555oo, 2025-03-11:
@dubhater 

What do you think about this patch? will the driver print the info needed if it is applied?
```patch
diff --git a/usb.c b/usb.c
index 92f4c06..b62e273 100644
--- a/usb.c
+++ b/usb.c
@@ -1095,8 +1095,11 @@ static int rtw_usb_switch_mode_new(struct rtw_dev *rtwdev)
 				     BIT_USB3_USB2_TRANSITION));
 
 	if (!can_switch) {
+		printk("rtw_usb: Switching to USB 3 mode unsupported by the chip\n");
+/*
 		rtw_dbg(rtwdev, RTW_DBG_USB,
 			"Switching to USB 3 mode unsupported by the chip\n");
+*/
 		return 0;
 	}
 
@@ -1143,14 +1146,20 @@ static int rtw_usb_switch_mode(struct rtw_dev *rtwdev)
 		return 0;
 
 	if (!rtwdev->efuse.usb_mode_switch) {
+		printk("rtw_usb: Switching to USB 3 mode disabled by chip's efuse\n");
+/*
 		rtw_dbg(rtwdev, RTW_DBG_USB,
 			"Switching to USB 3 mode disabled by chip's efuse\n");
+*/
 		return 0;
 	}
 
 	if (!rtw_switch_usb_mode) {
+		printk("rtw_usb: Switching to USB 3 mode disabled by module parameter\n");
+/*
 		rtw_dbg(rtwdev, RTW_DBG_USB,
 			"Switching to USB 3 mode disabled by module parameter\n");
+*/
 		return 0;
 	}
 
@@ -1288,7 +1297,8 @@ int rtw_usb_probe(struct usb_interface *intf, const struct usb_device_id *id)
 	ret = rtw_usb_switch_mode(rtwdev);
 	if (ret) {
 		/* Not a fail, but we do need to skip rtw_register_hw. */
-		rtw_dbg(rtwdev, RTW_DBG_USB, "switching to USB 3 mode\n");
+		printk("rtw_usb: switching to USB 3 mode\n")
+		// rtw_dbg(rtwdev, RTW_DBG_USB, "switching to USB 3 mode\n");
 		ret = 0;
 		goto err_destroy_rxwq;
 	}
```

dubhater, 2025-03-11:
@a5a5aa555oo Yes, printk will appear in dmesg even if CONFIG_RTW88_DEBUG is not on. But I think in this case all of those branches are skipped.

@Mastiffen I think it's a flaw in the automatic switching code. Try the patch from @a5a5aa555oo above and this:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/usb.c b/drivers/net/wireless/realtek/rtw88/usb.c
index 3c3a4b6cea92..0aa8621cb353 100644
--- a/usb.c
+++ b/usb.c
@@ -1130,6 +1130,14 @@ static int rtw_usb_switch_mode(struct rtw_dev *rtwdev)
 		return rtw_usb_switch_mode_new(rtwdev);
 }
 
+static void rtw_usb_reset_mode_switch(struct rtw_dev *rtwdev)
+{
+	if (rtw_usb3_chip_new(rtwdev->chip->id))
+		rtw_write32_clr(rtwdev, REG_PAD_CTRL2, BIT_MASK_USB23_SW_MODE_V1);
+	else if (rtw_usb3_chip_old(rtwdev->chip->id))
+		rtw_write8_clr(rtwdev, REG_HCI_OPT_CTRL, GENMASK(3, 2));
+}
+
 #define USB_REG_PAGE	0xf4
 #define USB_PHY_PAGE0	0x9b
 #define USB_PHY_PAGE1	0xbb
@@ -1305,6 +1313,8 @@ void rtw_usb_disconnect(struct usb_interface *intf)
 	rtwdev = hw->priv;
 	rtwusb = rtw_get_usb_priv(rtwdev);
 
+	rtw_usb_reset_mode_switch(rtwdev);
+
 	rtw_usb_cancel_rx_bufs(rtwusb);
 
 	rtw_unregister_hw(rtwdev, hw);
```

Mastiffen, 2025-03-11:
@dubhater Sorry for my ignorance (I only program simple scripts in LUA, Python and YAML), but I have no idea how I should patch it. Should I put that in a script and run it?

dubhater, 2025-03-11:
Save them as printk.diff and reset.diff, then apply them with patch:

```
cd /path/rtw88/
patch -i printk.diff
patch -i reset.diff
```

Then recompile, then reload the modules twice.

[32 more replies]