lwfinger/rtw88 #273: RTL8814AU: My Wi-Fi Direct isn't listed on Android devices

khuongtrinh2, 2024-12-24:
Hi everyone,

Currently, I'm trying to enable Wi-Fi Direct with rtl8814au driver (USB Wi-Fi: TP-LINK Archer T4U)
and have an issue when pairing both Board and Android devices:

- From My board, Wi-Fi Direct of Android Device can be listed on GUI and connect as usual. After connecting successfully, if it is GO (Group Ower), it can be listed on Android device:

```
12-24 06:14:15.914  2106  2134 I Tethering: WifiP2pAction: P2pInfo: groupFormed: true isGroupOwner: true groupOwnerAddress: null Group: network: DIRECT-BN-Android_mNNW
12-24 06:14:15.914  2106  2134 I Tethering:  GO: Device:
12-24 06:14:15.914  2106  2134 I Tethering:  deviceAddress: 02:00:00:00:00:00
12-24 06:14:15.914  2106  2134 I Tethering:  primary type: null
12-24 06:14:15.914  2106  2134 I Tethering:  secondary type: null
12-24 06:14:15.914  2106  2134 I Tethering:  wps: 0
12-24 06:14:15.914  2106  2134 I Tethering:  grpcapab: 0
12-24 06:14:15.914  2106  2134 I Tethering:  devcapab: 0
12-24 06:14:15.914  2106  2134 I Tethering:  status: 4
12-24 06:14:15.914  2106  2134 I Tethering:  wfdInfo: null
12-24 06:14:15.914  2106  2134 I Tethering:  vendorElements: null
12-24 06:14:15.914  2106  2134 I Tethering:  interface: p2p0
12-24 06:14:15.914  2106  2134 I Tethering:  networkId: 1
12-24 06:14:15.914  2106  2134 I Tethering:  frequency: 2412
12-24 06:14:15.915  2106  2134 I Tethering: [BpfCoordinator] Monitoring started
```

- From Android Phone, Wi-Fi Direct of My board cannot able to be see on GUI.

I want both my board and my Android device to be able to see each other before connecting.

Anyone has ideas or solution to fix this issue? 
Thank you so much for your help

dubhater, 2024-12-24:
Does it work with the other driver? https://github.com/morrownr/8814au

khuongtrinh2, 2024-12-24:
Thank you for your reply

I don't think it works with the driver you attached.
A person has issue similar to me [https://github.com/aircrack-ng/rtl8812au/issues/314](url)

dubhater, 2024-12-24:
Well, this is the current status of P2P support in rtw88: https://lore.kernel.org/linux-wireless/161bc4c6ffe14a36ad76f214fe533eef@realtek.com/

khuongtrinh2, 2024-12-25:
Hi @dubhater 
Although this information does not help me fix this issue but thank you for support.