lwfinger/rtw88 #274: Ambiguous Documentation

Boobies, 2024-12-26:
The README puts emphasis on doing a `make install` after kernel updates regardless of whether the user is using SecureBoot. It should either not specify the exact steps and instead suggest a pull and then refer to the installation process or, if the explicit `make install` command is reiterated (as it currently is) then it should clarify whether to use `install` or `sign-install` after kernel updates.

morrownr, 2024-12-29:
@Boobies 

Good catch. I just merged an attempt to fix this so please take a look and see what you think.

Boobies, 2025-01-02:
If it were up to me, I would just say to follow the usual installation instructions after doing a pull because for consistency, as it stands, we're refered to them for SecureBoot but they are explicitly repeated for the case with no SecureBoot. That said, I'm satisfied with the commit because all the information is there and I'm just being pedantic.