lwfinger/rtw88 #277: rtl8821cu cannot register as wifi dongle

briansune, 2024-12-27:
@dubhater @morrownr 

I am working on ARM 32bit CPU and discover a issue that also can be found on Google but no actual solution is given.

The dmesg as follows:

[   28.283096] rtw_core: loading out-of-tree module taints kernel.
[   36.959993] usbcore: registered new interface driver rtw_8821cu
[   68.189645] usb 1-1.1: USB disconnect, device number 3
[   94.534166] usb 1-1.1: new high-speed USB device number 4 using ci_hdrc
[   94.685906] usb-storage 1-1.1:1.0: USB Mass Storage device detected
[   94.693218] scsi host0: usb-storage 1-1.1:1.0
[   95.767026] scsi 0:0:0:0: CD-ROM            Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[   95.770080] scsi 0:0:0:0: Attached scsi generic sg0 type 5
[  991.469681] usb 1-1.1: USB disconnect, device number 4
[  993.834162] usb 1-1.2: new high-speed USB device number 5 using ci_hdrc
[  993.987547] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[  994.003238] scsi host0: usb-storage 1-1.2:1.0
[  995.047207] scsi 0:0:0:0: CD-ROM            Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[  995.050752] scsi 0:0:0:0: Attached scsi generic sg0 type 5
[ 1019.675588] usbcore: deregistering interface driver rtw_8821cu
[ 1029.338628] usbcore: registered new interface driver rtw_8821cu

We can see the USB is now entered a driver disk mode.
When trying the USB dongle in Windows I also see this disk and once you double click it. It will install the Windows driver w/o issue and auto turn into WIFI Dongle.

But when this happen in ARM+Ubuntu I have no clue how to make it back to normal WIFI dongle.

briansune, 2024-12-27:
Okay problem had solved. Found the proper procedure to resolve the disk.

Bus 001 Device 005: ID 0bda:1a2b Realtek Semiconductor Corp.
Bus 001 Device 002: ID 0424:2512 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

> sudo usb_modeswitch -KW -v 0bda -p 1a2b

```
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.5.2 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0x1a2b

StandardEject=1

Look for default devices ...
  found USB ID 0bda:1a2b
   vendor ID matched
   product ID matched
  found USB ID 0424:2512
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 006 on bus 001
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 8
Use endpoints 0x0b (out) and 0x8a (in)

USB description data (for identification)
-------------------------
Manufacturer: Realtek
     Product: DISK
  Serial No.: not provided
-------------------------
Sending standard EJECT sequence
Looking for active drivers ...
 OK, driver detached
Set up interface 0
Use endpoint 0x0b for message sending ...
Trying to send message 1 to endpoint 0x0b ...
 OK, message successfully sent
Read the response to message 1 (CSW) ...
 Response successfully read (13 bytes), status 1
Trying to send message 2 to endpoint 0x0b ...
 OK, message successfully sent
Read the response to message 2 (CSW) ...
 Response successfully read (13 bytes), status 0
Trying to send message 3 to endpoint 0x0b ...
 Sending the message returned error -1. Try to continue
Read the response to message 3 (CSW) ...
 Response reading failed (error -1)
 Device is gone, skip any further commands
-> Run lsusb to note any changes. Bye!
```

Bus 001 Device 007: ID 0bda:c820 Realtek Semiconductor Corp.
Bus 001 Device 002: ID 0424:2512 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

closed 2024-12-27 by briansune

dubhater, 2024-12-27:
In Arch Linux it's enough to install the usb_modeswitch package. After that it's all automatic.

briansune, 2024-12-27:
@dubhater 

For the 18.04 LTS Ubuntu it requires user to modify it like what I had mentioned.
There are other REALTEK chip also came with driver disk and need to convert back to normal chip mode