lwfinger/rtw88 #279: 8814au: injection doesn't work with channels above 11

reaveruser, 2024-12-30:
I am using this drivers from morroenr https://github.com/morrownr/8814au and the injection is working with all channels, however when I tested this driver, which is working fine, but the injection doesn't work with channels above 11 using ALFA AWUS1900

dubhater, 2024-12-30:
How do you administer the injection and how do you determine that it doesn't work?

reaveruser, 2024-12-30:
aireplay-ng -9

dubhater, 2024-12-30:
Seems to work for me:

```
23:18:40  Trying broadcast probe requests...
23:18:40  Injection is working!
23:18:42  Found 2 APs

23:18:42  Trying directed probe requests...
23:18:42  ... - channel: 116 - '...'
23:18:42  Ping (min/avg/max): 0.202ms/0.426ms/1.122ms Power: -80.53
23:18:42  30/30: 100%

23:18:42  ... - channel: 116 - '...'
23:18:42  Ping (min/avg/max): 0.251ms/0.425ms/1.265ms Power: -82.27
23:18:42  30/30: 100%
```

It also found the networks on channels 13, 40, 52, 56, 132.

It didn't find my network on channel 100 for some reason.

reaveruser, 2025-01-01:
That is strange? how did you test? did you enable monitor mode?

in my tests, any channel above 11 will not work, not even 36 to 48. And I have 2 8814au wifi adapters

**however, no issues using the driver from https://github.com/morrownr/8814au**

> └─$ sudo aireplay-ng wlan0mon -9                                                               
> 
> 07:30:55  Trying broadcast probe requests...
> 07:30:57  No Answer...
> 07:30:57  Found 0 APs
>                                                                                                                                                                                                         
> 
> └─$ sudo aireplay-ng wlan0mon -9 -a xx:xx:xx:xx:xx
> 07:30:58  Waiting for beacon frame (BSSID: xx:xx:xx:xx:xx) on channel 108
> 07:30:58  Trying broadcast probe requests...
> 07:31:00  No Answer...
> 07:31:00  Found 1 AP 
> 
> 07:31:00  Trying directed probe requests...
> 07:31:00  xx:xx:xx:xx:xx - channel: 0 - 'xxx'
> 07:31:06   0/30:   0%
> 
>                                                                                                                                                                                                         
> 
> └─$ sudo aireplay-ng wlan1mon -9                                                               
> 07:31:23  Trying broadcast probe requests...
> 07:31:24  No Answer...
> 07:31:24  Found 4 APs
> 
> 07:31:24  Trying directed probe requests...
> 07:31:24  xx:xx:xx:xx:xx - channel: 12 - 'xxx'
> 07:31:30   0/30:   0%
> 
> 07:31:31  xx:xx:xx:xx:xx - channel: 12 - 'xxx'
> 07:31:37   0/30:   0%
> 
> 07:31:37 xx:xx:xx:xx:xx - channel: 13 - 'xxx'
> 07:31:43   0/30:   0%
> 
> 07:31:43  xx:xx:xx:xx:xx - channel: 13 - 'xxx'
> 07:31:49   0/30:   0%
> 
> └─$ sudo aireplay-ng wlan1mon -9
> 07:34:07  Trying broadcast probe requests...
> 07:34:09  No Answer...
> 07:34:09  Found 2 APs
> 
> 07:34:09  Trying directed probe requests...
> 07:34:09  xx:xx:xx:xx:xx - channel: 48 - ''
> 07:34:15   0/30:   0%
> 
> 07:34:15  xx:xx:xx:xx:xx - channel: 48 - 'xxx'
> 07:34:21   0/30:   0%
> 
>                                                                                                                                                                                                         
> 
> └─$ sudo aireplay-ng wlan1mon -9
> 07:34:28  Trying broadcast probe requests...
> 07:34:30  No Answer...
> 07:34:30  Found 0 APs
>                                                                                                                                                                                                         
> 
> └─$ sudo aireplay-ng wlan0mon -9                     
> 07:34:45  Trying broadcast probe requests...
> 07:34:47  No Answer...
> 07:34:47  Found 0 APs
>                                                                                                                                                                                                         
> 
> └─$ sudo aireplay-ng wlan0mon -9
> 07:34:54  Trying broadcast probe requests...
> 07:34:56  No Answer...
> 07:34:56  Found 2 APs
> 
> 07:34:56  Trying directed probe requests...
> 07:34:56  xx:xx:xx:xx:xx - channel: 48 - ''
> 07:35:02   0/30:   0%
> 
> 07:35:02  xx:xx:xx:xx:xx - channel: 48 - 'xxx'
> 07:35:08   0/30:   0%

dubhater, 2025-01-01:
```sh
systemctl stop NetworkManager.service
airmon-ng start wlp3s0f3u4
airodump-ng wlp3s0f3u4 --channel 132
aireplay-ng -9 wlp3s0f3u4
```

Repeat the last two for the other channels.

reaveruser, 2025-01-01:
What distro/kernel are you using? can you please post `iw list` result?

I am testing using kali kernel 6.11.2 will all updates are installed up to date.

dubhater, 2025-01-01:
Arch Linux, currently 6.12.7.

<details>
<summary> iw list</summary>

```
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x7 RX 0x7
	Configured Antennas: TX 0x7 RX 0x7
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
	Band 1:
		Capabilities: 0x196f
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412.0 MHz [1] (20.0 dBm)
			* 2417.0 MHz [2] (20.0 dBm)
			* 2422.0 MHz [3] (20.0 dBm)
			* 2427.0 MHz [4] (20.0 dBm)
			* 2432.0 MHz [5] (20.0 dBm)
			* 2437.0 MHz [6] (20.0 dBm)
[... 151 lines cut ...]
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ POWERED_ADDR_CHANGE ]: can change MAC address while up
```
</details>

reaveruser, 2025-01-01:
Here is mine from this driver https://github.com/morrownr/8814au missing some features.

```
Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1863
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412.0 MHz [1] (20.0 dBm)
			* 2417.0 MHz [2] (20.0 dBm)
			* 2422.0 MHz [3] (20.0 dBm)
			* 2427.0 MHz [4] (20.0 dBm)
			* 2432.0 MHz [5] (20.0 dBm)
			* 2437.0 MHz [6] (20.0 dBm)
			* 2442.0 MHz [7] (20.0 dBm)
			* 2447.0 MHz [8] (20.0 dBm)
			* 2452.0 MHz [9] (20.0 dBm)
			* 2457.0 MHz [10] (20.0 dBm)
			* 2462.0 MHz [11] (20.0 dBm)
			* 2467.0 MHz [12] (20.0 dBm)
			* 2472.0 MHz [13] (20.0 dBm)
			* 2484.0 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x1863
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23
		VHT Capabilities (0x03c051b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 1300 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 1300 Mbps
		VHT extended NSS: not supported
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5180.0 MHz [36] (30.0 dBm)
			* 5200.0 MHz [40] (30.0 dBm)
			* 5220.0 MHz [44] (30.0 dBm)
			* 5240.0 MHz [48] (30.0 dBm)
			* 5260.0 MHz [52] (30.0 dBm) (no IR, radar detection)
			* 5280.0 MHz [56] (30.0 dBm) (no IR, radar detection)
			* 5300.0 MHz [60] (30.0 dBm) (no IR, radar detection)
			* 5320.0 MHz [64] (30.0 dBm) (no IR, radar detection)
			* 5500.0 MHz [100] (30.0 dBm) (no IR, radar detection)
			* 5520.0 MHz [104] (30.0 dBm) (no IR, radar detection)
			* 5540.0 MHz [108] (30.0 dBm) (no IR, radar detection)
			* 5560.0 MHz [112] (30.0 dBm) (no IR, radar detection)
			* 5580.0 MHz [116] (30.0 dBm) (no IR, radar detection)
			* 5600.0 MHz [120] (30.0 dBm) (no IR, radar detection)
			* 5620.0 MHz [124] (30.0 dBm) (no IR, radar detection)
			* 5640.0 MHz [128] (30.0 dBm) (no IR, radar detection)
			* 5660.0 MHz [132] (30.0 dBm) (no IR, radar detection)
			* 5680.0 MHz [136] (30.0 dBm) (no IR, radar detection)
			* 5700.0 MHz [140] (30.0 dBm) (no IR, radar detection)
			* 5720.0 MHz [144] (disabled)
			* 5745.0 MHz [149] (30.0 dBm)
			* 5765.0 MHz [153] (30.0 dBm)
			* 5785.0 MHz [157] (30.0 dBm)
			* 5805.0 MHz [161] (30.0 dBm)
			* 5825.0 MHz [165] (30.0 dBm)
			* 5845.0 MHz [169] (disabled)
			* 5865.0 MHz [173] (disabled)
			* 5885.0 MHz [177] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_wiphy_netns
		 * set_channel
		 * connect
		 * disconnect
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports SAE with AUTHENTICATE command
	Device supports scan flush.
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	Supported extended features:
```
What is your wifi adapter / chip? it seems it is not ALFA AWUS1900? am I right?

dubhater, 2025-01-01:
Right now it's Comfast CF-938AC. It has the same chip as the AWUS1900.

It would be more interesting to see your ``iw list`` with this driver.

reaveruser, 2025-01-01:
I will try tomorrow, as I am compiling now and I can't restart.

How do I know your was not AWUS1900?
`
max # scan SSIDs: 4` vs `max # scan SSIDs: 9` == less antennas

`VHT TX highest supported: 1170 Mbp`s vs `VHT RX highest supported: 1300 Mbps` == not same chip (device id)

dubhater, 2025-01-01:
You too will get 4 and 1170 with this driver.

reaveruser, 2025-01-03:
how did you compile the driver? did you encounter any error?

I had to make some links and remove these lines from `Makefile.modfinal` in `/usr/src/linux-kbuild-6.11.2/scripts/`
 
```
ifdef CONFIG_DEBUG_INFO_BTF_MODULES
	+$(if $(newer-prereqs),$(call cmd,btf_ko))
endif
```

Injection is still not working with any channel above 11 using this driver even after the last update 2 days ago https://github.com/lwfinger/rtw88/commit/22dc11c8e19f654642ee4ff98adf4a41cd098fe5

As I said there are no issues using the driver from https://github.com/morrownr/8814au

I guess it could be the kernel? maybe 6.12 will fix the issue, I will try testing using arch and 6.12 later

Here is `iw list` using this driver

```
Wiphy phy2
	wiphy index: 2
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x7 RX 0x7
	Configured Antennas: TX 0x7 RX 0x7
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x196f
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT Max RX data rate: 450 Mbps
		HT TX/RX MCS rate indexes supported: 0-23, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
[... 168 lines cut ...]
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ POWERED_ADDR_CHANGE ]: can change MAC address while up

```

dubhater, 2025-01-03:
I'm rarely testing this repository. I work directly in the kernel tree.

You have a lot of "no IR" there. Did you set the country code with ``iw reg set XX``? The other driver doesn't care about that because it takes the country code from a module parameter.

kimocoder, 2025-01-05:
I did some 8812au/8821/8814 drivers some years ago where these capabilities was added. [github.com/aircrack-ng](https://github.com/aircrack-ng) with the main focus was these capabilities. However, I have not had time to look at this yet, but it has entered a mac80211 driver and these capabilities has to be handled w/netlink and everything.. but Im happy to see these drivers joined the great company of a legend, right where 8814au should be.

1. Monitor Mode
````
Goal: Enable passive listening for all Wi-Fi traffic.
mac80211 Role:
mac80211 handles monitor mode directly with NL80211_IFTYPE_MONITOR.
Supports injection and passive mode configuration via netlink.
Driver Requirements:
Implement ieee80211_ops callbacks:
add_interface(): Add NL80211_IFTYPE_MONITOR.
config_interface(): Configure monitor filter flags for FCS, PLCP, or CTRL.
````

Set up hardware capability flags:
````
hw->flags |= IEEE80211_HW_RX_INCLUDES_FCS | IEEE80211_HW_SUPPORTS_MONITOR;
````
Adjust filter_flags in configure_filter callback for MONITOR mode.

2. Packet Injection
````
Goal: Inject crafted packets via software or raw radiotap headers.
mac80211 Role:
Use ieee80211_tx() to transmit crafted frames.
Handle radiotap headers via SKBs.
Driver Requirements:
Set IEEE80211_TX_CTL_INJECTED in SKB's control flags.
````

Support tx() callback in the driver:
````
static int driver_tx(struct ieee80211_hw *hw, struct ieee80211_tx_control *control,
                     struct sk_buff *skb);
````
Ensure proper DMA mapping for skb->data.

3. Radiotap Parsing (Iterator)
Goal: Decode radiotap headers for monitor mode or packet injection.
Implementation:
Use Linux kernel's radiotap iterator:
````
#include <net/ieee80211_radiotap.h>
struct ieee80211_radiotap_iterator iter;
ieee80211_radiotap_iterator_init(&iter, radiotap_hdr, len, NULL);
while (ieee80211_radiotap_iterator_next(&iter) == 0) {
    // Parse fields
}
````
Include support for radiotap extensions like TSFT, rate, and antenna.

4. 4-Byte Payload Handling
Goal: Support hardware or firmware's capability to add or strip 4-byte CRC/FCS.
Driver Changes:
Indicate CRC stripping in hardware flags:
````
hw->flags |= IEEE80211_HW_RX_INCLUDES_FCS;
````

If firmware strips CRC, adjust SKB lengths to avoid errors in higher layers:
````
skb_trim(skb, skb->len - 4);
````

5. Transmit (xmit)
Goal: Enable custom frame transmission in raw or processed format.
Driver Role:
Implement tx() or tx_queue_skb() callback:
````
static int driver_tx(struct ieee80211_hw *hw, struct ieee80211_tx_control *control,
                     struct sk_buff *skb) {
    // Handle SKB, map to DMA, and transmit.
}
````

6. SKB Management
Goal: Handle struct sk_buff efficiently during RX/TX.
Driver Tasks:
Allocate SKBs with appropriate headroom for hardware requirements.
Manage DMA mapping for TX and RX paths:
````
dma_map_single(dev, skb->data, skb->len, DMA_TO_DEVICE);
````

Migration Challenges
Differences in API Behavior:
````
cfg80211: Handles high-level operations like scanning and authentication.
mac80211: Delegates control to the driver for finer-grained features.
Hardware Offload Issues:

Older drivers rely heavily on firmware offloading.
Transitioning to mac80211 may expose firmware bugs or missing features.
Performance Impact:

mac80211 can introduce overhead for features like aggregation (A-MPDU).
Regulatory Compliance:

Ensure the driver adheres to regulatory domains set by cfg80211.
Testing and Debugging:

Use tools like iw and aircrack-ng for validating monitor mode and injection.
````

Debugging and Validation
````
Kernel Logs: Use dmesg to inspect errors during mode changes.
Injection Validation:
Test with tools like aireplay-ng and hping3.
Monitor Validation:
Use Wireshark or tcpdump to capture frames.
Ensure proper radiotap decoding.
````

reaveruser, 2025-01-06:
What ever region I use, all channel except 1 to 11 will be `no IR` and upon restart it will reset to 00 world wide.

I tried installing another kali with same results. It has something to do with the firmware not the driver.

I was waiting to test in archer, as it could be the kernel? maybe the 6.12 will fix the issue. I wil try later and reply.

However, using kali linux and all updates are installed, this driver/firmware will cause all channels to be `no IR` except 1 to 11

reaveruser, 2025-01-12:
wireless-regdb is installed by default in kali. I am waiting to test archer to confirm whether it is the distro/kernel or the firmware,

`iw reg set XX` works on kali, except it will change all the channels to `no IR `except 1 to 11. If it was not set it will be as the `iw list` posted before.

As I said, not even channels 36 to 48 are working, which means there is something wrong.

I will return once I test Arch distro

reaveruser, 2025-01-15:
The driver didn't work at all in Arch Linux and Ubuntu 24 (6.8 kernel)!

[image]

Is there any special thing needs to be done? Or am I missing something?

It works with Kali Linux, except of this issue.

a5a5aa555oo, 2025-01-16:
@reaveruser Did you load the module after installing the rtw88-dkms-git package?

`sudo modprobe rtw_8814au`

reaveruser, 2025-01-16:
I didn't unload the driver to load it. But here is the result

[image]

The driver is not working in Kali now. One of the new changes after the commit https://github.com/lwfinger/rtw88/commit/22dc11c8e19f654642ee4ff98adf4a41cd098fe5 cause it was working when tested 2 weeks ago https://github.com/lwfinger/rtw88/issues/279#issuecomment-2569243723

It is only 2 changes tough: https://github.com/lwfinger/rtw88/commit/1cc700d11f6826867972a2b1283bf11c37cdf498 and https://github.com/lwfinger/rtw88/commit/386382ace137f5209e8e83a4cf2e044bf52e8b38

a5a5aa555oo, 2025-01-16:
It looks like the driver is not installed, is there any error message during installation?

reaveruser, 2025-01-16:
@a5a5aa555oo  Found the reason. In VMware choosing USB 2.0 will not detect my wifi adapter. Changing it to USB 3.1 fixes the issue. which is super weird.

Now the WIFI is detected. Will test and return/

reaveruser, 2025-01-16:
@a5a5aa555oo  @dubhater  Finally I addressed the issue and I know 100% the reason why injection for channels above 11 will not work. Tested only in Kali.

It is because I am using 2 Wifi adapters at the same time. Testing while connecting only 1 wifi adapter will work.

```
└─$ sudo airmon-ng

PHY	Interface	Driver		Chipset

phy2	wlan0		rtw_8814au	Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac
phy3	wlan1		rtw_8822bu	TP-Link Archer T4U ver.3

```
```
└─$ sudo dmesg | grep rtw                                                                
[    4.091920] rtw_core: loading out-of-tree module taints kernel.
[    4.091926] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[    4.145029] rtw_8822bu 3-2:1.0: Firmware version 30.20.0, H2C version 14
[    4.739304] usbcore: registered new interface driver rtw_8822bu
[   47.530165] rtw_8822bu 3-2:1.0 wlan0mon: entered promiscuous mode
[   47.530227] rtw_8822bu 3-2:1.0 wlan0mon: entered allmulticast mode
[  113.021461] rtw_8814au 4-1:1.0: Firmware version 33.6.0, H2C version 6
[  113.283339] usbcore: registered new interface driver rtw_8814au
[  154.454070] rtw_8814au 4-1:1.0: Firmware version 33.6.0, H2C version 6
[  157.724353] rtw_8822bu 3-2:1.0: Firmware version 30.20.0, H2C version 14
```
                                                                              
setting `iw reg set xx` doesn't matter. An it will rest upon restart, but it doesn't matter. If I connect only 1 wifi adapter injection will work. Connecting 2 wifi adapters will not work,.

How ever, using the driver https://github.com/morrownr/8814au will not cause issues.

But why this driver will cause conflicts when connecting 2 wifi adapters? This a a strange behavior, which I will call a bug? what do you think?

dubhater, 2025-01-16:
I think you should try the same thing on real hardware.

reaveruser, 2025-01-22:
@dubhater  I am confused! I tested in a new VM machine, as usual kali installed with kernel 6.11.2 and it works fine with both adapters.

Tested again with my old kali and it is working now!

Not sure what happened. I tested many time and I got persisting results of injection not working. Now it is working with persisting results!?

I will close this issue now. I will open it again if I got issues.

Thanks for your help.

closed 2025-01-22 by reaveruser