lwfinger/rtw88 #280: RTL8812AU: left antenna is more useful than right antenna (RX)

wangwillian0, 2024-12-30:
Probably related: https://github.com/aircrack-ng/rtl8812au/issues/280

Tested in both managed and monitor mode with airodump, right antenna is almost useless in RX. I also tried https://github.com/morrownr/8812au-20210820 (without monitor mode because it doesnt support it), and I found a similar asymmetry when scanning APs using nmcli.

Maybe it's a hardware design, in this case sorry for the issue.

dubhater, 2024-12-30:
You cannot receive two spatial streams if one of the two antennas is useless. So, what kind of real RX speed do you see? Measured with iperf3 or speedtest dot net or something like that.

wangwillian0, 2024-12-31:
Sorry for the pseudo tests, it was because I could reach max speed without antenna from the previous setup. I changed places and here are the numbers:

both antennas: ~200mbps
only left antenna: ~110mbps
only right antenna: 65mbps
no antenna: ~25mbps

"almost useless" was probably too strong, but it's still very asymmetric. I tried to use the same antenna for both sides to avoid bias, but the only way to be sure it's not a hardware fault from my side is if another person tested it too.

dubhater, 2024-12-31:
It's probably just how the driver configures the hardware.

briansune, 2024-12-31:
> Sorry for the pseudo tests, it was because I could reach max speed without both antennas in the previous setup. I changed places and here are the numbers on a 5ghz network:
> 
> both antennas: ~200mbps only left antenna: ~110mbps only right antenna: ~65mbps no antenna: ~25mbps
> 
> "almost useless" was probably too strong, but it's still very asymmetric. I tried to use the same antenna for both sides to avoid bias, but the only way to be sure it's not a hardware fault from my side is if another person tested it too.

This is not a proper way to remove RF channel via hardware (unless you very sure the driver killed the RF channel) but still do the below method won't do any harm and make things more sure.

## Go buy a ipex/ULF connector then use 50ohm terminator on each side.

If this is still the same then either the board have a poor design on the transmission line balun etc.
Otherwise it is the driver.

wangwillian0, 2025-01-28:
Just tested with the 50ohm terminator and the results are still the same, this time I got +150mbps using only left antenna and <50mps using only the right one. I didn't get the chance to test it on Windows yet. It would be nice if other users of AWUS036ACH could help confirm this behavior.

briansune, 2025-01-28:
@wangwillian0 

This is very interesting result. You do measure it one time one each side?

What Band did you tested when the result is dropped?

5G or 2G or both band are tested?

This is very important. If I am correct, there is one antenna could implemented a LPF or BPF.

briansune, 2025-01-28:
@wangwillian0 

See this page:
https://www.olliw.eu/2020/wifibroadcast-openhd-dronebridge-5-ghz-wifi-cards/

Very interesting and I do see people mentioning switch hacks?

wangwillian0, 2025-01-30:
I measured a few times again on both 2.4ghz and 5ghz in different networks and not once the right antenna results were better than the left one, there were times where both results were bad and close to each other though.

One interesting test on 5ghz (confirmed using windows too):
only left antenna: 160 mbps download, 130 mbps upload
only right antenna: 135 mbps download, 12 mbps upload
Which makes me wonder if the TX is more impacted than RX (maybe the TX is the only impacted one and the bandwidths suffer because of ACKs?)

> Very interesting and I do see people mentioning switch hacks?

Hm maybe it's referring to taking the packets only from the receptor with best signal? 

Anyways, my main concern in this issue is if this asymmetry affects MIMO performance, if you have a better test for that, please tell me.

briansune, 2025-01-30:
@wangwillian0 

According to the link I enclosed I guess you are talking the left ant as the image ?
[image]

And the right ant as ?
[image]

If the above case is held, then it is most likely the transmission line do have a different layout from first place and the once with the longest even passes DCDC near component w/o shield. No spectrum to see the noise-floor so only guessing.

I read from internet that the power will be cap when the switch is push to NO-H side.
I am not sure this is the case. No power meter confirm the switch setting
But will this disable the gain somehow so the upload is load?

TODO: better to see the switch is pushed to H or not? No too sure

wangwillian0, 2025-01-30:
@briansune yeah, I used the left and right referenced the same as what you showed. And by switch you mean literally the physical switch? I think it's not present on the newer version of the same adapter that uses usb-c:

[image]
(https://www.rcgroups.com/forums/showthread.php?3880253-Ruby-Digital-FPV-System-(version-10-2-radio-links-updates)/page326, https://forum.openhdfpv.org/t/is-it-fake-awus036ach-c-usb-c/384)

I don't want to open the case and the shield off, but I can try to do take a better photo of my card if you think it will add some information

briansune, 2025-01-30:
@wangwillian0 

For the switch <- yes I am talking about the old version bottom right near the USB.

wangwillian0, 2025-01-30:
[image]
[image]

Took photos anyway because I was curious, there was actually no shield, I hope it's useful to someone in the future (also wondering if I can buy metal shields for it).

briansune, 2025-01-30:
@wangwillian0 

Okay now I see the full picture.

Back to the question, my experience tell me that if the chip itself do design a good MIMO such as 2T2R then both channel do have a impact (high chance of performance degradation) when STA+AP etc.

If I am the one trying to test this.
I will immediately remove the DC-block caps on each side and see if this is the case.:

RF ?? Left
[image]
RF ?? Right
[image]

But due to the difficulties I will suggest a easier way:

The only way to make thing very sure is to apply a "attenuator" between the 50R terminator.
This is why doing RF stuff is pricey and even 1 small terminator cost you much when the range is up to 50G and verified.

Could you see the dBm value software on either side of the antenna?

The LNA and PA is doing so strong even the transmission line itself at the entry section is able to pickup the signals.

Try a 60dB to future kill the signal further more as the transmission line is still picking up signal and the Front-end circuit is too good.

wangwillian0, 2025-01-30:
> Could you see the dBm value software on either side of the antenna?

What do you mean by dbm software value? the antennas are linear with 5dbi. The driver reports 22-23dbm (not sure what was the value during the test) with `iw dev`, but from https://github.com/lwfinger/rtw88/issues/111, I'm not sure this value represents the txpower that correctly.

> Try a 60dB to further kill the signal further more as the transmission line is still picking up signal and the Front-end circuit is too good.

Hmm, the idea is to attenuate the signal slowly at a longer physical distance or something like that? I can think about trying the attenuator in the future, I would need to buy it first.

briansune, 2025-01-30:
@wangwillian0 

To increase the measurement better (NOT suggested)
If I have no chance on using soldering to remove DC-Block capacitors.

I will stick aluminum foil all outside the plastic housing then make a good connection to ground of the SMA RF port and the USB port housing. This introduce a faraday cage shield the entire inside and the remain way to insert the signal is the SMA port ONLY.

The cheapest way and the best I can do with the 60dB attenuator is such way.

REDO all measurements.