lwfinger/rtw88 #283: rtw_8822cs / 6.12.1 : crash

chris4git, 2025-01-06:
Hi,
Context : Armbian bookworm v25.2 Linux 6.12.1-edge-rockchip64
After compilation and installation the loading fails badly :

```
Jan 06 12:01:15 X96 kernel: rtw_core: loading out-of-tree module taints kernel.
Jan 06 12:01:16 X96 kernel: rtw_8822cs mmc1:0001:1: Firmware version 9.9.15, H2C version 15
Jan 06 12:01:16 X96 kernel: rtw_8822cs mmc1:0001:1: WOW Firmware version 9.9.4, H2C version 15
Jan 06 12:01:16 X96 systemd[1]: Starting systemd-rfkill.service - Load/Save RF Kill Switch Status...
Jan 06 12:01:16 X96 systemd[1]: Started systemd-rfkill.service - Load/Save RF Kill Switch Status.
Jan 06 12:01:17 X96 kernel: ------------[ cut here ]------------
Jan 06 12:01:17 X96 kernel: Voluntary context switch within RCU read-side critical section!
Jan 06 12:01:17 X96 kernel: WARNING: CPU: 0 PID: 359 at kernel/rcu/tree_plugin.h:331 rcu_note_context_switch+0x404/0x4c4
Jan 06 12:01:17 X96 kernel: Modules linked in: rtw_8822cs(O) rtw_8822c(O) rtw_sdio(O) rtw_core(O) xt_nat xt_tcpudp veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo xt_addrtype nft_compat nf_tables nfnetlink br_netfilter bridge algif_hash algif_skcipher af_alg bnep hci_uart btqca mac80211 btrtl snd_soc_hdmi_codec btintel btbcm libarc4 snd_soc_simple_card snd_soc_spdif_tx bluetooth snd_soc_rockchip_spdif snd_soc_simple_card_utils snd_soc_rockchip_i2s_tdm ir_nec_decoder snd_soc_core snd_compress rc_beelink_gs1 snd_pcm_dmaengine snd_pcm panfrost snd_timer display_connector gpio_ir_recv snd gpu_sched drm_shmem_helper rc_core soundcore dw_hdmi_i2s_audio dw_hdmi_cec cpufreq_dt cfg80211 rfkill zram binfmt_misc dm_mod ip_tables x_tables autofs4 realtek dwmac_rk stmmac_platform stmmac pcs_xpcs [last unloaded: rtw88_core]
Jan 06 12:01:17 X96 kernel: CPU: 0 UID: 0 PID: 359 Comm: wpa_supplicant Tainted: G           O       6.12.1-edge-rockchip64 #1
Jan 06 12:01:17 X96 kernel: Tainted: [O]=OOT_MODULE
Jan 06 12:01:17 X96 kernel: Hardware name: Rockchip RK3566 BOX DEMO Board (DT)
Jan 06 12:01:17 X96 kernel: pstate: 604000c9 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jan 06 12:01:17 X96 kernel: pc : rcu_note_context_switch+0x404/0x4c4
Jan 06 12:01:17 X96 kernel: lr : rcu_note_context_switch+0x404/0x4c4
Jan 06 12:01:17 X96 kernel: sp : ffff8000855f3430
Jan 06 12:01:17 X96 kernel: x29: ffff8000855f3430 x28: ffff0001f738eec0 x27: 0000000000000001
Jan 06 12:01:17 X96 kernel: x26: 0000000000000000 x25: ffff8000855f3790 x24: ffff000106ab5780
Jan 06 12:01:17 X96 kernel: x23: 0000000000000000 x22: ffff000106ab5780 x21: ffff800081e3aba0
Jan 06 12:01:17 X96 kernel: x20: 0000000000000000 x19: ffff0001f738fd40 x18: ffff0001ff014fcc
Jan 06 12:01:17 X96 kernel: x17: ffff800175b23000 x16: ffff800080000000 x15: fffffffffffcbb9f
Jan 06 12:01:17 X96 kernel: x14: 0000000000000000 x13: 000000000000028a x12: 00000000ffffffea
Jan 06 12:01:17 X96 kernel: x11: 0000000000000001 x10: 0000000000000001 x9 : ffff800081c88d70
Jan 06 12:01:17 X96 kernel: x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
Jan 06 12:01:17 X96 kernel: x5 : ffff0001f737b488 x4 : 0000000000000000 x3 : ffff800175b23000
Jan 06 12:01:17 X96 kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff000106ab5780
Jan 06 12:01:17 X96 kernel: Call trace:
Jan 06 12:01:17 X96 kernel:  rcu_note_context_switch+0x404/0x4c4
Jan 06 12:01:17 X96 kernel:  __schedule+0xa0/0xadc
Jan 06 12:01:17 X96 kernel:  schedule+0x34/0x134
Jan 06 12:01:17 X96 kernel:  schedule_timeout+0x174/0x184
Jan 06 12:01:17 X96 kernel:  wait_for_completion+0x74/0x158
Jan 06 12:01:17 X96 kernel:  mmc_wait_for_req_done+0x2c/0xf8
Jan 06 12:01:17 X96 kernel:  mmc_wait_for_req+0xb0/0x100
Jan 06 12:01:17 X96 kernel:  mmc_io_rw_extended+0x1d0/0x2c0
Jan 06 12:01:17 X96 kernel:  sdio_io_rw_ext_helper+0x190/0x234
Jan 06 12:01:17 X96 kernel:  sdio_readl+0x30/0x5c
Jan 06 12:01:17 X96 kernel:  rtw_sdio_readl+0xbc/0xd4 [rtw_sdio]
Jan 06 12:01:17 X96 kernel:  rtw_sdio_read32+0xb8/0x204 [rtw_sdio]
Jan 06 12:01:17 X96 kernel:  rtw8822c_led_set+0x34/0x78 [rtw_8822c]
Jan 06 12:01:17 X96 kernel:  led_blink_setup+0x12c/0x138
Jan 06 12:01:17 X96 kernel:  led_blink_set+0x68/0xdc
Jan 06 12:01:17 X96 kernel:  led_blink_set_nosleep+0x60/0x88
Jan 06 12:01:17 X96 kernel:  led_trigger_blink+0x48/0x74
Jan 06 12:01:17 X96 kernel:  tpt_trig_timer+0xc8/0xe8 [mac80211]
Jan 06 12:01:17 X96 kernel:  ieee80211_mod_tpt_led_trig+0xb8/0xe4 [mac80211]
Jan 06 12:01:17 X96 kernel:  ieee80211_do_open+0x3f8/0x778 [mac80211]
Jan 06 12:01:17 X96 kernel:  ieee80211_open+0x5c/0x98 [mac80211]
Jan 06 12:01:17 X96 kernel:  __dev_open+0xfc/0x204
Jan 06 12:01:17 X96 kernel:  __dev_change_flags+0x190/0x208
Jan 06 12:01:17 X96 kernel:  dev_change_flags+0x24/0x6c
Jan 06 12:01:17 X96 kernel:  devinet_ioctl+0x550/0x5f8
Jan 06 12:01:17 X96 kernel:  inet_ioctl+0x1dc/0x1ec
Jan 06 12:01:17 X96 kernel:  sock_do_ioctl+0x48/0xf4
Jan 06 12:01:17 X96 kernel:  sock_ioctl+0x130/0x368
Jan 06 12:01:17 X96 kernel:  __arm64_sys_ioctl+0xa8/0xec
Jan 06 12:01:17 X96 kernel:  invoke_syscall+0x48/0x110
Jan 06 12:01:17 X96 kernel:  el0_svc_common.constprop.0+0x40/0xe8
Jan 06 12:01:17 X96 kernel:  do_el0_svc+0x20/0x2c
Jan 06 12:01:17 X96 kernel:  el0_svc+0x30/0xd0
Jan 06 12:01:17 X96 kernel:  el0t_64_sync_handler+0x13c/0x158
Jan 06 12:01:17 X96 kernel:  el0t_64_sync+0x190/0x194
Jan 06 12:01:17 X96 kernel: ---[ end trace 0000000000000000 ]---
Jan 06 12:01:17 X96 systemd-networkd[274]: wlan0: Link UP
Jan 06 12:01:17 X96 NetworkManager[774]: <info>  [1736161277.6999] rfkill2: found Wi-Fi radio killswitch (at /sys/devices/platform/fe2c0000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy1/rfkill2) (driver rtw_8822cs)
[... 111 lines cut ...]
Jan 06 12:01:18 X96 kernel:  call_timer_fn+0x1a0/0x1a8
Jan 06 12:01:18 X96 kernel:  expire_timers+0xc4/0x170
Jan 06 12:01:18 X96 kernel:  timer_expire_remote+0x124/0x144
Jan 06 12:01:18 X96 kernel:  tmigr_handle_remote+0x3c8/0x498
Jan 06 12:01:18 X96 kernel:  run_timer_softirq+0x38/0x44
Jan 06 12:01:18 X96 kernel:  handle_softirqs+0x108/0x338
Jan 06 12:01:18 X96 kernel:  __do_softirq+0x14/0x20
Jan 06 12:01:18 X96 kernel:  ____do_softirq+0x10/0x1c
Jan 06 12:01:18 X96 kernel:  call_on_irq_stack+0x24/0x4c
Jan 06 12:01:18 X96 kernel:  do_softirq_own_stack+0x1c/0x2c
Jan 06 12:01:18 X96 kernel:  irq_exit_rcu+0x80/0xb4
Jan 06 12:01:18 X96 kernel:  el1_interrupt+0x38/0x68
Jan 06 12:01:18 X96 kernel:  el1h_64_irq_handler+0x18/0x24
Jan 06 12:01:18 X96 kernel:  el1h_64_irq+0x64/0x68
Jan 06 12:01:18 X96 kernel:  default_idle_call+0x3c/0x144
Jan 06 12:01:18 X96 kernel:  do_idle+0x208/0x264
Jan 06 12:01:18 X96 kernel:  cpu_startup_entry+0x34/0x3c
Jan 06 12:01:18 X96 kernel:  secondary_start_kernel+0x124/0x144
Jan 06 12:01:18 X96 kernel:  __secondary_switched+0xb8/0xbc
Jan 06 12:01:18 X96 kernel: ---[ end trace 0000000000000000 ]---
Jan 06 12:01:21 X96 systemd[1]: systemd-rfkill.service: Deactivated successfully.

```

Any hints ?

dubhater, 2025-01-06:
Yes, I see the problem. Try it now.

briansune, 2025-01-17:
@chris4git 

I know this issue had closed.
And just want some info to debug or clear out some of SDIO devices.

May I ask what SDIO profile you are using on the RTL8822CS?
What PCBA or module are you using?
What OS you are deploying the RTW88?

Thank you for those info.

chris4git, 2025-01-18:
Hi,
happy to help if I can.

> May I ask what SDIO profile you are using on the RTL8822CS?
> What PCBA or module are you using?

Where can I find these informations ?

> What OS you are deploying the RTW88?
 
Debian stable (bookworm) v24.11.1 for Station M2 running Armbian Linux 6.12.9-current-rockchip64

Chris

briansune, 2025-01-18:
@chris4git 

Appreciate the helps.

> Where can I find these informations ?

Three things are needed.

1) I would need the simple image of the board aka the section of that RTL8822CS located.
2) Run and modify the follow bash script, change the location of the mmc# accordingly.

> cat check.sh
> chmod +x ./check.sh
> sudo ./check.sh

```
#!/bin/bash

cat /sys/class/mmc_host/mmc1/mmc1\:0001/uevent
cat /sys/kernel/debug/mmc1/ios
```

3) The device tree of the sdhc section that deploying the SDIO WIFI.

```
&sdhci1 {
...
...
};
```

These three things are good enough to see the profile.

chris4git, 2025-01-18:
> I would need the simple image of the board aka the section of that RTL8822CS located.

the box is well closed, there are no screws so I'm afraid of breaking something when opening it

> cat /sys/class/mmc_host/mmc1/mmc1\:0001/uevent
> cat /sys/kernel/debug/mmc1/ios

root@X96:~# cat /sys/class/mmc_host/mmc1/mmc1\:0001/uevent
MMC_TYPE=SDIO
SDIO_ID=024C:C822
SDIO_REVISION=0.0

root@X96:~# cat /sys/kernel/debug/mmc1/ios
clock:		50000000 Hz
actual clock:	50000000 Hz
vdd:		21 (3.3 ~ 3.4 V)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	2 (on)
bus width:	2 (4 bits)
timing spec:	2 (sd high-speed)
signal voltage:	1 (1.80 V)
driver type:	0 (driver type B)

>     The device tree of the sdhc section that deploying the SDIO WIFI.
there is no sdhc section in the dtb.
Here is sdmmc1 :

> 		sdmmc1 {
> 
> 			sdmmc1-bus4 {
> 				phandle = <0x5c>;
> 				rockchip,pins = <0x02 0x03 0x01 0xb3 0x02 0x04 0x01 0xb3 0x02 0x05 0x01 0xb3 0x02 0x06 0x01 0xb3>;
> 			};
> 
> 			sdmmc1-clk {
> 				phandle = <0x5e>;
> 				rockchip,pins = <0x02 0x08 0x01 0xb3>;
> 			};
> 
> 			sdmmc1-cmd {
> 				phandle = <0x5d>;
> 				rockchip,pins = <0x02 0x07 0x01 0xb3>;
> 			};
> 		};
> 

Is it what you expect ?

Chris

chris4git, 2025-01-18:
Sorry for the image.

Nothing found in the dts.

Here is the dts used :

[rk3566-box-demo_x96x6_24-custom.dts.txt](https://github.com/user-attachments/files/18464138/rk3566-box-demo_x96x6_24-custom.dts.txt)

Chris

briansune, 2025-01-18:
@chris4git 

Okay I guess these are all info that can gather. 
Appreciate all those info.

So instead of known issue 2.4G band.
You don't had any trouble on the Rockchip when RTW88 with 8822CS? 

What actual rockchip64 you are using?

briansune, 2025-01-18:
@chris4git 

Appreciate all the helps.

Model # rk3566
And that's are all info that can gather.

So do you had any trouble on ARM Cortex-A55 with kernel 6.12.1 on 8822CS with RTW88 driver?
Exclude the known issue of the 2.4G band issue.

chris4git, 2025-01-18:
This box is labeled X96-X6 and has a RK3566.
I have still an issue with the rtw88 driver : I can't get working the AP mode (the ssid is published but the dialog with the station don't go to the end)...

briansune, 2025-01-18:
@chris4git 

Okay, So I guess there are no issue on STA WIFI operations.

dubhater, 2025-01-18:
@chris4git Does this patch help with the AP mode?

https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/net/wireless/realtek?h=next-20250117&id=fb2fcfbe5eef9ae26b0425978435ae1308951e51

briansune, 2025-01-18:
@dubhater 

According to:

This is the equivalent of commit 28818b4d871b ("wifi: rtw88: usb: Fix
disconnection after beacon loss") for SDIO chips.
Tested on Pinephone (RTL8723CS), random disconnections became rare,
instead of a frequent nuisance.

This seems like resolving the issue I am encountering on RTL8822BS.
Once it connected it will able to link up for few seconds then it immediately disconnect.

chris4git, 2025-01-18:
@dubhater 

Unfortunately I'm not at home, I'll have to wait a few days before I can test it.

dubhater, 2025-01-18:
@chris4git Well, I applied it to this repository so you just have to pull.

chris4git, 2025-01-21:
@dubhater 
the patch don't help with AP mode : the station sees the ssid (test) but fails to authenticate (look at  22:38:57) :

> janv. 21 22:38:47 Latitude-7480 NetworkManager[1027]: <info>  [1737495527.6035] device (wlp2s0): supplicant interface state: disconnected -> authenticating
> janv. 21 22:38:47 Latitude-7480 NetworkManager[1027]: <info>  [1737495527.6036] device (p2p-dev-wlp2s0): supplicant management interface state: disconnected -> authenticating
> janv. 21 22:38:47 Latitude-7480 kernel: wlp2s0: send auth to 4c:24:ce:cd:dd:38 (try 1/3)
> janv. 21 22:38:47 Latitude-7480 wpa_supplicant[1126]: wlp2s0: Trying to associate with 4c:24:ce:cd:dd:38 (SSID='test' freq=2412 MHz)
> janv. 21 22:38:47 Latitude-7480 NetworkManager[1027]: <info>  [1737495527.6087] device (wlp2s0): supplicant interface state: authenticating -> associating
> janv. 21 22:38:47 Latitude-7480 NetworkManager[1027]: <info>  [1737495527.6087] device (p2p-dev-wlp2s0): supplicant management interface state: authenticating -> associating
> janv. 21 22:38:47 Latitude-7480 kernel: wlp2s0: authenticated
> janv. 21 22:38:47 Latitude-7480 kernel: iwlwifi 0000:02:00.0 wlp2s0: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
> janv. 21 22:38:47 Latitude-7480 kernel: wlp2s0: associate with 4c:24:ce:cd:dd:38 (try 1/3)
> janv. 21 22:38:47 Latitude-7480 kernel: wlp2s0: RX AssocResp from 4c:24:ce:cd:dd:38 (capab=0x411 status=0 aid=1)
> janv. 21 22:38:47 Latitude-7480 wpa_supplicant[1126]: wlp2s0: Associated with 4c:24:ce:cd:dd:38
> janv. 21 22:38:47 Latitude-7480 wpa_supplicant[1126]: wlp2s0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
> janv. 21 22:38:47 Latitude-7480 kernel: wlp2s0: associated
> janv. 21 22:38:47 Latitude-7480 NetworkManager[1027]: <info>  [1737495527.6250] device (wlp2s0): supplicant interface state: associating -> associated
> janv. 21 22:38:47 Latitude-7480 NetworkManager[1027]: <info>  [1737495527.6251] device (p2p-dev-wlp2s0): supplicant management interface state: associating -> associated
> 
> 
> janv. 21 22:38:57 Latitude-7480 wpa_supplicant[1126]: **wlp2s0: Authentication with 4c:24:ce:cd:dd:38 timed out.**
> janv. 21 22:38:57 Latitude-7480 kernel: wlp2s0: deauthenticating from 4c:24:ce:cd:dd:38 by local choice (Reason: 3=DEAUTH_LEAVING)
> janv. 21 22:38:57 Latitude-7480 wpa_supplicant[1126]: wlp2s0: CTRL-EVENT-DISCONNECTED bssid=4c:24:ce:cd:dd:38 reason=3 locally_generated=1
> janv. 21 22:38:57 Latitude-7480 wpa_supplicant[1126]: BSSID 4c:24:ce:cd:dd:38 ignore list count incremented to 2, ignoring for 10 seconds
> janv. 21 22:38:57 Latitude-7480 wpa_supplicant[1126]: wlp2s0: CTRL-EVENT-DSCP-POLICY clear_all
> janv. 21 22:38:57 Latitude-7480 NetworkManager[1027]: <info>  [1737495537.6914] device (wlp2s0): supplicant interface state: associated -> disconnected
> janv. 21 22:38:57 Latitude-7480 NetworkManager[1027]: <info>  [1737495537.6916] device (p2p-dev-wlp2s0): supplicant management interface state: associated -> disconnected
> janv. 21 22:38:57 Latitude-7480 NetworkManager[1027]: <info>  [1737495537.7904] device (wlp2s0): supplicant interface state: disconnected -> scanning
> janv. 21 22:38:57 Latitude-7480 NetworkManager[1027]: <info>  [1737495537.7906] device (p2p-dev-wlp2s0): supplicant management interface state: disconnected -> scanning
>

briansune, 2025-01-21:
@chris4git 

Same situation here AP on RTW88 not working, but vendor drivers can run no-issue.

https://github.com/briansune/rtw88-hw#sdio-devices-investigation-on-rtw88

chris4git, 2025-01-22:
@briansune 
How to help debug the issue ?

briansune, 2025-01-22:
@chris4git 

If you want a double checking on vendors driver give me an EMAIL I forward it to you.

I am deeply sorry that I only know hardware and able to provide hardware testing services. 

Just guide me on software side.

I have ZERO clue aka clueless on all kernel stuffs.

The headache I guess now is on the RTL8822BS modules. RTW88 have some issue on 5G STA.

I haven't tested other brand RTL8822CS.

Wish the all luck and resolve issue ASAP.