lwfinger/rtw88 #284: Clarity to identify which one to use

sanjarbek-ismatov, 2025-01-07:
Hi, I use realtek 8821ce wifi adapter on my AMD HP laptop.
`
lsmod | grep rtw                                                                             
rtw88_8821ce           12288  0
rtw88_8821c            98304  1 rtw88_8821ce
rtw88_pci              40960  1 rtw88_8821ce
rtw88_core            331776  2 rtw88_pci,rtw88_8821c
mac80211             1638400  2 rtw88_pci,rtw88_core
cfg80211             1396736  2 rtw88_core,mac80211

`
 I used linux before too, but Wifi was problematic. Then I found this dkms module: https://github.com/tomaspinho/rtl8821ce . I started using it. It worked normally, then it stopped recieveing updates, at the same time WiFi was fixed after the kernel 6.8.2. Now Wifi is working almost properly, but there are some signal sharp fluctations sometimes. Here is the fix: https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/?id=e1dfa21427baeb813f9a2f9ceab6b7d32c3ca425
The question is: where is the difference between rtw88 modules in the kernel, lwfinger/rtw88 and tomaspinho/rtl8821ce? Which one works better?

dubhater, 2025-01-07:
This repository has the same driver that is in your kernel, but it's newer.

https://github.com/tomaspinho/rtl8821ce is a completely different driver. Sometimes it's better, sometimes it's worse.

sanjarbek-ismatov, 2025-01-08:
> but it's newer.

Does newer mean better? Why is it seperated from the kernel source? Is it worth blacklisting the kernel modules and installing? 

> https://github.com/tomaspinho/rtl8821ce

It is not maintained any more. It was really helpful
I use Arch with KDE, the changes in the wifi signals was annoying, but I changed icons to papirus, not it's less-distracting.

sanjarbek-ismatov, 2025-01-08:
> This repository has the same driver that is in your kernel, but it's newer.
> 
> https://github.com/tomaspinho/rtl8821ce is a completely different driver. Sometimes it's better, sometimes it's worse.

Hey, you fixed my problem in kernel 6.8.x, right? I am fond of creating drivers, OS components. I am currently learning C, can you show me guidance? I mean, I do not know what to learn after C. Also, How do you know how to build or fix drivers for realtek, including mine? Your thoughts or suggestions would be really helpful!

dubhater, 2025-03-17:
> Also, How do you know how to build or fix drivers for realtek, including mine? Your thoughts or suggestions would be really helpful!

@sanjarbek-ismatov I used to have a laptop with RTL8821CE wifi. It had the same problem yours did. At the time I didn't know anything about wifi. Looking at the rtw88 code was pointless, it was all Greek to me. I also had a USB wifi adapter with the RTL8188FU chip which wasn't supported in the kernel.  I knew the rtl8xxxu driver in the kernel did support a few similar chips, so I decided to copy the most important parts from the RTL8188FU driver into rtl8xxxu and try to make it work. I thought that would be a good way to learn about the Realtek wifi chips and their drivers, while doing something fun and useful. It took several more unsupported chips, but eventually I learned enough to spot the problem with the RTL8821CE.

sanjarbek-ismatov, 2025-03-28:
> [@sanjarbek-ismatov](https://github.com/sanjarbek-ismatov) I used to have a laptop with RTL8821CE wifi. It had the same problem yours did. At the time I didn't know anything about wifi. Looking at [...]

Thanks for this suggestion. Did you know C well at that time?

dubhater, 2025-03-28:
I think so, yes.