lwfinger/rtw88 #285: rtw_8814au: error

avonder, 2025-01-07:
```
$ lsusb | grep 8814
Bus 004 Device 003: ID 2357:0106 TP-Link Archer T9UH v1 [Realtek RTL8814AU]
```

Repeats 160 times during connect:
```
Jan 07 22:05:53 avdr kernel: ------------[ cut here ]------------
Jan 07 22:05:53 avdr kernel: invalid arguments, band=1, bw=2, path=0, rate=44, ch=114
Jan 07 22:05:53 avdr kernel: WARNING: CPU: 2 PID: 1181 at /var/lib/dkms/rtw88/r588.1cc700d/build/phy.c:2195 rtw_get_tx_power_params+0x573/0x940 [rtw_core]
Jan 07 22:05:53 avdr kernel: Modules linked in: rtw_8814au(OE) rtw_8814a(OE) rtw_usb(OE) rtw_88xxa(OE) tun ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq cmac algif_hash algif_skcipher af_alg bnep usblp nls_iso8859_1 vfat fat amdgpu rt2800usb rt2x00usb rt2800lib snd_hda_codec_realtek rtw_core(OE) rt2x00lib snd_hda_codec_generic amd_atl intel_rapl_msr snd_hda_scodec_component snd_hda_codec_hdmi intel_rapl_common snd_hda_intel snd_intel_dspcfg amdxcp mac80211 btusb snd_intel_sdw_acpi eeepc_wmi drm_exec snd_virtuoso btrtl asus_wmi gpu_sched snd_hda_codec snd_oxygen_lib btintel drm_buddy platform_profile snd_mpu401_uart btbcm drm_suballoc_helper libarc4 snd_hda_core i8042 snd_rawmidi btmtk drm_ttm_helper pktcdvd kvm_amd ttm snd_hwdep snd_seq_device sparse_keymap ee1004 serio bluetooth mxm_wmi wmi_bmof cfg80211 igb snd_pcm drm_display_helper i2c_algo_bit joydev mousedev rfkill kvm snd_timer ptp cec sp5100_tco pps_core video rapl snd dca i2c_piix4 pcspkr soundcore i2c_smbus k10temp gpio_amdpt wmi gpio_generic mac_hid vboxnetflt(OE)
Jan 07 22:05:53 avdr kernel:  vboxnetadp(OE) vboxdrv(OE) pkcs8_key_parser sg crypto_user loop nfnetlink ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys trusted asn1_encoder tee sr_mod cdrom hid_logitech_hidpp usb_storage hid_logitech_dj hid_generic usbhid crct10dif_pclmul crc32_pclmul crc32c_intel dm_mod polyval_clmulni polyval_generic ghash_clmulni_intel sha512_ssse3 sha256_ssse3 sha1_ssse3 aesni_intel gf128mul nvme crypto_simd cryptd nvme_core ccp nvme_auth
Jan 07 22:05:53 avdr kernel: Unloaded tainted modules: rtw_usb(OE):2 rtw_8812a(OE):1 rtw_8812au(OE):1 rtw_8814a(OE):3 rtw_8814au(OE):3 [last unloaded: rtw_8814a(OE)]
Jan 07 22:05:53 avdr kernel: CPU: 2 UID: 0 PID: 1181 Comm: wpa_supplicant Tainted: G        W  OE      6.12.7-zen1-1-zen #1 b2ee061ecc6b06c215bd2e68132a2cd24af28da3
Jan 07 22:05:53 avdr kernel: Tainted: [W]=WARN, [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Jan 07 22:05:53 avdr kernel: Hardware name: System manufacturer System Product Name/PRIME X370-PRO, BIOS 6203 07/27/2023
Jan 07 22:05:53 avdr kernel: RIP: 0010:rtw_get_tx_power_params+0x573/0x940 [rtw_core]
Jan 07 22:05:53 avdr kernel: Code: 41 5f e9 20 b7 64 e3 44 8b 5c 24 28 44 8b 44 24 20 8b 4c 24 1c 89 f2 45 0f b6 cb 44 89 fe 48 c7 c7 20 0f 51 c1 e8 bd 2d 5f e2 <0f> 0b 48 8b 43 38 0f b6 68 45 48 8b 04 24 40 88 68 02 41 80 fc 03
Jan 07 22:05:53 avdr kernel: RSP: 0018:ffffa9d040e7f3a0 EFLAGS: 00010282
Jan 07 22:05:53 avdr kernel: RAX: 0000000000000000 RBX: ffff8a24b9ef2000 RCX: 0000000000000027
Jan 07 22:05:53 avdr kernel: RDX: ffff8a28ced218c8 RSI: 0000000000000001 RDI: ffff8a28ced218c0
Jan 07 22:05:53 avdr kernel: RBP: 0000000000000004 R08: ffffffffa5c751d8 R09: 000000010001a445
Jan 07 22:05:53 avdr kernel: R10: 0000000000000003 R11: 0000000000000003 R12: 000000000000002c
Jan 07 22:05:53 avdr kernel: R13: 0000000000000000 R14: 0000000000000004 R15: 0000000000000001
Jan 07 22:05:53 avdr kernel: FS:  000071201f918500(0000) GS:ffff8a28ced00000(0000) knlGS:0000000000000000
Jan 07 22:05:53 avdr kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 07 22:05:53 avdr kernel: CR2: 00007e7c060995f8 CR3: 000000014a8d8000 CR4: 0000000000350ef0
Jan 07 22:05:53 avdr kernel: Call Trace:
Jan 07 22:05:53 avdr kernel:  <TASK>
Jan 07 22:05:53 avdr kernel:  ? rtw_get_tx_power_params+0x573/0x940 [rtw_core 32d6d0bd89db97ea26d67d1e17fdf532edc78efa]
Jan 07 22:05:53 avdr kernel:  ? __warn.cold+0x93/0xed
Jan 07 22:05:53 avdr kernel:  ? rtw_get_tx_power_params+0x573/0x940 [rtw_core 32d6d0bd89db97ea26d67d1e17fdf532edc78efa]
Jan 07 22:05:53 avdr kernel:  ? report_bug+0xe7/0x210
Jan 07 22:05:53 avdr kernel:  ? handle_bug+0x58/0x90
Jan 07 22:05:53 avdr kernel:  ? exc_invalid_op+0x19/0xc0
Jan 07 22:05:53 avdr kernel:  ? asm_exc_invalid_op+0x1a/0x20
Jan 07 22:05:53 avdr kernel:  ? rtw_get_tx_power_params+0x573/0x940 [rtw_core 32d6d0bd89db97ea26d67d1e17fdf532edc78efa]
Jan 07 22:05:53 avdr kernel:  rtw_phy_get_tx_power_index+0x44/0x180 [rtw_core 32d6d0bd89db97ea26d67d1e17fdf532edc78efa]
Jan 07 22:05:53 avdr kernel:  rtw_phy_set_tx_power_level+0xdc/0x140 [rtw_core 32d6d0bd89db97ea26d67d1e17fdf532edc78efa]
Jan 07 22:05:53 avdr kernel:  rtw_set_channel+0xb8/0x130 [rtw_core 32d6d0bd89db97ea26d67d1e17fdf532edc78efa]
Jan 07 22:05:53 avdr kernel:  rtw_ops_config+0x81/0xc0 [rtw_core 32d6d0bd89db97ea26d67d1e17fdf532edc78efa]
Jan 07 22:05:53 avdr kernel:  ieee80211_emulate_add_chanctx+0x51/0xf0 [mac80211 1e989ccf5f7f4f524aa6634532be430758f8e7a8]
Jan 07 22:05:53 avdr kernel:  ieee80211_add_chanctx+0x57/0x130 [mac80211 1e989ccf5f7f4f524aa6634532be430758f8e7a8]
Jan 07 22:05:53 avdr kernel:  ieee80211_new_chanctx+0xf2/0x180 [mac80211 1e989ccf5f7f4f524aa6634532be430758f8e7a8]
Jan 07 22:05:53 avdr kernel:  _ieee80211_link_use_channel+0x33c/0x8c0 [mac80211 1e989ccf5f7f4f524aa6634532be430758f8e7a8]
Jan 07 22:05:53 avdr kernel:  ? srso_return_thunk+0x5/0x5f
Jan 07 22:05:53 avdr kernel:  ? cfg80211_find_elem_match+0x8a/0xb0 [cfg80211 6fe44167c41e3e5e4cd6ec1c4ee07e8ae9e1c98b]
Jan 07 22:05:53 avdr kernel:  ieee80211_prep_channel.isra.0+0x375/0x1790 [mac80211 1e989ccf5f7f4f524aa6634532be430758f8e7a8]
Jan 07 22:05:53 avdr kernel:  ieee80211_prep_connection+0x1d5/0x4a0 [mac80211 1e989ccf5f7f4f524aa6634532be430758f8e7a8]
Jan 07 22:05:53 avdr kernel:  ieee80211_mgd_auth+0x347/0x540 [mac80211 1e989ccf5f7f4f524aa6634532be430758f8e7a8]
Jan 07 22:05:53 avdr kernel:  ? regulatory_exit+0xa0/0x220 [cfg80211 6fe44167c41e3e5e4cd6ec1c4ee07e8ae9e1c98b]
Jan 07 22:05:53 avdr kernel:  cfg80211_mlme_auth+0xb7/0x1b0 [cfg80211 6fe44167c41e3e5e4cd6ec1c4ee07e8ae9e1c98b]
Jan 07 22:05:53 avdr kernel:  nl80211_authenticate+0x386/0x400 [cfg80211 6fe44167c41e3e5e4cd6ec1c4ee07e8ae9e1c98b]
Jan 07 22:05:53 avdr kernel:  genl_rcv_msg+0x2f1/0x460
Jan 07 22:05:53 avdr kernel:  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211 6fe44167c41e3e5e4cd6ec1c4ee07e8ae9e1c98b]
Jan 07 22:05:53 avdr kernel:  ? __pfx_nl80211_authenticate+0x10/0x10 [cfg80211 6fe44167c41e3e5e4cd6ec1c4ee07e8ae9e1c98b]
Jan 07 22:05:53 avdr kernel:  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211 6fe44167c41e3e5e4cd6ec1c4ee07e8ae9e1c98b]
Jan 07 22:05:53 avdr kernel:  ? srso_return_thunk+0x5/0x5f
Jan 07 22:05:53 avdr kernel:  ? __check_object_size+0x50/0x210
Jan 07 22:05:53 avdr kernel:  ? srso_return_thunk+0x5/0x5f
Jan 07 22:05:53 avdr kernel:  ? srso_return_thunk+0x5/0x5f
Jan 07 22:05:53 avdr kernel:  ? sk_getsockopt+0x6ce/0x12b0
Jan 07 22:05:53 avdr kernel:  ? __pfx_genl_rcv_msg+0x10/0x10
Jan 07 22:05:53 avdr kernel:  netlink_rcv_skb+0x53/0x100
Jan 07 22:05:53 avdr kernel:  genl_rcv+0x28/0x40
Jan 07 22:05:53 avdr kernel:  netlink_unicast+0x308/0x4f0
Jan 07 22:05:53 avdr kernel:  netlink_sendmsg+0x1fe/0x440
Jan 07 22:05:53 avdr kernel:  ____sys_sendmsg+0x39f/0x3d0
Jan 07 22:05:53 avdr kernel:  ? srso_return_thunk+0x5/0x5f
Jan 07 22:05:53 avdr kernel:  __sys_sendmsg+0x1d9/0x2b0
[... 80 lines cut ...]
Jan 07 22:07:42 avdr wpa_supplicant[1181]: wlan0: CTRL-EVENT-DISCONNECTED bssid=cc:2d:e0:ab:cd:ef reason=4 locally_generated=1
Jan 07 22:07:42 avdr wpa_supplicant[1181]: wlan0: Added BSSID cc:2d:e0:ab:cd:ef into ignore list, ignoring for 10 seconds
Jan 07 22:07:42 avdr NetworkManager[1080]: <info>  [1736251662.6471] device (wlan0): supplicant interface state: completed -> disconnected
Jan 07 22:07:42 avdr NetworkManager[1080]: <info>  [1736251662.6471] device (p2p-dev-wlan0): supplicant management interface state: completed -> disconnected
Jan 07 22:07:44 avdr ntpd[1191]: Deleting 21 wlan0, [192.168.88.10]:123, stats: received=0, sent=0, dropped=0, active_time=107 secs
Jan 07 22:07:46 avdr NetworkManager[1080]: <info>  [1736251666.0874] device (wlan0): supplicant interface state: disconnected -> scanning
Jan 07 22:07:46 avdr NetworkManager[1080]: <info>  [1736251666.0874] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Jan 07 22:07:50 avdr wpa_supplicant[1181]: wlan0: Removed BSSID cc:2d:e0:ab:cd:ef from ignore list (clear)
[... cut ...]
Jan 07 22:07:57 avdr NetworkManager[1080]: <warn>  [1736251677.8513] device (wlan0): link timed out.
Jan 07 22:07:57 avdr NetworkManager[1080]: <info>  [1736251677.8516] device (wlan0): state change: activated -> failed (reason 'ssid-not-found', managed-type: 'full')
Jan 07 22:07:57 avdr NetworkManager[1080]: <warn>  [1736251677.8519] device (wlan0): Activation: failed for connection '5G'
Jan 07 22:07:57 avdr NetworkManager[1080]: <info>  [1736251677.8523] device (wlan0): state change: failed -> disconnected (reason 'none', managed-type: 'full')
```

dubhater, 2025-01-07:
This is interesting. I think your AP is configured incorrectly. What channel number and width did you choose?

avonder, 2025-01-07:
It isn't. It also appears to not be uptime, but constant load. The adapter behaves like it's overheating, dying earlier after repeated testing; however, it's hitting the same temps and speeds under load on Windows with no issue.

dubhater, 2025-01-07:
I hope you like trying random patches. :)

Does this help?

```diff
diff --git a/phy.c b/phy.c
index a4c51cd..1ba9c93 100644
--- a/phy.c
+++ b/phy.c
@@ -797,6 +797,7 @@ static void rtw_phy_ra_track(struct rtw_dev *rtwdev)
 
 void rtw_phy_dynamic_mechanism(struct rtw_dev *rtwdev)
 {
+	return;
 	/* for further calculation */
 	rtw_phy_statistics(rtwdev);
 	rtw_phy_dig(rtwdev);
```

avonder, 2025-01-07:
It doesn't.

dubhater, 2025-01-07:
I pushed another patch. Pull and try again?

dubhater, 2025-01-07:
Can you show the output of ``lspci -nn``, please?

avonder, 2025-01-07:
Alright, that didn't do it.

```
$ lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Root Complex [1022:1480]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse IOMMU [1022:1481]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]
00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:05.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 61)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 0 [1022:1440]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 1 [1022:1441]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 2 [1022:1442]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 3 [1022:1443]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 4 [1022:1444]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 5 [1022:1445]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 6 [1022:1446]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 7 [1022:1447]
01:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983 [144d:a808]
02:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] X370 Series Chipset USB 3.1 xHCI Controller [1022:43b9] (rev 02)
02:00.1 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] X370 Series Chipset SATA Controller [1022:43b5] (rev 02)
02:00.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] X370 Series Chipset PCIe Upstream Port [1022:43b0] (rev 02)
03:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
03:02.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
03:03.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
03:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
03:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
03:07.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
06:00.0 PCI bridge [0604]: PLX Technology, Inc. PEX8112 x1 Lane PCI Express-to-PCI Bridge [10b5:8112] (rev aa)
07:04.0 Multimedia audio controller [0401]: C-Media Electronics Inc CMI8788 [Oxygen HD Audio] [13f6:8788]
08:00.0 USB controller [0c03]: ASMedia Technology Inc. ASM1143 USB 3.1 Host Controller [1b21:1343]
09:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
0b:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge [1022:1470] (rev c1)
0c:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge [1022:1471]
0d:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XL/XT [Radeon RX Vega 56/64] [1002:687f] (rev c1)
0d:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 HDMI Audio [Radeon Vega 56/64] [1002:aaf8]
0e:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Function [1022:148a]
0f:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Reserved SPP [1022:1485]
0f:00.1 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Cryptographic Coprocessor PSPCPP [1022:1486]
0f:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller [1022:149c]
0f:00.4 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse HD Audio Controller [1022:1487]
```

dubhater, 2025-01-07:
And if you plug it into the AMD controller?

It's a different model, but... https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1749961?comments=all

avonder, 2025-01-07:
All right, that did it. Thanks.

closed 2025-01-07 by avonder

dubhater, 2025-01-07:
That's good, but have you tried another driver, like this one? https://github.com/morrownr/8814au/

avonder, 2025-01-07:
It works, and on the ASMedia as well. I tried this one because it was giving me bad bandwidth, but that appears to be because of the channel thing.

avonder, 2025-01-07:
Throw any patch ideas you like.

dubhater, 2025-03-06:
> Jan 07 22:07:38 avdr kernel: xhci_hcd 0000:08:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
Jan 07 22:07:38 avdr kernel: xhci_hcd 0000:08:00.0: Looking for event-dma 00000000ffe3e420 trb-start 00000000ffe3e400 trb-end 00000000ffe3e400 seg-start 00000000ffe3e000 seg-end 00000000ffe3eff0
Jan 07 22:07:38 avdr kernel: xhci_hcd 0000:08:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
Jan 07 22:07:38 avdr kernel: xhci_hcd 0000:08:00.0: Looking for event-dma 00000000ffe3e430 trb-start 00000000ffe3e400 trb-end 00000000ffe3e400 seg-start 00000000ffe3e000 seg-end 00000000ffe3eff0

Can you still reproduce this error with the ASMedia controller?

dubhater, 2025-03-06:
If you can still reproduce the error here is another patch to try: [0001-wifi-rtw88-usb-Avoid-using-URB_ZERO_PACKET.patch.txt](https://github.com/user-attachments/files/19115940/0001-wifi-rtw88-usb-Avoid-using-URB_ZERO_PACKET.patch.txt)

avonder, 2025-03-25:
No effect.

dubhater, 2025-03-30:
Thanks for testing. Unfortunately that was my last idea.