lwfinger/rtw88 #287: 8821au - failed to get tx report from firmware

Boobies, 2025-01-10:
rooftopjoe@uranus:~$ sudo dmesg | grep rtw
[sudo] parola pentru rooftopjoe: 
[    4.541561] rtw_core: loading out-of-tree module taints kernel.
[    4.541566] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[    4.617742] rtw_8821au 1-10:1.0: Firmware version 42.4.0, H2C version 0
[    5.221474] usbcore: registered new interface driver rtw_8821au
[    5.436167] rtw_8821au 1-10:1.0 wlx40ed00e07c7e: renamed from wlan0
[   12.646689] Modules linked in: qrtr nvidia_drm(POE) drm_kms_helper nvidia_modeset(POE) binfmt_misc nvidia(POE) nls_ascii nls_cp437 vfat fat rtw_8821au(OE) rtw_8821a(OE) rtw_88xxa(OE) rtw_usb(OE) rtw_core(OE) intel_rapl_msr intel_rapl_common snd_soc_avs snd_soc_hda_codec mac80211 snd_hda_ext_core snd_soc_core snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi snd_usb_audio snd_hda_scodec_component x86_pkg_temp_thermal snd_compress intel_powerclamp snd_pcm_dmaengine coretemp crct10dif_pclmul snd_hda_intel ghash_clmulni_intel sha512_ssse3 snd_intel_dspcfg sha256_ssse3 snd_intel_sdw_acpi snd_usbmidi_lib snd_hda_codec sha1_ssse3 aesni_intel eeepc_wmi snd_rawmidi asus_wmi gf128mul libarc4 crypto_simd snd_seq_device cryptd snd_hda_core sparse_keymap mc rapl cfg80211 snd_hwdep intel_cstate platform_profile mei_hdcp mei_pxp snd_pcm intel_uncore battery iTCO_wdt snd_timer snd mei_me rfkill ee1004 intel_pmc_bxt wmi_bmof intel_wmi_thunderbolt serio_raw iTCO_vendor_support pcspkr mei watchdog intel_pmc_core soundcore
[ 6597.362673] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[ 6917.524266] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[ 7160.401423] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[ 7530.027107] rtw_8821au 1-10:1.0: firmware failed to leave lps state
[ 7533.363062] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[ 7535.411069] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[11996.452071] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[12318.411564] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[12320.459769] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[12322.443974] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[12324.364169] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[13505.510123] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[13812.200794] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[14348.529833] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[14739.440402] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[14748.427830] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[14749.387769] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[15239.447569] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[15242.391497] rtw_8821au 1-10:1.0: failed to get tx report from firmware
[15245.399418] rtw_8821au 1-10:1.0: failed to get tx report from firmware

dubhater, 2025-01-10:
I will guess that your network is in the 2 GHz band and you use Bluetooth headphones or speakers nearby.

I will further guess that the warning or whatever this is: "Modules linked in: qrtr nvidia_drm(POE)" was caused by Nvidia.

Boobies, 2025-01-11:
It's actually 5 GHz.

Boobies, 2025-01-16:
It's really impossible to use the 8821au, sometimes the disconnects happen every few seconds in a loop.

dubhater, 2025-01-16:
What's the signal strength?

Boobies, 2025-01-16:
It shows up as full but sometimes disconnects from the network and it might not even show up in GNOME's drop-down for a bit. Sometimes it will stay connected by to the network but I will not be able to reach the Internet. It generally lasts seconds but this sometimes happen in a loop dozens of times. Other devices work fine so it is not the network's fault. If you are confident that the driver is the issue, it might be a hardware issue with the USB dongle. Should I provide any extra info?

dubhater, 2025-01-16:
Does it help to turn off the power saving? ``iw wlan0 set power_save off``

Try the other driver in the same conditions: https://github.com/morrownr/8821au-20210708

Boobies, 2025-01-17:
Turning power saving off helped. I've been using it all day and got no disconnects.