lwfinger/rtw88 #288: RTL8821CS crashes on Kernel 6.1 ARM A35

briansune, 2025-01-10:
@dubhater 

I had finally get the board to test the SDIO RTL8821CS.
It crashes on RTW88 any patch can fix this?
Thank you

```
[  233.576722] skbuff: skb_over_panic: text:ffffffc000b16d98 len:11009 put:11009 head:ffffff800891e000 data:ffffff800891e040 tail:0x2b41 end:0x2c0 dev:<NULL>
[  233.576803] ------------[ cut here ]------------
[  233.576806] kernel BUG at net/core/skbuff.c:120!
[  233.576814] Internal error: Oops - BUG: 00000000f2000800 [#1] PREEMPT SMP
[  233.606532] Modules linked in: rtw_8821cs(O) rtw_8821c(O) rtw_sdio(O) rtw_core(O)
[  233.614013] CPU: 0 PID: 16 Comm: kworker/0:1 Tainted: G           O       6.1.111-rt42 #11
[  233.622248] Hardware name: Anlogic, DR1M90 FPSoc (DT)
[  233.627280] Workqueue: events sdio_irq_work
[  233.631468] pstate: 40000005 (nZcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  233.638406] pc : skb_panic+0x4c/0x50
[  233.641978] lr : skb_panic+0x4c/0x50
[  233.645546] sp : ffffff80000d7bb0
[  233.648848] x29: ffffff80000d7bc0 x28: 0000000000000000 x27: ffffff800891e22c
[  233.655969] x26: 0000000000002b20 x25: ffffff8001f4b000 x24: 0000000000000018
[  233.663090] x23: ffffff800d021e40 x22: 0000000000000018 x21: ffffff80000d7ca0
[  233.670210] x20: ffffff80000d7cd0 x19: ffffff800886f000 x18: 0000000000000060
[  233.677332] x17: cc345d1400640044 x16: 0000000000000015 x15: 663a617461642030
[  233.684452] x14: 3030653139383030 x13: 3030386666666666 x12: 663a646165682039
[  233.691573] x11: 3e4c4c554e3c3a76 x10: 6564203063327830 x9 : 3a646e6520313462
[  233.698695] x8 : ffffffc008fd7ed8 x7 : ffffff80000d79f0 x6 : ffffff801fea0958
[  233.705815] x5 : ffffff801fea0958 x4 : 0000000000000000 x3 : ffffff801fea3b88
[  233.712935] x2 : ffffff801fea3b88 x1 : ffffff800009cf80 x0 : 000000000000008e
[  233.720056] Call trace:
[  233.722494]  skb_panic+0x4c/0x50
[  233.725719]  skb_put+0x74/0x78
[  233.728769]  rtw_sdio_rx_skb+0x58/0xf8 [rtw_sdio]
[  233.733473]  rtw_sdio_rxfifo_recv+0x1d8/0x290 [rtw_sdio]
[  233.738779]  rtw_sdio_handle_interrupt+0x10c/0x190 [rtw_sdio]
[  233.744518]  process_sdio_pending_irqs+0x4c/0x180
[  233.749210]  sdio_irq_work+0x50/0x80
[  233.752777]  process_one_work+0x1fc/0x350
[  233.756783]  worker_thread+0x44/0x438
[  233.760440]  kthread+0x10c/0x118
[  233.763665]  ret_from_fork+0x10/0x20
[  233.767241] Code: a90023eb aa0a03e1 912b6120 940c48fa (d4210000)
[  233.773312] ---[ end trace 0000000000000000 ]---
[  233.773318] note: kworker/0:1[16] exited with irqs disabled
[  233.773365] note: kworker/0:1[16] exited with preempt_count 1
[  233.779482] ------------[ cut here ]------------
[  233.779494] WARNING: CPU: 0 PID: 0 at kernel/context_tracking.c:128 ct_kernel_exit.isra.6.constprop.8+0x98/0xa0
[  233.779527] Modules linked in: rtw_8821cs(O) rtw_8821c(O) rtw_sdio(O) rtw_core(O)
[  233.779559] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D    O       6.1.111-rt42 #11
[  233.779572] Hardware name: Anlogic, DR1M90 FPSoc (DT)
[  233.779578] pstate: 200000c5 (nzCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  233.779590] pc : ct_kernel_exit.isra.6.constprop.8+0x98/0xa0
[  233.779601] lr : ct_idle_enter+0x10/0x20
[  233.779611] sp : ffffffc008fc3d80
[  233.779616] x29: ffffffc008fc3d80 x28: 0000000001215ff0 x27: 000000001ffbf8b8
[  233.779635] x26: 000000001ed02368 x25: 0000000000000072 x24: 0000000000000000
[  233.779654] x23: ffffffc009123000 x22: ffffff801fecd5c0 x21: ffffffc008fc88e0
[  233.779673] x20: 0000000000000001 x19: ffffff801fea4e18 x18: 0000000000000000
[  233.779691] x17: 0000000000000000 x16: 0000000000000000 x15: fffffffe00147300
[  233.779710] x14: 0000000000000000 x13: 00000000000002fa x12: 0000000000000000
[  233.779728] x11: 0000000000000000 x10: 00000000000009e0 x9 : ffffffc008fc3d40
[  233.779746] x8 : ffffffc008fd0300 x7 : 0000000000000000 x6 : 00000001dea6c4e9
[  233.779764] x5 : 00ffffffffffffff x4 : 4000000000000002 x3 : ffffffc008fc3d80
[  233.779783] x2 : 4000000000000000 x1 : ffffffc008ea8000 x0 : ffffffc008ea8e18
[  233.779802] Call trace:
[  233.779806]  ct_kernel_exit.isra.6.constprop.8+0x98/0xa0
[  233.779818]  ct_idle_enter+0x10/0x20
[  233.779829]  default_idle_call+0x20/0x6c
[  233.779843]  do_idle+0xf4/0x120
[  233.779859]  cpu_startup_entry+0x38/0x40
[  233.779873]  kernel_init+0x0/0x130
[  233.779883]  arch_post_acpi_subsys_init+0x0/0x18
[  233.779897]  start_kernel+0x634/0x658
[  233.779907]  __primary_switched+0xb4/0xbc
[  233.779923] ---[ end trace 0000000000000000 ]---
```

dubhater, 2025-01-10:
Is it easy to reproduce?

briansune, 2025-01-10:
@dubhater 

Answer is YES.

Repeatable always.
The trigger method as follows:

load RTW88 drivers aka .ko
core, sdio, 8821c, 8821cs.

Then connect to any WIFI.
Immediately crashes.

briansune, 2025-01-14:
@dubhater 

There will be no crash when there are no WIFI is connected.

Once it try to connect always crashes W/O issue.

Had tried different MMC speed profile <- not really the main issue from first place as other module and work properly.

```
[   80.996932] ------------[ cut here ]------------
[   80.996935] kernel BUG at net/core/skbuff.c:120!
[   80.996943] Internal error: Oops - BUG: 00000000f2000800 [#1] PREEMPT SMP
[   81.026658] Modules linked in: rtw_8821cs(O) rtw_8821c(O) rtw_sdio(O) rtw_core(O) [last unloaded: rtw_8821cs(O)]
[   81.036824] CPU: 0 PID: 139 Comm: kworker/0:2 Tainted: G           O       6.1.111-rt42 #15
[   81.045146] Hardware name: Anlogic, DR1M90 FPSoc (DT)
[   81.050178] Workqueue: events sdio_irq_work
[   81.054366] pstate: 40000005 (nZcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   81.061304] pc : skb_panic+0x4c/0x50
[   81.064875] lr : skb_panic+0x4c/0x50
[   81.068444] sp : ffffff80027a7bb0
[   81.071746] x29: ffffff80027a7bc0 x28: 0000000000000000 x27: ffffff800224ddc4
[   81.078868] x26: 0000000000004058 x25: ffffff800270a700 x24: 0000000000000058
[   81.085990] x23: ffffff8005391e40 x22: 0000000000000058 x21: 0000000000000169
[   81.093111] x20: ffffff80027a7cd0 x19: ffffff800270ae00 x18: 0000000000000060
[   81.100232] x17: 0000000000000000 x16: 0000000000000000 x15: 663a617461642030
[   81.107353] x14: 3063643432323030 x13: 3a76656420306332 x12: 78303a646e652066
[   81.114474] x11: 3e4c4c554e3c3a76 x10: 6564203063327830 x9 : 3a646e6520666130
[   81.121595] x8 : ffffffc008fd7ed8 x7 : ffffff80027a79f0 x6 : ffffff801fea0958
[   81.128715] x5 : ffffff801fea0958 x4 : 0000000000000000 x3 : ffffff801fea3b88
[   81.135836] x2 : ffffff801fea3b88 x1 : ffffff8001fdea00 x0 : 000000000000008e
[   81.142958] Call trace:
[   81.145396]  skb_panic+0x4c/0x50
[   81.148620]  skb_put+0x74/0x78
[   81.151670]  rtw_sdio_rx_skb+0xbc/0xf8 [rtw_sdio]
[   81.156375]  rtw_sdio_rxfifo_recv+0x1d8/0x290 [rtw_sdio]
[   81.161681]  rtw_sdio_handle_interrupt+0x10c/0x190 [rtw_sdio]
[   81.167419]  process_sdio_pending_irqs+0x4c/0x180
[   81.172113]  sdio_irq_work+0x50/0x80
[   81.175680]  process_one_work+0x1fc/0x350
[   81.179685]  worker_thread+0x44/0x438
[   81.183343]  kthread+0x10c/0x118
[   81.186567]  ret_from_fork+0x10/0x20
[   81.190142] Code: a90023eb aa0a03e1 9126c120 940c6032 (d4210000)
[   81.196213] ---[ end trace 0000000000000000 ]---
co[   81.196219] note: kworker/0:2[139] exited with irqs disabled
[   81.196285] note: kworker/0:2[139] exited with preempt_count 1
[   81.202545] ------------[ cut here ]------------
[   81.202557] WARNING: CPU: 0 PID: 0 at kernel/context_tracking.c:128 ct_kernel_exit.isra.6.constprop.8+0x98/0xa0
[   81.202592] Modules linked in: rtw_8821cs(O) rtw_8821c(O) rtw_sdio(O) rtw_core(O) [last unloaded: rtw_8821cs(O)]
[   81.202629] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D    O       6.1.111-rt42 #15
[   81.202642] Hardware name: Anlogic, DR1M90 FPSoc (DT)
[   81.202649] pstate: 200000c5 (nzCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   81.202661] pc : ct_kernel_exit.isra.6.constprop.8+0x98/0xa0
[   81.202674] lr : ct_idle_enter+0x10/0x20
[   81.202685] sp : ffffffc008fc3d80
[   81.202690] x29: ffffffc008fc3d80 x28: 0000000001215ff0 x27: 000000001ffbf8b8
[   81.202710] x26: 000000001ed02488 x25: 0000000000000072 x24: 0000000000000000
[   81.202729] x23: ffffffc009123000 x22: ffffff801fecd5c0 x21: ffffffc008fc88e0
[   81.202748] x20: 0000000000000001 x19: ffffff801fea4e18 x18: 0000000000000000
[   81.202768] x17: 0000000000000000 x16: 0000000000000000 x15: 0000b5ec33999f90
[   81.202787] x14: 0000000000000062 x13: 0000000000000062 x12: 0000000000000000
[   81.202805] x11: 0000000000000000 x10: 00000000000009e0 x9 : ffffffc008fc3d40
[   81.202824] x8 : ffffffc008fd0300 x7 : 0000000000000000 x6 : 00000000af379ac6
[   81.202843] x5 : 00ffffffffffffff x4 : 4000000000000002 x3 : ffffffc008fc3d80
[   81.202862] x2 : 4000000000000000 x1 : ffffffc008ea8000 x0 : ffffffc008ea8e18
[   81.202882] Call trace:
[   81.202887]  ct_kernel_exit.isra.6.constprop.8+0x98/0xa0
[   81.202900]  ct_idle_enter+0x10/0x20
[   81.202912]  default_idle_call+0x20/0x6c
[   81.202927]  do_idle+0xf4/0x120
n[   81.202943]  cpu_startup_entry+0x38/0x40
[   81.202957]  kernel_init+0x0/0x130
[   81.202969]  arch_post_acpi_subsys_init+0x0/0x18
[   81.202984]  start_kernel+0x634/0x658
[   81.202996]  __primary_switched+0xb4/0xbc
[   81.203007] ---[ end trace 0000000000000000 ]---
```

briansune, 2025-01-19:
@dubhater 

The issue had been investigated on RTL8821CS.

Report: https://github.com/briansune/rtw88-hw/tree/main/rtl8821cs_arm_a35_hs

If the SDIO profile is set to low-speed the above issue will be easily triggered.
I haven't tested the profile between high-speed and low-speed, such as full-speed.
For low-speed profile it is still 4bit bus.

For SDIO profile see: https://www.prodigytechno.com/sdio-protocol

Both 5G&2.4G can connect normally under high-speed 50MHz clock (4bit bus).

So I guess there are some handshaking bugs like timeout of etc. issue on low-speed SDIO profile.

closed 2025-01-19 by briansune