lwfinger/rtw88 #291: RTL8822BS Cannot connect to WIFI

briansune, 2025-01-18:
@dubhater 

Any idea why it is kept unable to connect on STA?
All other USB and SDIO can connect to the same WIFI i.e. 8822CS 8723DS etc.

[   32.521284] rtw_core: loading out-of-tree module taints kernel.
[   33.336541] rtw_8822bs mmc1:0001:1: Firmware version 30.20.0, H2C version 14

MMC_TYPE=SDIO
SDIO_ID=024C:B822
SDIO_REVISION=0.0
clock:          50000000 Hz
actual clock:   50000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)

[   38.876559] wlan0: authenticate with 
[   39.576931] wlan0: send auth to  (try 1/3)
[   40.973670] wlan0: send auth to  (try 2/3)
[   41.965873] wlan0: send auth to  (try 3/3)
[   42.956801] wlan0: authentication with  timed out
[   43.808597] wlan0: authenticate with 
[   44.500882] wlan0: send auth to  (try 1/3)
[   44.511688] wlan0:  denied authentication (status 1)
[   48.828972] wlan0: authenticate with 
[   49.560886] wlan0: send auth to  (try 1/3)
[   50.957553] wlan0: send auth to  (try 2/3)
[   51.981646] wlan0: send auth to  (try 3/3)
[   52.972792] wlan0: authentication with  timed out
[   54.703375] wlan0: authenticate with 
[   55.404884] wlan0: send auth to  (try 1/3)
[   55.413017] wlan0:  denied authentication (status 1)
[   60.581687] wlan0: authenticate with 
[   61.304884] wlan0: send auth to  (try 1/3)
[   61.965527] wlan0: send auth to  (try 2/3)
[   62.989483] wlan0: send auth to  (try 3/3)
[   63.980778] wlan0: authentication with  timed out
[   68.595063] wlan0: authenticate with 
[   69.280888] wlan0: send auth to  (try 1/3)
[   69.285142] wlan0:  denied authentication (status 1)
[   83.443650] wlan0: authenticate with 
[   84.164885] wlan0: send auth to  (try 1/3)
[   84.973499] wlan0: send auth to  (try 2/3)
[   85.965473] wlan0: send auth to  (try 3/3)
[   86.988759] wlan0: authentication with  timed out
[   98.630182] wlan0: authenticate with 
[   99.296865] wlan0: send auth to  (try 1/3)
[   99.300635] wlan0:  denied authentication (status 1)

dubhater, 2025-01-18:
I don't know.

You can use another device in monitor mode to see if the RTL8822BS is transmitting anything.

briansune, 2025-01-18:
@dubhater 

Need more guidance on that test.
Please give more procedure that I can do to debug these situation.
 
I do see some USB and 8822BS do have such issue on STA.

briansune, 2025-01-18:
@dubhater 

Do RTW or you have any method to check the antenna power status etc?
Some module is 3-antenna some are 2-antenna configurations.

briansune, 2025-01-19:
@dubhater 

I had narrow down the issue to the frequency band that it trying to connect.
Before the main router is set as merged 2.4G+5G as same ESSID.
So there are no two separate xxx & xxx_5G.
So I had tried to split it back to two individual 5G and 2G ID.
Connect each freq band and test it out.

Result as follows:
2.4G no issue
5G cannot connect.

Please see if patch is need to fix such issue?

briansune, 2025-01-21:
@dubhater 

Okay the 5G also works and why previous 5G no working.

```
MMC_TYPE=SDIO
SDIO_ID=024C:B822
SDIO_REVISION=0.0
clock:          50000000 Hz
actual clock:   50000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    2 (sd high-speed)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```

Most likely is module issue, left side NOT ABLE link 5G network, right side ABLE to link 5G network.

[image]

briansune, 2025-01-21:
@dubhater 

Before start I would like to know this repository is able to run on kernel 6.1?

Thank you

briansune, 2025-01-22:
@dubhater 

Hope this helps and doing two module cross check again.

Why I didn't successfully compile the modified vendor driver from first place is because the Makefile having so trouble and I can't see any .ko file create. But the repository from RockChip do able to make a ko file out after code modifications.

briansune, 2025-01-22:
@dubhater 

That modified driver do able to run the AP no issue

Bluetooth interrupt test shows little drop in TRX.

```
ubuntu@briansune:~$ ./hostap.sh
Start AP @ WLAN0
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlan0 with hwaddr and ssid "AP-TEST"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
ubuntu@briansune:~$ iperf3 -s
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.175.86, port 55317
[  5] local 192.168.175.1 port 5201 connected to 192.168.175.86 port 55318
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.34 MBytes  19.6 Mbits/sec   10   67.0 KBytes
[  5]   1.00-2.00   sec  1.96 MBytes  16.4 Mbits/sec   11   69.9 KBytes
[  5]   2.00-3.00   sec  3.80 MBytes  31.9 Mbits/sec    0    104 KBytes
[  5]   3.00-4.00   sec  3.80 MBytes  31.9 Mbits/sec    0    128 KBytes
[  5]   4.00-5.00   sec  3.74 MBytes  31.3 Mbits/sec    1    130 KBytes
[  5]   5.00-6.00   sec  3.31 MBytes  27.8 Mbits/sec    0    130 KBytes
[  5]   6.00-7.00   sec  3.92 MBytes  32.9 Mbits/sec    0    130 KBytes
[  5]   7.00-8.00   sec  4.78 MBytes  40.1 Mbits/sec    0    133 KBytes
[  5]   8.00-9.00   sec  5.76 MBytes  48.3 Mbits/sec    0    133 KBytes
[  5]   9.00-10.00  sec  5.64 MBytes  47.3 Mbits/sec    0    133 KBytes
[  5]  10.00-11.00  sec  4.41 MBytes  37.0 Mbits/sec    0    133 KBytes
[  5]  11.00-12.00  sec  3.49 MBytes  29.3 Mbits/sec    0    133 KBytes
[  5]  12.00-13.00  sec  3.74 MBytes  31.3 Mbits/sec    0    133 KBytes
[  5]  13.00-14.00  sec  3.74 MBytes  31.3 Mbits/sec    0    133 KBytes
[  5]  14.00-15.00  sec  4.04 MBytes  33.9 Mbits/sec    0    133 KBytes
[  5]  15.00-16.00  sec  3.92 MBytes  32.9 Mbits/sec    0    133 KBytes
[  5]  16.00-17.00  sec  1.16 MBytes  9.76 Mbits/sec    1    133 KBytes
[  5]  17.00-18.00  sec  1.41 MBytes  11.8 Mbits/sec    0    133 KBytes
[  5]  18.00-19.00  sec   627 KBytes  5.14 Mbits/sec    3   75.6 KBytes
[  5]  19.00-20.00  sec   314 KBytes  2.57 Mbits/sec    0   85.5 KBytes
[  5]  19.00-20.00  sec   314 KBytes  2.57 Mbits/sec    0   85.5 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-20.00  sec  66.5 MBytes  27.9 Mbits/sec   26             sender
iperf3: the client has terminated
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
^Ciperf3: interrupt - the server has terminated
ubuntu@briansune:~$
```

briansune, 2025-01-22:
@dubhater

briansune, 2025-01-22:
@a5a5aa555oo 

Would you mind to see if I do any mistake or possible "interesting message"?

It would be deep appreciated.

Thank you

a5a5aa555oo, 2025-01-22:
I don't know what changes you have made, but I think your changes should be okay because the driver works.

briansune, 2025-01-22:
@a5a5aa555oo 

You observation is same as what I am trying to declared.

I have completely no clue those ctrl+c ctrl+v and recompiles again and again to clear out all errors. 

Just wait until it give me a .ko file 😂

briansune, 2025-01-24:
@dubhater @a5a5aa555oo 

I had ZERO idea now because I used a different test setup A53 to retest and assemble more 8822bs (same batch I buy from A vendor).

Result: 5G STA works and stable no any issue on RTW88.

I completely don't know it is a QC issue or EFUSE or any possible SDIO speed profile?

So meantime I also retested all and cap the speed to below 100MHz:

8723ds 8821cs 8822bs 8822cs

New sets of report had done:

8723: https://github.com/briansune/rtw88-hw/tree/main/rtl8723ds_arm_a53
8821: https://github.com/briansune/rtw88-hw/tree/main/rtl8821cs_arm_a53
8822b: https://github.com/briansune/rtw88-hw/tree/main/rtl8822bs_arm_a53
8822c: https://github.com/briansune/rtw88-hw/tree/main/rtl8822cs_arm_a53

I guess RTW88 should be fine in 8822BS and other SDIO devices exclude AP & 2.4G band issue.

closed 2025-01-24 by briansune