lwfinger/rtw88 #293: RTL8821CU no 5GHz at all

PD-Kerman, 2025-01-19:
Hi there

I've got a RTL8821CU here, but couldnt connect to / see a single SSID in 5ghz, not even any from my neighbors

changed all my router/ap(s) to channel 149 to avoid dfs, crda/wireless-regdb installed, using the latest driver from this repo, got headers installed first, then followed the guide and installed the driver under this very repository, but still no luck after all.

the no-ir mark in iwlist / iw phy constanly changes tho, it goes between completely without no-ir mark to all channels marked as no-ir mainly related to sleep hibernate reboot things like that, and as i remembered that it did connect to a 5ghz wifi once and the bitrate was promising, and even with no-ir marked at that time so i guess the accuracy of the result being shown in iw should be up for discission

and further more, just discovered some weird detection these days

its like, my ap(s) takes over several SSIDs, and there are one for 2.4band and one for 5ghz band

and as i discovered these days as i thought it would be no longer be able to detect any 5ghz signals,
weirdly when i check wavemon for scanning outputs, some 5ghz signals just sneaked out for like tens of seconds or so, and disappeared, and they were usually like without SSIDs or the SSIDs were for the signals on 2.4ghz band which should be wrong/impossible for such to be used

and one more thing is that
i went and borrowed a RTL8188GU dongle yesterday and tested with it
everything seemed to be working fine, its got all SSIDs on 5GHz scanned out, and most importanly
what i found was like
Those 5GHz signals which could not be found/recognized by RTL8821CU earlier all shown in wavemon with a Spectrum Mgmt mark which RTL8821CU happened to not having it/those signal detected, and all tests above were done with at least 1 nearby 5ghz ap set to channel 149 and/or channel 36

P.S. even though that it could mysteriously work with that specific 5GHz AP that all other AP/Devices cannot, not even with the exact bandwith & channel being set, it still had got weird/teribble non-comprehensive scanning results for all the time being

Nevertheless, the 8821CU ive got is still not being able to detect any 5ghz signals

Wish there would be some solutions

Any kinds of help will be appreciated !

And feel free to ask for any kind of logs/outputs

Thanks in advanced,
And best wishes to all contributors,
Best Regards

PD-Kerman, 2025-01-19:
Now attaching some related outputs: 

# 1: `iw phy`

`pd-kerman@archlinux ~ % sudo iw phy
[sudo] password for pd-kerman: 
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x1 RX 0x1
	Configured Antennas: TX 0x1 RX 0x1
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x196e
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412.0 MHz [1] (20.0 dBm)
			* 2417.0 MHz [2] (20.0 dBm)
			* 2422.0 MHz [3] (20.0 dBm)
			* 2427.0 MHz [4] (20.0 dBm)
			* 2432.0 MHz [5] (20.0 dBm)
[... 456 lines cut ...]
country CN: DFS-FCC
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5725 - 5850 @ 80), (N/A, 33), (N/A)
	(57240 - 59400 @ 2160), (N/A, 28), (N/A)
	(59400 - 63720 @ 2160), (N/A, 44), (N/A)
	(63720 - 65880 @ 2160), (N/A, 28), (N/A)

pd-kerman@archlinux ~ % 
`

# 6: `router admin info`

[image]

# 7: `scanning result B`

[image]

# 8: `wavemon`
 
[image]

[image]

PD-Kerman, 2025-01-19:
# Attaching more (rev. A)

# dkms status

[image]

briansune, 2025-01-19:
@PD-Kerman 

Report:
https://github.com/briansune/rtw88-hw/tree/main/rtl8821cu_arm_a35

I think I do successfully connected to 5G band.
But only Mbps performance is not that great.

dubhater, 2025-01-19:
@PD-Kerman Can you show lsusb with this RTL8188GU plugged in?

briansune, 2025-01-19:
> Now attaching some related outputs:
>
> # 1: `iw phy`
> `pd-kerman@archlinux ~ % sudo iw phy [sudo] password for pd-kerman: Wiphy phy0 wiphy index: 0 max # scan SSIDs: 4 max scan IEs length: 2243 bytes [...]

See that your iw list is almost same as my RTL8812AU dongle:

https://github.com/briansune/rtw88-hw/tree/main/rtl8812au_arm_a35

PD-Kerman, 2025-01-19:
# btw, now attaching more related outputs (rev. B)

[image]

[image]

PD-Kerman, 2025-01-19:
> @PD-Kerman 
> 
> Report:
> https://github.com/briansune/rtw88-hw/tree/main/rtl8821cu_arm_a35
> 
> I think I do successfully connected to 5G band.
> But only Mbps performance is not that great.
> 
> I guess do your hardware really populated 5G path from beginning?
> Open the housing and see the real hardware.

Well as I mentioned that it actually did connected to 5G band for like several times but with the same ap, and frankly the speed was actually pretty promising 

And yea I guess it should look just like its being shown in your doc there when it's been teared down 

> 
> See down-costed RTL8812AU:
> 
> https://github.com/lwfinger/rtw88/issues/246#issuecomment-2571406282
> 
> and 
> 
> https://github.com/briansune/rtw88-hw/tree/main/rtl8812au_arm_a35
> 

Oh cool, I’ll look into it
Although I’m not that familiar with all these wireless stuff lol I’ll try

PD-Kerman, 2025-01-19:
> @dubhater @PD-Kerman 
> 
> https://github.com/lwfinger/rtw88/issues/293#issuecomment-2600869890
> 
> See that the iw list from the above post is almost same as my RTL8812AU dongle:
> 
> https://github.com/briansune/rtw88-hw/tree/main/rtl8812au_arm_a35
> 

> Which are disabled most channels.

Oh uhm
So how/why was those being disabled and is there like any workarounds~? Is it like software or hardware...?

Sry I'm actually kind of a newbie when it comes to these wireless stuff lol I might need more context ig

Thank your very much for your reply and your info ! :D

PD-Kerman, 2025-01-19:
Oh and attaching some dmesg info as ive read about things that indicates some chip with efuse might leave some messages in the dmesg outputs, mines looks quite clean here

[image]

briansune, 2025-01-19:
@PD-Kerman @dubhater 

Well I am very poor in software stuffs.

But short test from another A9 ARM board:

```
wlx387accfc14ab  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 14 : 2.484 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Current Frequency:5.745 GHz

brian@localhost:~$ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=118 time=188 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=118 time=110 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=118 time=5.62 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=118 time=152 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=118 time=11.6 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=118 time=27.5 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=118 time=56.5 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=118 time=279 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=118 time=108 ms
64 bytes from 8.8.8.8: icmp_seq=10 ttl=118 time=217 ms

wlx387accfc14ab  IEEE 802.11  ESSID:""
          Mode:Managed  Frequency:5.745 GHz  Access Point: 
          Bit Rate=27 Mb/s   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=30/70  Signal level=-80 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```

Not sure what is your case. But with all do respect just wanna help.

PD-Kerman, 2025-01-19:
> @PD-Kerman @dubhater 
> 
> Well I am very poor in software stuffs.
> 
> But short test from another A9 ARM board:
> 
> ```
> wlx387accfc14ab  32 channels in total; available frequencies :
>           Channel 01 : 2.412 GHz
>           Channel 02 : 2.417 GHz
>           Channel 03 : 2.422 GHz
>           Channel 04 : 2.427 GHz
>           Channel 05 : 2.432 GHz
>           Channel 06 : 2.437 GHz
>           Channel 07 : 2.442 GHz
>           Channel 08 : 2.447 GHz
>           Channel 09 : 2.452 GHz
>           Channel 10 : 2.457 GHz
>           Channel 11 : 2.462 GHz
>           Channel 12 : 2.467 GHz
>           Channel 13 : 2.472 GHz
>           Channel 14 : 2.484 GHz
>           Channel 36 : 5.18 GHz
>           Channel 40 : 5.2 GHz
>           Channel 44 : 5.22 GHz
>           Channel 48 : 5.24 GHz
>           Channel 52 : 5.26 GHz
>           Channel 56 : 5.28 GHz
>           Channel 60 : 5.3 GHz
>           Channel 64 : 5.32 GHz
>           Channel 100 : 5.5 GHz
>           Channel 104 : 5.52 GHz
>           Channel 108 : 5.54 GHz
>           Channel 112 : 5.56 GHz
>           Channel 116 : 5.58 GHz
>           Channel 120 : 5.6 GHz
>           Channel 124 : 5.62 GHz
>           Channel 128 : 5.64 GHz
>           Channel 132 : 5.66 GHz
>           Channel 136 : 5.68 GHz
>           Current Frequency:5.745 GHz
> 
> brian@localhost:~$ ping 8.8.8.8
> PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
> 64 bytes from 8.8.8.8: icmp_seq=1 ttl=118 time=188 ms
> 64 bytes from 8.8.8.8: icmp_seq=2 ttl=118 time=110 ms
> 64 bytes from 8.8.8.8: icmp_seq=3 ttl=118 time=5.62 ms
> 64 bytes from 8.8.8.8: icmp_seq=4 ttl=118 time=152 ms
> 64 bytes from 8.8.8.8: icmp_seq=5 ttl=118 time=11.6 ms
> 64 bytes from 8.8.8.8: icmp_seq=6 ttl=118 time=27.5 ms
> 64 bytes from 8.8.8.8: icmp_seq=7 ttl=118 time=56.5 ms
> 64 bytes from 8.8.8.8: icmp_seq=8 ttl=118 time=279 ms
> 64 bytes from 8.8.8.8: icmp_seq=9 ttl=118 time=108 ms
> 64 bytes from 8.8.8.8: icmp_seq=10 ttl=118 time=217 ms
> 
> wlx387accfc14ab  IEEE 802.11  ESSID:""
>           Mode:Managed  Frequency:5.745 GHz  Access Point: 
>           Bit Rate=27 Mb/s   Tx-Power=20 dBm
>           Retry short limit:7   RTS thr:off   Fragment thr:off
>           Power Management:on
>           Link Quality=30/70  Signal level=-80 dBm
>           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
>           Tx excessive retries:0  Invalid misc:0   Missed beacon:0
> 
> brian@localhost:~$ lsmod
> Module                  Size  Used by
> rtw88_8821cu           12288  0
> rtw88_8821c            73728  1 rtw88_8821cu
> rtw88_usb              16384  1 rtw88_8821cu
> rtw88_core            147456  2 rtw88_usb,rtw88_8821c
> 
> brian@localhost:~$ lsusb -t
> /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ci_hdrc/1p, 480M
>     |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/4p, 12M
>         |__ Port 1: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
>         |__ Port 1: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
>         |__ Port 1: Dev 4, If 2, Class=Vendor Specific Class, Driver=rtw_8821cu, 12M
> ```
> 
> Not sure what is your case. But with all do respect just wanna help.

Oh sure, thanks again~
Well in my case it's just, there's some weird issue when it comes to my 8821cu and 5ghz networks so~
> 
> @PD-Kerman 
> 
> What is the cost to open the housing and verify the real hardware from beginning?

Well if you mean tear it down and see what's inside I might say I should be able to uhm get some references within hrs, it shouldn't cost much but, do you think it's related to the hardware~? 

Like, would it actually help and why~? :D

Thx~

PD-Kerman, 2025-01-19:
> @PD-Kerman @dubhater 
> 
> Well I am very poor in software stuffs.
> 
> But short test from another A9 ARM board:
> 
> ```
> wlx387accfc14ab  32 channels in total; available frequencies :
>           Channel 01 : 2.412 GHz
>           Channel 02 : 2.417 GHz
>           Channel 03 : 2.422 GHz
>           Channel 04 : 2.427 GHz
>           Channel 05 : 2.432 GHz
>           Channel 06 : 2.437 GHz
>           Channel 07 : 2.442 GHz
>           Channel 08 : 2.447 GHz
>           Channel 09 : 2.452 GHz
>           Channel 10 : 2.457 GHz
>           Channel 11 : 2.462 GHz
>           Channel 12 : 2.467 GHz
>           Channel 13 : 2.472 GHz
>           Channel 14 : 2.484 GHz
>           Channel 36 : 5.18 GHz
>           Channel 40 : 5.2 GHz
>           Channel 44 : 5.22 GHz
>           Channel 48 : 5.24 GHz
>           Channel 52 : 5.26 GHz
>           Channel 56 : 5.28 GHz
>           Channel 60 : 5.3 GHz
>           Channel 64 : 5.32 GHz
>           Channel 100 : 5.5 GHz
>           Channel 104 : 5.52 GHz
>           Channel 108 : 5.54 GHz
>           Channel 112 : 5.56 GHz
>           Channel 116 : 5.58 GHz
>           Channel 120 : 5.6 GHz
>           Channel 124 : 5.62 GHz
>           Channel 128 : 5.64 GHz
>           Channel 132 : 5.66 GHz
>           Channel 136 : 5.68 GHz
>           Current Frequency:5.745 GHz
> 
> brian@localhost:~$ ping 8.8.8.8
> PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
> 64 bytes from 8.8.8.8: icmp_seq=1 ttl=118 time=188 ms
> 64 bytes from 8.8.8.8: icmp_seq=2 ttl=118 time=110 ms
> 64 bytes from 8.8.8.8: icmp_seq=3 ttl=118 time=5.62 ms
> 64 bytes from 8.8.8.8: icmp_seq=4 ttl=118 time=152 ms
> 64 bytes from 8.8.8.8: icmp_seq=5 ttl=118 time=11.6 ms
> 64 bytes from 8.8.8.8: icmp_seq=6 ttl=118 time=27.5 ms
> 64 bytes from 8.8.8.8: icmp_seq=7 ttl=118 time=56.5 ms
> 64 bytes from 8.8.8.8: icmp_seq=8 ttl=118 time=279 ms
> 64 bytes from 8.8.8.8: icmp_seq=9 ttl=118 time=108 ms
> 64 bytes from 8.8.8.8: icmp_seq=10 ttl=118 time=217 ms
> 
> wlx387accfc14ab  IEEE 802.11  ESSID:""
>           Mode:Managed  Frequency:5.745 GHz  Access Point: 
>           Bit Rate=27 Mb/s   Tx-Power=20 dBm
>           Retry short limit:7   RTS thr:off   Fragment thr:off
>           Power Management:on
>           Link Quality=30/70  Signal level=-80 dBm
>           Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
>           Tx excessive retries:0  Invalid misc:0   Missed beacon:0
> 
> brian@localhost:~$ lsmod
> Module                  Size  Used by
> rtw88_8821cu           12288  0
> rtw88_8821c            73728  1 rtw88_8821cu
> rtw88_usb              16384  1 rtw88_8821cu
> rtw88_core            147456  2 rtw88_usb,rtw88_8821c
> 
> brian@localhost:~$ lsusb -t
> /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ci_hdrc/1p, 480M
>     |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/4p, 12M
>         |__ Port 1: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
>         |__ Port 1: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
>         |__ Port 1: Dev 4, If 2, Class=Vendor Specific Class, Driver=rtw_8821cu, 12M
> ```
> 
> Not sure what is your case. But with all do respect just wanna help.
> 
> @PD-Kerman 
> 
> What is the cost to open the housing and verify the real hardware from beginning?

Oh and uhm~ 
Btw is it just me or is it like

Aren't all those speedtest from your doc was like

Abnormally low~?

Like previously you mentioned that one of the test should be using 5G band 
But the bitrate was like :-/

briansune, 2025-01-19:
@PD-Kerman 

First reply to the hardware investigation ->

Need no agree with me as I have no idea what gear you are using from first place so misunderstanding might happen.
But opening a USB dongle only take less then 10 second to do.
WHY? Because it is a USB dongle and crashing it could introduce component/possible missing in my experience.
If it is onboard fixed gear aka soldered onto motherboard or any other configuration then you could be right that it is not worth to take that risk.
So how do you tell it could not be the case from the first place?
I don't think you are asking in a respected way but more like 

Second reply to the speed which you modified the post ->

I do mentioned the speed is bad. But it is connected to 5G band and i need to making stuff out from first place.
Do RTW88 perfect? I don't think so, so there might be performance degradation.

PD-Kerman, 2025-01-19:
Oh and uhm
Okey I think I get that now

So it's like some downcosted modules with some components missing and it was for that

Well uhm

As I mention earlier, this particular hardware was like, it did work with a particular AP in 5G band at a pretty high bitrate so~

Idk if that matters but, can some module with missing components could have done that~? :D

briansune, 2025-01-19:
@PD-Kerman 

Well you only need to miss 1 15-10pF capacitor to break the either 5G path. And USB dongle well hard to tell.
I am stepping out.
Should be most likely software issue.

Enjoy~ Need no agree with me.

briansune, 2025-01-19:
@PD-Kerman 

I really don't know software stuff well.

Hope this do useful.

[image]

PD-Kerman, 2025-01-19:
> @PD-Kerman 
> 
> First reply to the hardware investigation ->
> 
> Need no agree with me as I have no idea what gear you are using from first place so misunderstanding might happen.
Yea about that
The reason why I'm being kinda ambiguous about tearing it down and see the components inside was becuz it's made from a pre designed usb module with some customize design but that was all about how to make it inside and connected to the usb port which shouldn't affect anything, and it's been well tested , no need to say that I mentioned quite a few times that the 5G band was working previously, not for long, not with various devices but it did,
and yea it (the whole module) is inside my chassis so it should take me some time to open it and get it checked
> But opening a USB dongle only take less then 10 second to do.
> WHY? Because it is a USB dongle and crashing it could introduce component/possible missing in my experience.
Crashing it how I don't really get it here
> If it is onboard fixed gear aka soldered onto motherboard or any other configuration then you could be right that it is not worth to take that risk.
Yup as I said it actually is
> So how do you tell it could not be the case from the first place?
As I mentioned earlier, quite difficult to get it out and test and quite sure that it worked before so it shouldn't be my first concern, or as you mentioned not worth it for now, and it's like 11:30PM here so what I did was just confirming what we are up to so that I could decide whether take it apart or not
> I don't think you are asking in a respected way but more like challenging!
Well, sorry if you actually feel that way
My apologies 

But I really don't see where did I show anything in any disrespectful way, I thanked almost every info / reply youve provided and I'm just mostly confirming some infos

I would appreciate if you could point those disrespectful comments out for me, thx
And further more as you might already know that I'm absolutely not a native English speaker so if there are any disrespectful among all those comments, I sincerely apologize 
> 
> Second reply to the speed which you modified the post ->
> 
> I do mentioned the speed is bad. But it is connected to 5G band and i need no making stuff out from first place.
Okey, got it
Just confirming
Although such card in 5G band with such Bitrate is quite unbelievable, couldn't help thinking whether there are noises or something
> Do RTW88 perfect? I don't think so, so there might be performance degradation.
Yea, agreed, but not that much~?
> 
> I am stepping out hope you resolve your problem ASAP.
> 
Yea thank you for your support, have a nice day

PD-Kerman, 2025-01-19:
> @PD-Kerman 
> 
> Well you only need to miss 1 15-10pF capacitor to break the either 5G path. And USB dongle well hard to tell.
> I am stepping out.
> Should be most likely software issue.
> 
> Enjoy~ Need no agree with me.

Okey thank you very much that helps a lot,
I'll go check that out anyway then :D

briansune, 2025-01-19:
> But I really don't see where did I show anything in any disrespectful way, I thanked almost every info / reply youve provided and I'm just mostly confirming some infos
>
> I would appreciate if you [...]

Well I learnt a hard lesson never blame software before you check your hardware setup.
So different language different culture.
Challenging doesn't necessary meant disrespectful different story my friend.
I deeply hope RTW88 maintainers help you out soon.
You too have a nice day~

PD-Kerman, 2025-01-19:
> @PD-Kerman 
> 
> I really don't know software stuff well.
> 
> Hope this do useful.
> 
> [image]
> 
> 

Yea oh cool
So the Bitrate shown here is much higher than those previous references 

*offtopic*
And me neither,
Oh and I saw you work on FPGAs cool
I bought a Zynq-7020 board months ago and now I'm trying to complete the Microblaze session

And yea, have a nice day then

briansune, 2025-01-19:
> Yea oh cool So the Bitrate shown here is much higher than those previous references
>
> _offtopic_ And me neither, Oh and I saw you work on FPGAs cool I bought a Zynq-7020 board months ago and now I'm [...]

Because I simply use a 2G antenna on the quick test which just in front of my monitor. Reapply a proper 5G antenna of cause makes a different.

offtopic<-
You need no Microblaze when you got Cortex-A9 from first place go check out PYNQ forum.

PD-Kerman, 2025-01-19:
> Well I learnt a hard lesson never blame software before you check your hardware setup.

Oh thanks!~ 
I see, noted I'll keep that in mind!~
> So different language different culture.
Yea... :)
> Challenging doesn't necessary meant disrespectful different story my friend.
Oh gotcha
> I deeply hope RTW88 maintainers help you out soon.
Thx :)
> You too have a nice day~
Yea you too~

dubhater, 2025-01-19:
It's probably a driver issue.

@PD-Kerman Can you plug one of the devices at a time and show the output of this command for each device?

```
cat /sys/kernel/debug/ieee80211/phyX/rtw88/coex_info
```

Replace the X with the appropriate number.

By the way, what is the brand and model of the device with PID c820?

PD-Kerman, 2025-01-19:
> Because I simply use a 2G antenna on the quick test which just in front of my monitor. Reapply a proper 5G antenna of cause makes a different.
>
> offtopic<-
> You need no Microblaze when you got [...]

Oh, yea about those antennas
It's like what I heard was just
For laptops there were two pre-built antennas, one for rx and one for tx, plugged in main and aux, and either one should work just fine so...
And I've tried another antenna yea

I guess it's got something to do with the spectrum mgmt stuff I guess I'll wait for the software support first then, also considering 5ghz worked before :D 
Thx again

*offtopic*
Well yea I am aware of that, it's just some friend of mine told me that Microblaze uses beam/uram as caches, and it's insanely fast, comparing to the general pc, that kind of speed was like registers, and he said that could provide unexpected performance when executing some tasks, since the cache is much more responsive and that could highly optimize the pipeline, and speaking of which I mean microblaze should be more flexible while doing some tasks than the PS of zynq right

PD-Kerman, 2025-01-19:
> It's probably a driver issue.
>
> @PD-Kerman Can you plug one of the devices at a time and show the output of this command for each device?
>
> ```
> cat /sys/kernel/debug/ieee80211/phyX/rtw88/coex_info [...]

Okey I'm on it

PD-Kerman, 2025-01-19:
`
root@archlinux /sys/kernel/debug/ieee80211/phy1/rtw88 # cat coex_info  
**********************************************
		BT Coexist info 3
**********************************************
Mech/ RFE                                = Shared/ 2
Coex Ver/ BT Dez/ BT Rpt                 = 19092746/ 0x46/ 0x00000000 (Mismatch)
Role/ RoleSwCnt/ IgnWL/ Feature          = Master/ 0/ 0
WL FW/ BT FW/ BT FW Desired/ KT          = 24.11/ 0x0/ 0x0/ C
AFH Map                                  = 0/ 0/ 0/ ch-(3)
RSSI/ STA Channel                        = -44dBm/ 3
TP {Tx, Rx} = {0, 0}Mbps
[Tx Rate] =  MCS4     (0x10)
[Rx Rate] =  CCK_1M   (0x0)
**********************************************
		BT Status
**********************************************
BT status/ rssi/ retry/ pop              = non-conn/ -90dBm/ 0/ 0
Profiles                                 =  (multi-link 0)
Reinit/ Relink/ IgnWl/ Feature           = 0/ 0/ 0/ 0x00000000
Page/ Inq/ iqk/ iqk fail                 = 0/ 0/ 0/ 0
0xae/ 0xac/ score board (W->B)/ (B->W)   = 0x0000/ 0x0000/ 0x0006/ 0x0002
Hi-Pri TX/RX, Lo-Pri TX/RX               = 0/0, 0/0
WL_FW                                    = 00 00 00 00 00 00 00
BT_RSP                                   = 01 00 10 00 00 00 7f
BT_ACT                                   = 02 00 10 00 00 00 7f
**********************************************
		WiFi Status
**********************************************
Scanning                                 = 0
G_busy/ TX/ RX                           = 1/ TX 0 Mbps/ RX 0 Mbps
IPS/ Low Power/ PS mode                  = 0/ 0/ 1
Iface on Port (0)
	Beacon interval                  = 100
	Network Type                     = 2
	Peer   0
		RSSI                     = 54
		BW mode                  = 1
**********************************************
		Mechanism
**********************************************
TDMA                                     = 00 00 00 00 00 (case-0)
Coex Mode/Free Run/Timer base            = 2G1PORT/ No/ 0
Table/ 0x6c0/ 0x6c4/ 0x6c8               = 28(28)/ 0x66555555/ 0x6a5a6a5a/ 0xf0ffffff
0x778/ 0x6cc/ Run Count/ Reason          = 0x00000001/ 0x18000018/ 142/ reason (LPS)
AFH Map to BT                            = 00 00 00
AntDiv/ BtCtrlLPS/ g_busy                = Off/ 1
Null All/ Retry/ Ack/ BT Empty/ BT Late  = 0/ 0/ 0/ 0/ 0
Cnt TDMA Toggle/ Lk 5ms/ Lk 5ms on/ FW   = 19/ 0/ No/ 14
WL_TxPw/ BT_TxPw/ WL_Rx/ BT_LNA_Lvl      = 0/ 0/ Off/ 7
**********************************************
		HW setting
**********************************************
LTE Coex/ Path Owner                     = OFF/ BT
GNT_WL_CTRL/ GNT_BT_CTRL/ Dbg            = RF:HW_BB:HW/ RF:HW_BB:HW/ Off
GNT_WL/ GNT_BT                           = 1/ 0
CRC OK CCK/ OFDM/ HT/ VHT                = 52/ 0/ 2/ 0
CRC ERR CCK/ OFDM/ HT/ VHT               = 1/ 0/ 0/ 0
HiPr/ Locking/ Locked/ Noisy             = N/ N/ N/ 1
cb0/ cb4/ cba                            = 0x77775747/ 0x20000077/ 0x1
430/ 434/ 42a/ 426/ 45e[3]/ 454          = 0x1000000/ 0x8070504/ 0x2020/ 0x10/ 0x0/ 0x7005
4c[24:23]/ 64[0]/ 4c6[4]/ 40[5]/ RF_1    = 0x0/ 0x0/ 0x1/ 0x1/ 0xc828
550/ 522/ 953[1]/ c50/ 60a               = 0x1008/ 0xff/ 0x1/ 0x36/ 0x17
EVM A/ EVM B/ SNR A/ SNR B               = 0/ 0/ 0/ 0
CCK-CCA/CCK-FA/OFDM-CCA/OFDM-FA          = 1/ 13/ 15/ 0
CRC OK CCK/11g/11n/11ac                  = 52/ 0/ 2/ 0
CRC Err CCK/11g/11n/11ac                 = 1/ 0/ 0/ 0
root@archlinux /sys/kernel/debug/ieee80211/phy1/rtw88 # 

`

PD-Kerman, 2025-01-19:
> It's probably a driver issue.
>
> [@PD-Kerman](https://github.com/PD-Kerman) Can you plug one of the devices at a time and show the output of this command for each device?
>
> ```
> cat [...]

# for 8821cu
https://pastebin.com/fDPUbKcG

PD-Kerman, 2025-01-19:
# for 8188GU
## P.S. 8821CU isnt really that easy to remove so im uploading this for now as that as soon 8188gu got plugged in, phy2 was generated
ill see what can i do if this wont work
https://pastebin.com/wRWUaDRH

PD-Kerman, 2025-01-19:
[8188GU.txt](https://github.com/user-attachments/files/18469441/8188GU.txt)
[8821CU.txt](https://github.com/user-attachments/files/18469442/8821CU.txt)

PD-Kerman, 2025-01-19:
> It's probably a driver issue.
>
> @PD-Kerman Can you plug one of the devices at a time and show the output of this command for each device?
>
> ```
> cat /sys/kernel/debug/ieee80211/phyX/rtw88/coex_info [...]

If you mean 0bda:c820 in `lsusb -tv`

That should be 8821cu it self

dubhater, 2025-01-19:
> If you mean 0bda:c820 in `lsusb -tv`
> 
> That should be 8821cu it self

Yes, that one. I mean, what was written on the box when you bought it? If I wanted to buy that myself, how can I find it?

PD-Kerman, 2025-01-19:
> > If you mean 0bda:c820 in `lsusb -tv`
> > 
> > That should be 8821cu it self
> 
> Yes, that one. I mean, what was written on the box when you bought it? If I wanted to buy that myself, how can I find it?

Well uhm
It's actually quite a intriguing module tho

As mentioned earlier the module was pre designed, I guess you could just search rtl8821 module and get some kind of minimum tiny module

And what I've got here is like someone helped me to lead the usb signal out and plugged it somewhere inside the board so

Hmm so is it really like some kind of hardware issue then or~? :D

briansune, 2025-01-19:
@dubhater 

I guess he is talking about these kind of PCBA stamp-based mod onto motherboard.
https://github.com/briansune/rtw88-hw/tree/main/rtl8821cu_arm_a9_pcba

dubhater, 2025-01-19:
Ahhh, okay. I thought it was something like this:

[image]

Maybe it's not a driver issue after all. Try this driver: https://github.com/morrownr/8821cu-20210916

But you have to unplug the 0bda:c811. You can't have both connected at the same time when you use that driver.

PD-Kerman, 2025-01-19:
> @dubhater 
> 
> I guess he is talking about these kind of PCBA stamp-based mod onto motherboard.
> https://github.com/briansune/rtw88-hw/tree/main/rtl8821cu_arm_a9_pcba

@dubhater yeah pretty much exactly like that

PD-Kerman, 2025-01-19:
> Ahhh, okay. I thought it was something like this:
> 
> [image]
> 
> Maybe it's not a driver issue after all. Try this driver: https://github.com/morrownr/8821cu-20210916
> 
> But you have to unplug the 0bda:c811. You can't have both connected at the same time when you use that driver.

Oh I see
So there should be something todo with the hardware then~?

PD-Kerman, 2025-01-19:
Alright I'll go try that out

briansune, 2025-01-19:
@dubhater 

My deep bad to open up the can of worms.

Well I am afraid I loss interest on making more PCBA like these.
https://github.com/briansune/rtw88-hw/tree/main/rtl8821au_arm_a9

But @dubhater If you insist I can make it.
But @PD-Kerman must tell me what is the actual brand.
There are at least in my mind three major band that are making those.

PD-Kerman, 2025-01-19:
Alr then
I guess I should go check with the guy selling me these 
And I've got some photo of these stuff maybe reference-able?
[image]
[image]

briansune, 2025-01-19:
@dubhater 

Apart from that the RTL8821CU still got a performance degradation on the 5G band. Poor TRX rate.

briansune, 2025-01-19:
@dubhater 

I had compiled [morrownr](https://github.com/morrownr/8821cu-20210916) 

```
Testing download speed................................................................................
Download: 5.11 Mbit/s
Testing upload speed......................................................................................................
Upload: 6.45 Mbit/s
brian@localhost:~$ iwlist freq
lo        no frequency information.

sit0      no frequency information.

wlx387accfc14ab  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Current Frequency:5.745 GHz

wlx3a7accfc14ab  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
brian@localhost:~$ iwconfig wlx387accfc14ab
wlx387accfc14ab  IEEE 802.11  ESSID:""
          Mode:Managed  Frequency:5.745 GHz  Access Point: 
          Bit Rate=434 Mb/s   Tx-Power=13 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=44/70  Signal level=-66 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

Still poor TRX rate.

[image]

a5a5aa555oo, 2025-01-19:
@briansune I saw something interesting in your [test result](https://github.com/briansune/rtw88-hw/tree/main/rtl8821cu_arm_a9_pcba): your adapter runs in USB 1.1 mode.

[image]

briansune, 2025-01-19:
@a5a5aa555oo 

OMG you do got your point I completely forgot the hub is changed during BOM update.

That is why https://github.com/briansune/rtw88-hw/tree/main/rtl8821cu_arm_a35

5G got Download: 68.50 Mbit/s.

Shittttttttt ........ 

The BOM had replaced the 
Microchip USB3320
https://ww1.microchip.com/downloads/en/DeviceDoc/332xdb.pdf

 - x2-x3 more cost than

FE1.1S-BQFN24B
https://www.lcsc.com/datasheet/lcsc_datasheet_2410121852_Terminus-Tech-FE1-1S-BQFN24B_C6776948.pdf

Completely my bad forgotten this so important information.

briansune, 2025-01-19:
@dubhater 

Sorry about the wrong reporting on speed degradation on the PCBA of the RTL8821CU stamp-based module.
Close file.

Deep appreciate @a5a5aa555oo pointing out.

Even my self still not paying attention to all the details before blaming software.

[image]

briansune, 2025-01-19:
@a5a5aa555oo 

I guess this is issue about the USB hub driver or USB3320 driver having some issue.

Even the BOM replaced should able to support:

```
Fully compliant with Universal Serial Bus
Specification Revision 2.0 (USB 2.0);
□ Upstream facing port supports HighSpeed (480MHz) and Full-Speed
(12MHz) modes;
□ 4 downstream facing ports support
High-Speed (480MHz), Full-Speed
(12MHz), and Low-Speed (1.5MHz)
modes;
```

Same DUT only module replaced can enter High-Speed
https://github.com/briansune/rtw88-hw/tree/main/rtl8811cu_arm_a9

The board have two SD card image.
One is 6.6 Kernel and the other is 5.4 Kernel.

@a5a5aa555oo 

Really thanks for pointing it again.
I will investigate it again.

PD-Kerman, 2025-01-20:
> Ahhh, okay. I thought it was something like this:
> 
> [image]
> 
> Maybe it's not a driver issue after all. Try this driver: https://github.com/morrownr/8821cu-20210916
> 
> But you have to unplug the 0bda:c811. You can't have both connected at the same time when you use that driver.

## compiled and installed

`
pd-kerman@archlinux ~ % sudo iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CCMP-256 (00-0f-ac:10)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
		* CMAC-256 (00-0f-ac:13)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412.0 MHz [1] (16.0 dBm)
[... 127 lines cut ...]
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	Supported extended features:
		* [ RRM ]: RRM
pd-kerman@archlinux ~ % 

`

[image]

[image]

[image]

[image]

seems like still no luck(5G band) as shown, but somehow 2.4ghz was somehow faster than before ig (doesnt really matters)

and btw i went and asked about the brand of the PCBA and the guy sold me these said like they should all be the same so...

So where are we exactly at now~? is it most likely a softwave/driver issue or should it be the hardwave issue such as missing components for 5G band or is it like these PCBAs wont be considered in driver support at this time then ~? :D 

anyways i could always switch to my 8188GU dongle its just felt like i should report this and it sure, would be perfect if there will be a solution but i mean like i get it even if you guys think its not worth it to support ;)

briansune, 2025-01-20:
@PD-Kerman 

If you can take a more clear image I can do able to help you debug on the hardware side.
Because the first image you enclosed. Left IPEX connector is not showing but the second image is tell me that it is connected?
Why two different case on the same hardare.

Hard to tell.

briansune, 2025-01-20:
@dubhater @a5a5aa555oo

I had also test on the same board with kernel 6.6

https://github.com/briansune/rtw88-hw/tree/main/rtl8821cu_arm_a9_pcba_6#network-speed-test-via-ookla---band-5g

The speedtest-cli even better balance on TRX than kernel 5,4.

But the AP is dead after only few TRX. Cannot reconnect etc.

PD-Kerman, 2025-01-20:
> @PD-Kerman 
> 
> If you can take a more clear image I can do able to help you debug on the hardware side.
well sure i guess, ill try when i get back home
> Because the first image you enclosed. Left IPEX connector is not showing but the second image is telling me that it is connected?
well yea about that that could be a bit confusing i should've explained that
nah there's only a single IPEX connector, the other cable was just happened to be hovering there but nah its not connected, there's just one connector, and one should be more than sufficient 
> Why two different case on the same hardware.
nah just one
> 
> Hard to tell.
:D

briansune, 2025-01-20:
@PD-Kerman 

Am I really understand English?
>"hovering there" ????????

That is pure copper possible gold-plated.

PD-Kerman, 2025-01-20:
> @PD-Kerman 
> 
> Am I really understand English?
> >"hovering there" ????????
> 
> That is pure copper possible gold-plated.
> 

What...

its just
not connected

there arent such 'second IPEX connector' there and the other cable from the antenna is just happened to be there **not connected**

and those shots were from the seller and im pretty sure there's only one modle one with a single IPEX connector
theres no such module comes with two dont know if i made that clear

the plate stuff shoulf just be the IPEX connector from the cable side

[image]

briansune, 2025-01-20:
@PD-Kerman 

> no such module comes with two dont know if i made that clear
My friend it the module don't use diplexer it can be two antenna.

>---- hovering.

Could it possible short things out from time to time? Oh man, I completely give up on your case.

Just get real USB dongle it will make life much easier.

BTW I had no idea what you are trying to tell on the IPEX cable.

PD-Kerman, 2025-01-20:
> @PD-Kerman 
> 
> > no such module comes with two dont know if i made that clear
> 
> My friend it the module don't use diplexer it can be two antenna.
> 
yup i see
but i mean the module im using right now didnt come with two
> >---- hovering.
> 
> Could it possible short things out from time to time? Oh man, I completely give up on your case.
> 
oh man i just said that those shots were from the seller not me
i use tapes to prevent those from even happening
> Just get real USB dongle it will make life much easier.
> 
yea thats what i thought as well
> BTW I had no idea what you are trying to tell on the IPEX cable.
> 
i just meant that as the one ive got havent got two connectors on the module
i dont think the seller had made some other kind of similar module which ive checked

so although it looks like so
its nearly impossible that there are two IPEX connectors on board in that shot

so i mean it could just be the plate from the antenna cable but shot in some weird way so it looks like the connector/plate is on the board thats all its about

anyways yea its just my backup machine and its got 2.4g band working
makes no sense to dump these much time on it not to say its quite some niche soulution the module itself

so yea if there arent any specific error that could tell within the driver/software field

then it should be quite moot for any further discussions as we even including me dont really have much info about this module/solution/product

closing in 24hrs if no further chsnges

thank you guys for your participants and supports, its been my pleasure
>

PD-Kerman, 2025-01-20:
[image]

@briansune 

btw seller just provided some datasheet kind of docs

briansune, 2025-01-20:
@PD-Kerman 

You really like to argue so much in your life or just targeting me only (if so well I guess I just sorry that I enter your life from first place)?

I guess if you do know very well on these stuff that I guess you must sure what you are doing from first place.

The logic behind mentioning:

No module have two antenna path, this is possibility truth but once any order is placed to FN-LINK and do a 10k lot  and request split antenna.

Then your argument immediately dead.

Again If it is very sure it had no chance that module don't exist and happen in such way. Why your PCB reserve PI-Tuning path and a IPEX connector footprint from first place? Additional capacitor and IPEX connector must be fun to spend cost on.

Placing additional symbols and routing those components must be fun in your life when there are thousand of FAE ticket on your ticket system.

You logic flows are very interesting on hardware thought.

Have a great day and wish problem resolve ASAP.

briansune, 2025-01-20:
@PD-Kerman 

Hum....

Short search RTL8723DU:

[image]

Why ? Interesting....

PD-Kerman, 2025-01-20:
> @PD-Kerman 
> 
> You really like to argue so much in your life or just targeting me only (if so well I guess I just sorry that I enter your life from first place)?

Well what
i got to claim that i was just replying questions by questions from the first place

i really dont get how and why you will ever think its any kind of argue

you participated and i replied and appreciated i dont see any thing inappropriate
> 
> I guess if you do know very well on these stuff that I guess you must sure what you are doing from first place.

And i do not
i never stated that i made these modules and from the time being, the module refers to the pcba and the base pcb as a whole package thats the module i was refering

i dont know much about hardware yet, and if anything, it seems like you should be the one being helpful but snarky

first its between driver debugging and belike oh this is way to challenging cuz i dont fully know about the module solution and whether its been downcosted or components were missing

and its like stepping out but not fully

then its like okey i am planning to get more info of the module

then you went like opening a dongle takes only 10sec and something

i said for like multiple times that
it did connected to 5G band and worked well for some time but only with the very AP

its just what went wrong is like now it cannot properly search & connect to 5ghz and as the screenshots i provided, it sometimes can get 5g band signals scanned but wont exist for long and with wrong ssids but right channels, and 8188gu hadgot 5ghz ssids scanned and marked with spectrum mgmt so i was thinking about driver induced compatibility issues, thats what ive am talking about

and the whole module (base board and pcba) was designed and bought from a friend in the community
so again, i didnt designed the pcb and all i can tell you is that if you sre considering the phy rf signal path went wrong for 5G band, it was perfectly fine just weeks ago before i got my router/aps upgraded

and that should be something to discuss instead of finding hardware flaws that conflicts with the info already given we are mostly users instead of hardware engineers

and speaking of argue once again

from my perspective who just love to argue... shouldnt that be you

again you arent expecting for everyone reporting issues to be senior hardware engineers right?

i dont see any point of being snarky on teeny tiny states that you brought up and dont really have things to do with the issue it self

alright sure the pcba module it self could support 2IPEX connectors depends on the diplexer configirations

so what

does it have anything to do with this  issue it self?

and i mentioned multiple times that
5ghz works before, might still be now, but its having some serious compatibility issue with most APs

did you get that what does it mean by saying 5ghz worked before?

all these hardware diagnosis was just to confirm that there are nothing wrong with the rf path for 5ghz to the chip right

and im telling you again that 5G band worked before
all these should just be something thats moot

no need to say that i said yes to all hardware info check you asked for

i really dont see how and what these hardware stuff you are pulling is for and...

we are discussing under a driver project

consumers uses these drivers consumers provide reports

if it has something to do with the driver and its fixable, great lets put it on the solving schedule

if not, alright maybe the cause remains out of scope for driver dev teams or unknown or not worth supporting, they got feedbacks and issue closed

its that simple

and thank you for at least being helpful around here but your judgement would really be unecessary here

> The logic behind mentioning:
>
> No module have two antenna path, this is possibility truth but once any order is placed to FN-LINK and do a 10k lot  and request split antenna.
>
> Then your argument [...]
thx you too

briansune, 2025-01-20:
@PD-Kerman 

Thank you

PD-Kerman, 2025-01-20:
@briansune 

You're welcome

"USB 1.1"<-
"10Mbps but 5GHz"<-
"Reports but 5G Band never exceeds 100Mbps" <-

PD-Kerman, 2025-01-20:
Closing as discussion is moot and not scheduled for any fix

[4 more replies]

closed 2025-01-20 by PD-Kerman