lwfinger/rtw88 #295: Is RTW88_8822ce compatible with 6.11.2 kernel?

manudev-1, 2025-01-22:
Im trying to use rtw_8822ce 9.9.4 firmware version with kernel 6.11.x but even i can connect to my wifi, after a couple of minutes i cant anymore communicate and starting get "Destination Host Unreachable".

In my dmesg i can se:
```
[  966.246616] rtw_8822ce 0000:03:00.0: firmware failed to leave lps state
[  966.249981] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  966.253156] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  968.262771] rtw_8822ce 0000:03:00.0: firmware failed to leave lps state
[  970.246851] rtw_8822ce 0000:03:00.0: firmware failed to leave lps state
[  972.262956] rtw_8822ce 0000:03:00.0: firmware failed to leave lps state
[  972.266384] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  974.247033] rtw_8822ce 0000:03:00.0: firmware failed to leave lps state
[  974.250399] rtw_8822ce 0000:03:00.0: failed to send h2c command
[  976.266939] rtw_8822ce 0000:03:00.0: firmware failed to leave lps state
[  976.270292] rtw_8822ce 0000:03:00.0: failed to send h2c command
```

This problem start when i update and upgrade my kali, so i think its a kernel problem.

briansune, 2025-01-24:
@manudev-1 

Hello, I am encountering RTL8822CE RTL8821CE RTL8723DE not recognizing but only 8822BE.

Would you mind to share a bit some info which I hope it can debug my case a bit on RTW88 driver.

It is a PC or notebook or some kind of raspberry-pi like ARM device that you runs those PCIe card?

Thank you

manudev-1, 2025-01-24:
> @manudev-1
>
> Hello, I am encountering RTL8822CE RTL8821CE RTL8723DE not recognizing but only 8822BE.
>
> Would you mind to share a bit some info which I hope it can debug my case a bit on RTW88 [...]

Sure, It Is a notebook the 250 G9 I3 from Hp.
If you have any other question, let me know

briansune, 2025-01-24:
@manudev-1 

Sadly on me, not ARM-based device so I am completely stuck. Could be driver could be PCIe root complex not compatible with the card.

Really appreciate =]

Cont'd debug on my side.

dubhater, 2025-01-24:
@manudev-1 You could try the ``disable_lps_deep`` option of the rtw_core/rtw88_core module.

briansune, 2025-01-27:
@manudev-1 

Would you mind to print out the

> sudo lspci -vv

I think the link and the BAR could have some info that maybe help myside to debug why it is not working on 8822BE.

Thank you

manudev-1, 2025-01-27:
> @manudev-1 
> 
> Would you mind to print out the
> 
> > sudo lspci -vv
> 
> I think the link and the BAR could have some info that maybe help myside to debug why it is not working on 8822BE.
> 
> Thank you

now in my school it's hackton time, so i need my laptop working but i can try replicate the error on a virtual machine

briansune, 2025-01-27:
@manudev-1 

No PB just undergoing a smoke test not even sanity this stage.

Deep appreciate the helps.

Feel free to do it when you are free.

briansune, 2025-01-28:
@manudev-1 

I had fixed the problem on my side you can drop the lspci print.

Appreciate the helps. =]

CatOfBlades, 2025-02-16:
I installed arch on my HP 14-dq2089wm. 
Don't know a ton about drivers or anything and as far as I can tell the wifi works fine. But I am also getting the same "[I think this is a timestamp] rtw_8821ce 0000:01:00.0: mac power on failed"
"[I think this is a timestamp] rtw_8821ce 0000:01:00.0: failed to power on mac"
"[I think this is a timestamp] rtw_8821ce 0000:01:00.0: failed to send h2c command"
in the first tty shell when logged on as root. 
Started happening after a kernel update. 
Inturrupting as I am typing so I wanted to see if anyone has a solution. If theres any more information I can give to help I would like to.

CatOfBlades, 2025-02-16:
> I installed arch on my HP 14-dq2089wm. Don't know a ton about drivers or anything and as far as I can tell the wifi works fine. But I am also getting the same "[I think this is a timestamp] [...]

> [@manudev-1](https://github.com/manudev-1)
> 
> Would you mind to print out the
> 
> > sudo lspci -vv
> 
> I think the link and the BAR could have some info that maybe help myside to debug why it is not working on 8822BE.
> 
> Thank you

My lspci -vv shows:
`
0000:01:00.0 Network controller: Realtek Simiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
     DeviceName: Realtek Wireless LAN + BT
     Subsystem: Hewlett-Packard Company Device 884d
     !! Unknown header type 7f
     Interrupt: pin ? routed to IRQ 153
     Region 0: I/O ports at 3000 [size=256]
     Region 2: Memory at 84000000 (64-bit, non-prefetchable) [size=64K]
     Kernel driver in use: rtw_8821ce
     Kernel modules: rtw88_8821ce
`

briansune, 2025-02-16:
@CatOfBlades 

My case is not under x86 or x64 platform. My kernel build is based on ARM64 which is completely different.

I resolve my issue due to out-of-tree messing up.

But lspci -vv is very puzzling because if you see a normal lspci from my reports:

https://github.com/briansune/rtw88-hw#rtw-revision---e32c796572ee22ae172411fe5c1183921eca498e

I suggest you cross-check it with my report lspci which should shows all necessary working info from first place.

The driver could possibly have trouble on hot-plug so highly not suggest using in such way.

For BAR problem ARM do not support IO so ignore it w/o any possible operation issue.