lwfinger/rtw88 #298: RTL8822BE on ARM64 crash immediate after insmod

briansune, 2025-01-26:
@dubhater 

Please help and see if this is a driver issue thank you.

```
```

briansune, 2025-01-27:
@dubhater 

Updated kernel to 6.6

Same setup on the board using Intel WLAN card shows no issue

The crash messages had changed.

INTEL AC 3165

```
[    9.691278] Intel(R) Wireless WiFi driver for Linux
[    9.691476] pci 0000:00:00.0: enabling device (0000 -> 0002)
[    9.691495] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[    9.702303] iwlwifi 0000:01:00.0: Detected crf-id 0x0, cnv-id 0x0 wfpm id 0x0
[    9.702330] iwlwifi 0000:01:00.0: PCI dev 3165/4010, rev=0x210, rfid=0xd55555d5
[    9.759161] iwlwifi 0000:01:00.0: loaded firmware version 29.1044073957.0 7265D-29.ucode op_mode iwlmvm
[    9.878885] iwlwifi 0000:01:00.0: Detected Intel(R) Dual Band Wireless AC 3165, REV=0x210
[    9.920670] iwlwifi 0000:01:00.0: base HW address: bc:54:2f:5b:71:6f, OTP minor version: 0x0
[    9.988783] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[   10.015196] iwlwifi 0000:01:00.0 wlp1s0: renamed from wlan0
[   19.508038] iwlwifi 0000:01:00.0: Registered PHC clock: iwlwifi-PTP, with index: 0
[   19.813016] platform firmware:zynqmp-firmware:soc-nvmem: deferred probe pending
[   33.379459] pipewire[450]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[  412.366791] iwlwifi 0000:01:00.0: Unregistering PHC clock: iwlwifi-PTP, with index: 0
```

REALTEK 8822BE

```
[  465.664814] pci 0000:01:00.0: [10ec:b822] type 00 class 0x028000
[  465.664864] pci 0000:01:00.0: reg 0x10: [io  0x0000-0x00ff]
[  465.664908] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x0000ffff 64bit]
[  465.665101] pci 0000:01:00.0: supports D1 D2
[  465.665108] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[  465.665466] pci 0000:01:00.0: BAR 2: assigned [mem 0xe0000000-0xe000ffff 64bit]
[  465.665496] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0100]
[  465.665505] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0100]

[  524.565164] rtw_core: loading out-of-tree module taints kernel.
[  535.597923] ------------[ cut here ]------------
[  535.597940] WARNING: CPU: 0 PID: 691 at /net/mac80211/main.c:633 ieee80211_alloc_hw_nm+0x98/0x5f8
[  535.597966] Modules linked in: rtw_8822be(O+) rtw_8822b(O) rtw_pci(O) rtw_core(O) iwlmvm iwlwifi uio_pdrv_genirq
[  535.597998] CPU: 0 PID: 691 Comm: insmod Tainted: G           O       6.6.0-xilinx-gd8528968a1b3-dirty #1
[  535.598008] Hardware name: ZYNQ_MP (DT)
[  535.598012] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  535.598022] pc : ieee80211_alloc_hw_nm+0x98/0x5f8
[  535.598032] lr : rtw_pci_probe+0x5c/0x9fc [rtw_pci]
[  535.598050] sp : ffff800081c6b780
[  535.598054] x29: ffff800081c6b780 x28: ffff800081759c90 x27: 0000000000000000
[  535.598068] x26: ffff80008187abc0 x25: ffff0000090d47e8 x24: 0000000000000037
[  535.598082] x23: ffff00000af85000 x22: ffff00000af850a0 x21: ffff8000799fe000
[  535.598096] x20: ffff800079a00000 x19: ffff800079a00000 x18: 0000000000000020
[  535.598110] x17: 0000000000000068 x16: 00000000000000c0 x15: 0000000000000020
[  535.598123] x14: ffff80008183f408 x13: 0000000000000002 x12: 0000000000000000
[  535.598136] x11: ffff800081558200 x10: 000000000005f538 x9 : 0000000000000005
[  535.598150] x8 : ffff800081c6b798 x7 : 0000000000000000 x6 : ffff8000818b4750
[  535.598163] x5 : ffff00000af85184 x4 : 0000000000000000 x3 : ffff80007998d2f0
[  535.598177] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 000000000000ac48
[  535.598190] Call trace:
[  535.598194]  ieee80211_alloc_hw_nm+0x98/0x5f8
[  535.598205]  rtw_pci_probe+0x5c/0x9fc [rtw_pci]
[  535.598220]  pci_device_probe+0x9c/0x154
[  535.598230]  really_probe+0x148/0x2b0
[  535.598239]  __driver_probe_device+0x78/0x12c
[  535.598248]  driver_probe_device+0xd8/0x15c
[  535.598256]  __driver_attach+0x90/0x19c
[  535.598264]  bus_for_each_dev+0x7c/0xe0
[  535.598277]  driver_attach+0x24/0x30
[  535.598284]  bus_add_driver+0xe8/0x1f8
[  535.598292]  driver_register+0x5c/0x124
[  535.598301]  __pci_register_driver+0x4c/0x58
[  535.598313]  rtw_8822be_driver_init+0x28/0x1000 [rtw_8822be]
[  535.598329]  do_one_initcall+0x80/0x1c8
[  535.598339]  do_init_module+0x58/0x1e0
[  535.598351]  load_module+0x1b30/0x1cd4
[  535.598361]  init_module_from_file+0x88/0xcc
[  535.598372]  __arm64_sys_finit_module+0x184/0x368
[  535.598383]  invoke_syscall+0x54/0x120
[  535.598395]  el0_svc_common.constprop.0+0x40/0xe0
[  535.598406]  do_el0_svc+0x1c/0x28
[  535.598417]  el0_svc+0x40/0xe4
[  535.598429]  el0t_64_sync_handler+0x120/0x12c
[  535.598440]  el0t_64_sync+0x190/0x194
[  535.598449] ---[ end trace 0000000000000000 ]---
[  535.598457] rtw_8822be 0000:01:00.0: failed to allocate hw
[  535.603970] rtw_8822be: probe of 0000:01:00.0 failed with error -12

```

briansune, 2025-01-27:
@dubhater 

Forget it,

It is most likely the RTW88 backport issue.

I had tested RTW88 driver not backport one able to load w/o issue.

```

Module                  Size  Used by
rtw88_8822be           12288  0
rtw88_8822b           217088  1 rtw88_8822be
rtw88_pci              24576  1 rtw88_8822be
rtw88_core            188416  2 rtw88_pci,rtw88_8822b
uio_pdrv_genirq        12288  0

iwconfig
lo        no wireless extensions.

sit0      no wireless extensions.

end0      no wireless extensions.

wlp1s0    IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

iwlist wlp1s0 freq
wlp1s0    32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 14 : 2.484 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz

01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter
        Subsystem: Hewlett-Packard Company Realtek RTL8822BE 802.11ac 2x2 Wi-Fi + Bluetooth 4.2 Combo Adapter (MU-MIMO supported)
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 53
[... 55 lines cut ...]
[    2.501072] nwl-pcie fd0e0000.pcie:      MEM 0x00e0000000..0x00efffffff -> 0x00e0000000
[    2.509085] nwl-pcie fd0e0000.pcie:      MEM 0x0600000000..0x07ffffffff -> 0x0600000000
[    2.517215] nwl-pcie fd0e0000.pcie: Link is UP
[    2.522042] nwl-pcie fd0e0000.pcie: PCI host bridge to bus 0000:00
[    2.528228] pci_bus 0000:00: root bus resource [bus 00-ff]
[    2.533709] pci_bus 0000:00: root bus resource [mem 0xe0000000-0xefffffff]
[    2.540582] pci_bus 0000:00: root bus resource [mem 0x600000000-0x7ffffffff pref]
[    2.548090] pci 0000:00:00.0: [10ee:d011] type 01 class 0x060400
[    2.554162] pci 0000:00:00.0: PME# supported from D0 D1 D2 D3hot
[    2.561308] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    2.569410] pci 0000:01:00.0: [10ec:b822] type 00 class 0x028000
[    2.575452] pci 0000:01:00.0: reg 0x10: [io  0x0000-0x00ff]
[    2.581058] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x0000ffff 64bit]
[    2.588027] pci 0000:01:00.0: supports D1 D2
[    2.592292] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    2.599154] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    2.605785] pci 0000:00:00.0: BAR 8: assigned [mem 0xe0000000-0xe00fffff]
[    2.612580] pci 0000:01:00.0: BAR 2: assigned [mem 0xe0000000-0xe000ffff 64bit]
[    2.619902] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0100]
[    2.626170] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0100]
[    2.632784] pci 0000:00:00.0: PCI bridge to [bus 01]
[    2.637753] pci 0000:00:00.0:   bridge window [mem 0xe0000000-0xe00fffff]
[   10.184955] pci 0000:00:00.0: enabling device (0000 -> 0002)

```

dubhater, 2025-01-27:
I have no clue about the errors you got with kernel 5.4.

The error you got with kernel 6.6.0 is easier to understand, you just need to look at the kernel's source code. What is at line 633 in the file net/mac80211/main.c ?

briansune, 2025-01-28:
@dubhater 

at the line if

```
struct ieee80211_hw *ieee80211_alloc_hw_nm(size_t priv_data_len,
					   const struct ieee80211_ops *ops,
					   const char *requested_name)
{
	struct ieee80211_local *local;
	int priv_size, i;
	struct wiphy *wiphy;
	bool use_chanctx;

	if (WARN_ON(!ops->tx || !ops->start || !ops->stop || !ops->config ||
		    !ops->add_interface || !ops->remove_interface ||
		    !ops->configure_filter || !ops->wake_tx_queue))
		return NULL;
```

briansune, 2025-01-28:
@dubhater 

For 5.4 forgot it is is most likely a PCIe driver could be wrong on cross-compiling under ARM platform.

Should be good on using RTW88 backport on 8822BE.

It is likely need to compile the entire driver inside the kernel package during yocto.

Outside module kernel could somehow making some issues.

For 6.6 I think the case is the same.

```
ubuntu@zynqmp:~$ lsmod
Module                  Size  Used by
rtw_8822be             16384  0
rtw_8822b             221184  1 rtw_8822be
rtw_pci                32768  1 rtw_8822be
rtw_core              208896  2 rtw_8822b,rtw_pci
ubuntu@zynqmp:~$ sudo lspci -v
00:00.0 PCI bridge: Xilinx Corporation Device d011 (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 46
        Bus: primary=00, secondary=01, subordinate=0c, sec-latency=0
        I/O behind bridge: 00000000-00000fff
        Memory behind bridge: e0000000-e00fffff
        Capabilities: [40] Power Management version 3
        Capabilities: [60] Express Root Port (Slot-), MSI 00
        Capabilities: [100] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [10c] Virtual Channel
        Capabilities: [128] Vendor Specific Information: ID=1234 Rev=1 Len=018 <?>
        Kernel driver in use: pcieport

01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter
        Subsystem: Hewlett-Packard Company Device 831b
        Flags: bus master, fast devsel, latency 0, IRQ 50
        I/O ports at <unassigned> [disabled]
        Memory at e0000000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-b8-22-01
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
        Kernel driver in use: rtw_8822be
        Kernel modules: rtw_8822be

ubuntu@zynqmp:~$ dmesg | grep rtw
[    7.495998] rtw_8822be 0000:01:00.0: enabling device (0000 -> 0002)
[    7.528283] rtw_8822be 0000:01:00.0: Firmware version 30.20.0, H2C version 14
[    7.644039] rtw_8822be 0000:01:00.0 wlp1s0: renamed from wlan0
```

closed 2025-01-28 by briansune

briansune, 2025-01-28:
@dubhater 

This is the report on RTL8822BE: https://github.com/briansune/rtw88-hw/tree/main/rtl8822be_arm_a53

Working properly on ARM64 kernel 5.4

Same issue on AP and 2.4G band.

Good and stable