lwfinger/rtw88 #299: 8814AU reaver will fail and kill the driver until reboot!

reaveruser, 2025-01-26:
I am doing some test. No issues testing reaver with 8822bu with this driver, but with 8814au will fail immediately, the driver will crash, and will not work until reboot.

Here are the logs, hope it will help.

```
─$ sudo dmesg | grep rtw_88
[   52.115351] rtw_8814au 4-1:1.0: Firmware version 33.6.0, H2C version 6
[   52.405517] usbcore: registered new interface driver rtw_8814au
[   54.773631] Modules linked in: rtw_8814au(OE) rtw_8814a(OE) rtw_usb(OE) rtw_core(OE) mac80211 libarc4 cfg80211 snd_seq_dummy snd_hrtimer snd_seq_midi snd_seq_midi_event snd_seq rfkill qrtr sunrpc binfmt_misc intel_rapl_msr intel_rapl_common intel_uncore_frequency_common intel_pmc_core intel_vsec pmt_telemetry pmt_class rapl snd_ens1371 snd_ac97_codec ac97_bus vmw_balloon gameport snd_rawmidi snd_seq_device snd_pcm pcspkr snd_timer snd soundcore ac sg joydev evdev serio_raw msr efi_pstore configfs nfnetlink vsock_loopback vmw_vsock_virtio_transport_common vmw_vsock_vmci_transport vsock vmw_vmci ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid sd_mod crct10dif_pclmul crc32_pclmul crc32c_intel sr_mod cdrom vmwgfx drm_ttm_helper ghash_clmulni_intel ttm sha512_ssse3 ata_generic drm_kms_helper mptspi mptscsih mptbase ahci sha256_ssse3 ata_piix libahci xhci_pci uhci_hcd drm sha1_ssse3 xhci_hcd scsi_transport_spi libata ehci_pci psmouse ehci_hcd usbcore scsi_mod e1000 usb_common
[   54.774136] Modules linked in: rtw_8814au(OE) rtw_8814a(OE) rtw_usb(OE) rtw_core(OE) mac80211 libarc4 cfg80211 snd_seq_dummy snd_hrtimer snd_seq_midi snd_seq_midi_event snd_seq rfkill qrtr sunrpc binfmt_misc intel_rapl_msr intel_rapl_common intel_uncore_frequency_common intel_pmc_core intel_vsec pmt_telemetry pmt_class rapl snd_ens1371 snd_ac97_codec ac97_bus vmw_balloon gameport snd_rawmidi snd_seq_device snd_pcm pcspkr snd_timer snd soundcore ac sg joydev evdev serio_raw msr efi_pstore configfs nfnetlink vsock_loopback vmw_vsock_virtio_transport_common vmw_vsock_vmci_transport vsock vmw_vmci ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid sd_mod crct10dif_pclmul crc32_pclmul crc32c_intel sr_mod cdrom vmwgfx drm_ttm_helper ghash_clmulni_intel ttm sha512_ssse3 ata_generic drm_kms_helper mptspi mptscsih mptbase ahci sha256_ssse3 ata_piix libahci xhci_pci uhci_hcd drm sha1_ssse3 xhci_hcd scsi_transport_spi libata ehci_pci psmouse ehci_hcd usbcore scsi_mod e1000 usb_common
[   70.827458] rtw_8822bu 3-2:1.0: Firmware version 30.20.0, H2C version 14
[   71.409715] usbcore: registered new interface driver rtw_8822bu
[  101.571916] rtw_8822bu 3-2:1.0 wlan1mon: entered promiscuous mode
[  101.571920] rtw_8822bu 3-2:1.0 wlan1mon: entered allmulticast mode
[  107.310924] rtw_8814au 4-1:1.0 wlan0mon: entered promiscuous mode
[  107.310929] rtw_8814au 4-1:1.0 wlan0mon: entered allmulticast mode
[  117.173515] rtw_8814au 4-1:1.0: Firmware version 33.6.0, H2C version 6
[  118.207907] rtw_8814au 4-1:1.0: write register 0x1b80 failed with -71
[  118.208337] rtw_8814au 4-1:1.0: write register 0x1b80 failed with -71
[  118.216603] rtw_8814au 4-1:1.0: write register 0x1b80 failed with -71
[  118.411422] rtw_8814au 4-1:1.0: failed to do USB write, ret=-19
[  118.411428] rtw_8814au 4-1:1.0: failed to send h2c packet
[  118.411430] rtw_8814au 4-1:1.0: failed to do USB write, ret=-19
[  118.411431] rtw_8814au 4-1:1.0: failed to send h2c packet
[  118.921117] rtw_8814au 4-1:1.0: Firmware version 33.6.0, H2C version 6
[  146.266397] rtw_8814au 4-1:1.0: write register 0xc90 failed with -71
[  146.475323] rtw_8814au 4-1:1.0: failed to do USB write, ret=-19
[  146.475330] rtw_8814au 4-1:1.0: failed to send h2c packet
[  146.475332] rtw_8814au 4-1:1.0: failed to do USB write, ret=-19
[  146.475332] rtw_8814au 4-1:1.0: failed to send h2c packet
[  148.521830] rtw_8814au 4-1:1.0: Firmware version 33.6.0, H2C version 6
[  187.321608] rtw_8814au 4-1:1.0 wlan0mon: entered promiscuous mode
[  187.321612] rtw_8814au 4-1:1.0 wlan0mon: entered allmulticast mode
[  192.719678] rtw_8814au 4-1:1.0: Firmware version 33.6.0, H2C version 6
[  192.847821] rtw_8814au 4-1:1.0: read register 0x30 failed with -71
[  192.855159] rtw_8814au 4-1:1.0: read register 0x30 failed with -71
[  192.861906] rtw_8814au 4-1:1.0: read register 0x30 failed with -71
[  202.643703] rtw_8822bu 3-2:1.0: failed to get tx report from firmware
[  225.683368] rtw_8814au 4-1:1.0: read register 0x30 failed with -110
```

dubhater, 2025-01-26:
Try dmesg without grep.

dubhater, 2025-01-26:
Actually, nevermind that. Try with real hardware instead of a virtual machine.

reaveruser, 2025-01-26:
@dubhater  I have been testing since yesterday, not once I succeeded to do reaver with this driver. I restarted many many times. I might succeed later you don't know.

The issue is there were no issues at all using this driver https://github.com/morrownr/8814au at all. And I was using it for a long while in a VM.

there are lots of glitches using monitor mode even with 8822bu I never encounter them with the native kali driver (8822bu is detected by kali by default)

I agree with you, using VM is not as stable as a real hardware, I believe my issues before maybe cause I was using USB 2.0, once I change to to 3.1 it worked fine, not sure though, but I never had issue with this driver before https://github.com/morrownr/8814au

reaveruser, 2025-01-27:
@dubhater  I concede! you are 100% right

I launched the VM just to make the full log, and the first time I tried reaver it worked. I didn't do anything since I shutdown the VM yesterday. Not even a single settings. It is only I tested with a different router. The router I was testing with yesterday is no longer available to confirm.

But there are still lots of glitches in the monitor mode compared to the old driver from morrownr or even the driver from aircrack-ng. There are even more glitches with 8822bu that I never encountered with the native kali driver.

How ever it is hard to explain these glitches. I will report once I can confirm 100%

closed 2025-01-27 by reaveruser