lwfinger/rtw88 #300: RTL8723DS completely dead on 2.4G

briansune, 2025-01-27:
@dubhater 

c56536cfa522b42cfa73c2c4332000e6aa5bcf97

I had tested the patch complete dead.

After using nmtui and check if there are any channel to connect it is complete empty.
W/o rebooting and rmmod the new RTW88 driver.

I insmod the old 8723ds driver and able to connect to 2.4G w/o issue.

briansune, 2025-01-31:
@dubhater 

The nightmare returns.

This is why I cannot have any WIFI to connect.

If RTW88 don't pay full attention to the EFUSE and MAC address it will cont'd cont'd and cont'd happen again.

https://github.com/lwfinger/rtw88/issues/157#issuecomment-1821839999

[image]

[image]

[image]

dubhater, 2025-01-31:
Was this board working better before https://github.com/lwfinger/rtw88/commit/c56536cfa522b42cfa73c2c4332000e6aa5bcf97 ?

briansune, 2025-01-31:
@dubhater 

I can't see too much different on real WAN applications aka the speedtest-cli (other EFUSE correct boards).
I guess because the numbers is too small in different?

I still see a noticeable different on the Vendors driver.

Most likely people use real WAN on the board, too hard to tell on users side.

closed 2025-01-31 by briansune

dubhater, 2025-01-31:
How is that? 

[0001-wifi-rtw88-Handle-RTL8723D-with-blank-efuse.patch.txt](https://github.com/user-attachments/files/18624613/0001-wifi-rtw88-Handle-RTL8723D-with-blank-efuse.patch.txt)

briansune, 2025-02-01:
@dubhater 

Where to pull this? The trunk is local or?

dubhater, 2025-02-01:
You save the patch and apply it with ``patch -i file.txt``.

briansune, 2025-02-01:
@dubhater 

> patch -i file.txt

Done. Now it can scan WIFI channel and connect on the blank EFUSE board.

But the TRX still not as good as vendor one.

The speedtest shows a double or even more in some host compared to the vendor one.

briansune, 2025-02-01:
@dubhater 

I got to make "A" assumption.

The board and design used usually is internal LDO and as well as you got it.

So I have no idea what will happen if it is switched to DC/DC.

Because the WIFI power is not too worst so LDO do have a better noise-floor.

But application like battery-based one uses DCDC will it possibly create any other issue (NO IDEA)