lwfinger/rtw88 #301: rtw_8821au : wpa_supplicant negotiation failure

FelixBellaby, 2025-01-30:
The wpa_supplicant-2.11-4 based authorisation of the wifi connection is breaking down on Fedora rawhide

There is a discussion on bugzilla of a similar issue with the broadcom bcm4360 kernel driver:

https://bugzilla.redhat.com/show_bug.cgi?id=2302577

The broadcom bug report suggests a cludge to get around the problem:

# sudo iw dev wlp2s0 scan

This cludge also works in a similar way with the rtw_8821au driver:

# sudo iw dev  wlp14s0f3u1 scan

The bug report suggests that broadcom driver may have problems negotiating the max_scan_ie_len

I suppose that there might be a similar negotiation problem with the rtw_8821au driver.

dubhater, 2025-01-30:
It sounds like a wpa_supplicant problem. What is this driver supposed to do about it?

FelixBellaby, 2025-01-30:
I have not examined this bug in any detail, so I can not comment on the patches that might be needed. 

Some distributions have patched wpa_supplicant in the past to allow the Broadcom driver to function. Patches have also been applied to the Broadcom driver so it will work with an unpatched wpa_supplicant. I believe that the patches involved setting a suitable default value for a parameter that the hardware could not determine. The driver had set this value to 0, which confused the unpatched wpa_supplicant. Setting it to 512 in the driver (or wpa_supplicant) solved the problem. 

Since the unpatched wpa_supplicant works with all the kernel mainline drivers, I thought that patching the Broadcom driver made more sense.

I do not know whether the unpatched wpa_supplicant works with the other modules that you are shipping for other chipsets, but if the problem is limited to just the rtw_8821au, then it would make sense to patch this specific driver rather than patching wpa_supplicant.

FelixBellaby, 2025-01-31:
I can no longer replicate this problem. I suspect that wpa_supplicant or NetworkManager save enough details from the first successful activation of a connection to connect again without probing whatever was causing the original problem.

exu-g, 2025-02-02:
I experienced the same issues with this driver and wpa_supplicant.  
Switching to the (experimental) iwd backend in NetworkManager made it work, so I'd second that this is an issue in wpa_supplicant specifically.