lwfinger/rtw88 #303: Low signal strenght

MR-Diamond, 2025-02-05:
I switched from [https://github.com/morrownr/8814au/ ](https://github.com/morrownr/8814au/ )to this driver, for my usb wifi device `TP-Link Archer T9UH v1 [Realtek RTL8814AU`].
Seems to works good, but I noticed the following: in NetworkManager applet on the taskbar, the signal strenght is lower; with the previous driver i used to got around 99% - 96% signal strenght; now, instead, I see 72%.
Maybe this new driver is more accurate and is telling me the real signal strenght, I don't know.

Environment:
Manjaro Linux x86_64
Kernel 6.6.74-1
Network backend iwd 3.3-1

Working USB 3.0 ports.

Tell me if there is something more that I can check/provide.

dubhater, 2025-02-05:
Maybe the old driver is more accurate, maybe not. It's not the most readable code, so I just did more or less what the other rtw88 drivers do.

The raw numbers come from here:

https://github.com/morrownr/8814au/blob/2f02c0ed829523a89891cfeb7e4db3cd5103aeaf/hal/phydm/phydm_phystatus.c#L1190

They go through some hard to follow processing. Then it reports to userspace either this:

https://github.com/morrownr/8814au/blob/2f02c0ed829523a89891cfeb7e4db3cd5103aeaf/os_dep/linux/ioctl_cfg80211.c#L2313

or this:

https://github.com/morrownr/8814au/blob/2f02c0ed829523a89891cfeb7e4db3cd5103aeaf/os_dep/linux/ioctl_cfg80211.c#L2324

MR-Diamond, 2025-02-07:
I noticed another important thing: with this driver, most of the time, (I would say 8 on 10) my OS, sleep (suspend to ram) very faster, almost always instantaneously.

IRainman, 2025-02-11:
I also have an adapter like this, some EDUP AC1900, and I confirm that problem. It may be an issue in the https://github.com/morrownr/8814au/ driver because it isn't shown information about speed at all and may be shown only connected and disconnected status on the level.

MR-Diamond, 2025-02-19:
**UPDATE February 19, 2025:**

I also noticed that I have a lower link speed: with the previous driver from morrownr, I always got a speed of 1300 Mb/s, always.
With this driver, the speed oscillate between 970 Mb/s and 1170 Mb/s: rarely I get 1300 Mb/s.

I am in sync with the commits of this repo: every time that there is a commit, I update the package  `rtw88-dkms-git`

dubhater, 2025-02-19:
Actually that driver doesn't report the real link speed. You can see this if you plug the device into a USB 2 port. The driver transmits only 2 spatial streams in USB 2 mode, which means the TX rate can't be more than 866 MBit/s, but wavemon says 1300 MBit/s.

MR-Diamond, 2025-03-10:
Report after one month of usage:
My laptop is on all day, and I didn't encountered any issue with this driver/module, apart the reported very minor bugs.